raspberrypi/linux #2934: chmod issue

bobypierre, 2019-04-14:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Add a clear and concise description of what you think the bug is.

**To reproduce**
List the steps required to reproduce the issue.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

pelwell, 2019-04-14:
Do you have a card reader and a spare SD card? If so:

1. Install a matching Raspbian image on the spare card.
2. Boot from the new image, and put the card with the broken image in the card reader and plug it into a spare USB port.
3. Mount the second partition of the card on /mnt.
4. sudo cp -R /usr /mnt/usr
5. Power off the system and put the card from the reader back in your Pi.

This should fix your problem, but I won't be held responsible if it doesn't.

pelwell, 2019-04-14:
On second thoughts, replace the cp -R with a tar, in case there are symbolic links:
```
$ sudo tar cf - /usr | (cd /mnt/usr; tar xf -)
```

bobypierre, 2019-04-14:
hello,  thank you for your reply.

I can't run with hardware because I m in UK, and my raspi is in France and use only SSH protocol.

If I try :

pi@raspberrypi:~$ sudo tar cf - /usr | (cd /mnt/usr; tar xf -)

-bash: sudo: command not found
-bash: cd: /mnt/usr: No such file or directory
ERROR: ld.so: object '/usr/lib/arm-linux-gnueabihf/libarmmem.so' from /etc/ld.so.preload cannot be preloaded (cannot open shared object file): ignored.
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
pi@raspberrypi:~$

I don't know how to do.  
Do you have other idea?  
thank you , good by!

pelwell, 2019-04-14:
If you have no chance to access the Pi directly then I think you are stuck - you've removed the setuid bit from sudo, and I can't think of any way to put it back without using sudo - it would be a big security problem.

Are you familiar with "Chère Élise" ("There's a hole in my bucket")?

bobypierre, 2019-04-14:
😂 OK...I understand... 😕 😣 thank you very much ☺

closed 2019-04-14 by pelwell