raspberrypi/linux #2937: Unplug of ethernet cable is not detected

polomsky, 2019-04-15:
**Describe the bug**
If RPI is started with unplugged ethernet cable, after full boot it correctly detects that ethernet cable was connected or disconnected, e.g. `/sys/class/net/eth0/carrier` is changing content.
If RPI is started with plugged ethernet cable, it always responds that cable is connected.

**To reproduce**
1. turn on device with ethernet cable plugged in (ethernet must work)
2. watch /sys/class/net/eth0/carrier
3. unplug ethernet

**Expected behaviour**
`/sys/class/net/eth0/carrier` content should be `0`

**Actual behaviour**
unplug of ethernet is not detected but on ethernet card yellow light is switched off.
`/sys/class/net/eth0/carrier` content is `1`

**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
RPI3 model B+
* Which OS and version (`cat /etc/rpi-issue`)?
```
Raspberry Pi reference 2019-04-08
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, e910bf55ac92e27c6e77043e8995bcb5e0a2d742, stage2
```
* Which firmware version (`vcgencmd version`)?
```
Mar 27 2019 15:45:53 
Copyright (c) 2012 Broadcom
version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
```
Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
```

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

XECDesign, 2019-04-15:
Also reported here:
https://github.com/raspberrypi/firmware/issues/1100

SudhamaniB, 2019-07-30:
Hi Team,

I am not able to replicate the issue with the following system information. Update me if I am missing anything.

System:
1. RPI3 model B+
2. OS version
cat /etc/rpi-issue 
Raspberry Pi reference 2019-04-08
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, e910bf55ac92e27c6e77043e8995bcb5e0a2d742, stage4
3. Kernel version
Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

Thanks & Regards,
Sudhamani B

JamesH65, 2019-07-30:
It would be interesting to know if the problem is showing for anyone using the latest kernel (4.19)

SudhamaniB, 2019-07-31:
Hi Team,

I would like to know does anyone working on this issue?

Regards,
Sudhamani.

::DISCLAIMER::
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
The contents of this e-mail and any attachment(s) are confidential and intended for the named recipient(s) only. E-mail transmission is not guaranteed to be secure or error-free as information could be intercepted, corrupted, lost, destroyed, arrive late or incomplete, or may contain viruses in transmission. The e mail and its contents (with or without referred errors) shall therefore not attach any liability on the originator or HCL or its affiliates. Views or opinions, if any, presented in this email are solely those of the author and may not necessarily reflect the views or opinions of HCL or its affiliates. Any form of reproduction, dissemination, copying, disclosure, modification, distribution and / or publication of this message without the prior written consent of authorized representative of HCL is strictly prohibited. If you have received this email in error please delete it and notify the sender immediately. Before opening any email and/or attachments, please check them for viruses and other defects.
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

pelwell, 2019-07-31:
> > It would be interesting to know if the problem is showing for anyone using the latest kernel (4.19)

> I would like to know does anyone working on this issue?

I think @JamesH65 answered your question before you asked it - we don't work on problems unless we know they exist.

XECDesign, 2019-07-31:
The problem is still there with 4.19.58

jasonemann, 2019-08-23:
I'm seeing this problem on my Raspberry Pi 3 B+ as well.  I noticed it when power cycling a small switch with four RPi's connected caused all but the RPi3+ to request a new DHCP lease (The others are RPI3 non-plus).

```
root@rpi3p-01:~# uname -a
Linux rpi3p-01 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
root@rpi3p-01:~# vcgencmd version
Aug 15 2019 12:07:10
Copyright (c) 2012 Broadcom
version 0e6daa5106dd4164474616408e0dc24f997ffcf3 (clean) (release) (start_cd)
```

Booted with cable initially connected:

```
root@rpi3p-01:~# cat /sys/class/net/eth0/carrier
1
(Disconnected cable)
root@rpi3p-01:~# cat /sys/class/net/eth0/carrier
1
(Connected cable)
root@rpi3p-01:~# cat /sys/class/net/eth0/carrier
1
```

Booted with cable disconnected:

```
root@rpi3p-01:~# cat /sys/class/net/eth0/carrier
0
(Connected cable)
root@rpi3p-01:~# cat /sys/class/net/eth0/carrier
1
(Disconnected cable)
root@rpi3p-01:~# cat /sys/class/net/eth0/carrier
0
```

SudhamaniB, 2019-09-17:
Hi,

This problem is occurring so rarely on the following kernel
Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

Anyone working on this isse?

Regards,
Sudhamani

brubbel, 2020-01-25:
Still present in 4.19.93-v7+

JamesH65, 2020-01-27:
@brubbel Have you got a patch that shows that change in the driver?

paul-1, 2020-12-13:
@JamesH65 has anyone been able to resolve this issue.  This is still a problem on 5.4 kernels.

brubbel, 2020-12-14:
FWIW, I did not patch the driver but use a daemon script to detect this and restart the driver. Never seen the issue since then, so I don't know if the root problem has been resolved or not.

pelwell, 2020-12-14:
This is just a quick update to say that I can reproduce the problem and am looking for the right solution. As a workaround you could try putting "ethtool -d eth0" in /etc/rc.local (or running "sudo ethtool -d eth0" manually), and see if the carrier tracking is correct from then on, or if it reverts to the bad state after a while.

brubbel, 2020-12-14:
_If I remember correctly_ this occurs only once. After reset of the LAN78XX device all goes well.

pelwell, 2020-12-14:
Does that mean you will try my workaround instead of your script?

paul-1, 2020-12-15:
It helps until the interface is taken down.

Boot system with cable in.
sudo ethtool -d eth0
Remove Cable - Carrier goes to 0
Plugin cable - carrier goes to one

However

Remove Cable
ifconfig eth0 down
ifconfig eth0 up
plug cable in

/sys/class/net/eth0/carrier never goes back to a "1"   And the interface is dead until it is reset.  When resetting the USB interface, it has to happen when the cable is unplugged for it to work properly.

pelwell, 2020-12-15:
Thanks - that's a useful test case.

pelwell, 2020-12-15:
The carrier status disconnect is caused by the loss of interrupts from the PHY, which in turn is caused by resetting the MAC's view of the PHY interrupt status while an interrupt is pending. Until the lan78xx driver sees the interrupt status bit in the MAC it won't forward the interrupt to the PHY driver, and until it does that no further PHY interrupts will be generated.

`sudo ethtool -d eth0` breaks the deadlock by reading the PHY's interrupt status, but a proper solution is to explicitly acknowledge the PHY interrupts after clearing the bit in the MAC.

The fix is in rpi-5.4.y and rpi-5.10.y, and will appear in future kernel builds.

juergh, 2020-12-16:
Duplicates of this bug:
https://github.com/raspberrypi/firmware/issues/1100
https://github.com/raspberrypi/linux/issues/3939

pelwell, 2020-12-16:
The patch mentioned above is in the current rpi-update kernel, which has just switched to rpi-5.10.y.

paul-1, 2020-12-19:
Tested with Kernel built from latest 5.4.83, everything looks to be operating correctly.

Thanks.

closed 2020-12-19 by pelwell