raspberrypi/linux #2939: CONFIG_ZRAM=m # CONFIG_ZRAM_WRITEBACK is not set

StuartIanNaylor, 2019-04-19:
Can `CONFIG_ZRAM_WRITEBACK=y` as `echo /dev/sdxx > /sys/block/zramX/backing_dev` enables the write back also with further commands.
Zram is enabled as a module but for some reason CONFIG_ZRAM_WRITEBACK=n ?

JamesH65, 2020-02-13:
@pelwell @popcornmix Thoughts?

StuartIanNaylor, 2020-02-20:
Doesn't make sense to exclude a zram function that needs to be explicitly set to use.
The default is off unless as above echo /dev/sdxx > /sys/block/zramX/backing_dev is set when creating the zram device.
It will be exactly the same apart from if someone wants to use a backing_dev then they can

jan-imal, 2024-08-06:
I was just stumbling about this while testing some various scenarios on the rpi z2w.
Since this is open since 2019 and still not set (besides having CONFIG_ZRAM_MULTI_COMP=y would be nice too) - Is there any chance this might get changed? As already mentioned, it does not hurt and must be manually enabled, I would really appreciated to see this enabled in this module.

pelwell, 2024-08-06:
Adding ZRAM_WRITEBACK=y takes the _uncompressed_ ARM64 binary module from 46kB to 59kB. ZRAM_MULTI_COMP=y adds an extra 6kB - which is not extravagant. Both of them seem to do nothing without being enabled explicitly via sysfs attributes.

Happy with this, @popcornmix?

pelwell, 2024-08-06:
See https://github.com/raspberrypi/linux/pull/6304.

popcornmix, 2024-08-06:
I'm ok with this.

jan-imal, 2024-08-06:
Thank you! For enabling and the quick response!