raspberrypi/linux #2940: Very small portion of actual memory regonized on boot

craigphicks, 2019-04-20:
Raspberry Pi 3 B+

Two case (1) Openwrt, (2) Raspbian, on *same device*.
With Openwrt only ~108K of memory is avalable, but with Raspbian ~1000K (1GB)

Openwrt kernel version:
```
 Linux OpenWrt 4.9.152 #0 SMP Mon Jan 28 08:54:32 2019 aarch64 GNU/Linux
```
Raspbian kernel version
```
Linux rpi2 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
```

Opening of kernel log in case of openwrt:
```
Sat Apr 20 02:58:01 2019 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0
Sat Apr 20 02:58:01 2019 kern.notice kernel: [    0.000000] Linux version 4.9.152 (buildbot@builds-03.infra.lede-project.org) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r7627-753531d) ) #0 SMP Mon Jan 28 08:54:32 2019
Sat Apr 20 02:58:01 2019 kern.info kernel: [    0.000000] Boot CPU: AArch64 Processor [410fd034]
Sat Apr 20 02:58:01 2019 kern.info kernel: [    0.000000] cma: Reserved 16 MiB at 0x0000000006c00000
Sat Apr 20 02:58:01 2019 kern.debug kernel: [    0.000000] On node 0 totalpages: 32768
Sat Apr 20 02:58:01 2019 kern.debug kernel: [    0.000000]   DMA zone: 512 pages used for memmap
Sat Apr 20 02:58:01 2019 kern.debug kernel: [    0.000000]   DMA zone: 0 pages reserved
Sat Apr 20 02:58:01 2019 kern.debug kernel: [    0.000000]   DMA zone: 32768 pages, LIFO batch:7
Sat Apr 20 02:58:01 2019 kern.info kernel: [    0.000000] percpu: Embedded 20 pages/cpu @ffffffc007f7b000 s44056 r8192 d29672 u81920
Sat Apr 20 02:58:01 2019 kern.debug kernel: [    0.000000] pcpu-alloc: s44056 r8192 d29672 u81920 alloc=20*4096
Sat Apr 20 02:58:01 2019 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
Sat Apr 20 02:58:01 2019 kern.info kernel: [    0.000000] Detected VIPT I-cache on CPU0
Sat Apr 20 02:58:01 2019 kern.info kernel: [    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32256
Sat Apr 20 02:58:01 2019 kern.notice kernel: [    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0xec00000 vc_mem.mem_size=0x10000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 kgdboc=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
Sat Apr 20 02:58:01 2019 kern.info kernel: [    0.000000] PID hash table entries: 512 (order: 0, 4096 bytes)
Sat Apr 20 02:58:01 2019 kern.info kernel: [    0.000000] Dentry cache hash table entries: 16384 (order: 5, 131072 bytes)
Sat Apr 20 02:58:01 2019 kern.info kernel: [    0.000000] Inode-cache hash table entries: 8192 (order: 4, 65536 bytes)
Sat Apr 20 02:58:01 2019 kern.info kernel: [    0.000000] Memory: 91608K/131072K available (4094K kernel code, 280K rwdata, 908K rodata, 320K init, 295K bss, 23080K reserved, 16384K cma-reserved)
```

Opening of kernel log in case of Raspbian:
```
Nov 03 10:16:42 rpi2 kernel: Booting Linux on physical CPU 0x0
Nov 03 10:16:42 rpi2 kernel: Linux version 4.14.98-v7+ (dom@dom-XPS-13-9370) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1200 SMP Tue Feb 12 20:27:48 GMT 2019
Nov 03 10:16:42 rpi2 kernel: CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Nov 03 10:16:42 rpi2 kernel: CPU: div instructions available: patching division code
Nov 03 10:16:42 rpi2 kernel: CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Nov 03 10:16:42 rpi2 kernel: OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
Nov 03 10:16:42 rpi2 kernel: Memory policy: Data cache writealloc
Nov 03 10:16:42 rpi2 kernel: cma: Reserved 8 MiB at 0x3ac00000
Nov 03 10:16:42 rpi2 kernel: On node 0 totalpages: 242688
Nov 03 10:16:42 rpi2 kernel: free_area_init_node: node 0, pgdat 80c85780, node_mem_map ba39f000
Nov 03 10:16:42 rpi2 kernel:   Normal zone: 2133 pages used for memmap
Nov 03 10:16:42 rpi2 kernel:   Normal zone: 0 pages reserved
Nov 03 10:16:42 rpi2 kernel:   Normal zone: 242688 pages, LIFO batch:31
Nov 03 10:16:42 rpi2 kernel: percpu: Embedded 17 pages/cpu @ba348000 s38720 r8192 d22720 u69632
Nov 03 10:16:42 rpi2 kernel: pcpu-alloc: s38720 r8192 d22720 u69632 alloc=17*4096
Nov 03 10:16:42 rpi2 kernel: pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Nov 03 10:16:42 rpi2 kernel: Built 1 zonelists, mobility grouping on.  Total pages: 240555
Nov 03 10:16:42 rpi2 kernel: Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=ce32252f-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Apr 17 12:17:01 rpi1 kernel: PID hash table entries: 4096 (order: 2, 16384 bytes)
Apr 17 12:17:01 rpi1 kernel: Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
Apr 17 12:17:01 rpi1 kernel: Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
Apr 17 12:17:01 rpi1 kernel: Memory: 940232K/970752K available (7168K kernel code, 576K rwdata, 2076K rodata, 1024K init, 698K bss, 22328K reserved, 8192K cma-reserved)

```

One thing that is noticeable is how the kenrnel command lines are differ in their `vc_mem_size` parameter: 
- openwrt:         vc_mem.mem_size=0x10000000   (that is 256 M)
- raspbian-lite:  vc_mem.mem_size=0x40000000   (that is 1 G)

In fact `vc_mem` is supposed to be Video Codec memory, but the parameter value appears to be (estimated) memory.  (Why would 1G be passed as memory for Video Codec? - so it seems likely that despite the name it is indicating total memory.)

It seems the kernel command line values are determined by `/boot/bootcode.bin` and/or `start.elf`.  Those are Broadcom proprietary software, so there is no easy way to figure out exactly what is going on. 

Any assistance or advice would be appreciated.

craigphicks, 2019-04-20:
The problem was that newer versions of these Broadcom proprietary files changed but didn't support back compatibility:

- bcm2710-rpi-3-b.dtb
- bcm2710-rpi-3-b-plus.dtb
- bootcode.bin
- fixup_cd.dat
- fixup.dat
- start_cd.elf
- start.elf

I changed to a version dated June, 2018.  Unfortunately there is no version control to make it clear when a new version set will break existing kernels.  Also there is no documentation about this particular change at all.

6by9, 2019-04-21:
Sorry, this appears to be an issue with openwrt, not the RaspberryPi kernel.
The last pull from upstream I can see in our rpi-4.9.y kernel branch is 4.9.80 in Feb 2018. At that point efforts then moved on to 4.14 (and are now about to move on to 4.19). Where they manage to build a 4.9.152 kernel from is therefore their problem.

Version control for all firmware releases is in https://github.com/raspberrypi/firmware.
bcm2710-*.dtb are the compiled versions of https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/bcm2710-rpi-3-b.dts and counterparts.

I suspect that the firmware you have installed predates the release of the Pi3B+, therefore the default behaviour is to assume an original Pi model B which had 256MB. ```vcgencmd version``` should report the firmware version and build date.
We do work very hard to ensure backwards compatibility, but there is no way to support new hardware without updating the firmware as parameters such as SDRAM timings and GPIO allocations can change.
And actually retaining backwards compatibility without duplicating large blocks of code is hard!

andrum99, 2019-04-21:
Just to clarify: the correct place to report this problem is the openwrt bug tracker. (Also, as you have discovered: it's not actually a problem in Linux, but in the firmware).

craigphicks, 2019-05-04:
Thank you for the very detailed "beyond the call of duty" answer and your are correct.  I am marking this issue as closed.

closed 2019-05-04 by craigphicks