raspberrypi/linux #2942: ax88179_178a.c  need to update to solve the USB audio streaming.

allocom, 2019-04-25:
We done extensive test with ax88179_178a.c  ASIX USB ethernet module and noticed USB Audio distortion issue, then contacted ASIX and got updated driver , Can we commit the update driver through this github, please suggest.
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Add a clear and concise description of what you think the bug is.

**To reproduce**
List the steps required to reproduce the issue.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

pelwell, 2019-04-25:
ax88179_178a.c is an unmodified upstream driver. Updates should be sent upstream, ideally by ASIX themselves.

AlloKatana, 2019-04-25:
Hi pelwell 

   The linux driver that Asix is providing upstream is tailored to x86 systems. The new driver was  done for ARM CPUs. We are in touch with Asix team and we can get them to commit the driver themselves. The question is , can we ( actually asix ) have a optimized driver for ARM soc ?

pelwell, 2019-04-25:
In what way is it optimised for ARM CPUs?

AlloKatana, 2019-04-25:
I have asked Asix for details . They will provide them tomorrow .

P33M, 2019-05-03:
I am also intrigued.

AlloKatana, 2019-05-03:
We are in continuous discussion with Asix , they have asked for some time to complete testing

macmpi, 2019-05-07:
@allocom does it still happen with [latest rpi-update](https://github.com/Hexxeh/rpi-firmware/commit/18e0a0f9a31e7a3a47d9c4301c7705b980ab0516) firmware?

AlloKatana, 2019-05-31:
Hello , Asix is very unresponsive regarding the driver submission to Rasbian . 
 (even though they made a new driver)

In our testing (with with 5 different dongles ) a major improvement is found. Speeds are higher , there is no issues with USB audio.

  Can we submit ourselves ?

P33M, 2019-05-31:
Assuming you have distribution rights, then please do so.

allocom, 2019-06-05:
@P33M  initiated new pull request (#2999 ).

pelwell, 2019-06-05:
Thanks - we are aware of it. The new driver doesn't use the usbnet framework, so comparing it with the original is not going to be straightforward.

pelwell, 2019-09-16:
Both my previous comments still stand:
> ax88179_178a.c is an unmodified upstream driver. Updates should be sent upstream, ideally by ASIX themselves.

> The new driver doesn't use the usbnet framework, so comparing it with the original is not going to be straightforward.

Being so different from the upstream driver makes it more likely that a future upstream change will break the downstream patch, hence our reluctance to merge it here.

seniorgod, 2019-10-29:
Dear all, 

sorry for my question. Some people i know want to buy the new allo USBSig. But because we don't want to buy hardware without any proper software support, we want to know if there is a chance for that pull request to be merged into kernel soon. Is it?

Thanks for your answer

pelwell, 2019-10-29:
Until Allo or Asix come up with a patch that has a chance of being accepted upstream, the answer no.

JamesH65, 2020-02-13:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

drewdogg, 2020-03-12:
I came across this issue in my search for a fix to improve the throughput on my USB Ethernet adapter using the ASIX chipset connected to my Pi2B.  Would it be feasible to distribute this "improved" driver as an optional package instead?

popcornmix, 2020-03-12:
Not by us, no.

drewdogg, 2020-03-12:
OK, thank you for your prompt reply!  I will try to make some headway to narrow down a minimal patch/fix that would have a better chance at being accepted upstream.  This is really specific to operating in USB 2.0 High Speed mode.  Super Speed mode works fine on other USB 3.0 (or whatever they are calling it now) hosts.

MichaIng, 2023-05-06:
Is this still an issue? There have been a bunch of commits 2022 upstream: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/log/drivers/net/usb/ax88179_178a.c
And @allocom @AlloKatana you do not provide those driver builds for recent RPi kernels anymore, but at least there was no report about broken audio quality done to us.

seniorgod, 2023-05-06:
I still have problems with my allo and sound output via usb… Most of the time I use an older raspi because it works without problems. If I remember correct, there is no new kernel from allo including those patches. Correct?

MichaIng, 2023-05-06:
> If I remember correct, there is no new kernel from allo including those patches.

No, there never was. ASIX updated the driver to better support ARM in general and Allo produced kernel module builds based on the changed sources for RPi. But they do not build them anymore, it seems. Probably rebasing the commits onto Linux 6.1 leads to conflicts (maybe (some of) the commits from last summer that I linked) which need to be resolved, being the reason why no updated driver builds are provided anymore.

A pity that ASIX seems to not send the update upstream, so this issue will never really be solved and requires ongoing efforts. Would be good indeed if someone with knowledge would identify the minimal required change to fix the audio issue and sent this upstream. Not sure how good the changes are if not ASIX itself does it, but otherwise at least there is a patch which can be easier rebased onto newer Linux versions. I could then help to setup a GitHub Actions workflow to have the kernel modules built automatically and host them as GitHub releases.

inca21, 2023-11-20:
Hello I have the Usbridge Signature from Allo and the drivers cannot be updated anymore. So the cracks and distortion in the sound came back. It seems that Allo shut down and the sudo command I was using doesn't work. Have you found a way to fix that ?

MichaIng, 2023-11-20:
Based on the error log from [here](https://audiophilestyle.com/forums/topic/57258-allo-usbridge-signature/page/55/#comment-1225275), it is simple to fix:
Here: https://github.com/raspberrypi/linux/pull/2999/files#diff-0ac966c0229e99d07523378c08151f97e24c8406d087714de6eb7248eb41f42eR2697
```c
netif_napi_add(netdev, &dev->napi, ax88179_poll, AX88179_NAPI_WEIGHT);
```
=>
```c
netif_napi_add(netdev, &dev->napi, ax88179_poll);
```
required as of this upstream change: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?id=b48b89f9c189d24eb5e2b4a0ac067da5a24ee86d
```
However, this shows quite well why RPi maintainers are not keen to add 3rd party drivers, respectively such which were not upstreamed, to their own branch. When not in upstream code, the maintenance burden would remain at RPi, who do not have a direct commercial interest in it.

What do you mean by "Allo shut down"? The company does practically not exist anymore? That would explain why they never added builds for this driver since Linux 6.1 was released, even that the fix is trivial.

inca21, 2023-11-21:
Thanks Michalng, yes I had confirmation from the my local shop who sells Allo products. They stopped their activity :/
I don't have dietpi but volumio, if you can find an easy process to do so it'll be much appreciated ! I'm really not a dev, I can copy paste in the terminal but I have no idea what I'm doing :)
The usbridge sig is a very fine device and other than that perfectly working