raspberrypi/linux #2945: File gets blocked on a git push

Dark-Sky, 2019-04-28:
With the rpi-5.1.y branch in the .gitignore file *.dtb* prevents the scripts/Makefile.dtbinst file from getting pushed to a repo.

lategoodbye, 2019-05-19:
Please report this issue to linux-kbuild@vger.kernel.org and the relevant maintainers.

popcornmix, 2019-05-21:
Actually that is a downstream change.
@pelwell we have:
```
diff --git a/.gitignore b/.gitignore
index 97ba6b79834c..d99301c57cff 100644
--- a/.gitignore
+++ b/.gitignore
@@ -15,7 +15,7 @@
 *.bin
 *.bz2
 *.c.[012]*.*
-*.dtb
+*.dtb*
 *.dtb.S
```
I guess this is to ignore .dtbo files? Should we drop it or change to:
```
*.dtb
*.dtbo
```

pelwell, 2019-05-21:
Adding `*.dtbo` explicitly is better.

popcornmix, 2019-05-21:
Added to 4.19, 5.0 and 5.1 trees.
@Dark-Sky does this solve your issue?

Dark-Sky, 2019-05-21:
It looks good to me I had temporarily done this here because I have seen in the past some .dtb's with an extension for boards with different revisions and did not remember if that was true for .dtbo's also:

!scripts/Makefile.dtbinst

Thanks!

closed 2019-05-21 by popcornmix