raspberrypi/linux #2952: On-board audio device not found - 4.19.37

moodeaudio, 2019-05-01:
**Describe the bug**
Bumping from 4.19.34 to 4.19.37 via rpi-update results in Pi On-board audio device not being found.

**To reproduce**
* Stretch Lite 2018-11-13
* Pi 3B+
* Start with kernel 4.19.34 then bump to 4.19.37 via rpi-update
* Run commands amixer and aplay -l

**Expected behaviour**
pi@moode:~ $ amixer
Simple mixer control 'PCM',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback -10239 - 400
  Mono: Playback 400 [100%] [4.00dB] [on]

pi@moode:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

**Actual behaviour**
pi@moode:~ $ amixer
amixer: Mixer attach default error: No such file or directory

pi@moode:~ $ aplay -l
aplay: device_list:270: no soundcards found...

**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi-3B+

* Which OS and version (`cat /etc/rpi-issue`)?
pi@moode:~ $ cat /etc/rpi-issue 
Raspberry Pi reference 2018-11-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7e0c786c641ba15990b5662f092c106beed40c9f, stage2

* Which firmware version (`vcgencmd version`)?
pi@moode:~ $ vcgencmd version
Mar 27 2019 15:45:53 
Copyright (c) 2012 Broadcom
version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start)

* Which kernel version (`uname -a`)?
pi@moode:~ $ uname -a
Linux moode 4.19.37-v7+ #1214 SMP Tue Apr 30 17:20:43 BST 2019 armv7l GNU/Linux

**Logs**

```
pi@moode:~ $ lsmod
Module                  Size  Used by
brcmfmac              311296  0
brcmutil               16384  1 brcmfmac
sha256_generic         20480  0
cfg80211              614400  1 brcmfmac
bcm2835_codec          36864  0
bcm2835_v4l2           45056  0
v4l2_mem2mem           24576  1 bcm2835_codec
[... 289 lines cut ...]
[    4.328341] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[    4.428501] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[    4.428513] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[    4.431881] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[    4.431897] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[    4.435490] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[    4.435502] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[    5.536260] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    5.536291] brcmfmac: power management disabled
[    5.745414] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    6.617542] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
```

**Additional context**
pi@moode:~ $ cat /boot/config.txt
disable_splash=1
hdmi_drive=2
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=audio=on
#dtoverlay=pi3-disable-wifi
#dtoverlay=pi3-disable-bt

pi@moode:~ $ cat /boot/cmdline.txt
net.ifnames=0 dwc_otg.lpm_enable=0 console=serial0,115200 console=tty1 root=PARTUUID=7ee80803-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait

6by9, 2019-05-02:
I think we've lost the reverts on switching bcm2835-audio to being a platform driver. My tree from before the latest rebase included
```
commit a5fa30fd15bb368b0a5d7e622921e919ffe40f0e
Author: Dave Stevenson <dave.stevenson@raspberrypi.org>
Date:   Tue Dec 4 20:41:19 2018 +0000

    Revert "staging: bcm2835-audio: Drop DT dependency"
    
    This reverts commit 933bc853bb764e476b0b0f633588f46d20f1f76a.
    
    Signed-off-by: Dave Stevenson <dave.stevenson@raspberrypi.org>

commit 52a9a321d867e44248ef98d98c1d7cadcedc1c41
Author: Dave Stevenson <dave.stevenson@raspberrypi.org>
Date:   Tue Dec 4 19:40:12 2018 +0000

    Revert "staging: vchiq_arm: Register a platform device for the audio driver"
    
    This reverts commit ab59590ed562b89db51fe46cee5db96b9bc5abd8.
    
    Issues have been observed in LibreElec as this was unconditionally
    loading the audio driver instead of having the DT parameter to
    enable it.
    
    Includes a partial revert of 2147700eb7a1b9e55e0684f0749114ce35d61571
    which fixed up the error handling.
    
    Signed-off-by: Dave Stevenson <dave.stevenson@raspberrypi.org>
```

popcornmix, 2019-05-02:
Linux tree does now include "Revert "staging: bcm2835-audio: Drop DT dependency""
which wasn't present in last rpi-update build. I believe that restores audio (when new rpi-update kernel is pushed).

popcornmix, 2019-05-02:
Can you test again after an rpi-update?

moodeaudio, 2019-05-02:
That fixed it :-)

closed 2019-05-02 by moodeaudio