raspberrypi/linux #2958: Cherrypick important w1 fixes

manio, 2019-05-10:
Hi,
@pelwell, can you please cherry-pick those three fixes which are upstreamed already?
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/w1?id=49695ac46861180baf2b2b92c62da8619b6bf28f
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/w1?id=62909da8aca048ecf9fbd7e484e5100608f40a63
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/w1?id=5cb27d30fc3a281e830a2099d520b469e2b82008

It is working wrongly (w1 protocol errors) without this in certain circumstances (eg single DS2408 slave on w1 bus).

pelwell, 2019-05-15:
Hi @manio - sorry for the delay. I've merged those patches into rpi-4.19.y, but beware - I think the first patch contains a typo. Where the code says:
```
#ifdef fCONFIG_W1_SLAVE_DS2408_READBACK
```
Note the `f` in front of `CONFIG_`. I think it should say:
```
#ifdef CONFIG_W1_SLAVE_DS2408_READBACK
```
Watch out for (or create - I think you are the original author) an upstream fix for this (if indeed it is an error) and we can cherry-pick that as well.

manio, 2019-05-16:
@pelwell 
Thank you! You are damn right about this typo! At least three developers was reviewing this patch and didn't spot this until you! :)
Yes, I am the original author but then it was reworked by Jean-Francois Dagenais several times. As it is now commited in main kernel tree I just created another patch for fixing this as we cannot prepare another version. Big thanks!
https://lore.kernel.org/lkml/20190516120116.22262-1-manio@skyboo.net/T/#u

Regarding this patch here in the rpi kernel - please do what is more proper in your development flow. Please either fix this typo yourself during/after cherry-picking or please just apply my fix. I hope it will be merged upstream without any objections but it will surely take some time...

regards!

pelwell, 2019-05-16:
For back-ports we like to keep the same number of commits as appear upstream so that they are skipped without conflicts when we port our downstream patches to an upstream kernel that already includes them. If it's alright with you I'll wait until your fix is accepted, if only to get a commit hash for it.

BTW Your patch should really have a "Fixes: " tag to make it clear it isn't a new feature and to ensure it is replicated everywhere the original patch ends up.

manio, 2019-05-16:
I see. I've just posted PATCH v2 with the "Fixes" tag. I am now closing this issue and I will reply here with the commit hash when it will upstreamed :)
Thanks!

pelwell, 2019-05-24:
Fix cherry-picked - see 1694cf06.

afullagar, 2019-05-31:
Having an issue with a temp sensor /sys/bus/w1/devices/28-0417c1d913ff/w1_slave (
DS18B20)

Had to revert to 4.14.98 for now as the temp reading stays at 86 degrees celsius. After roll back now ok.

manio, 2019-05-31:
@afullagar 
Why are you thinking that this is related to this issue. This changes doesn't touch the w1_therm at all...

manio, 2019-06-23:
@pellwell
Please also cherry-pick those ones (mainly fixes for my DS2413 module):
https://git.kernel.org/pub/scm/linux/kernel/git/gregkh/char-misc.git/commit/drivers/w1?h=char-misc-next&id=ae2ee27aa985232f66421d7cd1c7f4b87c7dba7d
https://git.kernel.org/pub/scm/linux/kernel/git/gregkh/char-misc.git/commit/drivers/w1?h=char-misc-next&id=c50d09a86172073f55ebac0b92ad5a75907d64e7
https://git.kernel.org/pub/scm/linux/kernel/git/gregkh/char-misc.git/commit/drivers/w1?h=char-misc-next&id=3856032a0628e6b94badb9131a706dda185e071d
https://git.kernel.org/pub/scm/linux/kernel/git/gregkh/char-misc.git/commit/drivers/w1?h=char-misc-next&id=aacd152ecd7b18af5d2d96dea9e7284c1c93abea
https://git.kernel.org/pub/scm/linux/kernel/git/gregkh/char-misc.git/commit/drivers/w1?h=char-misc-next&id=0e3743d870711ae4daf1e7170c8d9381564e244d

pelwell, 2019-06-25:
Cherry-picked to the 4.19 tree.

manio, 2019-06-25:
Great! Thank you! :)

closed 2019-06-25 by manio