raspberrypi/linux #2959: USB serial outputs strings from system's memory to ttyUSB ports

AntonioSanta, 2019-05-10:
**Describe the bug**
I have Raspberry Pi 3 model B Kernel version 4.14.13. I connected an USB HighSPeed hub and 4 USB to doubleUART CP2105 converters attached to it. at the UART end side nothing talks as endpoint devices are switched off.
I have a simple cpp app that uses usb serial driver system calls to open() and then write() every 10ms always the same message. Message is just a simple sequence of 0x0D bytes , length is 1024 bytes but I observed same effects using shorter length, 21 bytes.
I have a Beagle USB monitoring the messages , It shows that sometimes messages are corrupted by some bytes that are not supposed to be there. At first sight these bytes are random, but inspecting later their content they seem to be revealing some on device's memory content as when observed in text mode some strings make sense and are related to root certificates folder onboard (!), time of the day or other info possibly, there are also not clearly meaningful other hex strings.
When opening ttyUSB com ports I check result returned is OK. So I am expecting then no other driver being able to use and overwrite on that port.
This is confirmed because  I don't see anything on the log from the Beagle analyzer if I just open the com ports without calling write(). So it seems calls to write() are needed to trigger these strings to come out.
When I call the write() function number of bytes reported to be written are as expected.
I use a statically allocated byte buffer for each of the com ports. I set their content at the startup ( all bytes are set to  0x0D ) of the app and then app code does not modify their content anymore. No moving pointers are used at app level.
I printout to a log from the app the content supposed to be written and this is never wrong.
You can see the strings if you open the .bin files dumps with notepad  and scrolling down the file as the corrects byte sequences are 0x0d adding just newlines chars.otehrwise you can use hex editor to inspect the content.

**To reproduce**
connect at least 2 USB to uart converters of the cp2105 type. these converters have a double endpoint as they have 2 uarts, so at least 4 endpoints in total. With lower number of devices connected I haven't observed the effect I describe here.
These devices are connected to the USB connector of the PI through a USB Highspeed external hub ( see output from usb-devices command attached ).

**Expected behaviour**
On the USB analyzer logger I expect to see just the original data output from the app.
This happens correctly if I use another "master device" ( my PC)  instead of Raspberry PI.

**Actual behaviour**
I see usually the correct content if I inspect the log from the USB analyzer , but sometimes content is corrupted by strings that sometimes can be identified as meaningful of some of the system's memory.

**System**
Aug 16 2018 17:30:30 
Copyright (c) 2012 Broadcom
version 31e0613622dc2f2463bf3dd74e6c897d91201a4d (clean) (release)
Raspberry Pi reference 2018-03-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 00013d7972122d1304aacda8fff5098f073ceb43, stage5
Linux devpi 4.14.62-v7+ #1134 SMP Tue Aug 14 17:10:10 BST 2018 armv7l GNU/Linux

similar results with another RaspberryPi3 modelB with Yocto generated image , and Kernel version 4.14.13.

I can see there is an ongoing fix on the otg driver that might be related to this and an open issue that could be related but from its
description I am not sure :  ( [https://github.com/raspberrypi/linux/issues/2140](url) )

[RaspberryPiSystemInfo.txt](https://github.com/raspberrypi/linux/files/3166014/RaspberryPiSystemInfo.txt)
[image]
[usb_devices.txt](https://github.com/raspberrypi/linux/files/3166017/usb_devices.txt)

[dumps.zip](https://github.com/raspberrypi/linux/files/3166012/dumps.zip)

P33M, 2019-05-10:
From the dumps provided, it looks like your 4-port hub is a single-TT hub (Alt setting of 0 => single-TT). Using a single-TT hub with that many endpoints is going to be very suboptimal. 

From your dumps, it looks like multiples of 32 bytes get corrupted which smells like a cache-related issue. 

- Can you post a full lsusb -v output?
- Does the issue go away with dwc_otg.fiq_fsm_mask=0x6 in cmdline.txt?

AntonioSanta, 2019-05-10:
- Can you post a full lsusb -v output?
file attached here below
[lsusb.txt](https://github.com/raspberrypi/linux/files/3166644/lsusb.txt)

- Does the issue go away with dwc_otg.fiq_fsm_mask=0x6 in cmdline.txt?
yes it does! at least apparently it does. I cannot reproduce the issue after adding that line. I am running a test now but no corrupted data output so far.
what does this modification mean ?

P33M, 2019-05-10:
The mask setting changes which types of USB transfer are handled by FIQ code and which ones aren't. Bit 0 of the mask corresponds to bulk/control split transactions. Cache effects are usually time-dependent - by changing the mask a different code path is taken which means there is an implicit change in timing between when a buffer is written to by application code, and the buffer's physical address being fed to the USB controller hardware.

URBs should be presented to the host controller driver in the correct state, i.e. should have called dma_sync_*_for_device()/usb_alloc_coherent() and set URB_NO_TRANSFER_DMA_MAP, or rely on the host controller driver to do the mapping. If either the device driver or the HCD is doing this wrong, then you will get random stale data in main memory as the actual data is still in the cache. Given that it's very infrequent, I suspect that there's a misuse of a coherent buffer somewhere.

AntonioSanta, 2019-05-10:
thanks, that makes sense, this flag then allows to work around the problem, at the moment.
I will keep the test running for a while.
You wrote:
_URBs should be presented to the host controller driver in the correct state, i.e. should have called dma_sync_*_for_device()/usb_alloc_coherent() and set URB_NO_TRANSFER_DMA_MAP, or rely on the host controller driver to do the mapping. If either the device driver or the HCD is doing this wrong, then you will get random stale data in main memory as the actual data is still in the cache. Given that it's very infrequent, I suspect that there's a misuse of a coherent buffer somewhere._

For curiosity In which file does this URB/buffer exchange happens ?
I see there are changes reported here 
[(https://github.com/raspberrypi/linux/commit/74845608962cb535232878e81aee7c699639ab89)]
that talk about memory corruption but I guess I have already these on the build I am using, apart from the modifications done very recently.

P33M, 2019-05-13:
Strictly speaking it's not the host controller driver itself, but the layer above -
https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/usb/core/hcd.c#L1500

- What USB-serial hardware are you using and where can I buy a representative device?
- Can you post your test script - as the corrupt data will be present in the USB packet stream I won't need the device at the other end of the UART.

AntonioSanta, 2019-05-14:
_What USB-serial hardware are you using and where can I buy a representative device?_
I am using CP2105 dual UART converter. 
[https://www.silabs.com/products/interface/usb-bridges/classic-usb-bridges/device.cp2105](url)
the two UART-USB endopoints in each of these converters are slightly different, the buffer reserved for USB is different in size.

_Can you post your test script - as the corrupt data will be present in the USB packet stream I won't need the device at the other end of the UART._
attached below is a simplified version of the test. I checked and it reproduces the problem when I remove that   _dwc_otg.fiq_fsm_mask=0x6_ line you told me to add to cmdline.txt.
There are 8 threads , one for each of the ttyUSB ports. 
Every thread opens a port and the writes to it 1024 bytes , then sleep for 10 ms ( but can be 1 ms as well ) and then loops again to repeat writing. I haven't added here code to close ports on exit.
Could it be the app or some settings when building it causing this ? I build this by VisualStudio 2017 but I have used this environment for a while without seeing other problems like this at runtime.

thank you

[main.zip](https://github.com/raspberrypi/linux/files/3176841/main.zip)

P33M, 2019-08-02:
Time to dig this one up again - @AntonioSanta does the issue happen with both USB-UART adapters plugged into the Pi, or do the affected devices need to be attached to the single-TT hub?

AntonioSanta, 2019-08-22:
the issue happened whichever USB the device is plugged in, also directly to the PI USB sockets, it was not only on the single-TT hub plugged in converters. I cannot do further tests at the moment.

P33M, 2020-01-14:
Please re-test with latest rpi-update firmware.

AntonioSanta, 2020-02-06:
sorry but I cannot test this as I don't have this device setup anymore, changed company.

P33M, 2020-02-06:
In that case, I am closing this issue as nobody else has reported this symptom and it's going to be impossible to progress the diagnosis. If someone else with a CP2105 usb-serial chipset can duplicate the fault seen here, then I will reopen it.

closed 2020-02-06 by P33M