raspberrypi/linux #2963: TFTP/NFS booted Pi 3 Model B+ freezes when using official V4L2 driver

jorisvervuurt, 2019-05-11:
When booting a Raspberry Pi 3 Model B+ via TFTP and NFS (no SD card), there seems to be an issue with the official V4L2 driver. When trying to access `/dev/video`, the entire Pi locks up.

I have the following setup:
- Ubiquiti EdgeRouter X
- Synology DiskStation DS715
- Raspberry Pi 3 Model B+

I have configured a number of VLANs on the EdgeRouter:
- Synology DiskStation DS715 is on the 192.168.10.x subnet (IP 192.168.10.6)
- Raspberry Pi 3 Model B+ is on the 192.168.40.x subnet (IP 192.168.40.2)

I use a clean install of the latest Raspbian Stretch Lite image.
The Synology NAS acts as a TFTP server (for the boot files) and NFS server (for the root filesystem). `bootcode.bin` is in the TFTP root directory, the rest of the boot files are in a `<serial>/` directory.
I have configured the Ubiquiti router’s DHCP server to add option 43 and provide the correct IP to the TFTP server. Booting the Pi works perfectly fine without any errors. I’ve edited `/etc/fstab` to mount the `<serial>/` boot file directory as `/boot` (this allows me to edit the boot files from the network-booted Pi).

I have added `bcm2835-v4l2` to the `/etc/modules` file.
After a reboot, `/dev/video0` is available, so it appears the driver has loaded correctly.

I use `v4l2rtspserver` to serve an RTSP stream of `/dev/video0`, which works perfectly fine when the Pi is booted from an SD card (both the boot files and the root filesystem).
As soon as I switch to full TFTP and NFS boot without an SD card, executing `v4l2rtspserver` completely locks up the Pi. I have confirmed that this is not a permissions problem.
This is where things become weird, because `raspivid` seems to work just fine. After running `raspivid`, the `v4l2rtspserver` command works just fine as well.

I’ve tried running `sudo rpi-update` to update to the bleeding-edge boot files, but unfortunately that hasn’t fixed the issue.

I then came across these two issues on GitHub:
https://github.com/raspberrypi/linux/issues/2335
https://github.com/raspberrypi/linux/issues/2528

So, I created `bcmhi.c` with the following content:
```
#include "bcm_host.h"

int main(int argc, char* argv)
{
  bcm_host_init();
  return 0;
}
```

I then compiled it using `sudo gcc bcmhi.c -I/opt/vc/include -L/opt/vc/lib -lbcm_host -o /usr/local/bin/bcmhi`. When I executed `bcmhi && v4l2rtspserver`, the Pi no longer locks up and everything works as it should.

**So, in short, the issue is that `bcm_host_init()` needs to be called before `/dev/video0` is usable. However, the weird thing is that this is only needed when the Pi is booted from TFTP/NFS.**

Any ideas what this could be?

6by9, 2019-05-11:
As requested in the template for issues, please provide your kernel and firmware version.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

The fix for #2335 and #2528 was merged May 2018 so I was believing the issue was resolved.

jorisvervuurt, 2019-05-11:
Thanks for your reply.
Here’s the info you requested:

- Which model of Raspberry Pi?
Pi3B+

- Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-04-08
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, e910bf55ac92e27c6e77043e8995bcb5e0a2d742, stage2

- Which firmware version (`vcgencmd version`)?
Mar 27 2019 15:48:59 
Copyright (c) 2012 Broadcom
version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start_x)

- Which kernel version (`uname -a`)?
Linux rpi-at 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

The issue as mentioned in the other threads indeed seems to be fixed, as loading the module (either via `modprobe` or via `/etc/modules`) no longer locks up the Pi. This issue however, seems to be somewhat related to the other two issues (same `bcm_host_init()` workaround)...

Thanks again!

vintozver, 2019-08-14:
Having the same issue on both 3B and 3B+

alkisg, 2020-01-19:
I too have the same issue on both rpi2 and rpi3b+, but not on rpi4.
I commented on the "original thread" at https://github.com/raspberrypi/linux/issues/2335#issuecomment-576011598.

JamesH65, 2020-02-13:
@alkisg Are you using the latest Raspbian Buster?

alkisg, 2020-02-13:
@JamexH65, I had the latest Raspbian Buster when I commented 25 days ago.

If I recall correctly, there was a kernel update since then, do you want me to test with that one too?

JamesH65, 2020-02-13:
So still present with that Buster then, might be worth am apt full-upgrade but suspect it's still broken. Thanks for the report.

alkisg, 2020-02-14:
@JamesH65, I just ran `apt full-upgrade` in my Buster installation and I **cannot reproduce this anymore** with the latest 4.19.97-v7+ kernel.
I tested 3 boots on an rpi3b+ and 2 boots on an rpi2. They booted fine over NFS-RW without having to specify `modprobe.blacklist=bcm2835_v4l2` in cmdline.txt.

Thanks a lot, seems like problem solved; hopefully the others will confirm it as well.

JamesH65, 2020-02-14:
Great, will give it a few days, if nothing bad turns up, will close.

closed 2025-10-19 by jorisvervuurt