raspberrypi/linux #2964: Downloads Stall on 4.19.40 - smsc95xx driver.

paul-1, 2019-05-11:
**Describe the bug**
We use Cloudflare CDN on our site.  Downloads from the Cloudflare servers stall since #2950.  Downloads from other locations that I've tested do not seem to be affected.
**To reproduce**
Use wget to download a file, stalls typically occur before 1MB of data is downloaded.  Any larger file served from Cloudflare will stall (100% of the time by my tests), here is a test link.  https://repo.picoreplayer.org/insitu/piCorePlayer4.1.0/piCorePlayer4.1.0.zip

**System**
* Which model of Raspberry Pi?
This only affects rpi B, B+, 2B, 3B boards using the ethernet connection.   3B+, Wifi, or other network devices are not affected.

* Which OS and version (`cat /etc/rpi-issue`)?
Fails: 
Raspberry Pi reference 2019-04-08
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, e910bf55ac92e27c6e77043e8995bcb5e0a2d742, stage4

Working: 4.19.37 kernel provided by, rpi-update 6cf2788e74b497c109865d052cd6fa68d094cf38 

* Which firmware version (`vcgencmd version`)?
Mar 27 2019 15:45:53
Copyright (c) 2012 Broadcom
version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberrypi 4.19.40-v7+ #1217 SMP Tue May 7 17:22:08 BST 2019 armv7l GNU/Linux

bulletmark, 2019-05-11:
I can confirm this serious bug. I update my Arch ARM Raspberry Pi every day and saw this bug introduced on the `2019-05-03 linux-raspberrypi 4.19.37-1 -> 4.19.37-2` [update](https://archlinuxarm.org/packages/armv7h/linux-raspberrypi/log?id=4daa98bc0dcb219b1f67f46abf968dccfa85e267).

When `4.19.40-1` was updated a few days later I checked and saw it still had the bug. I run a small web server on the Pi and the bug causes the served web pages to stall and lock up. Accessing the web server on the local network is fine, accessing it via the internet via some routes causes the lockup.  E.g. accessing the PI from my laptop via my phone internet network always locks up but proxying my browser from there via my internet VPS ssh server would always work(?!). I had not yet determined a pattern and had not seen anybody else complaining so was hesitant to raise an issue although I confirmed that the bug occurred on multiple model 2B RPi and using multiple different web servers and web frameworks. I switched back and forward many times between many kernel versions to absolutely confirm the bug was introduced on the `4.19.37-1 -> 4.19.37-2` update.

tnikitin, 2019-05-12:
Same when updating the PHP packages from the Sury.org repo.

Running curl https://packages.sury.org/php/pool/main/p/php7.3/php7.3-cli_7.3.5-1%2B0~20190503093827.38%2Bstretch~1.gbp60a41b_i386.deb will stall after about 180kb.

Reverted to the stock kernel (4.14.98) and all issues were resolved.

Doridian, 2019-05-13:
Could possibly be related to (or the same issue as) what I've seen: https://github.com/raspberrypi/linux/issues/2965

P33M, 2019-05-13:
> Could possibly be related to (or the same issue as) what I've seen: #2965
Different hardware. The 3B+ uses a lan7800-derived controller, the lan951x is used on Pi1-3B.

I've been able to replicate this locally - it looks like outbound tx checksums get corrupted in specific situations. Given that we've gone and changed the overall length of a transmit packet, I wonder if the somewhat mysterious code here 
https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/net/usb/smsc95xx.c#L2103
is now failing for "short" packets.

P33M, 2019-05-13:
Even more mysterious, in 5.2:
https://github.com/torvalds/linux/blob/master/drivers/net/usb/smsc95xx.c#L2012

P33M, 2019-05-13:
I've spent some time unpicking this and there's bugs sitting on top of more bugs.
There is a regression in the latest rpi-update kernel.
1. TCP ack packets sent with no trailing options have the checksum in the last 4 bytes of the packet
2. It appears that Cloudflare doesn't specify any TCP options, so minimum-length frames of 54 bytes are sent to the hardware
3. Many other sites end up with a TCP timestamp option appended, so the checksum is now offset from the last 4 bytes.
4. The regression doesn't appear on v6 networks. The checksum is still in the last 4 bytes of the packet - but v6 checksumming is disabled due to _another_ bug with UDP checksum processing.

The comment in 4.19 is wrong - it's not the length that matters, it's the existence of the checksum in the last 4 bytes of the frame to be transmitted. It just so happens that for TCP and UDP packets, the checksum is in the header and so doing the length check automatically strips hardware checksumming out. 5.2's comment is getting closer - but it's the preamble that needs to be aligned (and will be if the datasheet instructions are followed).

That said, I have no idea how tx checksum offload is even working as intended. The datasheet specifies the following command sequence for tx checksum offload:
- Send a command word A/B/checksum preamble sequence with data start offset = 0, FS=1, LS=0, FL=frame_length, buffer size of 4 and tx checksum enable=1
- Send a command word A/B/data sequence with FS=0, frame length identical to the previous transmission and tx checksum enable = 1.

Whereas the driver is just inserting the checksum preamble right before the data. This also begs the question why *any* of our TCP transmit packets work with the current state of the world, as the padding is being inserted between tx_cmd_b and the preamble, not between the preamble and packet data.

I think for now, this needs a revert and more in-depth investigation.

JamesH65, 2019-05-13:
Might be worth reporting this upstream, the Microchip guys are active on netdev IIRC.

bulletmark, 2019-05-15:
The Arch `linux-raspberrypi` kernel package update today [includes this revert commit](https://archlinuxarm.org/packages/armv7h/linux-raspberrypi/log?id=5f5a74a5733bd6233fb5589bab6b893d5a067693) and I can confirm the problem is now gone. Thanks.

tnikitin, 2019-05-16:
Same here, updated to the latest version and the problems is gone. Thanks!

closed 2019-05-24 by P33M