raspberrypi/linux #2966: cma_alloc: alloc failed, req-size: 2025 pages, ret: -12

StuartIanNaylor, 2019-05-12:
cma_alloc: alloc failed, req-size: 2025 pages, ret: -12

pelwell, 2019-05-12:
A bit more context would be helpful:

1. Is there any additional hardware attached?

2. What was the Pi doing at the time?

3. Was there any additional indication of what failed?

StuartIanNaylor, 2019-05-12:
I presumed it would be common after an rpi-update its just vanilla pi3 and displayed on boot but part of the kernel log.

May 12 12:30:24 raspberrypi kernel: [    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=bdedf1a2-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
May 12 12:30:24 raspberrypi kernel: [    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
May 12 12:30:24 raspberrypi kernel: [    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
May 12 12:30:24 raspberrypi kernel: [    0.000000] Memory: 939088K/970752K available (8192K kernel code, 627K rwdata, 2164K rodata, 1024K init, 821K bss, 23472K reserved, 8192K cma-reserved)
May 12 12:30:24 raspberrypi kernel: [    0.000000] Virtual kernel memory layout:
May 12 12:30:24 raspberrypi kernel: [    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
May 12 12:30:24 raspberrypi kernel: [    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
May 12 12:30:24 raspberrypi kernel: [    0.000000]     vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
May 12 12:30:24 raspberrypi kernel: [    0.000000]     lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
May 12 12:30:24 raspberrypi kernel: [    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
May 12 12:30:24 raspberrypi kernel: [    0.000000]       .text : 0x(ptrval) - 0x(ptrval)   (9184 kB)
May 12 12:30:24 raspberrypi kernel: [    0.000000]       .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
May 12 12:30:24 raspberrypi kernel: [    0.000000]       .data : 0x(ptrval) - 0x(ptrval)   ( 628 kB)
May 12 12:30:24 raspberrypi kernel: [    0.000000]        .bss : 0x(ptrval) - 0x(ptrval)   ( 822 kB)
May 12 12:30:24 raspberrypi kernel: [    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
May 12 12:30:24 raspberrypi kernel: [    0.000000] ftrace: allocating 26226 entries in 77 pages
May 12 12:30:24 raspberrypi kernel: [    0.000000] rcu: Hierarchical RCU implementation.
May 12 12:30:24 raspberrypi kernel: [    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
May 12 12:30:24 raspberrypi kernel: [    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
May 12 12:30:24 raspberrypi kernel: [    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
May 12 12:30:24 raspberrypi kernel: [    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
May 12 12:30:24 raspberrypi kernel: [    0.000018] Switching to timer-based delay loop, resolution 52ns
May 12 12:30:24 raspberrypi kernel: [    0.000276] Console: colour dummy device 80x30
May 12 12:30:24 raspberrypi kernel: [    0.000295] console [tty1] enabled
May 12 12:30:24 raspberrypi kernel: [    0.000337] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
May 12 12:30:24 raspberrypi kernel: [    0.000353] pid_max: default: 32768 minimum: 301
May 12 12:30:24 raspberrypi kernel: [    0.000694] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
May 12 12:30:24 raspberrypi kernel: [    0.000709] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
May 12 12:30:24 raspberrypi kernel: [    0.001622] CPU: Testing write buffer coherency: ok
May 12 12:30:24 raspberrypi kernel: [    0.002082] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
May 12 12:30:24 raspberrypi kernel: [    0.002731] Setting up static identity map for 0x100000 - 0x10003c
May 12 12:30:24 raspberrypi kernel: [    0.002879] rcu: Hierarchical SRCU implementation.
May 12 12:30:24 raspberrypi kernel: [    0.003667] smp: Bringing up secondary CPUs ...
May 12 12:30:24 raspberrypi kernel: [    0.004500] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
May 12 12:30:24 raspberrypi kernel: [    0.005410] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
May 12 12:30:24 raspberrypi kernel: [    0.006266] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
May 12 12:30:24 raspberrypi kernel: [    0.006386] smp: Brought up 1 node, 4 CPUs
May 12 12:30:24 raspberrypi kernel: [    0.006398] SMP: Total of 4 processors activated (153.60 BogoMIPS).
May 12 12:30:24 raspberrypi kernel: [    0.006404] CPU: All CPU(s) started in HYP mode.
May 12 12:30:24 raspberrypi kernel: [    0.006409] CPU: Virtualization extensions available.
May 12 12:30:24 raspberrypi kernel: [    0.007387] devtmpfs: initialized
May 12 12:30:24 raspberrypi kernel: [    0.019253] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
May 12 12:30:24 raspberrypi kernel: [    0.019492] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
May 12 12:30:24 raspberrypi kernel: [    0.019510] futex hash table entries: 1024 (order: 4, 65536 bytes)
May 12 12:30:24 raspberrypi kernel: [    0.020144] pinctrl core: initialized pinctrl subsystem
May 12 12:30:24 raspberrypi kernel: [    0.021010] NET: Registered protocol family 16
May 12 12:30:24 raspberrypi kernel: [    0.024064] DMA: preallocated 1024 KiB pool for atomic coherent allocations
May 12 12:30:24 raspberrypi kernel: [    0.029834] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[... cut ...]
May 12 12:30:24 raspberrypi kernel: [    0.354041] libphy: Fixed MDIO Bus: probed
May 12 12:30:24 raspberrypi kernel: [    0.354153] usbcore: registered new interface driver lan78xx
May 12 12:30:24 raspberrypi kernel: [    0.354212] usbcore: registered new interface driver smsc95xx
May 12 12:30:24 raspberrypi kernel: [    0.354229] dwc_otg: version 3.00a 10-AUG-2012 (platform bus)
May 12 12:30:24 raspberrypi kernel: [    0.382151] dwc_otg 3f980000.usb: base=(ptrval)
May 12 12:30:24 raspberrypi kernel: [    0.582411] Core Release: 2.80a

pelwell, 2019-05-13:
Thanks for reporting this - see https://github.com/raspberrypi/linux/pull/2970.

jeffersonchua, 2019-06-08:
Same issue on Linux 5.2.0-rc3

   cma: cma_alloc: alloc failed, req-size: 2016 pages, ret: -12

Thanks,
Jeff

tmm1, 2019-08-22:
Can be closed?

jeffersonchua, 2019-08-23:
Yes. Rather seeing the error, I just patch this locally ...
--- linux/drivers/base/dd.c     2019-08-13 17:40:14.967391440 +0800
+++ linux/drivers/base/dd.c     2019-08-13 17:41:20.107392180 +0800
@@ -574,6 +574,7 @@
                break;
        default:
                /* driver matched but the probe failed */
+               if(strcmp(drv->name, "bcm2835_audio"))
                printk(KERN_WARNING
                       "%s: probe of %s failed with error %d\n",
                       drv->name, dev_name(dev), ret);
--- linux/mm/cma.c      2019-08-13 19:41:02.440001040 +0800
+++ linux/mm/cma.c      2019-08-13 19:41:49.710001577 +0800
@@ -479,6 +479,7 @@
        }

        if (ret && !no_warn) {
+               if(strcmp(__func__, "cma_alloc"))
                pr_err("%s: alloc failed, req-size: %zu pages, ret: %d\n",
                        __func__, count, ret);
                cma_debug_show_areas(cma);
--- linux/drivers/staging/vc04_services/bcm2835-audio/bcm2835.c 2019-08-14
17:38:46.682156366 +0800
+++ linux/drivers/staging/vc04_services/bcm2835-audio/bcm2835.c 2019-08-14
17:38:55.922156471 +0800
@@ -300,7 +300,6 @@
        err = of_property_read_u32(dev->of_node, "brcm,pwm-channels",
                                   &numchans);
        if (err) {
-               dev_err(dev, "Failed to get DT property
'brcm,pwm-channels'");
                return err;
        }

My best.,
Jeff

JamesH65, 2020-02-13:
@6by9 Is this still present?

6by9, 2020-02-13:
No, fixed.

JamesH65, 2020-02-13:
Closing this issue as questions answered/issue resolved.

closed 2020-02-13 by JamesH65