raspberrypi/linux #2971: SIGILL/ILL_ILLOPC

weber-software, 2019-05-14:
**Is this the right place for my bug report?**
Unsure. It's related to the kernel but i'm using the mainline 5.1.0 kernel.

**Describe the bug**
The system is unstable. Sometimes the system doesn't correctly boot up, but most of the time user processes fail.
It seems to be related to syscalls. `stress -c 4` is working fine. `find /` fails.

**To reproduce**
1. Take a original raspbian image
2. cross compile a mainline 5.1.0 Kernel ([Dockerfile](https://github.com/raspberrypi/linux/files/3177522/Dockerfile.txt))
3. Copy Kernel + Modules to SD
4. Add to `/boot/config.txt`:
```
kernel=zImage
device_tree=bcm2837-rpi-3-b-plus.dtb
```
5. Boot the RPI and run `strace find /`

**Expected behaviour**
It should just print out a lot of file names

**Actual behaviour**
Either my ssh session just gets closed or the `find`
 process gets killed undetermined with signal `SIGILL`

```
getdents64(6, /* 4 entries */, 32768)   = 112
getdents64(6, /* 0 entries */, 32768)   = 0
close(6)                                = 0
close(4)                                = 0
fcntl64(8, F_DUPFD_CLOEXEC, 0)          = 4
write(1, "/proc/38/net/nfsfs/servers\n", 27/proc/38/net/nfsfs/servers
) = 27
--- SIGILL {si_signo=SIGILL, si_code=ILL_ILLOPC, si_addr=0xb6edbff0} ---
+++ killed by SIGILL +++
Illegal instruction
```

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
I tried it with Pi3B+ and Pi3B

* Which OS and version (`cat /etc/rpi-issue`)?
```
Raspberry Pi reference 2019-04-08
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, e910bf55ac92e27c6e77043e8995bcb5e0a2d742, stage2
```
* Which firmware version (`vcgencmd version`)?
```
Mar 27 2019 15:45:53 
Copyright (c) 2012 Broadcom
version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start)
```
* Which kernel version (`uname -a`)?
`Linux raspberrypi 5.1.0 #1 SMP Tue Jan 1 00:00:00 UTC 2019 armv7l GNU/Linux`

**Logs**

The warning pops up because SMP is disabled by the defconfig.

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.1.0 (root@buildkitsandbox) (gcc version 5.4.1 20160919 (15:5.4.1+svn241155-1)) #1 Tue Jan 1 00:00:00 UTC 2019
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] Memory policy: Data cache writeback
[    0.000000] Ignoring RAM at 0x30000000-0x3b400000
[... 225 lines cut ...]
[    5.259750] uart-pl011 3f201000.serial: no DMA platform data
[    5.280192] bcm2835-power bcm2835-power: Timeout waiting for grafx power OK
[    5.420098] Bluetooth: hci0: BCM: chip id 107
[    5.420424] Bluetooth: hci0: BCM: features 0x2f
[    5.421956] Bluetooth: hci0: BCM4345C0
[    5.421969] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0000
[    5.682263] bcm2835-power bcm2835-power: Timeout waiting for grafx power OK
[    5.776339] bcm2835-power bcm2835-power: Timeout waiting for grafx power OK
[    5.790400] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.791865] bcm2835-power bcm2835-power: Timeout waiting for grafx power OK
[    5.818891] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,3-model-b-plus.txt failed with error -2
[    5.970795] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.991203] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[    6.390017] Bluetooth: hci0: BCM4345C0 (003.001.025) build 0280
[    8.083163] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    8.893594] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SS
[    9.077329] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[    9.176604] bcm2835_audio bcm2835_audio: card created with 8 channels
[    9.176867] bcm2835-power bcm2835-power: Timeout waiting for grafx power OK
[    9.286520] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
```

**Additional context**
I tried different raspberries and different power supplies.
As soon as i change back to the original (rpi) kernel, everything is stable.

pelwell, 2019-05-14:
I'm wondering if it is significant that it fails while reading /proc. Does `strace find /usr` also fail? How about `find /usr`?

weber-software, 2019-05-14:
Its not significant. For me it seems like any program that uses syscalls extensively crashes after a while. `apt-get update` also crashes most of the time, `top` crashes, the `ssh` process crashes. But `find` is the process that crashes the fastest.

weber-software, 2019-05-14:
I'm currently trying different kernel configs. I noticed that different configurations have an influence on how fast the `find` fails. But until now i'm unable to find a configuration that is rock solid and i have no clue why there is this instability.

lategoodbye, 2019-05-15:
Maybe related, there seems to be an issue with the new power driver:
`bcm2835-power bcm2835-power: Timeout waiting for grafx power OK`

Where does the dtb file come from? From 5.1.0 or 4.19?

weber-software, 2019-05-15:
`bcm2837-rpi-3-b-plus.dtb` is from the 5.1.0

lategoodbye, 2019-05-15:
I don't get the timeout with my RPi 3 B+, but i can reproduce a similiar unstable behavior of userspace programs.

lategoodbye, 2019-05-16:
Currently i'm very busy, could you please test the following combinations with your RPi 3:
Linux 5.0 / bcm2835_defconfig
Linux 5.1 / multi_v7_defconfig

weber-software, 2019-05-16:
5.0.16 / bcm2835_defconfig - unstable
5.1.2 / multi_v7_defconfig - stable

lategoodbye, 2019-05-16:
I guess the bcm2835_defconfig doesn't work reliable with RPI 3. Yesterday i tested a recent linux-next with bcm2835_defconfig on a RPi Zero which worked flawless. Since multi_v7_defconfig is actually intended for RPI 2/3 please use this instead and manually enable vc04_services from staging.

weber-software, 2019-05-17:
Okay thank you very much! Maybe i will look into which configuration is exactly causing the issue.

closed 2019-05-17 by weber-software

lategoodbye, 2019-05-17:
Btw mainline issues should be reported here:
linux-rpi-kernel@lists.infradead.org (moderated list)