raspberrypi/linux #2976: Using mailbox property tag 0x00030010 hangs the vcio interface

ptesarik, 2019-05-17:
Any attempt to use property tag 0x00030010 (RPI_FIRMWARE_EXECUTE_CODE) hangs the vcio interface.

```
pi@raspberrypi:~ $ strace /opt/vc/bin/vcmailbox 0x00030010 4 4 0
[…]
open("/dev/vcio", O_RDONLY|O_LARGEFILE) = 3
ioctl(3, _IOC(_IOC_READ|_IOC_WRITE, 0x64, 0x00, 0x04)
```

The process hangs here in 'D' state and cannot be killed. Any attempt to use the vcio interface hangs, even kernel-internal use. I reproduced the hang here with incorrect arguments, but the same thing happens with correct arguments (i.e. 7 words and a valid VPU function address).

With wrong arguments, the kernel should report an error. With correct arguments, the system should execute the function (or maybe rather not, because the interface is available to regular users and can easily crash VCOS).

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Raspberry Pi 3B.

* Which OS and version (`cat /etc/rpi-issue`)?
`Raspberry Pi reference 2019-04-08`
`Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, `
`e910bf55ac92e27c6e77043e8995bcb5e0a2d742, stage2`

* Which firmware version (`vcgencmd version`)?
`Mar 27 2019 15:45:53 `
`Copyright (c) 2012 Broadcom`
`version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start)`

* Which kernel version (`uname -a`)?
`Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux`

**Logs**
```
[  980.797770] INFO: task kworker/0:0:884 blocked for more than 120 seconds.
[  980.797782]       Tainted: G         C      4.14.98-v7+ #1200
[  980.797787] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  980.797794] kworker/0:0     D    0   884      2 0x00000000
[  980.797823] Workqueue: events get_throttled_poll
[  980.797856] [<807a2ed0>] (__schedule) from [<807a3538>] (schedule+0x50/0xa8)
[  980.797874] [<807a3538>] (schedule) from [<807a39b0>] (schedule_preempt_disabled+0x18/0x1c)
[  980.797895] [<807a39b0>] (schedule_preempt_disabled) from [<807a52b8>] (__mutex_lock.constprop.3+0x190/0x58c)
[  980.797913] [<807a52b8>] (__mutex_lock.constprop.3) from [<807a57d0>] (__mutex_lock_slowpath+0x1c/0x20)
[  980.797931] [<807a57d0>] (__mutex_lock_slowpath) from [<807a5830>] (mutex_lock+0x5c/0x60)
[  980.797949] [<807a5830>] (mutex_lock) from [<80640540>] (rpi_firmware_transaction+0x44/0xac)
[  980.797969] [<80640540>] (rpi_firmware_transaction) from [<806406a0>] (rpi_firmware_property_list+0xf8/0x208)
[  980.797986] [<806406a0>] (rpi_firmware_property_list) from [<80640814>] (rpi_firmware_property+0x64/0x84)
[  980.798004] [<80640814>] (rpi_firmware_property) from [<806409e8>] (rpi_firmware_get_throttled+0x124/0x214)
[  980.798022] [<806409e8>] (rpi_firmware_get_throttled) from [<80640b6c>] (get_throttled_poll+0x28/0x54)
[  980.798040] [<80640b6c>] (get_throttled_poll) from [<80137a40>] (process_one_work+0x158/0x454)
[  980.798055] [<80137a40>] (process_one_work) from [<80137da0>] (worker_thread+0x64/0x5b8)
[  980.798070] [<80137da0>] (worker_thread) from [<8013de24>] (kthread+0x13c/0x16c)
[  980.798088] [<8013de24>] (kthread) from [<801080ac>] (ret_from_fork+0x14/0x28)
[  980.798100] INFO: task vcmailbox:1006 blocked for more than 120 seconds.
[  980.798107]       Tainted: G         C      4.14.98-v7+ #1200
[  980.798112] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  980.798116] vcmailbox       D    0  1006   1004 0x00000010
[  980.798140] [<807a2ed0>] (__schedule) from [<807a3538>] (schedule+0x50/0xa8)
[  980.798160] [<807a3538>] (schedule) from [<807a711c>] (schedule_timeout+0x1d0/0x3f4)
[  980.798177] [<807a711c>] (schedule_timeout) from [<807a41bc>] (wait_for_common+0xc0/0x184)
[  980.798194] [<807a41bc>] (wait_for_common) from [<807a42a0>] (wait_for_completion+0x20/0x24)
[  980.798211] [<807a42a0>] (wait_for_completion) from [<80640564>] (rpi_firmware_transaction+0x68/0xac)
[  980.798229] [<80640564>] (rpi_firmware_transaction) from [<806406a0>] (rpi_firmware_property_list+0xf8/0x208)
[  980.798246] [<806406a0>] (rpi_firmware_property_list) from [<8053b0a8>] (vcio_device_ioctl+0xd8/0x1c0)
[  980.798264] [<8053b0a8>] (vcio_device_ioctl) from [<8029f27c>] (do_vfs_ioctl+0xac/0x7c4)
[  980.798280] [<8029f27c>] (do_vfs_ioctl) from [<8029f9d8>] (SyS_ioctl+0x44/0x6c)
[  980.798296] [<8029f9d8>] (SyS_ioctl) from [<80108198>] (__sys_trace_return+0x0/0x10)
[ 1103.676314] INFO: task kworker/0:0:884 blocked for more than 120 seconds.
[ 1103.676325]       Tainted: G         C      4.14.98-v7+ #1200
[ 1103.676331] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 1103.676338] kworker/0:0     D    0   884      2 0x00000000
[ 1103.676368] Workqueue: events get_throttled_poll
[ 1103.676403] [<807a2ed0>] (__schedule) from [<807a3538>] (schedule+0x50/0xa8)
[... 91 lines cut ...]
[ 1349.441109] [<807a57d0>] (__mutex_lock_slowpath) from [<807a5830>] (mutex_lock+0x5c/0x60)
[ 1349.441126] [<807a5830>] (mutex_lock) from [<80640540>] (rpi_firmware_transaction+0x44/0xac)
[ 1349.441144] [<80640540>] (rpi_firmware_transaction) from [<806406a0>] (rpi_firmware_property_list+0xf8/0x208)
[ 1349.441162] [<806406a0>] (rpi_firmware_property_list) from [<80640814>] (rpi_firmware_property+0x64/0x84)
[ 1349.441179] [<80640814>] (rpi_firmware_property) from [<806409e8>] (rpi_firmware_get_throttled+0x124/0x214)
[ 1349.441197] [<806409e8>] (rpi_firmware_get_throttled) from [<80640b6c>] (get_throttled_poll+0x28/0x54)
[ 1349.441212] [<80640b6c>] (get_throttled_poll) from [<80137a40>] (process_one_work+0x158/0x454)
[ 1349.441225] [<80137a40>] (process_one_work) from [<80137da0>] (worker_thread+0x64/0x5b8)
[ 1349.441238] [<80137da0>] (worker_thread) from [<8013de24>] (kthread+0x13c/0x16c)
[ 1349.441254] [<8013de24>] (kthread) from [<801080ac>] (ret_from_fork+0x14/0x28)
[ 1349.441265] INFO: task vcmailbox:1006 blocked for more than 120 seconds.
[ 1349.441272]       Tainted: G         C      4.14.98-v7+ #1200
[ 1349.441277] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 1349.441282] vcmailbox       D    0  1006   1004 0x00000010
[ 1349.441305] [<807a2ed0>] (__schedule) from [<807a3538>] (schedule+0x50/0xa8)
[ 1349.441322] [<807a3538>] (schedule) from [<807a711c>] (schedule_timeout+0x1d0/0x3f4)
[ 1349.441339] [<807a711c>] (schedule_timeout) from [<807a41bc>] (wait_for_common+0xc0/0x184)
[ 1349.441356] [<807a41bc>] (wait_for_common) from [<807a42a0>] (wait_for_completion+0x20/0x24)
[ 1349.441373] [<807a42a0>] (wait_for_completion) from [<80640564>] (rpi_firmware_transaction+0x68/0xac)
[ 1349.441391] [<80640564>] (rpi_firmware_transaction) from [<806406a0>] (rpi_firmware_property_list+0xf8/0x208)
[ 1349.441409] [<806406a0>] (rpi_firmware_property_list) from [<8053b0a8>] (vcio_device_ioctl+0xd8/0x1c0)
[ 1349.441427] [<8053b0a8>] (vcio_device_ioctl) from [<8029f27c>] (do_vfs_ioctl+0xac/0x7c4)
[ 1349.441442] [<8029f27c>] (do_vfs_ioctl) from [<8029f9d8>] (SyS_ioctl+0x44/0x6c)
[ 1349.441457] [<8029f9d8>] (SyS_ioctl) from [<80108198>] (__sys_trace_return+0x0/0x10)
```

ptesarik, 2019-05-17:
Oh, I have to take back the part with correct parameters. It's just that the firmware code is apparently relocated at runtime, so the function address I got from `start.elf` was invalid.

I can confirm that the interface works properly when given a valid function address. It's also possible to pass less parameters than 7 (which might be added to documentation). Here a successful call to the _signed modulo_ function:

```
pi@raspberrypi:~ $ /opt/vc/bin/vcmailbox 0x00030010 12 12 0x3ec5090c 42 10
0x00000024 0x80000000 0x00030010 0x0000000c 0x80000004 0x00000002 0x0000002a 0x0000000a 0x00000000 
```

You can see the two parameters (0x2a and 0xa), preceded by the result (2).

JamesH65, 2020-02-13:
Not sure anything can be done here to protect against bad addresses being passed in - how do you know it's a bad address? Or that the parameters are incorrect? So...

This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

ptesarik, 2020-02-26:
I agree. The user of the API must ensure that all parameters are valid. After all, there are many other ways an OS kernel can crash the firmware (e.g. misprogramming a DMA controller).

It's a bit unfortunate that the raw API is exposed to user-space, but the permissions can be restricted to privileged users, so yeah, let's close the issue.

closed 2020-02-26 by ptesarik