raspberrypi/linux #2978: "brcmf_sdio_readframes: RXHEADER FAILED: -84" spam fills root partition

ali1234, 2019-05-18:
**Describe the bug**
After some time the wifi stops working and the following messages are spammed several hundred times per second to /var/log/syslog and /var/log/kern.log, causing the root partition to become completely full.

```
May  1 13:34:01 ender3 kernel: [4635658.324289] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
May  1 13:34:01 ender3 kernel: [4635658.325477] brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -84
May  1 13:34:01 ender3 kernel: [4635658.325480] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
May  1 13:34:01 ender3 kernel: [4635658.326677] brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -84
```
**To reproduce**
Just leave the current release of Raspbian running while connected to wifi for a couple of weeks.

**System**
RPI 3B

Raspberry Pi reference 2018-11-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7e0c786c641ba15990b5662f092c106beed40c9f, stage2

Linux ender3 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

Feb 12 2019 19:42:42 
Copyright (c) 2012 Broadcom
version 8eff5e4023657a8b3b59e1f90dc966f62d74908c (clean) (release) (start)

**Logs**
The full log is over 6 GB. I will attempt to extract relevant information from it when I can get a copy from the device.

JamesH65, 2019-05-18:
Since this seems unusual (ie others haven't reported it), it might be worth describing your network in more detail - ie routers, other devices that might be causing traffic etc.

ali1234, 2019-05-18:
I checked the logs and there was nothing interesting. I simply starts printing the two errors over and over again at some point and keeps doing it until the disk is full.

Network details: I have two Homehub 5a running openwrt. One is the internet gateway and the other is acting as a wireless extender. You can see the specs of them here: https://openwrt.org/toh/bt/homehub_v5a . They both use the same SSID on both radios - in fact on the wireless side the configuration is identical due to being deployed from the same script. The rest of the network is the usual computers, phones, printers etc.

lategoodbye, 2019-05-21:
At least the log message should be rate limited.

@ali1234 So you are not able to reproduce this issue within a day after a reboot?

ali1234, 2019-05-23:
Unfortunately not. The Pi was on for about 5 weeks before it happened.

Billybangleballs, 2019-05-31:
May 31 21:26:01 raspberrypi kernel: [770935.997625] brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
May 31 21:26:01 raspberrypi kernel: [770935.997639] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK

 22:56:56 up 8 days, 23:39,  3 users,  load average: 0.54, 0.31, 0.36

I didn't get a disc-full, but I was disassociated at the time if that has any bearing on the issue.

jonhadfield, 2020-06-30:
```
Jun 29 23:09:19 localhost kernel: [476354.512737] brcmfmac: brcmf_sdio_readframes: RXHEADER FAILED: -110
Jun 29 23:09:19 localhost kernel: [476354.519352] brcmfmac: brcmf_sdio_rxfail: abort command, terminate frame, send NAK
```

 16:11:31 up 6 days,  5:21,  2 users,  load average: 0.24, 0.24, 0.19

No other issues reported around that time. 
Pi 3B running:
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=19.10
DISTRIB_CODENAME=eoan
DISTRIB_DESCRIPTION="Ubuntu 19.10"

Linux pi2 5.3.0-1027-raspi2 #29-Ubuntu SMP Wed May 27 10:47:51 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux

lategoodbye, 2020-06-30:
@jonhadfield How do you operate your RPi 3 (e.g. headless AP)?
Was this the first time the issue occured?

jonhadfield, 2020-06-30:
@lategoodbye I'm just booting into a terminal and running a dashboard (https://wtfutil.com/) that runs continuously on an 11" screen attached via HDMI. I'm using an ethernet cable, and not WIFI, to connect to the network.