raspberrypi/linux #2979: Hostapd + 4.19.40 + Edimax EW-7811Un fails to accept client connections

moodeaudio, 2019-05-18:
**Describe the bug**
Kernel 4.19.40 + Pi-2B + Edimax EW-7811Un adapter results in hostapd failing to accept client connections. The error message on the client is "Incorrect password". Reverting to 4.14.98 kernel and no issues.

**To reproduce**
1. Download and install moOde 5.2 image onto SDCard
https://github.com/moode-player/moode/releases/download/r52prod-fix/moode-r52.zip
2. Boot up a Pi-2B with Edimax EW-7811Un adapter and Ethernet cable plugged in
3. SSH pi@moode (pwd = moodeaudio)
4. Activate hostapd mode manually and then optionally restart hostapd in debug mode

sudo sed -i "/AP mode/,/$p/ d" /etc/dhcpcd.conf
sudo sed -i "$ a#AP mode\ninterface wlan0\nstatic ip_address=172.24.1.1/24\nnohook wpa_supplicant" /etc/dhcpcd.conf
sudo systemctl daemon-reload
sudo systemctl restart dhcpcd
sudo systemctl start hostapd
sudo systemctl start dnsmasq

5. Verify a few things

a. ifconfig (verify that 172.24.1.1 address assigned to wlan0)
b. On a client device for example an iPhone verify that SSID Moode appears in the list of available WiFi networks

6. Optionally restart hostapd in debug logging mode

sudo systemctl stop hostapd
sudo hostapd -d /etc/hostapd/hostapd.conf

**Expected behaviour**
iPhone client is able to successfully connect to AP SSID = Moode using password = moodeaudio. Sometimes 2 attempts are required.

**Actual behaviour**
iPhone client receives error message "Incorrect password". Any number of repeated attempts to connect to AP SSID using the correct password result in the same error message. This also appears to hang the network interfaces and the Pi must be power-down rebooted.

**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi-2B
* Which OS and version (`cat /etc/rpi-issue`)?
Raspbian Stretch Lite 
pi@rp6:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2018-11-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7e0c786c641ba15990b5662f092c106beed40c9f, stage2
* Which firmware version (`vcgencmd version`)?
pi@rp6:~ $ vcgencmd version
Mar 27 2019 15:45:53 
Copyright (c) 2012 Broadcom
version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start)
* Which kernel version (`uname -a`)?
pi@rp6:~ $ uname -a
Linux rp6 4.19.40-v7+ #1217 SMP Tue May 7 17:22:08 BST 2019 armv7l GNU/Linux
**Logs**
Logs and additional information on the issue are in moOde Support Forum thread
http://moodeaudio.org/forum/showthread.php?tid=1399&pid=10581#pid10581

paul-1, 2019-05-23:
Do you have a console connection, I’m seeing a segfault.  I didn’t have time to go any further.

moodeaudio, 2019-05-23:
Post the segfault.

paul-1, 2019-05-23:
Sorry meant a kernel panic.....im a bit tired, fighting a cold.  I'll have to setup a serial console to be able to capture it....or take a picture.  Will do it when I can, just didn't know if were seeing the same thing.

moodeaudio, 2019-05-23:
Thanks @paul-1. I don't have any way to do a serial connection to capture a crash but I was able to get what appears to be the last part of it using a 3B with a 7" Touch. It happens exactly when the client connects to the AP.

[image]

paul-1, 2019-05-23:
You should be able add pause_on_oops=xx  to cmdline.txt (Where xx is number of seconds to pause)  That will keep the top of the oops from scrolling off the screen.

moodeaudio, 2019-05-23:
Ok, I'll try in a bit.

moodeaudio, 2019-05-23:
The pause_on_oops did not seem to work. The screen still scrolled. I ended up taking a movie and the  first few lines of the oops were more or less clear.

[image]

pelwell, 2019-05-23:
Try adding "framebuffer_height=2160” to config.txt - you may get the whole report on screen at once.

moodeaudio, 2019-05-23:
Gave it a try but there was still scrolling past the top. I'd guess maybe 2-3 pages.

jan-h-1970, 2019-05-25:
Raspberry 2Bv1.2 with hostapd running 
vcgencmd version
Mar 27 2019 15:45:53 
Copyright (c) 2012 Broadcom
version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start)
uname -a
Linux raspi-test 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux
lsusb (of the access point adapters)
ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
or
ID 0bda:8178 Realtek Semiconductor Corp. RTL8192CU 802.11n WLAN Adapter
crash when client tries to connect to hostapd
[  216.702035] ------------[ cut here ]------------
[  216.717282] kernel BUG at mm/slub.c:3904!
[  216.731696] Internal error: Oops - BUG: 0 [#1] SMP ARM
[  216.747355] Modules linked in: bridge stp llc ipv6 8192cu sha256_generic sg cfg80211 rfkill cdc_ether r8152 uas bcm2835_v4l2(C) bcm2835_codec(C) v4l2_mem2mem v4l2_common videobuf2_vmalloc bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_memops snd_bcm2835(C) videobuf2_v4l2 raspberrypi_hwmon videobuf2_common hwmon snd_pcm videodev media vc_sm_cma(C) snd_timer snd uio_pdrv_genirq fixed uio ip_tables x_tables
[  216.807958] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        4.19.42-v7+ #1219
[  216.828282] Hardware name: BCM2835
[  216.843934] PC is at kfree+0x1a8/0x1e8
[  216.860272] LR is at nl80211_send_station+0xaa4/0xca4 [cfg80211]
[  216.878661] pc : [<80294314>]    lr : [<7f280794>]    psr: 40000113
[  216.897420] sp : 80d01920  ip : 0024770a  fp : 80d0194c
[  216.915110] r10: b25e107c  r9 : 00000000  r8 : 80d01a00
[  216.932766] r7 : 80d01970  r6 : c0d3a000  r5 : 7f280794  r4 : b25e1014
[  216.951784] r3 : bd859c28  r2 : 00000000  r1 : 00000000  r0 : c0d3a000
[  216.970896] Flags: nZcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  216.990584] Control: 10c5383d  Table: 3c3dc06a  DAC: 00000055
[  217.008853] Process swapper/0 (pid: 0, stack limit = 0xcb9efe30)
[  217.027457] Stack: (0x80d01920 to 0x80d02000)
[  217.044438] 1920: 0000005a 805137f8 b25e1014 80d04d48 ba8b1780 80d01970 80d01a00 00000000
[  217.065340] 1940: 80d019ac 80d01950 7f280794 80294178 00000005 00000013 80d019ac b25e1034
[  217.086182] 1960: 00000012 80d04d48 b25e1014 b25e1030 73657469 82080174 00d019a4 ca1bf7c7
[  217.107043] 1980: 801ffc74 00000000 babca000 b27c6ad2 80d01a00 b97f99a0 00480020 b97f9800
[  217.127989] 19a0: 80d019fc 80d019b0 7f280f10 7f27fcfc 00000000 b97f9800 babca000 b27c6ad2
[  217.149025] 19c0: 80d01a00 80d019d0 7f363960 ba8b1780 b8be18a4 00000000 00000000 00000000
[  217.170129] 19e0: c0d3b000 c0d3b918 00000076 c0e12474 80d01acc 80d01a00 7f396728 7f280e94
[  217.191335] 1a00: 00000000 00000000 801019c0 c0e02360 c0d3a000 c0e02360 c0e12474 c0e02374
[  217.212633] 1a20: 000000dd b8be18a4 80d01a44 80d01a38 7f38383c 7f3831d4 80d01a5c 80d01a48
[  217.233975] 1a40: 7f36e9a4 7f383830 c0d3a000 c0e02360 80d01a74 80d01a60 7f34cf40 7f36e964
[  217.255426] 1a60: c0d3a000 c0d3d3cc b27c6ae4 0000005a 7f352a44 7f34ceec c0e02374 c0e12474
[  217.276861] 1a80: 80d01a9c 00001a90 8082bfcc c0016674 00000018 50001aa0 010102f2 ca1bf7c7
[  217.298285] 1aa0: 00000001 c0d3a000 b27c6ac8 00000000 c0d3b000 ca1bf7c7 c0d3a000 b27c6ac8
[  217.319830] 1ac0: 80d01c8c 80d01ad0 7f356968 7f3966b8 bc801e40 80292c5c 80d01b2c 00000014
[  217.341497] 1ae0: b27c6aff 80d01af0 0000005a c0e129c4 00000007 00000000 00000000 00000010
[  217.363326] 1b00: 00000010 b27c6ae6 00000007 b27c6aef 00000008 00000000 00000000 00000000
[  217.385091] 1b20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  217.406741] 1b40: 00000000 00000000 00000000 b27c6af9 00000004 00000000 00000000 b27c6aff
[  217.428252] 1b60: 00000014 b27c6b31 00000007 00000000 00000000 00000000 00000000 00000000
[  217.449777] 1b80: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  217.471191] 1ba0: 00000000 b27c6b15 0000001a 00000000 00000000 00000000 00000000 02f25000
[  217.492466] 1bc0: bc230100 80d04d48 00000000 968b8482 2418120c 6c604830 80d01c94 00000000
[  217.513737] 1be0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  217.534960] 1c00: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  217.556017] 1c20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  217.576942] 1c40: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 ca1bf7c7
[  217.597791] 1c60: 00000496 b27c6acc c0e0e1c0 7f3a64f8 c0d3a000 b27c6acc b27c6b44 00000001
[  217.618840] 1c80: 80d01cbc 80d01c90 7f34bcd8 7f3560c4 ffffffff fffff7c7 ffffffff ca1bf7c7
[  217.639870] 1ca0: b27c6ac8 c0d3a000 c0e0e1c0 c0e12474 80d01cf4 80d01cc0 7f34ca0c 7f34bc40
[  217.660993] 1cc0: 7f35dc14 fffff5a4 ffffffff ca1bf7c7 b27c6ac8 c0e0e1c0 c0e12474 c0d3a000
[  217.682212] 1ce0: c0d3a000 00000076 80d01d14 80d01cf8 7f35de74 7f34c8b4 00000000 c0e0e1c0
[  217.703517] 1d00: b27c6ac8 00000000 80d01d34 80d01d18 7f35dfac 7f35dd28 c0d3a000 c0e0e1c0
[  217.724787] 1d20: c0e0e1c0 b27c6ac8 80d01d54 80d01d38 7f35ea2c 7f35dea4 00000064 00000100
[  217.746111] 1d40: c0e0e1c0 ba8b16c0 80d01dac 80d01d58 7f38484c 7f35e9cc c0d3b690 c0d3a000
[  217.767449] 1d60: c0d3b750 baa68900 c0d3d000 babca800 c0d3b694 b8bfc060 00000100 b8bfc048
[  217.788890] 1d80: 80d01dac c0d3b72c c0d3b730 00000000 bcee7298 00000000 00000006 00000040
[  217.810365] 1da0: 80d01dd4 80d01db0 80126c34 7f3845e0 80d03098 00000006 80d00028 00000100
[  217.831859] 1dc0: bc810180 00000018 80d01de4 80d01dd8 80126d14 80126bdc 80d01e4c 80d01de8
[... cut ...]
[  220.056445] [<80109a08>] (arch_cpu_idle) from [<8082bcd4>] (default_idle_call+0x34/0x48)
[  220.079967] [<8082bcd4>] (default_idle_call) from [<801524bc>] (do_idle+0xec/0x17c)
[  220.103043] [<801524bc>] (do_idle) from [<8015280c>] (cpu_startup_entry+0x28/0x2c)
[  220.126008] [<8015280c>] (cpu_startup_entry) from [<8010fcc8>] (secondary_start_kernel+0x134/0x140)
[  220.150533] [<8010fcc8>] (secondary_start_kernel) from [<0010268c>] (0x10268c)
[  220.173211] ---[ end Kernel panic - not syncing: Fatal exception in interrupt ]---
[  220.196361] ------------[ cut here ]------------

jan-h-1970, 2019-05-25:
Raspberry 2Bv1.2 with hostapd running
vcgencmd version
Mar 27 2019 15:45:53
Copyright (c) 2012 Broadcom
version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start)
uname -a
Linux xxxxx 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux
This Kernel version is ok:
Linux xxxxx 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
lsusb (of the access point adapters)
ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
or
ID 0bda:8178 Realtek Semiconductor Corp. RTL8192CU 802.11n WLAN Adapter

crash when client tries to connect to hostapd

[ 216.702035] ------------[ cut here ]------------
[ 216.717282] kernel BUG at mm/slub.c:3904!
[ 216.731696] Internal error: Oops - BUG: 0 [#1] SMP ARM
[ 216.747355] Modules linked in: bridge stp llc ipv6 8192cu sha256_generic sg cfg80211 rfkill cdc_ether r8152 uas bcm2835_v4l2(C) bcm2835_codec(C) v4l2_mem2mem v4l2_common videobuf2_vmalloc bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_memops snd_bcm2835(C) videobuf2_v4l2 raspberrypi_hwmon videobuf2_common hwmon snd_pcm videodev media vc_sm_cma(C) snd_timer snd uio_pdrv_genirq fixed uio ip_tables x_tables
[ 216.807958] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G C 4.19.42-v7+ #1219
[ 216.828282] Hardware name: BCM2835
[ 216.843934] PC is at kfree+0x1a8/0x1e8
[ 216.860272] LR is at nl80211_send_station+0xaa4/0xca4 [cfg80211]
[ 216.878661] pc : [<80294314>] lr : [<7f280794>] psr: 40000113
[ 216.897420] sp : 80d01920 ip : 0024770a fp : 80d0194c
[ 216.915110] r10: b25e107c r9 : 00000000 r8 : 80d01a00
[ 216.932766] r7 : 80d01970 r6 : c0d3a000 r5 : 7f280794 r4 : b25e1014
[ 216.951784] r3 : bd859c28 r2 : 00000000 r1 : 00000000 r0 : c0d3a000
[ 216.970896] Flags: nZcv IRQs on FIQs on Mode SVC_32 ISA ARM Segment user
[ 216.990584] Control: 10c5383d Table: 3c3dc06a DAC: 00000055
[ 217.008853] Process swapper/0 (pid: 0, stack limit = 0xcb9efe30)
[ 217.027457] Stack: (0x80d01920 to 0x80d02000)
[ 217.044438] 1920: 0000005a 805137f8 b25e1014 80d04d48 ba8b1780 80d01970 80d01a00 00000000
[ 217.065340] 1940: 80d019ac 80d01950 7f280794 80294178 00000005 00000013 80d019ac b25e1034
[ 217.086182] 1960: 00000012 80d04d48 b25e1014 b25e1030 73657469 82080174 00d019a4 ca1bf7c7
[ 217.107043] 1980: 801ffc74 00000000 babca000 b27c6ad2 80d01a00 b97f99a0 00480020 b97f9800
[ 217.127989] 19a0: 80d019fc 80d019b0 7f280f10 7f27fcfc 00000000 b97f9800 babca000 b27c6ad2
[ 217.149025] 19c0: 80d01a00 80d019d0 7f363960 ba8b1780 b8be18a4 00000000 00000000 00000000
[ 217.170129] 19e0: c0d3b000 c0d3b918 00000076 c0e12474 80d01acc 80d01a00 7f396728 7f280e94
[ 217.191335] 1a00: 00000000 00000000 801019c0 c0e02360 c0d3a000 c0e02360 c0e12474 c0e02374
[ 217.212633] 1a20: 000000dd b8be18a4 80d01a44 80d01a38 7f38383c 7f3831d4 80d01a5c 80d01a48
[ 217.233975] 1a40: 7f36e9a4 7f383830 c0d3a000 c0e02360 80d01a74 80d01a60 7f34cf40 7f36e964
[ 217.255426] 1a60: c0d3a000 c0d3d3cc b27c6ae 0000005a 7f352a44 7f34ceec c0e02374 c0e12474
[ 217.276861] 1a80: 80d01a9c 00001a90 8082bfcc c0016674 00000018 50001aa0 010102f2 ca1bf7c7
[ 217.298285] 1aa0: 00000001 c0d3a000 b27c6ac8 00000000 c0d3b000 ca1bf7c7 c0d3a000 b27c6ac8
[ 217.319830] 1ac0: 80d01c8c 80d01ad0 7f356968 7f3966b8 bc801e40 80292c5c 80d01b2c 00000014
[ 217.341497] 1ae0: b27c6aff 80d01af0 0000005a c0e129c4 00000007 00000000 00000000 00000010
[ 217.363326] 1b00: 00000010 b27c6ae6 00000007 b27c6aef 00000008 00000000 00000000 00000000
[ 217.385091] 1b20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 217.406741] 1b40: 00000000 00000000 00000000 b27c6af9 00000004 00000000 00000000 b27c6aff
[ 217.428252] 1b60: 00000014 b27c6b31 00000007 00000000 00000000 00000000 00000000 00000000
[ 217.449777] 1b80: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 217.471191] 1ba0: 00000000 b27c6b15 0000001a 00000000 00000000 00000000 00000000 02f25000
[ 217.492466] 1bc0: bc230100 80d04d48 00000000 968b8482 2418120c 6c604830 80d01c94 00000000
[ 217.513737] 1be0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 217.534960] 1c00: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 217.556017] 1c20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 217.576942] 1c40: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 ca1bf7c7
[ 217.597791] 1c60: 00000496 b27c6acc c0e0e1c0 7f3a64f8 c0d3a000 b27c6acc b27c6b44 00000001
[ 217.618840] 1c80: 80d01cbc 80d01c90 7f34bcd8 7f3560c4 ffffffff fffff7c7 ffffffff ca1bf7c7
[ 217.639870] 1ca0: b27c6ac8 c0d3a000 c0e0e1c0 c0e12474 80d01cf4 80d01cc0 7f34ca0c 7f34bc40
[ 217.660993] 1cc0: 7f35dc14 fffff5a4 ffffffff ca1bf7c7 b27c6ac8 c0e0e1c0 c0e12474 c0d3a000
[ 217.682212] 1ce0: c0d3a000 00000076 80d01d14 80d01cf8 7f35de74 7f34c8b4 00000000 c0e0e1c0
[ 217.703517] 1d00: b27c6ac8 00000000 80d01d34 80d01d18 7f35dfac 7f35dd28 c0d3a000 c0e0e1c0
[ 217.724787] 1d20: c0e0e1c0 b27c6ac8 80d01d54 80d01d38 7f35ea2c 7f35dea4 00000064 00000100
[ 217.746111] 1d40: c0e0e1c0 ba8b16c0 80d01dac 80d01d58 7f38484c 7f35e9cc c0d3b690 c0d3a000
[ 217.767449] 1d60: c0d3b750 baa68900 c0d3d000 babca800 c0d3b694 b8bfc060 00000100 b8bfc048
[ 217.788890] 1d80: 80d01dac c0d3b72c c0d3b730 00000000 bcee7298 00000000 00000006 00000040
[ 217.810365] 1da0: 80d01dd4 80d01db0 80126c34 7f3845e0 80d03098 00000006 80d00028 00000100
[... 151 lines cut ...]
[... cut ...]
[ 221.908113] [<8082bde0>] (_raw_read_lock) from [<806a1644>] (ledtrig_cpu+0xb4/0xf0)
[ 221.932816] [<806a1644>] (ledtrig_cpu) from [<80109a78>] (arch_cpu_idle_exit+0x1c/0x20)
[ 221.957872] [<80109a78>] (arch_cpu_idle_exit) from [<801524d0>] (do_idle+0x100/0x17c)
[ 221.982767] [<801524d0>] (do_idle) from [<8015280c>] (cpu_startup_entry+0x28/0x2c)
[ 222.007459] [<8015280c>] (cpu_startup_entry) from [<808253e4>] (rest_init+0xbc/0xc0)
[ 222.032338] [<808253e4>] (rest_init) from [<80c00fb0>] (start_kernel+0x484/0x4b4)
[ 222.056912] ---[ end trace 9ec9408f0ced3506 ]---

jan-h-1970, 2019-05-29:
same to kernel: Linux xxxxx **4.19.46-v7+ #1230** SMP Tue May 28 16:18:48 BST 2019 armv7l GNU/Linux
modules lsmod: "cfg80211              614400  1 8192cu"

[   72.158114] ------------[ cut here ]------------
[   72.172957] kernel BUG at mm/slub.c:3904!
[   72.187058] Internal error: Oops - BUG: 0 [#1] SMP ARM
[   72.202361] Modules linked in: bridge stp llc ipv6 8192cu sg sha256_generic cfg80211 uas rfkill bcm2835_codec(C) bcm2835_v4l2(C) snd_bcm2835(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) snd_pcm videobuf2_dma_contig v4l2_common videobuf2_vmalloc videobuf2_memops snd_timer videobuf2_v4l2 raspberrypi_hwmon videobuf2_common hwmon snd videodev media vc_sm_cma(C) uio_pdrv_genirq fixed uio ip_tables x_tables
[   72.260856] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        4.19.46-v7+ #1230
[   72.280879] Hardware name: BCM2835
[   72.296208] PC is at kfree+0x1a8/0x1e8
[   72.312186] LR is at nl80211_send_station+0xaa4/0xca4 [cfg80211]
[   72.330112] pc : [<802949fc>]    lr : [<7f243908>]    psr: 40000113
[   72.348361] sp : 80d01968  ip : 002475fc  fp : 80d01994
[   72.365536] r10: bbd5107c  r9 : 00000000  r8 : 80d01a48
[   72.382640] r7 : 80d019b8  r6 : c0d1c000  r5 : 7f243908  r4 : bbd51014
[   72.401169] r3 : bd8597f0  r2 : 00000000  r1 : 00000000  r0 : c0d1c000
[   72.419691] Flags: nZcv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[   72.439038] Control: 10c5383d  Table: 36b6806a  DAC: 00000055
[   72.457119] Process swapper/0 (pid: 0, stack limit = 0x82dce9f4)
[   72.475347] Stack: (0x80d01968 to 0x80d02000)
[   72.491756] 1960:                   0000005a 80513fe0 bbd51014 80d04d48 b7c6f240 80d019b8
[   72.512193] 1980: 80d01a48 00000000 80d019f4 80d01998 7f243908 80294860 00000005 00000013
[   72.532752] 19a0: 80d019f4 bbd51034 00000012 80d04d48 bbd51014 bbd51030 73657469 82080174
[   72.553345] 19c0: 00d019ec ca1bf7c3 80200324 00000000 ba61d800 b6bec052 80d01a48 bc2679a0
[   72.574038] 19e0: 00480020 bc267800 80d01a44 80d019f8 7f244084 7f242e70 00000000 bc267800
[   72.594715] 1a00: ba61d800 b6bec052 80d01a48 80d01a18 7f326960 b7c6f240 b87078a4 00000000
[   72.615371] 1a20: 00000000 00000000 c0d1d000 c0d1d918 00000076 c0d5dda4 80d01b14 80d01a48
[   72.636114] 1a40: 7f359728 7f244008 00000000 00000000 c0d4da14 c0d4da00 c0d1c000 c0d4da00
[   72.656959] 1a60: c0d5dda4 c0d4da14 000000dd b87078a4 80d01a8c 80d01a80 7f34683c 7f3461d4
[   72.677866] 1a80: 80d01aa4 80d01a90 7f3319a4 7f346830 c0d1c000 c0d4da00 80d01abc 80d01aa8
[   72.698894] 1aa0: 7f30ff40 7f331964 c0d1c000 c0d1f3cc b6bec064 0000005a 7f315a44 7f30feec
[   72.720001] 1ac0: c0d4da14 c0d5dda4 80d01ae4 00001ad8 8082ca2c c00166b8 00000018 50001ae8
[   72.741236] 1ae0: 010102f2 ca1bf7c3 00000001 c0d1c000 b6bec048 00000000 c0d1d000 ca1bf7c3
[   72.762552] 1b00: c0d1c000 b6bec048 80d01cd4 80d01b18 7f319968 7f3596b8 80d01b6c 80293344
[   72.783865] 1b20: 80d01b74 00000014 b6bec07f 80d01b38 0000005a c0d5e2f4 00000007 00000000
[   72.805141] 1b40: 00000000 00000010 00000010 b6bec066 00000007 b6bec06f 00000008 00000000
[   72.826446] 1b60: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   72.847750] 1b80: 00000000 00000000 00000000 00000000 00000000 b6bec079 00000004 00000000
[   72.869009] 1ba0: 00000000 b6bec07f 00000014 b6bec0b1 00000007 00000000 00000000 00000000
[   72.890363] 1bc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   72.911658] 1be0: 00000000 00000000 00000000 b6bec095 0000001a 00000000 00000000 00000000
[   72.932876] 1c00: 00000000 02f25000 bc240100 80d04d48 00000000 968b8482 2418120c 6c604830
[   72.954045] 1c20: 80d01cdc 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   72.975196] 1c40: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   72.996175] 1c60: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   73.016987] 1c80: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   73.037739] 1ca0: 00000000 ca1bf7c3 00000084 b6bec04c c0d546f8 7f3694f8 c0d1c000 b6bec04c
[   73.058578] 1cc0: b6bec0c4 00000001 80d01d04 80d01cd8 7f30ecd8 7f3190c4 ffffffff fffff7c3
[   73.079471] 1ce0: ffffffff ca1bf7c3 b6bec048 c0d1c000 c0d546f8 c0d5dda4 80d01d3c 80d01d08
[   73.100493] 1d00: 7f30fa0c 7f30ec40 7f320c14 ffff25a4 ffffffff ca1bf7c3 b6bec048 c0d546f8
[   73.121607] 1d20: c0d5dda4 c0d1c000 c0d1c000 00000076 80d01d5c 80d01d40 7f320e74 7f30f8b4
[   73.142818] 1d40: 00000000 c0d546f8 b6bec048 00000000 80d01d7c 80d01d60 7f320fac 7f320d28
[   73.164022] 1d60: c0d1c000 c0d546f8 c0d546f8 b6bec048 80d01d9c 80d01d80 7f321a2c 7f320ea4
[   73.185237] 1d80: 00000064 00000100 c0d546f8 b7c6fc00 80d01df4 80d01da0 7f34784c 7f3219cc
[   73.206480] 1da0: c0d1d690 c0d1c000 c0d1d750 ba41a3c0 c0d1f000 ba61e000 c0d1d694 b871c060
[   73.227845] 1dc0: 00000100 b871c048 80d01df4 c0d1d72c c0d1d730 00000000 bcee7298 00000000
[   73.249287] 1de0: 00000006 00000040 80d01e1c 80d01df8 80126c78 7f3475e0 80d03098 00000006
[   73.270766] 1e00: 80d00030 00000100 bc810180 00000018 80d01e2c 80d01e20 80126d58 80126c20
[   73.292289] 1e20: 80d01e94 80d01e30 80102388 80126d3c 80d01e54 80d01e40 00000001 00200102
[   73.313875] 1e40: 80d03d00 ffffa701 00000009 80c92a00 80d8ec58 80d04d70 80c8a308 80d03080
[   73.335462] 1e60: 80d01e30 00000007 8017e870 80c929e8 00000000 00000000 00000001 bc810180
[   73.357165] 1e80: 80d00000 80c63a38 80d01ea4 80d01e98 80126920 80102204 80d01ecc 80d01ea8
[... cut ...]
[   77.251325] [<801019bc>] (__irq_svc) from [<80109a08>] (arch_cpu_idle+0x34/0x4c)
[   77.275695] [<80109a08>] (arch_cpu_idle) from [<8082c734>] (default_idle_call+0x34/0x48)
[   77.300805] [<8082c734>] (default_idle_call) from [<80152500>] (do_idle+0xec/0x17c)
[   77.325451] [<80152500>] (do_idle) from [<80152850>] (cpu_startup_entry+0x28/0x2c)
[   77.349999] [<80152850>] (cpu_startup_entry) from [<80825e44>] (rest_init+0xbc/0xc0)
[   77.374806] [<80825e44>] (rest_init) from [<80c00fb0>] (start_kernel+0x484/0x4b4)
[   77.399322] ---[ end trace 5ccbbc48e3a2f844 ]---

jan-h-1970, 2019-05-30:
**Seems to me 8192cu module never worked as an access point with a 4.9.xx kernel.**
I checked these kernel versions, each of them with a kernel panic:
4.19.46-v7+ #1230
4.19.42-v7+ #1219
4.19.42-v7+ #1218
4.19.37-v7+ #1214
4.19.32-v7+ #1210
4.19.29-v7+ #1207
4.19.25-v7+ #1204

jan-h-1970, 2019-05-30:
**Seems to me 8192cu module never worked as an access point with a 4.19.xx kernel.**
I checked these kernel versions, each of them with a kernel panic:
4.19.46-v7+ #1230
4.19.42-v7+ #1219
4.19.42-v7+ #1218
4.19.37-v7+ #1214
4.19.32-v7+ #1210
4.19.29-v7+ #1207
4.19.25-v7+ #1204

klimchuk, 2019-07-02:
I believe issue is more related to driver rather than operating system
https://github.com/pvaret/rtl8192cu-fixes/issues/149

macmpi, 2019-07-07:
Yes, most probably a driver/hostapd combination issue than kernel.

8192cu has problematic driver(s) and may need customized version of `hostapd` if [it does not support cfg80211](https://github.com/volumio/Build/blob/2f09878fca434492a3a1c8d64c992c38a304bdcb/volumio/bin/hotspot.sh#L8-L13)

Also, "regular" `hostapd` needed some tweaks for 4.19: Foundation's Buster release [fixed it](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=224931&p=1434906&hilit=hostapd#p1437905).
If you use a custom `hostapd` due to your driver, or sub-par `hostapd` version, you may need to fix `hostapd` accordingly.

moodeaudio, 2019-07-08:
@klimchuk - I would tend to agree that the issue is in the driver given that kernel panic occurs in 4.19 and no issues in 4.14. Same version of Raspbian, hostapd and hostapd config.

@macmpi - I've not experienced any issues with hostapd, 8192cu and cfg80211 over the years. It's only since the 4.19 branch that connection attempts to the AP cause kernel crash.

macmpi, 2019-07-08:
@moodeaudio - Looking @ first post, it occurs you are running Raspbian Stretch Lite: so **it's very likely** `hostpad` **version running is not suitable for 4.19.x**, unless you managed to install Buster's version, or built your own based on same sources, which you never said.

So, I advise you try on stock Buster distribution first. If it works and you still need to use Stretch, then try to get `hostapd` sources from Buster version, and recompile for your distribution.

moodeaudio, 2019-07-08:
Yes, our project has built hostapd from sources since v2.4 as I recall. I've included the build recipe and hostapd.conf. The 2.7 binary that was compiled on Stretch works fine in Buster but we also experience same kernel crash on connection attempt when using Edimax adapter. This is with 4.19.56 kernel.

@jan-h-1970 posted two complete crash dumps so I think it's just a matter of someone with enough experience analyzing them, isolating the issue and hopefully determining root cause.

If there is anything that should be adjusted in the build recipe or config I'll be happy to test. 

```
BUILD

# get missing lib package
sudo apt-get install libnl-3-dev

# unrelated to the above, create a link to an existing lib
cd /lib/arm-linux-gnueabihf/
sudo ln -s libnl-genl-3.so.200.22.0 libnl-genl-3.so

# download sources
cd ~
curl https://w1.fi/releases/hostapd-2.7.tar.gz -o hostapd-2.7.tar.gz
tar xf hostapd-2.7.tar.gz
cd hostapd-2.7/hostapd

# configure build
cp defconfig .config
sed -i 's/^#CONFIG_DRIVER_NL80211=y/CONFIG_DRIVER_NL80211=y/g' .config

# enable 802.11n and 802.11ac
sed -i 's/^#CONFIG_IEEE80211N=y/CONFIG_IEEE80211N=y/g' .config
sed -i 's/^#CONFIG_IEEE80211AC=y/CONFIG_IEEE80211AC=y/g' .config

# enable automatic channel selection
sed -i 's/^#CONFIG_ACS=y/CONFIG_ACS=y/g' .config

# compile
make

# install 2.7 binaries
sudo cp hostapd /usr/sbin/
sudo cp hostapd_cli /usr/sbin/

CONFIG

#########################################
# This file is automatically generated by
# the player Network configuration page. 
#########################################

# Interface and driver
interface=wlan0
driver=nl80211

# Wireless settings
ssid=Moode
hw_mode=g
channel=6

# Security settings
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_passphrase=moodeaudio
rsn_pairwise=CCMP
```

macmpi, 2019-07-08:
Buster version is not plain 2.7: you may want to get [hostapd (2:2.7+git20190128+0c1e29f-6)](https://packages.debian.org/buster/hostapd) sources, assuming that's the one that Raspbian ships.

moodeaudio, 2019-07-08:
Do you know what wold be different in Buster hostapd?

I just tested the Buster hostapd package and it exhibits the exact same symptoms when trying to connect to AP using Edimax adapter, the Pi hangs (kernel crashes). No issues with Pi integrated adapter.

macmpi, 2019-07-08:
And from baseline [Raspbian Lite Buster image](https://downloads.raspberrypi.org/raspbian_lite_latest) ?

moodeaudio, 2019-07-08:
pi@rp2:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2019-06-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage2

pi@rp2:~ $ vcgencmd version
Jun 26 2019 17:47:16 
Copyright (c) 2012 Broadcom
version a59fb7a74180be0111dbc5c18a37ec6df86f14a3 (clean) (release) (start)

pi@rp2:~ $ uname -a
Linux rp2 4.19.56+ #1242 Wed Jun 26 17:24:44 BST 2019 armv6l GNU/Linux

macmpi, 2019-07-08:
What if AP security is disabled (WPA2)?

macmpi, 2019-07-08:
Sorry for this stupid question, but I assume you blacklisted `rtl8192cu`, right ? (so that `8192cu` is used instead)
It does not seem to be blacklisted by default...

npreiser, 2019-07-08:
I see the same issue as @moodeaudio ,  I also tried disabling the security, and the same kernel fault occurs.. 
I confirmed that the rtl8192cu is blacklisted.

moodeaudio, 2019-07-08:
Where to look for blacklisting?

popcornmix, 2019-07-08:
See ``/etc/modprobe.d/blacklist-rtl8192cu.conf`` for example of blacklisting a module.

npreiser, 2019-07-08:
what I am seeing is this:,  which I assume means that the 8192 is blacklisted??, 

pi@raspberrypi:~ $ cd /etc/modprobe.d/
pi@raspberrypi:/etc/modprobe.d $ ls
blacklist-rtl8192cu.conf  blacklist-rtl8xxxu.conf  ipv6.conf
pi@raspberrypi:/etc/modprobe.d $

moodeaudio, 2019-07-08:
Thx, confirmed the file is present

```
pi@rp3:~ $ cat /etc/modprobe.d/blacklist-rtl8192cu.conf
blacklist rtl8192cu
```

npreiser, 2019-07-08:
As a side note, my hw guy here had an alternative LM811 wifi module,  which I just compiled the driver for, and it works just fine with the 6/20 raspbian lie.  . .https://github.com/lwfinger/rtl8723bu  ... uses the rtl8723bu...   we are most likley going to move forward with that..

jan-h-1970, 2019-07-10:
I can confirm the kernel crash comes still with raspbian buster lite (2019-06-20), it's independent of HW, I tested 2Bv1.2, 3B, 3B+.
Module rtl8192cu is blacklisted by default file /etc/modprobe.d/blacklist-rtl8192cu.conf
blacklist rtl8192cu

# dpkg -l hostapd
ii  hostapd        2:2.7+git20190128+0c1e29f-6 armhf        IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
# hostapd -v
hostapd v2.8-devel

# cat /etc/rpi-issue
Raspberry Pi reference 2019-06-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage2
# vcgencmd version
Jun 20 2019 16:12:41
Copyright (c) 2012 Broadcom
version a59fb7a74180be0111dbc5c18a37ec6df86f14a3 (clean) (release) (start)
# uname -a
Linux rpi 4.19.50-v7+ #896 SMP Thu Jun 20 16:11:44 BST 2019 armv7l GNU/Linux

same with the latest kernel
# cat /etc/rpi-issue
Raspberry Pi reference 2019-06-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage2
# vcgencmd version
Jul  9 2019 14:40:53
Copyright (c) 2012 Broadcom
version 6c3fe3f096a93de3b34252ad98cdccadeb534be2 (clean) (release) (start)
# uname -a
Linux rpi 4.19.57-v7+ #1244 SMP Thu Jul 4 18:45:25 BST 2019 armv7l GNU/Linux

error is the same at all ... kernel BUG at mm/slub.c:3904! ...
Latest 4.14 kernel still works with buster/hostapd (Linux rpi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux).

I can send another kernel crash dump output if it's helpful.
I've no idea whether it is a kernel, a 8192cu/cfg80211 module, or a hostapd problem.

jan-h-1970, 2019-07-10:
I can confirm the kernel crash comes still with raspbian buster lite (2019-06-20), it's independent of HW, I tested 2Bv1.2, 3B, 3B+.
Module rtl8192cu is blacklisted by default file /etc/modprobe.d/blacklist-rtl8192cu.conf
blacklist rtl8192cu

\# dpkg -l hostapd
ii  hostapd        2:2.7+git20190128+0c1e29f-6 armhf        IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator
\# hostapd -v
hostapd v2.8-devel

\# cat /etc/rpi-issue
Raspberry Pi reference 2019-06-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage2
\# vcgencmd version
Jun 20 2019 16:12:41
Copyright (c) 2012 Broadcom
version a59fb7a74180be0111dbc5c18a37ec6df86f14a3 (clean) (release) (start)
\# uname -a
Linux rpi 4.19.50-v7+ #896 SMP Thu Jun 20 16:11:44 BST 2019 armv7l GNU/Linux

same with the latest kernel
\# cat /etc/rpi-issue
Raspberry Pi reference 2019-06-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage2
\# vcgencmd version
Jul  9 2019 14:40:53
Copyright (c) 2012 Broadcom
version 6c3fe3f096a93de3b34252ad98cdccadeb534be2 (clean) (release) (start)
\# uname -a
Linux rpi 4.19.57-v7+ #1244 SMP Thu Jul 4 18:45:25 BST 2019 armv7l GNU/Linux

error is the same at all ... kernel BUG at mm/slub.c:3904! ...
Latest 4.14 kernel still works with buster/hostapd (Linux rpi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux).

I can send another kernel crash dump output if it's helpful.
I've no idea whether it is a kernel, a 8192cu/cfg80211 module, or a hostapd problem.

sphakka, 2019-08-22:
Still broken with:
```
# uname -a
Linux localhost 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
```

badbat75, 2019-09-16:
Still broken in 4.19.69 .

moodeaudio, 2019-09-23:
Your fix also works for me.

I did notice though that the driver no longer supports scanning while running with hostapd. Scan works fine however when the adapter is connected to WiFi Router. Very odd.

```
pi@rp7:~ $ sudo iwlist wlan0 scan
wlan0     Interface doesn't support scanning : Operation not supported
````

itsme009, 2019-09-27:
Hi,
Similar problem for me:
The attempt to create an AP with the Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS] results in some kind of crash. I have not found any records whats going on but after the first attempt to connect to the AP I can not access the raspi anymore and more worse it seems that my whole network is getting some problems that I can resolve by unplug the power connection from my raspi.
The only possibility that I have found yet to create a AP is to remove the two blacklist files from /etc/modprobe.d/.
Just a remark: The idea is to use two Edimax WLAN access point for my home network and a guest network, the build in device is used to connect to the internet using a WIFI-hotspot.
Best regards
Uli

jm-cjp, 2019-10-07:
Using a model B+ Rev 1.2 with Realtek RTL8188CUS and deleting /etc/modprobe.d/blacklist-rtl8192cu.conf  worked wonderfully for me.  Appreciate the info and work above.

sergiodobrasil, 2019-10-08:
Hi,
@DannyMTK you make my day ! 
same issue with the driver 8192cu and the some workaround 
_remove blacklist-rtl8* files from /etc/modprobe.d/_
fixed the kernel panic !

i will share my setup in order to help to solve the issue in case of...

Rapsberry = RP1
usb wifi adaptater : TP-Link TL-WN822n V3

`uname -a`
Linux WIFICave 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019 armv6l GNU/Linux

`lsb_release -a`
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 10 (buster)
Release:	10
Codename:	buster

`lsusb`
Bus 001 Device 005: ID 0bda:8178 Realtek Semiconductor Corp. RTL8192CU 802.11n WLAN Adapter

`modprobe -c | grep -i p8178`
alias usb:v07B8p8178d*dc*dsc*dp*ic*isc*ip*in* rtl8192cu
alias usb:v07B8p8178d*dc*dsc*dp*ic*isc*ip*in* 8192cu
alias usb:v0BDAp8178d*dc*dsc*dp*ic*isc*ip*in* rtl8192cu
alias usb:v0BDAp8178d*dc*dsc*dp*ic*isc*ip*in* 8192cu

`apt-cache policy firmware-realtek`
firmware-realtek:
  Installé : 1:20190114-1+rpt2
  Candidat : 1:20190114-1+rpt2

`lsmod | grep 8192`
rtl8192cu              90112  0
rtl_usb                20480  1 rtl8192cu
rtl8192c_common        65536  1 rtl8192cu
rtlwifi               114688  3 rtl_usb,rtl8192c_common,rtl8192cu
mac80211              692224  3 rtl_usb,rtlwifi,rtl8192cu

`apt-cache policy dnsmasq`
dnsmasq:
  Installé : 2.80-1+rpt1
  Candidat : 2.80-1+rpt1

`apt-cache policy hostapd`
hostapd:
  Installé : 2:2.7+git20190128+0c1e29f-6+deb10u1
  Candidat : 2:2.7+git20190128+0c1e29f-6+deb10u1

`sudo iwlist wlan0 scan`
wlan0     Interface doesn't support scanning : Operation not supported

`watch -d -n 3 "iw dev wlan0 station dump; iwconfig wlan0; cat /proc/net/wireless"`

Station 14:9f:3c:e3:98:23 (on wlan0)
        inactive time:  110 ms
        rx bytes:       671319
        rx packets:     3418
        tx bytes:       6824529
        tx packets:     5907
        tx retries:     0
        tx failed:      0
        rx drop misc:   10
        signal:         -26 dBm
        signal avg:     -25 dBm
        tx bitrate:     54.0 MBit/s
        rx bitrate:     48.0 MBit/s
        authorized:     yes
        authenticated:  yes
        associated:     yes
        preamble:       short
        WMM/WME:        no
        MFP:            no
        TDLS peer:      no
        DTIM period:    2
        beacon interval:100
        short slot time:yes
        connected time: 229 seconds
wlan0     IEEE 802.11  Mode:Master  Tx-Power=20 dBm
          Retry short limit:7   RTS thr=2347 B   Fragment thr:off
          Power Management:off

Sergio

julius59, 2019-11-03:
Exact same issue and solution here (Raspberry 1 Model B + TP-Link TL-WN823N V2)

Thanks @DannyMTK !

brad, 2019-12-10:
I'm still unable to connect to the AP, even after fiddling with the blacklist. Just me? Kernel 4.19.75 Where is the log where people are seeing the kernel panic? Is it the logs seen when running hostapd in debugging mode?

hamishmb, 2020-02-12:
Hi there,

I've recently been trying to use this to make an up to date tutorial for setting these adaptors up on Raspbian (https://www.hamishmb.com/blog/how-to-make-a-raspi-ap-with-a-rtl8xxx-wifi-adaptor/) but I get a kernel panic as soon as anything connects to the AP. Is this a resolved/known issue or anything?

I might well be doing something wrong so if you have any advice please let me know :)

faf0, 2020-04-18:
@hamishmb Have you tried to use `driver=nl80211` instead of `driver=rtl871xdrv`? Using `driver=nl80211` worked for me out of the box without patching drivers or hostapd.

hamishmb, 2020-04-21:
Interesting, what chipset and kernel version was this with?

macmpi, 2020-04-21:
since 4.4.13 as [per this](https://github.com/oblique/create_ap/issues/249#issue-220466152) and referred details.

hamishmb, 2020-04-22:
Causes a kernel panic for me when I try to connect, much like with the modified hostapd :/

LewisLebentz, 2020-11-22:
@hamishmb did you ever get this working? Came across your blog while trying to get mine setup. Tried multiple different patches of hostapd, followed numerous different tutorials but can't get anything working on the Edimax-7811UN.

I am currently using the latest version of Raspbian. Tempted to try downgrading to a 5yr old one just to see if it makes a difference.

hamishmb, 2020-11-23:
Nope, never managed to get it working or complete that post, unfortunately.

I did once having it working on Raspbian Wheezy I think, so I guess it must not work with newer kernels of something. Unfortunately I'm not sure exactly what I did back then.

I'm not even sure if the repos are still up for older versions to be honest, maybe there's something like old-releases.ubuntu.com but for Raspbian.

ktp420, 2020-12-15:
FYI, I just fixed the kernel panic on connection, which seems to have started with kernel >=4.18.  If anyone is interested in testing it you can build the module for here:

https://github.com/ktp420/rtl8192cu-fixes/tree/fix_ap_crash

hamishmb, 2020-12-22:
I'd love to try it, but sods law happened: I let someone borrow the relevant WiFi dongle and they broke it.

My replacment doesn't use a Realtek chipset, but I'll have a shot at fixing the old one soon probably. If I get it working, I'll definitely try this out.