raspberrypi/linux #2981: I2c channel 0 not working on kernel 4.19

girante, 2019-05-21:
After the latest kernel update, the i2c devices that were previously working stopped being detected. 

I am currently using Raspberry pi Zero w, not sure about this also being an issue on other RPi models.

Pre-conditions:
- RPi zero W
- One or more i2c devices connected on the i2c channel 0 
- update to latest with apt update/upgrade
- Raspbian will be on kernel 4.19 after the reboot

Steps for reproduction:

1. Edit /boot/config.txt and add line:
"dtparam=i2c_vc=on"

2. Reboot

3. Run "i2cdetect -y 0"
- No devices are detected - 

4. Revert back to the latest 4.14 kernel with:
"sudo rpi-update a08ece3d48c3c40bf1b501772af9933249c11c5b"

5. Reboot

6. Run "i2cdetect -y 0"
- Your devices are detected and interaction with them is possible -

6by9, 2019-05-21:
Can you post the output of ```raspi-gpio get``` please?

pelwell, 2019-05-21:
I can confirm the problem - the firmware in the 4.19 image is leaving I2C0 mapped to GPIOs 28 & 29 as well as the expected GPIOs 0 & 1, which breaks the interface.
The problem can be temporarily fixed using:
```
$ raspi-gpio set 28-29 ip
```

6by9, 2019-05-21:
Ah, the kernel will be automatically probing for the camera as bcm2835-v4l2 is now a platform driver off vchiq.
Whilst there are times where it is useful to enable i2c0 and yet still have the firmware able to talk to it, I suspect it may be time to mask off firmware access should it be enabled in DT.

6by9, 2019-05-21:
Note that I'd only expect this will to happen if the camera is enabled (ie start_x.elf is in use). Without that the camera subsystem should never try probing things as no cameras are supported.

pelwell, 2019-05-21:
I'm not using start_x - this is a fresh Raspbian install with an rpi-updated kernel.

6by9, 2019-05-21:
My initial guess may be wrong then.
Quickest test would be to blacklist bcm2835-v4l2 and see if the GPIOs are left in the expected state.

pelwell, 2019-05-21:
In case it helps, here's the end of the firmware log:
```
002901.470: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
002924.388: Device tree loaded to 0x1bfea000 (size 0x5f2e)
002926.017: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
004669.711: vchiq_core: vchiq_init_state: slot_zero = 0x5b580000, is_master = 1
004674.580: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
004680.206: TV service:host side not connected, dropping notification 0x00000002, 0x00000001, 0x00000010
014806.176: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 40, power enable gpio = 44
015106.967: camsubs: Camera not found
015107.055: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 40, power enable gpio = 44
015407.845: camsubs: Camera not found
015407.940: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 40, power enable gpio = 44
015408.700: camsubs: Camera not found
015408.779: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 40, power enable gpio = 44
015409.552: camsubs: Camera not found
015409.644: camsubs: Looking for camera 0: i2c_port = 0, led gpio = 40, power enable gpio = 44
015410.403: camsubs: Camera not found
```

6by9, 2019-05-21:
OK, so camera_subsystem is the culprit, and is running even though NUM_CAMERAS seems to be 0 based on "vcgencmd get_camera".

pelwell, 2019-05-21:
Blacklisting bcm2835-v4l2 seems to be a workaround.

6by9, 2019-05-21:
Just firing up the debugger to see why camera_subsystem seems to be running when the number of supported cameras is 0.

girante, 2019-05-21:
[Result from unning raspi-gpio get on kernel 4.14]:
BANK0 (GPIO 0 to 27):
GPIO 0: level=1 fsel=4 alt=0 func=SDA0
GPIO 1: level=1 fsel=4 alt=0 func=SCL0
GPIO 2: level=1 fsel=4 alt=0 func=SDA1
GPIO 3: level=1 fsel=4 alt=0 func=SCL1
GPIO 4: level=1 fsel=0 func=INPUT
GPIO 5: level=1 fsel=0 func=INPUT
GPIO 6: level=0 fsel=1 func=OUTPUT
GPIO 7: level=1 fsel=1 func=OUTPUT
GPIO 8: level=1 fsel=0 func=INPUT
GPIO 9: level=0 fsel=0 func=INPUT
GPIO 10: level=0 fsel=0 func=INPUT
GPIO 11: level=0 fsel=0 func=INPUT
GPIO 12: level=1 fsel=0 func=INPUT
GPIO 13: level=0 fsel=0 func=INPUT
GPIO 14: level=1 fsel=2 alt=5 func=TXD1
GPIO 15: level=1 fsel=2 alt=5 func=RXD1
GPIO 16: level=1 fsel=0 func=INPUT
GPIO 17: level=0 fsel=0 func=INPUT
GPIO 18: level=0 fsel=0 func=INPUT
GPIO 19: level=0 fsel=0 func=INPUT
GPIO 20: level=0 fsel=0 func=INPUT
GPIO 21: level=1 fsel=0 func=INPUT
GPIO 22: level=0 fsel=0 func=INPUT
GPIO 23: level=1 fsel=0 func=INPUT
GPIO 24: level=1 fsel=0 func=INPUT
GPIO 25: level=1 fsel=1 func=OUTPUT
GPIO 26: level=0 fsel=0 func=INPUT
GPIO 27: level=0 fsel=0 func=INPUT
BANK1 (GPIO 28 to 45):
GPIO 28: level=1 fsel=0 func=INPUT
GPIO 29: level=1 fsel=0 func=INPUT
GPIO 30: level=0 fsel=7 alt=3 func=CTS0
GPIO 31: level=0 fsel=7 alt=3 func=RTS0
GPIO 32: level=1 fsel=7 alt=3 func=TXD0
GPIO 33: level=1 fsel=7 alt=3 func=RXD0
GPIO 34: level=0 fsel=7 alt=3 func=SD1_CLK
GPIO 35: level=1 fsel=7 alt=3 func=SD1_CMD
GPIO 36: level=1 fsel=7 alt=3 func=SD1_DAT0
GPIO 37: level=1 fsel=7 alt=3 func=SD1_DAT1
GPIO 38: level=1 fsel=7 alt=3 func=SD1_DAT2
GPIO 39: level=1 fsel=7 alt=3 func=SD1_DAT3
GPIO 40: level=0 fsel=1 func=OUTPUT
GPIO 41: level=1 fsel=1 func=OUTPUT
GPIO 42: level=0 fsel=0 func=INPUT
GPIO 43: level=0 fsel=4 alt=0 func=GPCLK2
GPIO 44: level=0 fsel=1 func=OUTPUT
GPIO 45: level=1 fsel=1 func=OUTPUT
BANK2 (GPIO 46 to 53):
GPIO 46: level=1 fsel=0 func=INPUT
GPIO 47: level=0 fsel=1 func=OUTPUT
GPIO 48: level=0 fsel=4 alt=0 func=SD0_CLK
GPIO 49: level=1 fsel=4 alt=0 func=SD0_CMD
GPIO 50: level=1 fsel=4 alt=0 func=SD0_DAT0
GPIO 51: level=1 fsel=4 alt=0 func=SD0_DAT1
GPIO 52: level=1 fsel=4 alt=0 func=SD0_DAT2
GPIO 53: level=1 fsel=4 alt=0 func=SD0_DAT3

[Result from unning raspi-gpio get on kernel 4.19]:
BANK0 (GPIO 0 to 27):
GPIO 0: level=1 fsel=0 func=INPUT
GPIO 1: level=1 fsel=0 func=INPUT
GPIO 2: level=1 fsel=4 alt=0 func=SDA1
GPIO 3: level=1 fsel=4 alt=0 func=SCL1
GPIO 4: level=1 fsel=0 func=INPUT
GPIO 5: level=1 fsel=0 func=INPUT
GPIO 6: level=0 fsel=1 func=OUTPUT
GPIO 7: level=1 fsel=1 func=OUTPUT
GPIO 8: level=1 fsel=0 func=INPUT
GPIO 9: level=0 fsel=0 func=INPUT
GPIO 10: level=0 fsel=0 func=INPUT
GPIO 11: level=0 fsel=0 func=INPUT
GPIO 12: level=0 fsel=0 func=INPUT
GPIO 13: level=0 fsel=0 func=INPUT
GPIO 14: level=1 fsel=2 alt=5 func=TXD1
GPIO 15: level=1 fsel=2 alt=5 func=RXD1
GPIO 16: level=1 fsel=0 func=INPUT
GPIO 17: level=0 fsel=0 func=INPUT
GPIO 18: level=0 fsel=0 func=INPUT
GPIO 19: level=0 fsel=0 func=INPUT
GPIO 20: level=0 fsel=0 func=INPUT
GPIO 21: level=1 fsel=0 func=INPUT
GPIO 22: level=0 fsel=0 func=INPUT
GPIO 23: level=1 fsel=0 func=INPUT
GPIO 24: level=1 fsel=0 func=INPUT
GPIO 25: level=1 fsel=1 func=OUTPUT
GPIO 26: level=0 fsel=0 func=INPUT
GPIO 27: level=0 fsel=0 func=INPUT
BANK1 (GPIO 28 to 45):
GPIO 28: level=1 fsel=4 alt=0 func=SDA0
GPIO 29: level=1 fsel=4 alt=0 func=SCL0
GPIO 30: level=0 fsel=7 alt=3 func=CTS0
GPIO 31: level=0 fsel=7 alt=3 func=RTS0
GPIO 32: level=1 fsel=7 alt=3 func=TXD0
GPIO 33: level=1 fsel=7 alt=3 func=RXD0
GPIO 34: level=0 fsel=7 alt=3 func=SD1_CLK
GPIO 35: level=1 fsel=7 alt=3 func=SD1_CMD
GPIO 36: level=1 fsel=7 alt=3 func=SD1_DAT0
GPIO 37: level=1 fsel=7 alt=3 func=SD1_DAT1
GPIO 38: level=1 fsel=7 alt=3 func=SD1_DAT2
GPIO 39: level=1 fsel=7 alt=3 func=SD1_DAT3
GPIO 40: level=0 fsel=1 func=OUTPUT
GPIO 41: level=1 fsel=1 func=OUTPUT
GPIO 42: level=0 fsel=0 func=INPUT
GPIO 43: level=1 fsel=4 alt=0 func=GPCLK2
GPIO 44: level=0 fsel=1 func=OUTPUT
GPIO 45: level=1 fsel=1 func=OUTPUT
BANK2 (GPIO 46 to 53):
GPIO 46: level=1 fsel=0 func=INPUT
GPIO 47: level=0 fsel=1 func=OUTPUT
GPIO 48: level=0 fsel=4 alt=0 func=SD0_CLK
GPIO 49: level=1 fsel=4 alt=0 func=SD0_CMD
GPIO 50: level=1 fsel=4 alt=0 func=SD0_DAT0
GPIO 51: level=1 fsel=4 alt=0 func=SD0_DAT1
GPIO 52: level=1 fsel=4 alt=0 func=SD0_DAT2
GPIO 53: level=1 fsel=4 alt=0 func=SD0_DAT3

Differences are what pelwell already pointed out: " the firmware in the 4.19 image is leaving I2C0 mapped to GPIOs 28 & 29 as well as the expected GPIOs 0 & 1,"

girante, 2019-05-21:
> Blacklisting bcm2835-v4l2 seems to be a workaround.

How does one do that?

6by9, 2019-05-21:
>>    Blacklisting bcm2835-v4l2 seems to be a workaround.

> How does one do that?

Create a file in /etc/modprobe.d (needs sudo) with a name such as bcm2835-v4l2.conf. The contents needs to read
```
blacklist bcm2835-v4l2
```

We're investigating why the firmware is actually probing for the camera when the support infrastructure isn't there.

6by9, 2019-05-21:
It didn't help that ```vcgencmd get_camera``` was lying to me about the number of supported cameras :-(

I believe I have a fix, so that will be up for internal review and release in due course.

Thanks for your report.

jaxxzer, 2021-04-21:
Is there any update or workaround for this?
> Blacklisting bcm2835-v4l2 seems to be a workaround.

I'm using the 5.10 kernel from the latest raspbian release and ran `raspi-update`.
I've put `dtparam=i2c_vc=on` in config.txt and blacklisted `bcm2835_v4l2`.

GPIO0 and GPIO1 remain configured as INPUT after boot. Further, I am unable to configure GPIO0 or GPIO1 as alternate function 0 (sda/scl0) using `raspi-gpio`.

pelwell, 2021-04-22:
The use of the i2cmux driver means that GPIOs 0&1 will remain as inputs _until /dev/i2c-0 is used_.

Try this (assuming `dtparam=i2c_vc=on` is in config.txt):
1. Reboot.
2. `raspi-gpio get 0-1,44-45`
3. `i2cdetect -y 0`
4. `raspi-gpio get 0-1,44-45`
5. `i2cdetect -y 10`
6. `raspi-gpio get 0-1,44-45`

See how I2C0 is initially enabled on 44-45, using i2c-0 switches it to 0&1, and i2c-10 switches it back again?

N.B. No blacklisting is needed.

6by9, 2021-04-22:
Unsurprisingly things change between 4.19 and 5.10, however the changes are generally documented in the [README](https://github.com/raspberrypi/linux/blob/rpi-5.10.y/arch/arm/boot/dts/overlays/README) for all the overlays, a copy of which is installed in /boot on all standard Pi images.

I2C controller 0 can be routed through to either GPIOs 0&1 on pins 27&28 of the 40 pin header, or to the camera and display ports. 5.4 and above kernels use the i2cmux_pinctrl kernel module to do that automagically, with GPIOs 0&1 being /dev/i2c-0, and the camera/display being /dev/i2c-10. The default muxing doesn't matter.

And Phil just beat me to hitting "comment".

6by9, 2021-04-22:
The base issue of the firmware probing the camera even when not running start_x was fixed back in May 2019, so closing this issue.

closed 2021-04-22 by 6by9