raspberrypi/linux #2982: Alsa amixer 4.00dB gain at 100%

entroPyth, 2019-05-21:
On Raspberry Pi 2 setting volume to 100% with amixer results in 4.00dB gain, which I believe should be 0.00dB.

Example:
    pi@moode:~ $ amixer -c 0 sset "'PCM'" 100%
    Simple mixer control 'PCM',0
    Capabilities: pvolume pvolume-joined pswitch pswitch-joined
    Playback channels: Mono
    Limits: Playback -10239 - 400
    Mono: Playback 400 [100%] [4.00dB] [on]

popcornmix, 2019-05-21:
This is deliberate.
The analogue audio on a Pi is quite quiet when used with non-amplifying headphones, and often audio files aren't full range so having a bit of extra boost is useful.

closed 2020-01-14 by P33M

chapterjason, 2020-12-17:
@popcornmix But how to change it?

popcornmix, 2020-12-18:
You can run alsamixer and adjust the volume.

chapterjason, 2020-12-29:
@popcornmix I mean how to change the 1-100% volume - db gain mapping to 0db gain at 100%

popcornmix, 2020-12-30:
Change the 400 to 0 and rebuild:
https://github.com/raspberrypi/linux/blob/40cc64a98a55ea75a011a1793675a788f63bad09/drivers/staging/vc04_services/bcm2835-audio/bcm2835-ctl.c#L12