raspberrypi/linux #2983: HMDI audio dropouts / distortion under certain scenarios in kernel 4.19

blue-smoke, 2019-05-21:
After the recent kernel update to 4.19 via apt-get update, my audio becomes distorted.
Playback of MP3 via VLC has brief dropouts and some clicking can intermittently be heard.
Remote desktop to a Windows PC via X-RDP results in the sound being distorted such that it no longer resembles the sound being played (eg sound of emptying recycle bin)

rolling back to 4.14 thus : 
`sudo rpi-update a08ece3d48c3c40bf1b501772af9933249c11c5b`
largely (but not completely) fixes the problem.  Some brief dropout in VLC playback, but X-RDP sound is working.

A clean install of Raspbian entirely cures the problem, but when certain additional software is installed via apt-get install, the problem returns.

Pre-conditions :

Raspberry Pi 3B+ booting from SD card, with Raspbian installed on external SSD

Fresh install of Raspbian
apt-get upgrade and reboot so that 4.19 kernel is in use

How to reproduce the problem : 

On desktop, play a known good MP3 file using VLC.  Note that it sounds good with **no** stuttering, distortion or dropouts.

Install the additional software that I use on my Pi : 

paste this list into a file pkglist.txt 
`acl
adduser
adwaita-icon-theme
akonadi-backend-mysql
akonadi-server
alacarte
alpine
alpine-doc
alsa-base
alsa-utils
ant
ant-optional
apt
apt-listchanges
apt-transport-https
apt-utils
aptitude
aptitude-common
aspell
aspell-en
autoconf
automake
autopoint
autotools-dev
avahi-daemon
base-files
base-passwd
bash
bash-completion
bind9-host
binutils
blends-tasks
blt
bluealsa
bluej
bluez
bluez-firmware
bsdmainutils
bsdutils
build-essential
bzip2
ca-certificates
catdoc
chromium-browser
chromium-browser-l10n
chromium-codecs-ffmpeg-extra
[... 1711 lines cut ...]
xserver-xorg-video-fbturbo
xxd
xz-utils
yelp
yelp-xsl
zenity
zenity-common
zip
zlib1g:armhf
zlib1g-dev:armhf
`
Use this command to install the list of software : 
`sudo apt-get install $(cat /home/pi/pkglist.txt | awk '{print $1}')`

Reboot the Pi

Play the same MP3 file as earlier, and note that there is dropout at the start of play, and additional (random?) dropout and clicks throughout.

Revert back to previous 4.14.98 kernel : 

`sudo rpi-update a08ece3d48c3c40bf1b501772af9933249c11c5b`

Reboot

Play the same MP3 file using VLC.  Note that there is brief dropout at start of file, but that continued playback is now without problem.

XECDesign, 2019-05-21:
And if you remove pulseaudio?

blue-smoke, 2019-05-21:
> And if you remove pulseaudio?

It works :)  Audio is fine now under 4.19.42.

I hadn't knowingly installed pulseaudio, I don't know where it came from (or what it does?)

Thanks for the quick response

XECDesign, 2019-05-21:
It's listed as a recommended package of some software, so it will sometimes get pulled in automatically when you install other things.

popcornmix, 2019-05-21:
Pulseaudio is generally undesirable if you don't need it (it's quite heavyweight, handling audio resampling/mixing/conversion).

But I'm still curious if the kernel made things worse.
You say the last 4.14 kernel is fine (a08ece3d48c3c40bf1b501772af9933249c11c5b).
Is the first 4.19 kernel good or bad (152b6b858d2f6b94fa398fa4545d5e130ab24434).

You can test with 
```sudo rpi-update a08ece3d48c3c40bf1b501772af9933249c11c5b```

blue-smoke, 2019-05-22:
The new kernel has made things worse when pulseaudio is installed.

Version                  | MP3 dropouts in 1st 2 secs | MP3 dropouts & clicks | rdp sound ok | rdp sound bad
------------------------|------------------------------------|-----------------------------|------------------|--------------------
4.14.98-v7+ #1200|                Y                          |                                    |         Y           |
4.19.23-v7+ #1202|                Y                          |                                    |         Y           |
4.19.42-v7+ #1218|                Y                          |               Y                   |                      |            Y

With pulseaudio removed, all kernel versions above perform fine, there are no audio dropouts AT ALL when playing mp3, and rdp sound is good.

notes : MP3 dropouts in 1st 2 secs means that there are 2, 3 or 4 brief dropouts when the file starts to play.  After this playback is unaffected

MP3 dropouts and clicks means that there are additional dropouts throughout the playback, and there are brief clicks or pops that resemble a vinyl record

RDP sound bad means that using rdesktop to remotely access a Windows PC on the Pi has strangely distorted sound.  Recycle bin empty noise sounds like a buzzing insect, default windows ping sounds like a scifi effect!

Let me know if there are further tests I can do.

popcornmix, 2019-05-22:
Okay, so something got worse between 1202 (4.19.23) and 1218 (4.19.42).
Can you confirm the version that made it worse?

4.19.37 (https://github.com/Hexxeh/rpi-firmware/commit/6cf2788e74b497c109865d052cd6fa68d094cf38) is perhaps the most likely update as it involved bringing in some upstream alsa fixes.

Can you check if that had the bad audio and whether the previous build (https://github.com/Hexxeh/rpi-firmware/commit/4bfbd6d739ade814c999a6e6b80e9318919ea460) was good?

blue-smoke, 2019-05-22:
Version                  | MP3 dropouts in 1st 2 secs | MP3 dropouts & clicks | rdp sound ok | rdp sound bad
------------------------|------------------------------------|-----------------------------|------------------|--------------------
4.14.98-v7+ #1200|                Y                          |                                    |         Y           |
4.19.23-v7+ #1202|                Y                          |                                    |         Y           |
4.19.36-v7+ #1213|                 Y                         |                                    | did not test    | did not test
4.19.37-v7+ #1214|          NO SOUND                |         NO SOUND        |  did not test   |  did not test
4.19.37-v7+ #1215|                Y                           |              Y                   | did not test    | did not test
4.19.42-v7+ #1218|                Y                          |               Y                   |                      |            Y

With 4.19.37 1214 there was no sound at all, and the volume control had a red X on it.
So I tried the next commit, taking me to 4.19.37 1215 and it's there that the mp3 playback gets the crackles and pops and far more dropouts.

The Windows PC isn't immediately available today so I didn't test the rdesktop sound for these 2 new tests.

blue-smoke, 2019-05-26:
Version                  | MP3 dropouts in 1st 2 secs | MP3 dropouts & clicks | rdp sound ok | rdp sound bad
------------------------|------------------------------------|-----------------------------|------------------|--------------------
4.14.98-v7+ #1200|                Y                          |                                    |         Y           |
4.19.23-v7+ #1202|                Y                          |                                    |         Y           |
4.19.36-v7+ #1213|                 Y                         |                                    |         Y           | 
4.19.37-v7+ #1214|          NO SOUND                |         NO SOUND        |  did not test   | did not test 
4.19.37-v7+ #1215|                Y                           |              Y                   |                      |            Y
4.19.42-v7+ #1218|                Y                          |               Y                   |                      |            Y

I've re-run the tests of 1213, 1215 and 1218, using rdesktop to remotely connect to a Windows PC.  The table above is updated.

So, it seems the sound problem got worse between 1213 and 1215 when using Raspbian with Pulseaudio installed.  

The rdesktop sound is very distorted 1215 onwards when Pulseaudio is present.

bftsystems, 2019-07-22:
I have encountered a similar issue, but a much simpler scenario, which may help troubleshooting.   Raspberry Pi B running Raspbian Lite Stretch, using 'aplay' to output notifications via the 3.5mm jack. Ran apt-get upgrade and updated to the 4.19 kernel tree, immediately started getting audio truncation issues (1-2 seconds of clip dropped on playback). Did some troubleshooting, and reverting to kernel 4.19.34 resolves the issues for me. Excluding kernel updates at the moment until this is fixed. 

Let me know if there's anything I can do to assist in resolving this issue.

dimitry-ishenko, 2019-08-03:
Sorry for dumb question, but do the numbers 1202, 1214, etc refer to? They link to PRs from 2015, which doesn't make sense.

I am also experiencing crackling when playing audio through HDMI with pulseaudio and would like to try one of the kernels. If I understand it correctly, running `sudo rpi-update c50c65e52bad69e8fa7d6576d9e2399eade7faf6` should give me the 4.19.36 kernel, which works well for @blue-smoke?

pelwell, 2019-08-03:
> Sorry for dumb question, but what do the numbers 1202, 1214, etc refer to? They link to PRs from 2015, which doesn't make sense.

They are kernel build numbers. Each kernel build is counted, and although the count resets when you move to a new directory or computer we try to adjust them to make a single sequence.

dimitry-ishenko, 2019-08-03:
No support for RPi4  before 4.19.55... :unamused: 

To elaborate, I've got an RPi4 running pulseaudio and likewise experience crackles when playing audio through HDMI. I had to manually enable HDMI output in `etc/pulse/default.pa`:
```init
load-module module-alsa-sink name=HDMI device=hw:0,1 channels=6
```
Tried various fixes suggested for pulse (eg, add `tsched=0`, disable `module-suspend-on-idle`, tried 2 channels instead of 6, etc) to handle crackling, but to no avail.

If I play audio straight through ALSA (eg, `mpv --audio-device=alsa/hw:0,1 ...`), bypassing pulse, the sound is OK.

I am set up to compile RPi kernel from source. @popcornmix if you are able to point me to the right commits, I can try reverting them to see if I can figure out the problem.

dimitry-ishenko, 2019-08-11:
Here is also PulseAudio logs with `tsched=0` [pulse_tsched=0.log](https://github.com/raspberrypi/linux/files/3490197/pulse_tsched.0.log).

I noticed the following in the logs:
`Using 4.0 fragments of size 13248 bytes (25.03ms), buffer size is 52896 bytes (99.95ms)`

Could it be that 25ms fragments are too small?

I will try playing around with the buffer and period settings later tonight, but if anyone sees any or suspect things, please holler.

sakaki-, 2019-08-27:
Just a quick question / possible workaround: per Darksky's suggestion on the Manjaro forum [here](https://forum.manjaro.org/t/manjaro-arm-19-08-released/99031/72), do the audio issues still occur under `pulseaudio` if you set `tsched=0` in `/etc/pulse/default.pa` as follows (and reboot)?

```ini
### Automatically load driver modules depending on the hardware available
.ifexists module-udev-detect.so
load-module module-udev-detect tsched=0
.else
```

Best, sakaki

bftsystems, 2019-09-01:
As this issue is still open, did more testing this weekend and confirmed the ALSA 5.1 kernel "back ports" in Raspbian kernel 4.19.37 are the culprit (at least for aplay on the command line - sound is perfect right up to 4.19.36+ on all my systems). Looking at the mainline Linux kernel, there are plenty of additional ALSA fixes in 5.2. Not being a kernel expert, nothing in the changelog seems like an obvious fix to this issue, so hoping someone with more knowledge of the ALSA code can look at it...

dimitry-ishenko, 2019-09-01:
@bftsystems have you tried setting it to 8 channels (7.1) to see if that makes a difference?

bftsystems, 2019-09-01:
@dimitry-ishenko I'm running Raspbian-Lite, so pulseaudio isn't even installed. Since kernel 4.19.37 in May, I've been having clipping issues using plain old 'aplay' from the terminal. Only resolution so far has been to roll-back to 4.19.36.  I've actually downloaded the kernel code and am building the kernel from source with the alsa fixes I found in the Linux 5.2 kernel, and will report back...

dimitry-ishenko, 2019-09-01:
@bftsystems I meant `aplay -c 8 ...`

bftsystems, 2019-09-02:
@dimitry-ishenko I tried that, and it didn't work. Interesting update though. I downloaded a copy or the latest kernel (4.19.69) from git, and the most recent official raspbian kernel source files which worked for me (linux-raspberrypi-kernel_1.20190401-1). I pulled the bcm2835-audio source files from the old kernel (everything under 'drivers/staging/vc04_services/bcm2835-audio/') and replaced the files in the new kernel. I rebuilt and reinstalled this new kernel on my 3B, and here's the thing, after rebooting, the sound is still "clipped" when played using aplay, so these changes don't seem to have caused the issue.

I looked into the change logs further, and noticed there was a further commit on May 2nd to fix the audio module not loading properly (issue labeled as 'Revert staging: bcm2835-audio: Drop DT dependency'). I reverted these changes as well, rebooted, and now my sound is working PERFECTLY under kernel 4.19.69.  So, it seems something in this DT dependency fix to the firmware blobs is actually what is causing the audio issues, not the back-ported alsa patches in kernel 4.19.36.

Not sure where to take this from here, or who to report this issue to get attention on it. Any thoughts???

dimitry-ishenko, 2019-09-02:
@bftsystems could you point me to that commit? I am no kernel expert, but I'd like to take a peek

bftsystems, 2019-09-02:
@dimitry-ishenko Well, I have good news and bad news. To confirm my findings, I downloaded a fresh copy of kernel 4.19.69, and reverted the ALSA code changes from Raspbian pull request #2948 (https://github.com/raspberrypi/linux/pull/2948). But to my surprise, after rebuilding the kernel, this fixed the sound issues without any additional changes. When I did this yesterday, it didn't fix the audio issues, so I must have missed something in my first attempt.  

The good news is that this confirms the sound issues are introduced by one of the patches in PR #2948. Now we just need to review the individual commits in PR #2948 and identify which is causing the issue.

dimitry-ishenko, 2019-09-03:
@bftsystems I would try running `git bisect` to see if you can nail it the exact commit. And then go from there. There are 44 commits in the PR, so should be able to hit it in around 6 hops.

bftsystems, 2019-09-04:
@dimitry-ishenko Git bisect...where have you been all my life? Thanks for the tip. After about 5 tries, I was able to isolate commit 'bcc5def6f0e80412edc624649c11e6f4dcf6643c' (bcm2835-audio: Operate non-atomic PCM ops) as the culprit. Unfortunately, it's the biggest commit in the group and makes a bunch of code changes which can't be reverted without affecting newer commits.  Should I submit an issue against this specific commit, or contact the author directly?

pelwell, 2019-09-04:
What do you think, @tiwai?

dimitry-ishenko, 2019-09-04:
Well at least now we know who the culprit is... ;)

bftsystems, 2019-09-11:
@pelwell @dimitry-ishenko @tiwai On a hunch, I have confirmed the issue seems to be in the new draining logic. I changed the code to bypass the new draining logic (see code change below) and this resolves the issue for me. WAV files played via aplay on the command line now play successfully without being truncated before the end of the sound clip.

```
case SNDRV_PCM_TRIGGER_DRAIN:
    alsa_stream->draining = true;
    /* return bcm2835_audio_drain(alsa_stream); */
    return 0;
```

I commented out `return bcm2835..` and restored the `return 0;` code to skip the new draining logic. As mentioned, this resolves the issue for me, not sure what else this would break for other folks.

JamesH65, 2019-09-11:
@bftsystems Since you are building kernels, and have a test setup, can you try, in drivers/staging/vc04_services/bcm2835-audio/bcm2835-vchiq.c, the following change

```
int bcm2835_audio_drain(struct bcm2835_alsa_stream *alsa_stream)
{
	struct vc_audio_msg m = {
		.type = VC_AUDIO_MSG_TYPE_STOP,
		.stop.draining = 1,  **<<<<<<<< change this to a 0.**
	};

	return bcm2835_audio_send_msg(alsa_stream->instance, &m, false);
}
```
Note, this is not a fix, as this I think needs to be changed in firmware, but should indicate a smoking gun.

bftsystems, 2019-09-11:
@JamesH65 James, I reverted my code changes in bcm2835-pcm.c and applied the changes you suggested. The issue is back, sound is once again clipped/truncated using aplay.  I hope that gives you the information you were looking for...

tiwai, 2019-09-11:
It implies that the chip (firmware) doesn't behave as expected regarding the draining.
The simplest workaround would be to remove SNDRV_PCM_INFO_TRIGGER_DRAIN from snd_bcm2835_playback_hw.info and snd_bcm2835_playback_spdif_hw.info flags.

popcornmix, 2019-09-11:
@tiwai I guess the question is what is the expected behaviour of the call?

The firmware code has audioplus which is quite an advanced sound layer that accepts inputs from multiple clients (there may be up to 8 alsa clients, plus openmax/mmal audio_render clients) and handles resampling and mixing together, plus dma to hdmi or pwm device.

The audioplus dma buffer allows up to a few hundred milliseconds of audio data (after the resampling and mixing). Once submitted to audioplus you can't retract your samples without flushing other clients audio (as it's already been mixed).

audioserver is what the alsa driver talks to (but not openmax/mmal). To reduce the amount of unretractable audio, it has its own queue of audio samples from each client and only submits them to audioplus when there is less than 50ms of latency on that channel.

``VC_AUDIO_MSG_TYPE_STOP`` causes audioserver to stop pushing audio from its queue to audioplus. If ``.stop.draining = 1`` then it also discards its current queue. As said before it can't retract the (up to) 50ms of audio already submitted to audioplus.

bftsystems, 2019-09-12:
@popcornmix Alright, last update on this today. I reverted all my code changes and only removed the SNDRV_PCM_INFO_DRAIN_TRIGGER flag as suggested by @tiwai, and the problem is resolved (as expected). So where do we go from here? I have confirmed this issue on an RPIB, W and 3B+, so it is not hardware specific. Should a commit be submitted to have the DRAIN_TRIGGER flag removed until a permanent fix is available? I'm no kernel developer, so looking to the experts for direction on this one...

tiwai, 2019-09-12:
Yes, the driver code needs the fix so that the existing devices work.  I'm going to submit the fix to upstream.
Once when the firmware gets some updates, we may consider restoring the feature, but I don't think it'd be needed.  After all, the drain-trigger is merely a would-be-nice feature for the device that supports such behavior.

popcornmix, 2019-09-24:
I've cherry-picked @tiwai's patch and it is in latest rpi-update kernel.
Can any users with issues test it and report back?

bftsystems, 2019-09-25:
Just ran rpi-update and rebooted. Everything working perfect, no sound issues to report. Here's the output of `uname -a` for confirmation.

Linux rpi3 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux

blue-smoke, 2019-10-16:
I'm a bit late coming back to this, sorry.  The original test Pi I was using has been repurposed, and I've been away.

So, to test the updates I installed the latest Raspbian with desktop on a Pi3B+ and let it update so it's running latest stock releases.

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux

```
MP3 playback perfect
install rdesktop
Use rdesktop to connect to Win 7 box. Sound ok - Win 7 Sample music "Kalimba - Mr Scruff" plays OK via remote desktop.

install pulseaudio
reboot

mp3 playback stutter at start for 1-2 seconds then playback is ok
rdesktop sound very distorted - sample music on Win7 box is unlistenable

So at this stage, things are improved with 4.19.75 #1270 .  The MP3 playback, which previously was stuttering and popping, is now just briefly stuttering as it begins playing, then settles down.  This is how it always used to be with Pulseaudio installed.

BUT the rdesktop sound is still very distorted.  Playing the Win7 sample music is unlistenable.  It's metallic (not simply low bitrate sounding, just.. weird)

I then tried rpi-update 

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.79-v7+ #1273 SMP Fri Oct 11 18:13:16 BST 2019 armv7l GNU/Linux
```

mp3 playback stutter at start for 1-2 seconds then playback is ok (as before)
But a new problem - rdesktop screen corrupted and unusable.

I rebooted the Windows 7 box, still the corrupted screen with rdekstop from the pi with 4.19.79 #1273

When I reverted back with
```
pi@raspberrypi:~ $ sudo apt-get install --reinstall raspberrypi-bootloader raspberrypi-kernel

pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux

```
mp3 playback stutter at start (as before)
rdesktop screen now OK, sound still distorted (as before)

And finally, removing pulseaudio fixes all the sound issues.

So..

MP3 playback is better with the current kernel + pulseaudio
rdesktop still has distorted sound with current kernel + pulseaudio
removing / not installing pulseaudio makes mp3 playback and rdesktop sound perfect
and something in 4.19.79 #1273 breaks rdesktop and the display is corrupted

rogergithubbing, 2020-11-28:
Hello, I don't know if the problems described in this forum could have the same origin (I'd made the latests posts there explaining that I have the same problem as the initial posting person):

https://www.raspberrypi.org/forums/viewtopic.php?f=43&t=256402&p=1769475#p1769475

In any case, what could be done to fix the issue described in the included link ? I'm very confused about this problem. How can I uninstall 'pulseaudio' to check if this fixes the issue ?