raspberrypi/linux #2984: vc4 kms issue for a DSI 4 lanes panel

aromanro, 2019-05-22:
Description:

I have a DSI panel, 4 lanes, running on DSI1 on compute module 3 that sits in the dev io board. I had to write a device driver for it, I managed to make it work (including the touch), but I have issues with it.

For some reason, the screen is 'teared' from time to time, in the sense that the image (or a part of it), 'jumps' sideways a distance that is proportional with hback-porch or the whole hfront-porch + sync + hback-porch size.

I tried to alter the horizontal timings, I've got it working with very different timings that I have for the panel, but that jump still occurs. I tried various flags for the 'mode', no matter what I did I couldn't get rid of those 'jumps'. Some settings do seem to be better than others, but no matter what I tried, I couldn't get rid of them completely. It seems to be cpu% related, if I open up a static web page and I let it just sit there opened, it doesn't seem to occur so much. The cpu% does not need to be high or sustained, sometimes even if I open up the desktop menu I get those jumps.

System:

Compute module 3 on the development IO board

cat /etc/rpi-issue 
Raspberry Pi reference 2019-04-08
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, e910bf55ac92e27c6e77043e8995bcb5e0a2d742, stage4

vcgencmd version
Feb 12 2019 22:12:51 
Copyright (c) 2012 Broadcom
version 8eff5e4023657a8b3b59e1f90dc966f62d74908c (clean) (release) (start)

uname -a
Linux raspberrypi 4.19.40-v7+ #8 SMP Thu May 16 13:39:13 EEST 2019 armv7l GNU/Linux

Please help! I'm not sure if I'm doing something wrong or it's something in the vc4 kms driver...

aromanro, 2019-05-22:
vcdbg log msg does this:

Failed to allocate -505734863 bytes for message buffer

aromanro, 2019-05-22:
in vc4_dsi.c, vc4_dsi_encoder_mode_fixup function is called several times. It ends up incrementing adjusted_mode->hsync_end and adjusted_mode->hsync_start several times, so they get bigger than htotal. Is this the intent?

I changed 

```
	adjusted_mode->hsync_end += adjusted_mode->htotal - mode->htotal;
	adjusted_mode->hsync_start += adjusted_mode->htotal - mode->htotal;
```

with

```
	adjusted_mode->hsync_end = mode->hsync_end + adjusted_mode->htotal - mode->htotal;
	adjusted_mode->hsync_start = mode->hsync_start + adjusted_mode->htotal - mode->htotal;
```

with no visible differences in behavior.

popcornmix, 2019-05-22:
Your code change is a no-op, so there won't be any change in behaviour.
@anholt should adjusted_mode->hsync_end/djusted_mode->hsync_start be reset somewhere if this gets called more than once?

aromanro, 2019-05-23:
Some more info on this, I don't know how relevant it is: I can drive the panel at 30 fps with no issue.

My guess is that the timings I have are not good because of the adjustments I mentioned above.
Is there a recipe to compute the proper timings? By looking into other panels drivers, I can see that the clock set is not the true 'pixel clock', but somehow higher.

aromanro, 2019-05-24:
Some more info about the issue:

I tried adjusting the timings and after a lot of attempts, I simply set back porch to 200. That's way too much, it wouldn't be possible to drive the panel with a vrefresh of 60. But with that setting, the issue went very hard to reproduce, I almost thought I got rid of it entirely.

Then I thought that adjusting the clock up to 83333 would be too high and that would create issues for the panel, so I tried to stay under the lower value where it gets adjusted, that is, 66666.
I can get that way to vrefresh 50 or even 55 and with some timings, the issue is hard to reproduce.

I'm not sure if this is an issue with the panel or with the video driver...

anholt, 2019-05-28:
The adjusted_mode should be reset at the start of modesetting every time:

```
	for_each_new_crtc_in_state(state, crtc, new_crtc_state, i) {
		if (!new_crtc_state->mode_changed &&
		    !new_crtc_state->connectors_changed)
			continue;

		drm_mode_copy(&new_crtc_state->adjusted_mode, &new_crtc_state->mode);
	}
```

aromanro, 2019-05-28:
I displayed the values after adjustment and they do get bigger and bigger as that function is called...

Do you have any suggestion why I have this odd behavior? I cannot get rid of it entirely and having a back porch set to 220 or so is odd (I really have no ideea what this is doing).

anholt, 2019-07-30:
My suggestion is that you trace how you're getting an adjusted_mode passed in to the adjust mode function that's already been adjusted.  At the start of a modeset I would expect it to have been reset to the original mode.

aromanro, 2019-08-01:
I already did this change:
```
	adjusted_mode->hsync_end = mode->hsync_end + adjusted_mode->htotal - mode->htotal;
	adjusted_mode->hsync_start = mode->hsync_start + adjusted_mode->htotal - mode->htotal;
```
but the tearing issue is still there, probably finding why it's not reset won't help with the original issue...

aromanro, 2020-12-16:
I'm closing this as probably won't find the cause here. I 'solved' it by dropping the refresh rate to 50Hz. The method works either by changing the clock setting or increasing sensibly the back porch.

6by9, 2020-12-16:
Hi @aromanro.
I'm looking at exactly the same bit of code myself in trying to get some other DSI devices working. I'll agree with you that ```vc4_dsi_encoder_mode_fixup ``` is being called twice with the same ```adjusted_mode```, and I ended up with a timing where ```hsync_end``` and ```hsync_start``` were greater than ```hsync_total``` - not good.

I'm still investigating where/why/what/how the state resets should happen, but for reference the two calls have the call stacks
```
[    9.825026] [<bf2d3dac>] (vc4_dsi_encoder_mode_fixup [vc4]) from [<bf25aa94>] (drm_atomic_helper_check_modeset+0x768/0xaf8 [drm_kms_helper])
[    9.825241] [<bf25a32c>] (drm_atomic_helper_check_modeset [drm_kms_helper]) from [<bf25c194>] (drm_atomic_helper_check+0x20/0xa4 [drm_kms_helper])
[    9.825435] [<bf25c174>] (drm_atomic_helper_check [drm_kms_helper]) from [<bf2db9b4>] (vc4_atomic_check+0x270/0x44c [vc4])
[    9.825739] [<bf2db744>] (vc4_atomic_check [vc4]) from [<bf1ad524>] (drm_atomic_check_only+0x488/0x930 [drm])
[    9.826184] [<bf1ad09c>] (drm_atomic_check_only [drm]) from [<bf1ad9ec>] (drm_atomic_commit+0x20/0x60 [drm])
[    9.826623] [<bf1ad9cc>] (drm_atomic_commit [drm]) from [<bf1c42fc>] (drm_client_modeset_commit_atomic+0x204/0x244 [drm])
[    9.827053] [<bf1c40f8>] (drm_client_modeset_commit_atomic [drm]) from [<bf1c442c>] (drm_client_modeset_commit_locked+0x6c/0x194 [drm])
[    9.827384] [<bf1c43c0>] (drm_client_modeset_commit_locked [drm]) from [<bf265184>] (drm_fb_helper_pan_display+0xe0/0x200 [drm_kms_helper])
[    9.827506] [<bf2650a4>] (drm_fb_helper_pan_display [drm_kms_helper]) from [<c07c648c>] (fb_pan_display+0xcc/0x154)
[    9.827541] [<c07c63c0>] (fb_pan_display) from [<c07d4c18>] (bit_update_start+0x24/0x40)
[    9.827566] [<c07d4bf4>] (bit_update_start) from [<c07d06ec>] (fbcon_switch+0x420/0x480)
[    9.827590] [<c07d02cc>] (fbcon_switch) from [<c081dc0c>] (redraw_screen+0x180/0x218)
[    9.827624] [<c081da8c>] (redraw_screen) from [<c081e7d8>] (do_bind_con_driver+0x350/0x3d4)
[    9.827651] [<c081e488>] (do_bind_con_driver) from [<c081ec40>] (do_take_over_console+0x174/0x20c)
[    9.827684] [<c081eacc>] (do_take_over_console) from [<c07d12b4>] (do_fbcon_takeover+0x94/0xf0)
[    9.827716] [<c07d1220>] (do_fbcon_takeover) from [<c07d4370>] (fbcon_fb_registered+0x150/0x154)
[    9.827740] [<c07d4220>] (fbcon_fb_registered) from [<c07c7428>] (register_framebuffer+0x1bc/0x2cc)
[    9.827858] [<c07c726c>] (register_framebuffer) from [<bf263e14>] (__drm_fb_helper_initial_config_and_unlock+0x3e0/0x57c [drm_kms_helper])
[    9.828073] [<bf263a34>] (__drm_fb_helper_initial_config_and_unlock [drm_kms_helper]) from [<bf2646ac>] (drm_fbdev_client_hotplug+0x100/0x180 [drm_kms_helper])
[    9.828287] [<bf2645ac>] (drm_fbdev_client_hotplug [drm_kms_helper]) from [<bf265358>] (drm_fbdev_generic_setup+0xb4/0x1a4 [drm_kms_helper])
[    9.828474] [<bf2652a4>] (drm_fbdev_generic_setup [drm_kms_helper]) from [<bf2d3460>] (vc4_drm_bind+0x1fc/0x208 [vc4])
[    9.828571] [<bf2d3264>] (vc4_drm_bind [vc4]) from [<c0843e3c>] (try_to_bring_up_master+0x178/0x1c8)
[    9.828597] [<c0843cc4>] (try_to_bring_up_master) from [<c0844338>] (component_master_add_with_match+0xe0/0x114)
[    9.828699] [<c0844258>] (component_master_add_with_match) from [<bf2d3514>] (vc4_platform_drm_probe+0xa
[    9.828794] [<bf2d346c>] (vc4_platform_drm_probe [vc4]) from [<c084daf0>] (platform_drv_probe+0x7c/0xb4)[    9.828805]  r10:00000000 r9:0000002b r8:bf302150 r7:c13fe100 r6:bf302150 r5:c1a5fc10
[    9.828828] [<c084da74>] (platform_drv_probe) from [<c084b96c>] (really_probe+0x220/0x3d0)
[    9.828851] [<c084b74c>] (really_probe) from [<c084bd0c>] (driver_probe_device+0x6c/0xc4)
[    9.828885] [<c084bca0>] (driver_probe_device) from [<c084bec4>] (__device_attach_driver+0x98/0xa8)
[    9.828907] [<c084be2c>] (__device_attach_driver) from [<c084971c>] (bus_for_each_drv+0x88/0xcc)
[    9.828929] [<c0849694>] (bus_for_each_drv) from [<c084b6c4>] (__device_attach+0xf0/0x158)
[    9.828953] [<c084b5d4>] (__device_attach) from [<c084bef0>] (device_initial_probe+0x1c/0x20)
[    9.828977] [<c084bed4>] (device_initial_probe) from [<c084a830>] (bus_probe_device+0x94/0x9c)
[    9.828991] [<c084a79c>] (bus_probe_device) from [<c084ad30>] (deferred_probe_work_func+0x74/0xa4)
[    9.829017] [<c084acbc>] (deferred_probe_work_func) from [<c023e9f8>] (process_one_work+0x180/0x4d0)
[    9.829041] [<c023e878>] (process_one_work) from [<c023fabc>] (worker_thread+0x54/0x5ac)
[    9.829074] [<c023fa68>] (worker_thread) from [<c02457a8>] (kthread+0x13c/0x168)
[    9.829104] [<c024566c>] (kthread) from [<c02000ec>] (ret_from_fork+0x14/0x28)
[    9.829242] [drm:vc4_dsi_encoder_mode_fixup [vc4]] *ERROR* Requested mode is clock 27500, htotal 542, sync_end 530, sync_start 518
[    9.829327] [drm:vc4_dsi_encoder_mode_fixup [vc4]] *ERROR* Adjusted mode is clock 31250, htotal 615, sync_end 603, sync_start 591. Adjustment 73

[   17.783996] [<bf2d3dac>] (vc4_dsi_encoder_mode_fixup [vc4]) from [<bf25aa94>] (drm_atomic_helper_check_modeset+0x768/0xaf8 [drm_kms_helper])
[   17.784217] [<bf25a32c>] (drm_atomic_helper_check_modeset [drm_kms_helper]) from [<bf25c194>] (drm_atomic_helper_check+0x20/0xa4 [drm_kms_helper])
[   17.784421] [<bf25c174>] (drm_atomic_helper_check [drm_kms_helper]) from [<bf2db9b4>] (vc4_atomic_check+0x270/0x44c [vc4])
[   17.784730] [<bf2db744>] (vc4_atomic_check [vc4]) from [<bf1ad524>] (drm_atomic_check_only+0x488/0x930 [drm])
[... cut ...]
[   17.788796] [drm:vc4_dsi_encoder_mode_fixup [vc4]] *ERROR* Requested mode is clock 27500, htotal 542, sync_end 530, sync_start 518
[   17.788894] [drm:vc4_dsi_encoder_mode_fixup [vc4]] *ERROR* Adjusted mode is clock 31250, htotal 615, sync_end 676, sync_start 664. Adjustment 73
```

I do also have the Pi 7" display running, so I'll test with that and see whether the adjustment is being made twice there.
If it's a small adjustment then it's likely not to be noticeable if called twice, but start and end exceeding total is going to be messy.

6by9, 2020-12-16:
With CM4 and both HDMI and Pi 7" DSI panel connected I get 3 calls to the vc4_dsi_encoder_mode_fixup. Each one increments the porches.

[```mode_fixup```](https://elixir.bootlin.com/linux/latest/source/drivers/gpu/drm/drm_atomic_helper.c#L403) loops through the CRTCs in the state and resets ```adjusted_mode``` iff either ```new_crtc_state->mode_changed``` or ```new_crtc_state->connectors_changed``` are set.

It then loops through the connectors in the state and UNCONDITIONALLY calls ```funcs->mode_fixup``` on all the ```new_crtc_state``` of the connectors.

It then loops the crtcs again and calls funcs->mode_fixup iff  either ```mode_changed``` or ```connectors_changed``` are set.

It feels like the connectors loop should also check ```new_crtc_state->mode_changed``` or ```new_crtc_state->connectors_changed``` before calling fixups.

```
diff --git a/drivers/gpu/drm/drm_atomic_helper.c b/drivers/gpu/drm/drm_atomic_helper.c
index dbac8d4..25c63d8 100644
--- a/drivers/gpu/drm/drm_atomic_helper.c
+++ b/drivers/gpu/drm/drm_atomic_helper.c
@@ -430,6 +435,11 @@ mode_fixup(struct drm_atomic_state *state)
                new_crtc_state =
                        drm_atomic_get_new_crtc_state(state, new_conn_state->crtc);
 
+               if (!new_crtc_state->mode_changed &&
+                   !new_crtc_state->connectors_changed) {
+                       continue;
+               }
+
                /*
                 * Each encoder has at most one connector (since we always steal
                 * it away), so we won't call ->mode_fixup twice.
```

I'll ping Maxime for comment.

6by9, 2020-12-16:
Reopening as I'm investigating (sorry it's taken so long for us to be debugging DSI).

aromanro, 2020-12-16:
An information that might or might not help: since I implemented that panel driver, the hardware we're using has changed. The panel now has some other chip. Of course, I had to change the initialization sequence in order to have it working again... and again it exhibits the same issue at 60 Hz. At 50 Hz it works fine (or at least the issue is much, much harder to reproduce).

6by9, 2020-12-16:
Can I ask what resolution these panels are please? Ideally the full timings.

A shift sounds like the pixel valve messing up the timings, but the DSI link is slightly remoted from the raw pixel timings.

aromanro, 2020-12-16:
Both are 800x1280. 

I would have to dig a little for the timings of the old one, but the new one has for horizontal 18, 18, 18 respectively for front porch, sync len and back porch and for vertical, 30, 4, 10 for vfront porch, vsync len and vback porch respectively.
The clock for 60 Hz I tried to set at 67842 (I calculated it from htotal * vtotal * frame_rate / 1000). 

As far as I know the clock gets adjusted by the kms driver upwards towards an available value and it also adjusts the other timings accordingly (I think the front porch is enlarged, at least, to still obtain the required refresh rate).

6by9, 2020-12-16:
Based on that resolution I'd guess it to be one of two driver chips then, although there may be more.
I've had one of those (ILI9881) working at 720x1280 quite happily over two data lanes. I can't say that I'd noticed issues of tearing/sideways shifting.

Yes the KMS driver will increase the DSI link frequency to a suitable divisor of the incoming clock (2GHz IIRC), and then compensates for the clock speed change by increasing the front porch. 

Do note that I'm testing against the 5.10 kernel branch and on CM4. It shouldn't make much difference overall though as the DSI peripheral hasn't changed.

aromanro, 2020-12-16:
The last one is indeed based on ILI9881.