raspberrypi/linux #2985: New 5.2.0-rc1 kernel compile issues

Dark-Sky, 2019-05-22:
I have 2 issues cross compiling the new 5.2.0-rc1 kernel. Tried arch linux cross compiler gcc version 9.1.0 and the one from arm developer version 7 and get the same error here with rtl8192cu:

> drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/os_intfs.c:1013:22: error: initialization of 'u16 (*)(struct net_device *, struct sk_buff *, struct net_device *)' {aka 'short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *)'} from incompatible pointer type 'u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))' {aka 'short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))'} [-Werror=incompatible-pointer-types]
>  1013 |  .ndo_select_queue = rtw_select_queue,
>       |                      ^~~~~~~~~~~~~~~~
> drivers/net/wireless/realtek/rtl8192cu/os_dep/linux/os_intfs.c:1013:22: note: (near initialization for 'rtw_netdev_ops.ndo_select_queue')

The second error involves bcm2835-mmc.c and bcm2835-sdhost.c. It seems that mmiowb() was removed from include/asm-generic/io.h but not from bcm2835-mmc.c and bcm2835-sdhost.c:

https://github.com/raspberrypi/linux/commit/01e3b958efe85a26d9b1b77be3a0a1491bb4cb36#diff-134992f403b7974bf7cdfcf9c0e53e3a

I was able to get it to compile by re-entering the defines from io.h that was in an older kernel and it compiled but have not tested the kernel yet to see if all is good as it appears you may be going in a different direction now. 

In the pic below of meld I moved over the missing defines for mmiowb() but I did not pay attention and moved over the __io_br() define also so I have no idea or not if the missing __io_br() would have errored out also:

[image]

popcornmix, 2019-05-22:
Ah, I had fixed mmiowb locally but that didn't make it into the pushed version.
The rtl8192 one was only a warning with my compiler. I've added a patch that avoids that.
Try now.

Dark-Sky, 2019-05-22:
It compiled like a champ.

The only problem I have now is my Logitech wireless mouse does not work with this kernel but it is an upstream issue. It will not work on my x64 either with the new 5.2.0-rc1 kernel. Modules seem to load and no errors; guess maybe some other modules depends are not getting called.

Thanks again!

Dark-Sky, 2019-05-22:
When I went to compile for AARM64 This error came up:

>   CC      arch/arm64/kernel/asm-offsets.s
> In file included from ./arch/arm64/include/asm/io.h:209,
>                  from ./include/linux/scatterlist.h:9,
>                  from ./include/linux/dma-mapping.h:11,
>                  from arch/arm64/kernel/asm-offsets.c:24:
> ./include/asm-generic/io.h: In function 'inb':
> ./include/asm-generic/io.h:57:24: error: implicit declaration of function '__io_br' [-Werror=implicit-function-declaration]
>  #define __io_pbr()     __io_br()
>                         ^~~~~~~
> ./include/asm-generic/io.h:465:2: note: in expansion of macro '__io_pbr'
>   __io_pbr();
>   ^~~~~~~~
> cc1: some warnings being treated as errors

Dark-Sky, 2019-05-23:
@popcornmix disregard the last error it was my bad. I forgot to revert the changes I mentioned earlier. I reverted the changes I had done and the aarch64 kernel compiled just fine.

I am closing this as all got compiled.

Thanks for your help.

closed 2019-05-23 by Dark-Sky

Dark-Sky, 2019-05-23:
Well I closed it too soon. It had started the second stage building the modules when I closed it so I thought all was ok. It had a warning at the end something about dual modules with the same name:

> warning: same basename if the following are built as modules:
>   arch/arm64/lib/crc32.ko
>   lib/crc32.ko