raspberrypi/linux #2986: /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq reports wrong frequency when throttling

JsBergbau, 2019-05-23:
/sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq  shows wrong CPU frequency, takes throttling not into account
OS / Environment

Raspbian
uname -r
4.19.42-v7+

Steps To Reproduce

Execute sysbench --num-threads=4 --test=cpu --cpu-max-prime=200000 run to heat CPU up. When it reaches 80 °C CPU frequency gets throttled (on Pi 3B+ beginning with 60 °C frequency gets reduced from 1400 MHz to 1200 MHz). cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq results always in 1400 MHz That always the unthrottled frequency. On Raspberry the currenct frequency can be measured via /opt/vc/bin/vcgencmd measure_clock arm 

Expected behavior

report real frequency with respecting throttling 

This issue comes from https://github.com/netdata/netdata/issues/6069 where it was stated that it is a bug in firmware of raspberry Pi

giuliomoro, 2024-01-19:
this seems to still be the case on the Raspberry Pi 5 

```

Raspberry Pi 5 Model B Rev 1.0
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"

Raspberry Pi reference 2023-10-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 962bf483c8f326405794827cce8c0313fd5880a8, stage4

Linux raspberrypi 6.1.65-v8-16k-_giulio-rt+ #6 SMP PREEMPT_RT Thu Dec  7 01:20:31 UTC 2023 aarch64 GNU/Linux
Revision        : c04170
Serial          : f4f12a9b9e056981
Model           : Raspberry Pi 5 Model B Rev 1.0
Throttled flag  : throttled=0xe0000
Camera          : vc_gencmd_read_response returned -1 error=1 error_msg="Command not registered"

```

pelwell, 2024-01-20:
Explain why this behaviour is a problem for you.

JsBergbau, 2024-01-20:
Isn't that obivious when something wrong is displayed?

pelwell, 2024-01-20:
The figure is the frequency that the governor has selected for the main CPU clock. In this case it has selected 1400, and that is the figure reported here. However, the governor is not aware that the actual frequency is lower because the firmware is trying to prevent the SoC from melting.

The governor interface to the system is probably (it's worth confirming) not designed to double-check that the value it has requested is not being honoured, hence the discrepancy. I can tell you how to disable the firmware's thermal throttling, or you can investigate ways to configure Linux's built-in thermal throttling to achieve the same effect, or perhaps just try to cool your SoC more effectively - it will last longer. We have no incentive to change the current behaviour.

popcornmix, 2024-01-20:
You can use `vcgencmd measure_clock arm` to report the actual frequency.
`/sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq` will report the frequency requested by the arm (without throttling) and there's no simple way of changing that.

closed 2024-01-20 by pelwell

giuliomoro, 2024-01-20:
> Explain why this behaviour is a problem for you.

The OP referenced this comment https://github.com/netdata/netdata/issues/6069#issuecomment-494886253 which stated that this value is supposed to reflect the actual hw frequency. So I thought I'd chime in and say that years after this issue had been opened and overlooked, the issue was still there.

Now, after [reading further ](https://www.kernel.org/doc/html/v6.1/admin-guide/pm/cpufreq.html), I know that:

> scaling_cur_freq
> Current frequency of all of the CPUs belonging to this policy (in kHz).
> In the majority of cases, this is the frequency of the last P-state requested by the scaling driver from the hardware using the scaling interface provided by it, which may or may not reflect the frequency the CPU is actually running at (due to hardware design and other limitations).

So I no longer see this as an issue.

I'll open a separate issue for `cpuinfo_cur_freq`.