raspberrypi/linux #2988: [4.19.42] bcm2835-aux-uart 3f215040.serial: irq not found - -517

WaeCo, 2019-05-24:
I am working an CM3 and after switching kernel from `4.14.97-v7+` to `4.19.42-v7+` (tag: raspberrypi-kernel_1.20190517-1) aux uart stopped working.
dmsg repeatedly shows:
> bcm2835-aux-uart 3f215040.serial: irq not found - -517

From config.txt
```
enable_uart=1
core_freq=250
dtoverlay=uart1,txd1_pin=14,rxd1_pin=15
```

cmdline.txt includes `console=serial1,115200`.

Before the update kernel messages where logged to the aux uart and a serial getty was spawned on /dev/ttyS0.

After upgrade non of the above worked and when trying to access /dev/ttyS0 I get 
```
$ sudo stty -F /dev/ttyS0
stty: /dev/ttyS0: Input/output error
```

uart0 is not affected by the update and continued to work normally.

Not sure if related but I also get other IRQ related messages in dmsg:
```
[    0.005999] arch_timer: WARNING: Invalid trigger for IRQ161, assuming level low
[    0.006004] arch_timer: WARNING: Please fix your firmware
```

That made me wonder and I tried different firmware versions 
- master (commit 69a759d39aa802f65d0cc578361ff73e9dcbedc4, Mar 27 2019 15:45:53 version 2e98b31d18547962e564bdf88e57b3df7085c29b)
- stable (commit e1900836948f6c6bdf4571da1b966a9085c95d37, Mar 27 2019 15:45:53 version 2e98b31d18547962e564bdf88e57b3df7085c29b)
- next (commit 3bcd3cdccc36babf6579574d49f9c1e7d05c9afc, Feb 12 2019 22:12:51 version 8eff5e4023657a8b3b59e1f90dc966f62d74908c)

But the result was always the same.

full dmsg log: https://pastebin.com/FW8kJ99S

6by9, 2019-05-24:
-517 is EPROBE_DEFER - it's waiting for the resource (an irq in this case) to become available before it will allow the device to probe fully.

lategoodbye, 2019-05-25:
I think the problem is that the irq workaround isn't applied anymore https://github.com/raspberrypi/linux/commit/ecb6033f19bcdd44cfee7891700dda085c61148f but the devicetree blob still expected it.

WaeCo, 2019-05-27:
Indeed using the dtb compiled by the kernel fixed the issue. I guess the firmware repo is just not in sync.

closed 2019-05-27 by WaeCo