raspberrypi/linux #2989: i2c-rtc overlay no longer loading rtc-abx80x after upgrade to kernel 4.19

clystron, 2019-05-27:
**Describe the bug**
I have been using the i2c-rtc overlay with an ab1805 RTC ("dtoverlay=i2c-rtc,abx80x"), after upgrading the kernel from 4.14 (raspberrypi-kernel=1.20190401-1) to 4.19 (raspberrypi-kernel=1.20190517-1) the module for the RTC does not get loaded any more. It looks like something is up with the MODALIAS and therefor the module is not found, see below for an output of udevadm monitor -p -k -u when manually loading the overlay for both kernels

**Expected behaviour** (Kernel 4.14)
<pre>
monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[65.567354] add      /devices/platform/soc/3f804000.i2c/i2c-1/1-0069 (i2c)
ACTION=add
DEVPATH=/devices/platform/soc/3f804000.i2c/i2c-1/1-0069
MODALIAS=i2c:abx80x
OF_COMPATIBLE_0=abracon,abx80x
OF_COMPATIBLE_N=1
OF_FULLNAME=/soc/i2c@7e804000/abx80x@69
OF_NAME=abx80x
SEQNUM=1251
SUBSYSTEM=i2c

KERNEL[65.575146] add      /module/rtc_abx80x (module)
ACTION=add
DEVPATH=/module/rtc_abx80x
SEQNUM=1252
SUBSYSTEM=module

UDEV  [65.578348] add      /module/rtc_abx80x (module)
ACTION=add
DEVPATH=/module/rtc_abx80x
SEQNUM=1252
SUBSYSTEM=module
USEC_INITIALIZED=65577929
net.ifnames=0

KERNEL[65.580915] add      /devices/platform/soc/3f804000.i2c/i2c-1/1-0069/rtc/rtc0 (rtc)
ACTION=add
DEVNAME=/dev/rtc0
DEVPATH=/devices/platform/soc/3f804000.i2c/i2c-1/1-0069/rtc/rtc0
MAJOR=253
MINOR=0
SEQNUM=1253
SUBSYSTEM=rtc

UDEV  [65.581144] add      /devices/platform/soc/3f804000.i2c/i2c-1/1-0069 (i2c)
ACTION=add
DEVPATH=/devices/platform/soc/3f804000.i2c/i2c-1/1-0069
MODALIAS=i2c:abx80x
OF_COMPATIBLE_0=abracon,abx80x
OF_COMPATIBLE_N=1
OF_FULLNAME=/soc/i2c@7e804000/abx80x@69
OF_NAME=abx80x
SEQNUM=1251
SUBSYSTEM=i2c
USEC_INITIALIZED=65571180
net.ifnames=0

KERNEL[65.581328] add      /bus/i2c/drivers/rtc-abx80x (drivers)
ACTION=add
DEVPATH=/bus/i2c/drivers/rtc-abx80x
SEQNUM=1255
SUBSYSTEM=drivers

UDEV  [65.583138] add      /bus/i2c/drivers/rtc-abx80x (drivers)
ACTION=add
DEVPATH=/bus/i2c/drivers/rtc-abx80x
SEQNUM=1255
SUBSYSTEM=drivers
USEC_INITIALIZED=65582468
net.ifnames=0

UDEV  [66.837197] add      /devices/platform/soc/3f804000.i2c/i2c-1/1-0069/rtc/rtc0 (rtc)
ACTION=add
DEVLINKS=/dev/rtc
DEVNAME=/dev/rtc0
DEVPATH=/devices/platform/soc/3f804000.i2c/i2c-1/1-0069/rtc/rtc0
MAJOR=253
MINOR=0
SEQNUM=1253
SUBSYSTEM=rtc
USEC_INITIALIZED=65583082
net.ifnames=0</pre>

**Actual behaviour** (Kernel 4.19)
<pre>monitor will print the received events for:
UDEV - the event which udev sends out after rule processing
KERNEL - the kernel uevent

KERNEL[52.549315] add      /devices/platform/soc/3f804000.i2c/i2c-1/1-0069 (i2c)
ACTION=add
DEVPATH=/devices/platform/soc/3f804000.i2c/i2c-1/1-0069
MODALIAS=of:N<NULL>T<NULL>Cabracon,abx80x
OF_COMPATIBLE_0=abracon,abx80x
OF_COMPATIBLE_N=1
OF_FULLNAME=/soc/i2c@7e804000/abx80x@69
OF_NAME=<NULL>
SEQNUM=1321
SUBSYSTEM=i2c

UDEV  [52.561359] add      /devices/platform/soc/3f804000.i2c/i2c-1/1-0069 (i2c)
ACTION=add
DEVPATH=/devices/platform/soc/3f804000.i2c/i2c-1/1-0069
MODALIAS=of:N<NULL>T<NULL>Cabracon,abx80x
OF_COMPATIBLE_0=abracon,abx80x
OF_COMPATIBLE_N=1
OF_FULLNAME=/soc/i2c@7e804000/abx80x@69
OF_NAME=<NULL>
SEQNUM=1321
SUBSYSTEM=i2c
USEC_INITIALIZED=52555233
net.ifnames=0</pre>

**System**
Pi3B

<pre>cat /etc/rpi-issue
Raspberry Pi reference 2019-04-08
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, e910bf55ac92e27c6e77043e8995bcb5e0a2d742, stage2

vcgencmd version
Mar 27 2019 15:45:53
Copyright (c) 2012 Broadcom
version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start)

uname -a
Linux raspberrypi 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux
</pre>

clystron, 2019-05-27:
Almost forgot:

I can load the module manually with modprobe and everything works fine but that should not be required with devicetree.

pelwell, 2019-05-27:
The failure to load the abx80x module is cause by upstream commit af503716ac1444db61d80cb6d17cfe62929c21df. The commit message asserts that no in-kernel drivers should be adversely affected by the change, but you appear to have found a counter-example. Personally, I think the fact that each driver has to register `i2c:` and `of:` aliases is ugly - the matter should have been handled in the I2C framework - but that's where we are. A quick glance suggests the problem still exists in the 5.2 kernel.

A bit of scripted greping says the following drivers are also affected:
```
a8293
ad714x_i2c
af9033
arizona_i2c
at24
au8522_decoder
cs53l32a
cx25840
cxd2099
cxd2820r
ds1621
ds2482
dvb_pll
e4000
edt_ft5x06
fc2580
goodix
hdc100x
helene
jc42
lgdt330x
m88rs6000t
mcp3422
mn88472
mn88473
msp3400
mt2060
mt9v011
mxl301rf
ov7640
pwm_pca9685
qm1d1b0004
qm1d1c0042
radio_si470x_i2c
radio_tea5764
rtc_fm3130
rtc_m41t80
rtc_max6900
rtc_pcf8563
rtc_pcf8583
rtc_x1205
s2250
saa7115
saa7706h
sc16is7xx
sht21
sht3x
shtc1
si2157
si2165
snd_soc_adau1701
snd_soc_adau1977_i2c
sony_btf_mpx
sp2
ssd1307fb
tc90522
tda10071
tda18212
tda18250
tef6862
ts2020
tsl4531
tua9001
tuner
tw2804
tw9903
tw9906
uda1342
veml6070
wl1273_core
wm8775
```
Of those, only 7 are currently used by overlays:
```
ds1621
goodix
hdc100x
mcp3422
sht3x
tsl4531
veml6070
```
The quick fix is to revert the offending patch - it does revert cleanly - but ultimately either the abx80x driver has to advertise `of:` aliases or the I2C framework has to be enhanced to take care of it.

pelwell, 2019-05-27:
Do you have any suggestions or comments, @martinezjavier?

martinezjavier, 2019-05-27:
@pelwell I think that the correct fix is to add a proper OF device ID table to the abx80x.

clystron, 2019-05-28:
I just saw that my original post did not show the real MODALIAS, this is how it looks exactly:
`MODALIAS=of:N<NULL>T<NULL>Cabracon,abx80x`

Does that look like a "good" OF alias? The NULLs do seem strange to me...

pelwell, 2019-05-28:
It does look strange, but that's how all OF aliases turn out - it's nothing to be concerned about.

OdysseusGE, 2019-07-21:
This issue also breaks module auto loading for i2c sound codecs like the cs4271.
What is the expected way for the snd_soc_cs4271_i2c module to be automatically loaded by udev?

pelwell, 2019-07-22:
Yes, this issue also affects other codecs. There is more discussion and a proposed fix in https://github.com/raspberrypi/linux/issues/3061. SPI could potentially have the same problem, although the framework isn't currently forcing the use of "of:" aliases, but if it did the same patch to i2cprobe will fix it.

spambake, 2019-07-22:
The SuperAudioBoard uses I2C not SPI.  The proposed fix repairs the problem with my spin off of the SuperAudioBoard which uses the SuperAudioboard overlay.