raspberrypi/linux #2992: Simple question: Is vc4hdmi ALSA sound card supposed to work by now?

vanfanel, 2019-05-28:
Hi there,

I use the vc4 overlay, and I get two ALSA "sound cards": "ALSA", which is created by the bcm2835 module (dtparam=audio=on), and "vc4hdmi".
The "ALSA" named card works, of course. But the "vc4hdmi" card does not: alsamixer reports it has no controls, and no software can output audio using that card.
I can get ONLY the "vc4hdmi" card if I disable "dtparam=audio=on" or otherwise prevent the bcm2835 module from loading, and then ALSA only reports the "vc4hdmi" card on the system, but even if it is the only card, it does not work either (no controls, libraries fail to open audio device, etc).

So, in a nutshell, does the "vc4hdmi" ALSA card work already?
Thing is, Eric Anholt already reported it is in kernel years ago, but he did not mention that it should be working.

I am using kernel rpi 4.19.37.
I am willing to update to a more recent kernel if I know for sure the vc4hdmi card works.

Thanks!

vanfanel, 2019-05-30:
Updated to latest kernel, and still the "vc4hdmi" card does not work at all.

popcornmix, 2019-05-30:
vc4hdmi is an upstream driver and not something we actively use. (bcm2835 is the supported option).

I have heard reports of people using using vc4hdmi, so it should be possible. e.g.
https://github.com/raspberrypi/linux/issues/2868

vanfanel, 2019-05-30:
@popcornmix : So using the upstream kernel instead of the raspberrypi kernel in this repo is better suited for VC4 graphics + vc4hdmi ALSA card?

6by9, 2019-05-30:
Isn't this that the vc4hdmi driver requires IEC958 packed data (```SNDRV_PCM_FMTBIT_IEC958_SUBFRAME_LE```), not raw PCM? 
There should be an asound.rc that sets up plughw to to the correct encapsulation of frames. The raw format doesn't permit mixing or volume control as those functions don't support the format.

With a Raspbian build I've just run ```aplay -v foo.wav``` and it's worked fine with ```dtoverlay=vc4-kms-v3d``` and no ```dtparam=audio=on```.
Output is 
```
pi@raspberrypi:~ $ aplay -v foo.wav
Playing WAVE 'foo.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Channels 6
ALSA <-> PulseAudio PCM I/O Plugin
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 6
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 24000
  period_size  : 6000
  period_time  : 125000
  tstamp_mode  : NONE
  tstamp_type  : GETTIMEOFDAY
  period_step  : 1
  avail_min    : 6000
  period_event : 0
  start_threshold  : 24000
  stop_threshold   : 24000
  silence_threshold: 0
  silence_size : 0
  boundary     : 1572864000
q^CAborted by signal Interrupt...
aplay: pcm_write:2053: write error: Interrupted system call
```

```
pi@raspberrypi:~ $ aplay -L -v
default
    Playback/recording through the PulseAudio sound server
null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
sysdefault:CARD=vc4hdmi
    vc4-hdmi, 
    Default Audio Device
front:CARD=vc4hdmi,DEV=0
    vc4-hdmi, 
    Front speakers
iec958:CARD=vc4hdmi,DEV=0
    vc4-hdmi, 
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=vc4hdmi,DEV=0
    vc4-hdmi, 
    Direct sample mixing device
dsnoop:CARD=vc4hdmi,DEV=0
    vc4-hdmi, 
    Direct sample snooping device
hw:CARD=vc4hdmi,DEV=0
    vc4-hdmi, 
    Direct hardware device without any conversions
plughw:CARD=vc4hdmi,DEV=0
    vc4-hdmi, 
    Hardware device with all software conversions
usbstream:CARD=vc4hdmi
    vc4-hdmi
    USB Stream Output
```
I suspect the plughw:CARD=vc4hdmi,DEV=0 entry is the critical one.

popcornmix, 2019-05-30:
> @popcornmix : So using the upstream kernel instead of the raspberrypi kernel in this repo is better suited for VC4 graphics + vc4hdmi ALSA card?

You will likely get the same behaviour with upstream or raspberry pi kernel - we have all the upstream drivers present. I'm just saying that the downstream bcm2835 driver is the default one in raspbian and the one we have most familiarity with. Others may be in a better position to say how to make vc4hdmi work.

vanfanel, 2019-05-30:
@6by9 : Can you please post your asound.rc then, please?
Is it usable as /etc/asound.conf?

6by9, 2019-05-30:
Not currently, but it's just the standard one from Raspbian.
Does ```aplay -L -v``` give a different output for you?

6by9, 2019-05-31:
It looks like my Raspbian system is using Pulseaudio to do the conversion (encapsulation) rather than plugins within alsa.

The upstream file that Eric references in https://github.com/anholt/linux/issues/152 is https://git.alsa-project.org/?p=alsa-lib.git;a=blob;f=src/conf/cards/vc4-hdmi.conf;h=027804a145d1ed9e4fe33dbb5fa7ef278230c328;hb=HEAD
There was a discussion at the time as to whether that was the correct way to do it, but I can't track that down at the moment.

vanfanel, 2019-05-31:
@6by9 
Pulseaudio... I will not use that. I would rather stop using audio on GNU/Linux at all.
Thanks a lot for the file!
Where should I put that on my Debian system? (I am not using Raspbian)

6by9, 2019-05-31:
> Pulseaudio... I will not use that. I would rather stop using audio on GNU/Linux at all.

Your choice, but then please don't look for much support here.

Thread found, but no real resolution. https://www.spinics.net/lists/alsa-devel/msg60478.html

vanfanel, 2019-05-31:
@6by9 
No need to add another abstraction layer to my system, ALSA does the job perfectly well.
Why is this the wrong place for support if I do not use PA? PA is not part of the Raspberry Pi Linux kernel.

popcornmix, 2019-05-31:
Probably worth reading the alsa thread. Bits that seemed important:

```
PulseAudio on the desktop seems to work fine.  aplay at the console
fails:

aplay -v  ~/LRMonoPhase4.wav
Playing WAVE '/home/anholt/LRMonoPhase4.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
aplay: set_params:1297: Sample format non available
Available formats:
- IEC958_SUBFRAME_LE

but if you specify iec958 it works:

aplay -v -D iec958  ~/LRMonoPhase4.wav 
Playing WAVE '/home/anholt/LRMonoPhase4.wav' : Signed 16 bit Little Endian, Rate 48000 Hz, Stereo
IEC958 subframe conversion PCM (IEC958_SUBFRAME_LE)
Its setup is:
  stream       : PLAYBACK
```

and:
```
And I'm afraid that dmix won't work with IEC958 subframe type.  dmix
can deal with the direct hw access, and it's limited with the normal
linear PCM types.  In theory, we can apply the dmix-style plugin onto
iec958 subframes, but the current code doesn't do it.

That said, the default should be rather passing iec958 or hdmi without
the mixing.  If the stream mixing is required, you need to use some
sound server instead.
```

vanfanel, 2021-11-03:
The ALSA sound card works, but HW only supports that IEC958 subframe audio format.

closed 2021-11-03 by vanfanel