raspberrypi/linux #2993: lirc_rpi got replaced by gpio-ir and gpio-ir-tx, how do I upgrade?

elmicha, 2019-05-29:
I found that lirc-rpi has been replaced by gpio-ir and gpio-ir-tx, and I found that I have to use "dtoverlay=gpio-ir-tx,gpio_pin=17" in /boot/config.txt to get the transmitter working like before. But I can't figure out how the receiver is supposed to work now.

I tried various values for the gpio_pull and rc-map-name options, but none worked.

**Describe the bug**
I rebooted. All IR transmission and receiving broke. I googled and found how to replace the transmission capability with gpio-ir-tx and it worked. I googled again for the receiving part and nothing worked.

**To reproduce**
Upgrade from a kernel before 4.19 to 4.19.x when you were happily using LIRC to send and receive IR codes.

**Expected behaviour**
Everything works like before, maybe after a bit of googling and replacing some module names and options.

**Actual behaviour**
Both transmitting and receiving broke without warning. No useful documentation to be found anywhere.

**System**
pi@rpi ~ $ raspinfo
raspinfo: command not found
pi@rpi ~ $ apt search raspinfo
Sortierung... Fertig
Volltextsuche... Fertig

pelwell, 2019-05-29:
That's a good question. I've had no reason to use IR remotes with a Pi except for testing, and although I did some work with the LIRC driver once I've not played with the replacement - gpio-ir.

I managed to find this LibreElec IR guide which may help: https://wiki.libreelec.tv/infrared_remotes#ir_remote_configuration

I also think that if you read that and come back with more questions then @HiassofT may be able to give you a few pointers.

HiassofT, 2019-05-29:
both gpio-ir and gpio-ir-tx will provide a /dev/lircX device so if you like you can use lircd as before - but you'll need to add udev rules to create symlinks for the receive/transmit device nodes if you enable both at the same time as the /dev/lircX numbers aren't guaranteed to be fixed (0 could be receive and 1 transmit, or the other way round). You'll also need to run 2 lircd instances (one for receive, one for transmit/blast).

For receiving I'd recommend to drop lircd and switch to in-kernel decoding / ir-keytable configuration. The majority of consumer remotes are supported and you only need to use lircd for a few non-standard remotes.

You can also use ir-ctl to transmit/blast IR codes instead of using lircd/irsend

Search the RPi forums for more details, I posted some info there

elmicha, 2019-05-30:
Thank you! I want to keep using lircd, because I already configured a few remotes with it.
For anyone who wants to keep using lircd for a transmitter and a receiver, here's what I did.

In /boot/config.txt remove the lirc-rpi line and add gpio-ir for the receiver and gpio-ir-tx or pwm-ir-tx for the transmitter:

    # dtoverlay=lirc-rpi
    dtoverlay=gpio-ir,gpio_pin=17
    # dtoverlay=gpio-ir-tx,gpio_pin=18
    dtoverlay=pwm-ir-tx,gpio_pin=18

I exchanged the gpio_pin numbers (and the wires of the transmitter and receiver), because pwm-ir-tx doesn't work with gpio pin 17. See /boot/overlays/README for the details.

Remove the lirc_rpi module from /etc/modules.

Add these rules in /etc/udev/rules.d/71-lirc.rules to get stable /dev/lirc-rx and /dev/lirc-tx device names:

    ACTION=="add", SUBSYSTEM=="lirc", DRIVERS=="gpio_ir_recv", SYMLINK+="lirc-rx"
    ACTION=="add", SUBSYSTEM=="lirc", DRIVERS=="gpio-ir-tx", SYMLINK+="lirc-tx"
    ACTION=="add", SUBSYSTEM=="lirc", DRIVERS=="pwm-ir-tx", SYMLINK+="lirc-tx"

Change the device and listening address in /etc/lirc/lirc_options.conf:

    device          = /dev/lirc-rx
    listen          = 0.0.0.0:8766

Copy lirc_options.conf to lirc_tx_options.conf and edit these lines:

    device          = /dev/lirc-tx
    output          = /var/run/lirc/lircd-tx
    pidfile         = /var/run/lirc/lircd-tx.pid
    listen          = 0.0.0.0:8765
    connect         = 127.0.0.1:8766

Create /etc/systemd/system/lircd-tx.service (from the output of `systemctl cat lircd`) and edit it to be:

    [Unit]
    Documentation=man:lircd(8)
    Documentation=http://lirc.org/html/configure.html
    Description=Second lircd, the transmitter
    Wants=lircd-setup.service
    After=network.target lircd-setup.service lircd.service

    [Service]
    Type=simple
    ExecStart=/usr/sbin/lircd --nodaemon --options-file /etc/lirc/lirc_tx_options.conf
    ; User=lirc
    ; Group=lirc

    ; Hardening opts, see systemd.exec(5). Doesn't add much unless
    ; not running as root.
    ;
    ; # Required for dropping privileges in --effective-user.
    ; CapabilityBoundingSet=CAP_SETEUID
    ; MemoryDenyWriteExecute=true
    ; NoNewPrivileges=true
    ; PrivateTmp=true
    ; ProtectHome=true
    ; ProtectSystem=full

    [Install]
    WantedBy=multi-user.target

Create /etc/systemd/system/lircd-tx.socket (from the output of `systemctl cat lircd.socket`) and edit it:

    [Socket]
    ListenStream=/run/lirc/lircd-tx

    [Install]
    WantedBy=sockets.target
    Also=lircd-tx.service

Start lircd-tx:

    sudo systemctl daemon-reload
    sudo systemctl start lircd-tx
    sudo systemctl enable lircd-tx

Create /usr/local/bin/irsend and make it executable:

     #! /bin/sh
     exec /usr/bin/irsend --device=/var/run/lirc/lircd-tx "$@"

closed 2019-05-30 by elmicha

majki09, 2019-05-30:
Hello. I'm also here with the same problem for receiving IR signal. After a week of banging my head against the wall I've noticed that:

- no IR signal is comming to RPi - I've checked the RAW signal with `ir-ctl -d /dev/lirc0 -r`
- my IR receiver (3-pinned AT138B connected to GPIO18) is "not enabled" - which means no **3.3V power** and no **pull-up** (my receiver requires pull-up to work).

So I'd rather look for the reason here and it doesn't make sense to go further with any configuration.

@elmicha, I suppose the problem is still there, can you please try the same with `ir-ctl -d /dev/lirc0 -r` ? This will check IR signal is indeed comming to your RPi, it's like `mode2` in lircd...
Also, you need to make sure that the **gpio-ir** overlay has been loaded with
`sudo vcdbg log msg`

For now I need to revert back my kernel to previous stable (4.14.y) with lircd **working** and I'm waiting for stable solution in new kernel release. You can downgrade your kernel using following command:
`sudo rpi-update c30ae2bb624f7fd60fcbedff950cc4361c8d2aec`

carloselectro, 2019-08-10:
Thank you for this Elmicha! 
I can finally make the IR receiving working in Buster and my raspberry 4.
Were you able to record new remotes with this command:?
irrecord -d /dev/lirc-rx

josla972, 2020-01-12:
This forum thread summarizes everything very well:

https://www.raspberrypi.org/forums/viewtopic.php?t=235256

Had to patch and rebuild the 4.19 kernel to get my AC control working again because of the #define LIRCBUF_SIZE limitation. Sounds like this is going to be fixed in future kernel versions but that is probably going to take a long while. Not really happy about this surprise.

xgridder, 2020-02-18:
Guys, I am giving up, tried everything but I can't make my transmitter to work.:)  I am using Raspbian Buster Lite, kernel  4.19. Lirc version is 0.10.1. I configured lirc as per @elmicha instructions and it works ok for receiver but for transmitter is not. The transmitter has been soldered exactly as per this instruction and identical parts: https://www.instructables.com/id/Raspberry-Pi-Zero-Universal-Remote/ 

I checked resistors and transistors with multi-meter and they are definitely ok.

Uncommitted these lines in /boot/config.txt as well:

`# Uncomment this to enable infrared communication.
dtoverlay=gpio-ir,gpio_pin=17
dtoverlay=gpio-ir-tx,gpio_pin=23`

The lirc log shows:

`Feb 18 02:29:49.054307 lircd: Notice: accepted new client on /var/run/lirc/lircd-tx
Feb 18 02:29:49.054836 lircd: Info: Cannot configure the rc device for /dev/lirc-tx
Feb 18 02:29:49.078133 lircd: Info: removed client`

In /var/log/messages:

`Feb 18 02:29:49 irsend: lirc_command_run: data:END, status:0`

I grabbed by Sony TV KEY_POWER button and irw reacts to its push. Now I want to send the same code with transmitter but nothing happens after irsend SEND_ONCE sony KEY_POWER

So, I don't really know where I need to dig in to resolve this issue. Any suggestions?