raspberrypi/linux #2995: dtoverlay w1-gpio command fails but config.txt still works

msiegen, 2019-06-02:
I am enabling the OneWire bus by running a command:
`/usr/bin/dtoverlay w1-gpio gpiopin=18`

After a system update, this stopped working. Editing /boot/config.txt to add the following line makes it work again:
`dtoverlay=w1-gpio,gpiopin=18`

I expected that running the dtoverlay command should work, even if the entry in config.txt is absent.

Logs are from a Pi1B but this also affects Pi1B+.
```
# cat /etc/rpi-issue 
Raspberry Pi reference 2019-04-08
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, e910bf55ac92e27c6e77043e8995bcb5e0a2d742, stage2
# uname -a
Linux devpi 4.19.42+ #1219 Tue May 14 21:16:38 BST 2019 armv6l GNU/Linux
# vcgencmd version
Mar 27 2019 15:45:53 
Copyright (c) 2012 Broadcom
version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start)
```

After running dtoverlay, the directory /sys/bus/w1/devices exists but is empty, notably missing the w1_bus_master1 symlink. dmesg contains:
```
[   88.155065] Driver for 1-wire Dallas network protocol.
[   88.168474] w1-gpio onewire@12: prop pinctrl-0 index 0 invalid phandle
[   88.168517] w1-gpio: probe of onewire@12 failed with error -22
```

Verbose invocation:
```
# dtoverlay -v w1-gpio gpiopin=18
run_cmd: which dtoverlay-pre >/dev/null 2>&1 && dtoverlay-pre
DTOVERLAY[debug]: Found override gpiopin
DTOVERLAY[debug]:   override gpiopin: cell target gpios @ offset 4 (size 4)
DTOVERLAY[debug]:   override gpiopin: cell target reg @ offset 0 (size 4)
DTOVERLAY[debug]:   override gpiopin: cell target brcm,pins @ offset 0 (size 4)
DTOVERLAY[debug]:   override gpiopin: cell target reg @ offset 0 (size 4)
DTOVERLAY[debug]: delete_node(/__symbols__)
DTOVERLAY[debug]: Wrote 1000 bytes to '/tmp/.dtoverlays/1_w1-gpio.dtbo'
DTOVERLAY[debug]: Wrote 1012 bytes to '/sys/kernel/config/device-tree/overlays/1_w1-gpio/dtbo'
run_cmd: which dtoverlay-post >/dev/null 2>&1 && dtoverlay-post
```

pelwell, 2019-06-03:
After reproducing this on a 4.19 kernel, and verifying that 4.14 worked, it took a lot of digging through the internals of kernel overlay application to pinpoint the breakage.

Commit f96278810150fc39085d1872e5b39ea06366d03e prevents properties from overlay nodes being duplicated directly into the kernel tree - instead, they are supposed to be added one at a time as a kind of mini overlay in a way which allows them to be subsequently removed. The patch was added to the 5.0 kernel along with 8814dc46bd9e347d4de55ec5bf8f16ea54470499, which explicitly copies the name and phandle (*) values (which are deliberately filtered by the usual property setting mechanism).

The former patch must have been considered enough of a bug fix to cause it to be back-ported (5b3109eb89dd800354481b76a1b856d303d3d2a8) into 4.19 - a Long-Term Stable kernel. Unfortunately the followup wasn't, which broke legitimate links between overlay fragments. In the case of the `w1-gpio` overlay, the lack of a phandle in the `w1_pins` node meant that reference from the `pinctrl-0` property of `onewire@0` didn't work, resulting in the error you saw:
```
[   88.168474] w1-gpio onewire@12: prop pinctrl-0 index 0 invalid phandle
```

The fix is (unsurprisingly) to back-port the other patch into rpi-4.19.y, which has now been done - see https://github.com/raspberrypi/linux/commit/8bff48328aff10c1e9c134a55337620929e449c2.

Thanks for reporting this, and so clearly. I'm surprised it hasn't come up before, because many other overlays will have been affected. I guess runtime overlay application is not widely used. I know there are limitations - some subsystems don't use the notification hooks that would allow them to detect an overlay being applied, so some overlays are effectively non-functional at runtime - but it can save lots of reboots when changing configurations.

(*) phandles are integers that act as bookmarks in a Flattened Device Tree file. Whenever the `dtc` compiler finds a label on a node it assigns it a unique phandle. Any reference to that label is replaced with the phandle value. When the FDT file is processed references can be matched with their target by searching for a node with the given phandle value.

popcornmix, 2019-06-03:
Latest rpi-update should have the fix.

turmary, 2019-06-19:
https://github.com/respeaker/seeed-voicecard/issues/166