raspberrypi/linux #2996: Pi3B+: very slow ethernet rx, unless switch has flow control turned on

sltam, 2019-06-02:
I just set up a file server on a Pi3B+ and uploading is very slow. I isolated the problem into slow ethernet. Then I stumbled upon #2449 and someone suggested enabling Flow Control on the switch (it's a Ubiquiti USW managed switch), and bravo it works. These are what I saw before and after turning on flow control: (fir is a mac at 10.0.0.6, and 10.0.0.16 is the pi)

```
sltam@fir 2019-06-02 01:52:17 ~$ iperf3 -c 10.0.0.16
Connecting to host 10.0.0.16, port 5201
[  5] local 10.0.0.6 port 51740 connected to 10.0.0.16 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  4.20 MBytes  35.2 Mbits/sec                  
[  5]   1.00-2.00   sec  5.74 MBytes  48.2 Mbits/sec                  
[  5]   2.00-3.00   sec  4.14 MBytes  34.7 Mbits/sec                  
[  5]   3.00-4.00   sec  4.62 MBytes  38.8 Mbits/sec                  
[  5]   4.00-5.00   sec  4.09 MBytes  34.3 Mbits/sec                  
[  5]   5.00-6.00   sec  4.09 MBytes  34.3 Mbits/sec                  
[  5]   6.00-7.00   sec  4.12 MBytes  34.6 Mbits/sec                  
[  5]   7.00-8.00   sec  4.22 MBytes  35.4 Mbits/sec                  
[  5]   8.00-9.00   sec  4.00 MBytes  33.6 Mbits/sec                  
[  5]   9.00-10.00  sec  3.98 MBytes  33.4 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec  43.2 MBytes  36.2 Mbits/sec                  sender
[  5]   0.00-10.00  sec  43.2 MBytes  36.2 Mbits/sec                  receiver

iperf Done.
```

```
sltam@fir 2019-06-02 01:52:40 ~$ iperf3 -c 10.0.0.16
Connecting to host 10.0.0.16, port 5201
[  5] local 10.0.0.6 port 51776 connected to 10.0.0.16 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  35.8 MBytes   301 Mbits/sec                  
[  5]   1.00-2.00   sec  35.9 MBytes   301 Mbits/sec                  
[  5]   2.00-3.00   sec  35.3 MBytes   296 Mbits/sec                  
[  5]   3.00-4.00   sec  34.6 MBytes   290 Mbits/sec                  
[  5]   4.00-5.00   sec  27.4 MBytes   230 Mbits/sec                  
[  5]   5.00-6.00   sec  34.9 MBytes   293 Mbits/sec                  
[  5]   6.00-7.00   sec  36.7 MBytes   308 Mbits/sec                  
[  5]   7.00-8.00   sec  37.4 MBytes   314 Mbits/sec                  
[  5]   8.00-9.00   sec  34.5 MBytes   289 Mbits/sec                  
[  5]   9.00-10.00  sec  36.6 MBytes   307 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec   349 MBytes   293 Mbits/sec                  sender
[  5]   0.00-10.00  sec   348 MBytes   292 Mbits/sec                  receiver

iperf Done.
```

Note that slowness only occurs when sending to the Pi, but not the other direction, i.e. if I add `-R` to the above commands, slowness doesn't happen.

**Expected behaviour**
Same max speed with and without flow control.

**Actual behaviour**
With flow control off, transfer is 10x slower.

**System**
There is no `raspinfo` command on my raspbian system, but I just did `rpi-update`

```
# cat /etc/apt/sources.list.d/raspi.list 
deb http://archive.raspberrypi.org/debian/ stretch main ui
```

* Which model of Raspberry Pi?
Pi3B+

* Which OS and version (`cat /etc/rpi-issue`)?
```
# cat /etc/rpi-issue 
Raspberry Pi reference 2017-04-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, acc4f56597e75225b93be3e1e20504a40565a9b9, stage5
```

* Which firmware version (`vcgencmd version`)?
```
# /opt/vc/bin/vcgencmd version
May 28 2019 16:21:27 
Copyright (c) 2012 Broadcom
version f37252430f9659d04032b6e0123d7950e3f9d92c (clean) (release) (start)
```

* Which kernel version (`uname -a`)?
```
# uname -a
Linux pi 4.19.46-v7+ #1230 SMP Tue May 28 16:18:48 BST 2019 armv7l GNU/Linux
```

**Logs**
Nothing interesting, except a couple of `Voltage normalised (0x00000000` and `Under-voltage detected! (0x00050005)`.

**Additional context**
```
# ethtool -a eth0
Pause parameters for eth0:
Autonegotiate:	on
RX:		on
TX:		on
RX negotiated:	on
TX negotiated:	on
```

```
# ethtool -i eth0
driver: lan78xx
version: 
firmware-version: 
expansion-rom-version: 
bus-info: usb-3f980000.usb-1.1.1
supports-statistics: yes
supports-test: no
supports-eeprom-access: yes
supports-register-dump: yes
supports-priv-flags: no
```

```
# ethtool -k eth0
Features for eth0:
Cannot get device udp-fragmentation-offload settings: Operation not supported
rx-checksumming: on
tx-checksumming: on
	tx-checksum-ipv4: off [fixed]
	tx-checksum-ip-generic: on
	tx-checksum-ipv6: off [fixed]
	tx-checksum-fcoe-crc: off [fixed]
	tx-checksum-sctp: off [fixed]
scatter-gather: on
	tx-scatter-gather: on
	tx-scatter-gather-fraglist: off [fixed]
tcp-segmentation-offload: off
	tx-tcp-segmentation: off [fixed]
	tx-tcp-ecn-segmentation: off [fixed]
	tx-tcp-mangleid-segmentation: off [fixed]
	tx-tcp6-segmentation: off [fixed]
udp-fragmentation-offload: off
generic-segmentation-offload: on
generic-receive-offload: on
large-receive-offload: off [fixed]
rx-vlan-offload: on
tx-vlan-offload: off [fixed]
ntuple-filters: off [fixed]
receive-hashing: off [fixed]
highdma: off [fixed]
rx-vlan-filter: on
vlan-challenged: off [fixed]
tx-lockless: off [fixed]
netns-local: off [fixed]
tx-gso-robust: off [fixed]
tx-fcoe-segmentation: off [fixed]
tx-gre-segmentation: off [fixed]
tx-gre-csum-segmentation: off [fixed]
tx-ipxip4-segmentation: off [fixed]
tx-ipxip6-segmentation: off [fixed]
tx-udp_tnl-segmentation: off [fixed]
tx-udp_tnl-csum-segmentation: off [fixed]
tx-gso-partial: off [fixed]
tx-sctp-segmentation: off [fixed]
tx-esp-segmentation: off [fixed]
tx-udp-segmentation: off [fixed]
fcoe-mtu: off [fixed]
tx-nocache-copy: off
loopback: off [fixed]
rx-fcs: off [fixed]
rx-all: off [fixed]
tx-vlan-stag-hw-insert: off [fixed]
rx-vlan-stag-hw-parse: off [fixed]
rx-vlan-stag-filter: off [fixed]
l2-fwd-offload: off [fixed]
hw-tc-offload: off [fixed]
esp-hw-offload: off [fixed]
esp-tx-csum-hw-offload: off [fixed]
rx-udp_tunnel-port-offload: off [fixed]
tls-hw-tx-offload: off [fixed]
tls-hw-rx-offload: off [fixed]
rx-gro-hw: off [fixed]
tls-hw-record: off [fixed]
```

```
# ethtool -S eth0
NIC statistics:
     RX FCS Errors: 0
     RX Alignment Errors: 0
     Rx Fragment Errors: 0
     RX Jabber Errors: 0
     RX Undersize Frame Errors: 0
     RX Oversize Frame Errors: 0
     RX Dropped Frames: 363
     RX Unicast Byte Count: 7825358317
     RX Broadcast Byte Count: 35190
     RX Multicast Byte Count: 2380172
     RX Unicast Frames: 5551863
     RX Broadcast Frames: 347
     RX Multicast Frames: 7816
     RX Pause Frames: 18
     RX 64 Byte Frames: 920
     RX 65 - 127 Byte Frames: 230755
     RX 128 - 255 Byte Frames: 7708
     RX 256 - 511 Bytes Frames: 2112
     RX 512 - 1023 Byte Frames: 8388
     RX 1024 - 1518 Byte Frames: 5310143
     RX Greater 1518 Byte Frames: 0
     EEE RX LPI Transitions: 0
     EEE RX LPI Time: 0
     TX FCS Errors: 0
     TX Excess Deferral Errors: 0
     TX Carrier Errors: 1
     TX Bad Byte Count: 0
     TX Single Collisions: 0
     TX Multiple Collisions: 0
     TX Excessive Collision: 0
     TX Late Collisions: 0
     TX Unicast Byte Count: 527398587
     TX Broadcast Byte Count: 1463
     TX Multicast Byte Count: 338950
     TX Unicast Frames: 1990556
     TX Broadcast Frames: 18
     TX Multicast Frames: 779
     TX Pause Frames: 1099056
     TX 64 Byte Frames: 2924
     TX 65 - 127 Byte Frames: 1709972
     TX 128 - 255 Byte Frames: 10869
     TX 256 - 511 Bytes Frames: 1274
     TX 512 - 1023 Byte Frames: 106
     TX 1024 - 1518 Byte Frames: 266207
     TX Greater 1518 Byte Frames: 1
     EEE TX LPI Transitions: 0
     EEE TX LPI Time: 0
```

6by9, 2019-06-02:
Documented at https://www.raspberrypi.org/documentation/faqs/#networking

Sorry, but your only other solution is do downgrade the link to 100Mbit/s.

The ethernet interface on all Pis is over a USB2 link, therefore you have a theoretical maximum of 480Mbit/s, but practically around 350Mbit/s. If you send data over the ethernet interface faster than that then the two options are to use flow control to back off the link, or to drop packets. The latter is the behaviour should you disable flow control.

JamesH65, 2019-06-03:
As 6by9 states, turning on flow control **IS** the correct fix for this problem. There is no other fix available/possible. Ethernet flow control was specifically created in order to fix this class of issues.

closed 2019-06-03 by JamesH65

vk5tu, 2019-06-03:
Just a heads-up that ethernet flow control is seen as a misfeature by many experts on end-to-end performance. It adds jitter to the packet drop signalling to TCP's pacing. So "is the correct fix" overstates the strength of the solution.

A question would be why the packet drop from the ethernet controller onto the bottleneck USB bus has such a high a jitter that the sending TCP can't get a correct estimate of the non-congesting transmission rate across the USB bus.  Does that controller retry multiple times, have exponential backoff, or some other media access algorithm which introduces substantial jitter?

JamesH65, 2019-06-03:
Feel free to examine the code and determine which of the suggested options might be causing the problem. Note that we simply use the drivers supplied by the manufacturer, with no changes, so as far as we are concerned, and after some testing, turning on flow control is our recomended fix. It's clear that it massively improves overall throughput (from 85MBits/s to about 250Mbits/s or more).

6by9, 2019-06-03:
For reference, the LAN7800 core that is used has a total of 12kB of receive FIFO. That's 98usecs should data be incoming at the full 1Gbit/s. Realistically any control mechanism therefore has to be built into the LAN adapter, which reduces the options down to flow control.

sltam, 2019-06-03:
Disclaimer: I am no networking expert, so I am not going to contribute to the technical discussion.

I was well aware of the USB 2.0 issue and I did read https://www.raspberrypi.org/documentation/faqs/#networking so I was set with an expectation of around 220–250Mbits/s, not 40.

The FAQ says: `To get the best performance, you should ensure that Ethernet flow control is turned ON on your router.`. If `turning on flow control IS the correct fix for this problem` is expected to get down to 40Mbps without flow control, it should read `To avoid horrible performance, [...]` instead.

So IMHO this issue needs to be either fixed in code, or in docs.

JamesH65, 2019-06-04:
Well, it cannot be fixed in code (AFAIK), and the docs already say you should turn on flow control, so I am not sure what else can be done.

sltam, 2019-06-04:
@JamesH65 As someone without much networking knowledge, the way I understood
> To get the best performance, you should ensure that Ethernet flow control is turned ON on your router.
is that throughput would be slightly affected without flow control. I did not expect a 10x slowdown to be the same issue, so I proceeded to ignore that for now and attempt to find the real culprit. Turned our I wasted hours until I finally stumbled upon #2449 which let me to try turning on flow control (which actually isn't a trivial thing to do: it's buried deep in some advanced settings in my UniFi system).

I mean yeah maybe I am stupid, but this FAQ could have been written better, right?

JamesH65, 2019-06-04:
To me its prety clear - to get best performance, turn on flow control. But I supose I coudl add something to indicate the symptoms of not having it on. Or you could provide you own PR on the docs repo with what you think it should say.