raspberrypi/linux #3002: When using kernel 4.19.X and LIRC, long IR sequences cannot be sent.

neuralassembly, 2019-06-05:
**Describe the bug**
When using kernel 4.19.X and sending IR commands with LIRC, the length of pulse-space IR sequence is restricted to be smaller than 256.
Therefore, the long IR sequence of remote controllers of air conditioners cannot be sent because its length is often longer than 256.

This is caused by the following line in [lirc_dev.c](https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/media/rc/lirc_dev.c).

`#define LIRCBUF_SIZE	256`

This definition did not exist in kernel 4.14.

I built a kernel 4.19.46-v7+ with LIRCBUF_SIZE=1024 and I confirmed that a long IR sequence can be sent correctly.

Therefore, I propose to set LIRCBUF_SIZE to the value 1024.

popcornmix, 2019-06-06:
This is an upstream linux file we don't modify here.
It would be best to fix this at source and we will get the fix once it is merged upstream.
Could you report it using this guild: https://www.kernel.org/doc/html/v4.19/admin-guide/reporting-bugs.html

neuralassembly, 2019-06-07:
Thanks. I e-mailed to the maintainer.

popcornmix, 2019-06-07:
Great. When a fix gets commited upstream let us know and we can pull it in.

neuralassembly, 2019-07-10:
I noticed that LIRCBUF_SIZE was increased to the value 1024 in the upstream file of master branch.

[https://github.com/torvalds/linux/commit/5c4c8b4a999019f19e770cb55cbacb89c95897bd#diff-3b71f634ae88214ee31a1b6c90f7df5c](https://github.com/torvalds/linux/commit/5c4c8b4a999019f19e770cb55cbacb89c95897bd#diff-3b71f634ae88214ee31a1b6c90f7df5c)

Is it possible to merge it? Or commit to the v4.19 tag is required?

realies, 2019-08-02:
When should we expect to see this merged @popcornmix?

RollMan, 2019-12-01:
If I see  `zcat /proc/config.gz|grep LIRC`, it seems that is embedded in the kernel (`CONFIG_LIRC=y`).

Is there any reason to embed that in the kernel? It would be useful if we can change buffer size by ourselves as module parameters (without building the whole kernel).

How about:

```
- CONFIG_LIRC=y
+ CONFIG_LIRC=m
```

MallocArray, 2020-02-21:
Any update on when this can be merged in?  As it stands, I'm unable to use 4.19 for both of my use cases of sending IR signals due to this limitation.

yipidee, 2020-04-09:
Is there any timeline on getting this merged? This really limits the utility of the rpi in air conditioning unit control

popcornmix, 2020-04-09:
Added https://github.com/raspberrypi/linux/commit/7f80928576fc456c0817df95ab6034003e846faa
will be in next rpi-update kernel.

yipidee, 2020-04-13:
Upgraded and tested with kernel 4.19.114-1-ARCH today. I am now able to send long IR signals like those used for AC units.

Thanks very much.

martynwheeler, 2020-04-22:
Just upgraded to 4.19.115+ and I am still sending garbled codes through the IR transmitter when sending NEC protocol codes.  Any ideas when this may be fixed please?

Thanks, Martyn

neuralassembly, 2020-04-22:
Patched version of lirc would be required. Please see:

https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=235256

martynwheeler, 2020-04-22:
still getting garbage using the patched version...  very frustrating....

neuralassembly, 2020-04-23:
Please note that the patched lirc would be downgraded when you execute "sudo apt dist-upgrade". In such a case, please install the patched deb files again.