raspberrypi/linux #3004: after reboot of the PiZeroW blocks many minutes on getrandom()

zoff99, 2019-06-07:
**Describe the bug**
after reboot of the PiZeroW it hangs about 5 minutes on the syscal
```getrandom()```

**To reproduce**
use libsodium call ```sodium_init()``` after reboot

**Expected behaviour**
on the rpi 3B+ this does not happen. i guess there is not enough entropy?
maybe something can be done to speed this up?
no mouse is connected to the PiZeroW

**Actual behaviour**
it takes about 5 minutes before the program continues

* Which model of Raspberry Pi?
PiZeroW

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-05-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 980bdb961bb72bd130f68fb2141a6cc095d1251d, stage2

* Which firmware version (`vcgencmd version`)?
Mar 27 2019 15:48:59 
Copyright (c) 2012 Broadcom
version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start_x)

* Which kernel version (`uname -a`)?
Linux rpi-B3121 4.19.42+ #1219 Tue May 14 21:16:38 BST 2019 armv6l GNU/Linux

**Logs**
dmesg:
```
[  210.074807] random: crng init done
[  210.074826] random: 7 urandom warning(s) missed due to ratelimiting
```

gdb:
```
(gdb) bt
#0  syscall () at ../sysdeps/unix/sysv/linux/arm/syscall.S:37
#1  0x004ec6f8 in _randombytes_linux_getrandom (size=<optimized out>, 
    buf=<optimized out>) at randombytes/sysrandom/randombytes_sysrandom.c:230
#2  randombytes_linux_getrandom (size=16, buf_=0xbed282b4)
    at randombytes/sysrandom/randombytes_sysrandom.c:247
#3  randombytes_sysrandom_init ()
    at randombytes/sysrandom/randombytes_sysrandom.c:267
#4  0x004ec8fc in randombytes_sysrandom_stir ()
    at randombytes/sysrandom/randombytes_sysrandom.c:295
#5  0x004eac1c in randombytes_stir () at randombytes/randombytes.c:91
#6  randombytes_init_if_needed () at randombytes/randombytes.c:49
#7  randombytes_stir () at randombytes/randombytes.c:89
#8  0x004eb9c8 in sodium_init () at sodium/core.c:40
```

zoff99, 2019-06-07:
```
#include <sodium.h>

int main(void)
{
    if (sodium_init() < 0) {
        /* panic! the library couldn't be initialized, it is not safe to use */
    }
    return 0;
}
```
```
gcc a.c -lsodium -o a
```

can be seen with this simple example

pelwell, 2019-06-07:
The system is probably stalling for lack of entropy. Please run:
```
$ sudo apt-get intall haveged
```
and try again.

zoff99, 2019-06-07:
@pelwell is that cryptographically secure?

zoff99, 2019-06-07:
there is a warning here:
https://wiki.archlinux.org/index.php/Haveged

XECDesign, 2019-06-07:
Recent images should have rng-tools installed also, so you're not relying purely on haveged, but also data coming in from hwrng and other sources the kernel uses.

zoff99, 2019-06-07:
i'll try rng-tools first. is the documentation on how secure rng-tools is?

XECDesign, 2019-06-07:
Depends on how much you trust the hw rng. I didn't see anything particularly conclusive while looking into it, but you can look at some of the links people have put here - https://security.stackexchange.com/questions/147482/why-isnt-rng-tools-installed-by-default-on-all-os

zoff99, 2019-06-07:
well at least rng-tools gets rid of the 5 minutes waiting.
so that's a good start.

thanks for the tips guys

jedisct1, 2019-06-07:
https://download.libsodium.org/doc/usage

zoff99, 2019-06-07:
@jedisct1 nice

zoff99, 2019-06-07:
since rng-tools solves my issue, and seems to be safe, i will close the issue.

closed 2019-06-07 by zoff99