raspberrypi/linux #3005: bcm2835_audio bcm2835_audio: Failed to get DT property 'brcm,pwm-channels'

jeffersonchua, 2019-06-08:
# dmesg
bcm2835_audio bcm2835_audio: Failed to get DT property 'brcm,pwm-channels'

... happens on Linux 5.2.0-rc3 with latest commit 060a2d44b2e6f0bc62137e11c89404e0a3c71a4f

How to get rid of this ...

Thanks,
Jeff

dxmann, 2019-08-01:
I have the same problem on 5.2.2.
Any news?

pelwell, 2019-08-01:
It's a consequence of upstream's push to make VCHIQ's clients platform drivers, which removes the ability to use Device Tree properties and forces the use of module parameters.

@lategoodbye can explain why this is a good thing.

dxmann, 2019-08-01:
ok, so how can I use fe-pi-audio? usually it is sufficent to load the dtoverlay on config.txt:
`dtoverlay=fe-pi-audio`

pelwell, 2019-08-01:
This error message comes from the driver that controls audio output via HDMI or the headphone socket - Fe-Pi is an I2S-based soundcard that should be completely unaffected.

pelwell, 2019-08-01:
Also, the i2s-mmap overlay no longer exists because it isn't needed.

Notwithstanding my earlier comment, in the case of the bcm2835-audio driver, the DT support that has been reverted back into downstream kernels since 5.0 doesn't seem to be doing anything useful (the brcm,pwm-channels property has the default value of 8), so it is doing more harm than good. Unless @popcornmix knows of any reason not to I'll get rid of them.

popcornmix, 2019-08-01:
@pelwell not entirely sure on the reasons for that revert, but if its doing no good then feel free to drop it.

pelwell, 2019-08-01:
It might have been picked up (again) as one of the mass "lostcommits" cherry-picks. I'll blacklist it to ensure it doesn't happen again.

dxmann, 2019-08-01:
so, is the error below related to missing audio devices?
```
[    4.283470] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[    4.285271] bcm2835_audio bcm2835_audio: Failed to get DT property 'brcm,pwm-channels'
[    4.296313] bcm2835_audio: probe of bcm2835_audio failed with error -22

```

pelwell, 2019-08-01:
It will break the analogue and (non-VC4) HDMI audio, but external soundcards should be unaffected.

dxmann, 2019-08-01:
ok, I'll check elsewhere

pelwell, 2019-08-01:
Ah - Fe-Pi uses the sgtl5000 codec, which means you'll be needing f6cc589, which hasn't yet been picked up by _one of the mass "lostcommits" cherry-picks_.

dxmann, 2019-08-01:
In the kernel checkout that I use, the fe-pi-audio-overlay has been patched with the [f6cc589](https://github.com/raspberrypi/linux/commit/f6cc5893c46f73e9e278abe96acf6c4e05b6d3e9) patch.

pelwell, 2019-08-01:
In that case I have no idea.

dxmann, 2019-08-01:
me too :) even with the kernel 4.19.60 I can't see any audio device. I'm using the rpi-kernel with voidlinux. I have installed alsa-lib alsa-utils and alsa-firmware. I don't know if I have missed any package. but it is not related to this issue....

pelwell, 2019-08-01:
Time for some systematic debugging:
1) Read the output of "sudo vcdbg log message" and check for errors with the overlay loading.
2) Run "raspi-gpio get" and confirm that GPIOs 28-31 are configured for I2S.
3) Run "dmesg -l err" and check for errors.
4) run "lsmod" and look for the codec and soundcard drivers.

pelwell, 2019-08-01:
Going back to the original topic for a moment, although the error message is fatal for the device instance, it doesn't break audio output. The reason is that two instances are being created - one via device tree that (with the revert patch) can find `brcm,pwm-channels`, and one created explicitly by VCHIQ as a platform device. We could undo the reversion of the upstream DT-exorcism commit and stop instantiating bcm2835-audio from DT, which is what upstream intends, but then we would lose `dtparam=audio=on/off` support.

dxmann, 2019-08-01:
ok

lategoodbye, 2019-08-11:
> It's a consequence of upstream's push to make VCHIQ's clients platform drivers, which removes the ability to use Device Tree properties and forces the use of module parameters.
> 
> @lategoodbye can explain why this is a good thing.

You already found one reason. Another one is that VCHIQ is a software interface to the firmware, so devicetree bindings will have a little chance to be accepted.

FunWithFlaggs, 2019-12-22:
Did i overread something or how can the original problem be solved?
`bcm2835_audio bcm2835_audio: Failed to get DT property 'brcm,pwm-channels'`

I try to get hwaccel video decoding + audio over hdmi but i can only get either of them, not both at the same time by either compiling the sources from here (to get hwaccel working, need `/dev/vcsm-cma`) or from https://git.archlinux.org/linux.git to get audio working.