raspberrypi/linux #3006: Could you please add CONFIG_NET_SCH_CAKE=m

wu-yue, 2019-06-08:
On  4.19.42-v7+ kernel, I see CONFIG_NET_SCH_CAKE is not set.

 CAKE (Common Applications Kept Enhanced) is a shaping-capable queue  discipline which uses both Active Queue Management and Fair Queuing.

It's a new thing merged into the kernel since 4.19 

Man page at man 8 tc-cake

I wish to have it because AQM and FQ could be loaded faster, as it has both of them in a single module. And it's NEW, I would like to try it.

pelwell, 2019-06-10:
We're open to new options and features, but I don't think it's reasonable to expect all users to be part of the experiment. Building your own kernel is quicker and easier than you might think - please follow the guide here: https://www.raspberrypi.org/documentation/linux/kernel/building.md

As a general rule of thumb, if a config option just causes a number of modules to be built then we will usually enable it fairly promptly; anything which makes the core kernel larger or slows it down, whether or not the new feature is used, will need much more justification.

If you find that the new options have a positive impact without penalising those who don't want or require the new behaviour then report back here with your findings. Until then I'm going to close the issue.

closed 2019-06-10 by pelwell

kucharskim, 2019-06-27:
CAKE is new network packet scheduler and is enabled in 4.19 in Debian Buster. It's `new` in kernel 4.19, as it was merged in that version into official Linux sources, but was in development for longer. For example Turris Omnia with their custom kernel 4.4 has CAKE module available. Main advantage which I'm looking for with CAKE is `autorate-ingress` capability, where scheduler tries to optimize packet scheduling based on dynamic bandwidth of the link. I'm targeting use of Raspberry Pi and newly released Raspian Buster with LTE modems, where bandwidth is dynamic. Buster is needed to have fresh enough iproute2 package.

kucharskim, 2019-07-15:
@pelwell, would you consider reopening this issue and enabling CAKE in further RPi kernels?