raspberrypi/linux #3007: pi3-disable-wifi overlay not working on 4.19.46

moodeaudio, 2019-06-08:
**Describe the bug**
Adding dtoverlay=pi3-disable-wifi to /boot/config.txt does not disable Pi-3 integrated WiFi adapter.

**To reproduce**
Add dtoverlay=pi3-disable-wifi to /bootconfig.txt then reboot.

**Expected behaviour**
Pi-3 integrated WiFi adapter is disabled and is not listed in ifconfig or iwconfig.

**Actual behaviour**
Adapter is still present in ifconfig
```
pi@moode:~ $ cat /boot/config.txt
disable_splash=1
hdmi_drive=2
dtparam=i2c_arm=on
dtparam=i2s=on
dtparam=audio=on
dtoverlay=pi3-disable-wifi
#dtoverlay=pi3-disable-bt
pi@moode:~ $ 
```
```
pi@moode:~ $ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.181  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::4cd8:7747:3040:1fdc  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:ec:46:b2  txqueuelen 1000  (Ethernet)
        RX packets 58  bytes 8292 (8.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 82  bytes 13082 (12.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 84  bytes 7606 (7.4 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 84  bytes 7606 (7.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b8:27:eb:b9:13:e7  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi-3B+
* Which OS and version (`cat /etc/rpi-issue`)?
pi@moode:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2018-11-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7e0c786c641ba15990b5662f092c106beed40c9f, stage2
* Which firmware version (`vcgencmd version`)?
pi@moode:~ $ vcgencmd version
May 28 2019 16:21:27 
Copyright (c) 2012 Broadcom
version f37252430f9659d04032b6e0123d7950e3f9d92c (clean) (release) (start)
* Which kernel version (`uname -a`)?
pi@moode:~ $ uname -a
Linux moode 4.19.46-v7+ #1230 SMP Tue May 28 16:18:48 BST 2019 armv7l GNU/Linux
**Logs**
```
pi@moode:~ $ dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.46-v7+ (dom@dom-XPS-13-9370) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1230 SMP Tue May 28 16:18:48 BST 2019
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[... 248 lines cut ...]
[    4.146393] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[    4.154323] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[    4.177031] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[    4.177048] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[    4.188791] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    4.202196] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[    4.202217] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[    4.221434] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[    4.221454] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[    4.259166] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    4.335516] brcmfmac: F1 signature read @0x18000000=0x15264345
[    4.347662] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    4.348203] usbcore: registered new interface driver brcmfmac
[    4.557219] random: crng init done
[    4.557235] random: 7 urandom warning(s) missed due to ratelimiting
[    4.578169] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    4.593033] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[    5.702527] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    5.702612] brcmfmac: power management disabled
[    5.998670] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    6.863812] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
pi@moode:~ $ 
```
**Additional context**
Possibly same for dtoverlay=pi3-disable-bt but I did not test this case.

pelwell, 2019-06-09:
This is a duplicate of https://github.com/raspberrypi/firmware/issues/1145, but as it's in a different repo you weren't to know, and `raspberrypi/linux` is actually a better home.

The bug was fixed in the firmware repo on June 6th, but that probably hasn't reached the Raspbian packages yet. In the meantime you can download the fixed version of the overlay [here](https://github.com/raspberrypi/firmware/blob/master/boot/overlays/pi3-disable-wifi.dtbo) - just copy it into `/boot/overlays`.

moodeaudio, 2019-06-09:
Thanks, that fixes the issue :-)

closed 2019-06-09 by moodeaudio