raspberrypi/linux #3010: Wireless clients in AP mode is very limited

martignoni, 2019-06-11:
I'm reporting this on Jamesh's advice (see [this forum discussion](https://www.raspberrypi.org/forums/viewtopic.php?f=63&t=241223&p=1472540&hilit=moodlebox#p1472522)).

**Describe the bug**
On RPi 3B+, on can expect an optimistic max. of 20-30 devices connected simultaneously via Wi-Fi.

**To reproduce**
Use case: the RPi serves as a wifi AP and a web server, that's all. No user is logged in the RPi in production use.

Nginx and php-fpm are optimized so that memory is not a problem. Flooding the web server using ApacheBench shows that memory is not a problem: available memory remains at around 600M at any time, even with 100 concurrent http requests! Swap is never used.

The limitation experienced is uniquely caused by concurrent wifi connections to the AP (hostapd). For that reason, I'm suspecting it's a wifi driver issue.

**Expected behaviour**
More client devices than a mere 20 can be simultaneously connected to the RPi in AP mode.

**Actual behaviour**
When about 20 devices are connected to the RPi in AP mode, the RPi freezes.

**System**

* Model: Pi3B+
* OS and version: Raspberry Pi reference 2019-04-08
* Firmware: version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start_cd)
* Kernel: Linux moodlebox 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux

**Logs**
Not applicable.

**Additional context**
This is the content of my `/etc/hostapd/hostapd.conf`:
```
# Set country code
country_code=US
# Name of the Wi-Fi interface
interface=wlan0
# Use the nl80211 driver
driver=nl80211
# Wi-Fi network name (SSID)
ssid2=4d6f6f646c65426f78
utf8_ssid=1
# Show or hide SSID
ignore_broadcast_ssid=0
# Use the 2.4GHz band
hw_mode=g
# The Wi-Fi channel
channel=11
# Enable 802.11n
ieee80211n=1
# Enable WMM
wmm_enabled=1
# Enable 40 MHz channels with short guard interval for 20 Mhz
#ht_capab=[MAX-AMSDU-3839][HT40+][SHORT-GI-20][SHORT-GI-40][DSSS_CCK-40]
ht_capab=[HT40][SHORT-GI-20][DSSS_CCK-40]
# Accept all MAC addresses
macaddr_acl=0
# Use WPA authentication
auth_algs=1
# Use WPA2
wpa=2
# Use a pre-shared key
wpa_key_mgmt=WPA-PSK
# The network passphrase
wpa_passphrase=moodlebox
# Use AES, instead of TKIP
rsn_pairwise=CCMP
# Enable hostapd_cli
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
```

Choupinounours, 2019-07-05:
Mhhh I meet the same problem ! Could be really useful to fix it !

marcusdiy, 2019-07-23:
I can connect only 9 devices with RPi 3B+ ...

marcusdiy, 2019-07-28:
Seems like every time, before reaching the quota, there is like some kind undervoltage issue...

Jul 28 20:49:07 dnsmasq-dhcp[896]: DHCPACK(ap0) 10.3.141.110 xx:xx:xx:xx:xx:xx HUAWEI_P9_lite
Jul 28 20:49:12 spherae kernel: [  440.556265] Under-voltage detected! (0x00050005)
Jul 28 20:49:16 spherae kernel: [  444.716262] Voltage normalised (0x00000000)
Jul 28 20:49:21 dnsmasq-dhcp[896]: DHCPDISCOVER(ap0) xx:xx:xx:xx:xx:xx
Jul 28 20:49:21 dnsmasq-dhcp[896]: DHCPOFFER(ap0) 10.3.141.108  xx:xx:xx:xx:xx:xx
Jul 28 20:49:21 dnsmasq-dhcp[896]: DHCPREQUEST(ap0) 10.3.141.108  xx:xx:xx:xx:xx:xx
Jul 28 20:49:21 dnsmasq-dhcp[896]: DHCPACK(ap0) 10.3.141.108  xx:xx:xx:xx:xx:xx Galaxy-S7
Jul 28 20:50:48 spherae kernel: [  536.727290] brcmfmac: send_key_to_dongle: wsec_key error (-52)

martignoni, 2019-07-28:
@marcusdiy You should absolutely be certain to use a adequate power supply and cable before any testing, so that not under-voltage issue occurs in these testing.

I'll edit my bug report to positively indicate that a limited number of clients occurs even when no under-voltage event occurs.

christian-westphal, 2019-07-30:
RPI is a very nice plateform for a lot of "boxes" (moodlebox,  piratebox etc) so this resolution willy be very usefull...

Thanks

marcusdiy, 2019-08-09:
I were able to connect more than 20+ devices using ubuntu mate + snap wifi-ap
It is some kind of software limit that we got there...

ezimmert, 2019-11-14:
Any news about this?

I manage to get exactly 19 connections on a Rpi 3b and 14 connections on a Rpi 3b+ with the following hostapd conf:

`interface=wlan0
driver=nl80211
ssid=xxxx
hw_mode=g
channel=7
macaddr_acl=0
auth_algs=3`

pelwell, 2019-11-14:
Both 43438 and 43455 are designed for "mobile or handheld wireless system[s]". As such, 19 and 14 sounds pretty good. I suspect they both have relatively small RAMs that also need to contain the firmware, so resources are going to be limited.

leond08, 2020-03-09:
any news about this?

martignoni, 2020-08-18:
Some reports seem to indicate that the issue is worse on RPi 4 hardware, where max. number of wireless clients is sometimes less than 10!

Ref: https://discuss.moodlebox.net/d/198-wifi-client-advice/6

ezimmert, 2020-08-18:
Same results here, 8 devices connected with a Rapsberry Pi 4 on Buster... but same with a 3b+
Having a few Rapsberry Pi in production for more than a year now with hostapd, the real dropdown in number of users connected simultaneously was seen by upgrading from Stretch (RPi3 and RPi3b+ only) to Buster.

fabrizioromanelli, 2020-10-10:
> Some reports seem to indicate that the issue is worse on RPi 4 hardware, where max. number of wireless clients is sometimes less than 10!
> 
> Ref: https://discuss.moodlebox.net/d/198-wifi-client-advice/6

In my case I have a Raspberry Pi 4, I have tried your same configuration for `/etc/hostapd/hostapd.conf` and can connect just 5 devices simultaneously. As soon as I disconnect a device, another one can connect without any problem. All the devices are in range.

greenorca, 2020-10-13:
Same problem here on Raspi4. Hostapd worked well when testing with 2 clients back home, failed utterly when using it in school during my Web-Dev class...

andrew-d-l-nelson, 2020-10-21:
I have also just encountered this issue with a Raspi4 (both 4 and 8 GB ram) using Buster. Can't get more than 7 other raspis to connect... Also, I don't know if this is pertinent, but it seems like power save mode is automatically enabled for the brcm wifi driver. I saw in earlier iterations that this may cause clients to get dropped?

brandonyb, 2020-11-04:
I am using Raspberry Pi 4 (1 GB Ram), and only able to connect 1 client to it...

guillain, 2020-12-19:
Hi,

Same issue, **no more than 8 devices** connected in same time on Raspi with RaspAP :(
- Raspi: 3B+(5.4.79-v7+ #1373 2020 armv7l)
- DC: Laboratory power supply (GPR-3030)
- Devices: ESP (mainly 8266 and 32cam)

Even if I can put some devices on sleep and wait events to wake them, others devices need permanent connection (like the camera).

Any idea/workaround to move forward ? because a apparently this will not be fixed in 2020 ^^

Thanks and take care!

pelwell, 2020-12-19:
I think there is no possibility of a "fix" - every change to the firmware (vulnerability patch, etc.) takes up more of the limited RAM, resulting in less space to host AP clients. If you really care about AP clients then you'll have to either use an old firmware or buy a USB dongle (which might give you higher performance as well).

ziesemer, 2020-12-19:
> I think there is no possibility of a "fix" - every change to the firmware (vulnerability patch, etc.) takes up more of the limited RAM, resulting in less space to host AP clients.

Can we please clarify, is this system RAM, or RAM specific to the chipset / wireless chip?  I.E., having a Pi with 1 vs. 8 GB of RAM doesn't seem to make any difference here?

pelwell, 2020-12-19:
The RAM on the WiFi part, into which the firmware is loaded, and which is also used for the running state of the device.

huan, 2020-12-20:
> buy a USB dongle (which might give you higher performance as well).

@pelwell using a usb dongle is considering an alternative solution for me.

Could you please to share which usb dongle can handle more wifi clients, with brand and model specs?

Thank you very much!

ziesemer, 2020-12-23:
> Could you please to share which usb dongle can handle more wifi clients, with brand and model specs?

FYI, there is the reference at https://elinux.org/RPi_USB_Wi-Fi_Adapters .  It would be nice to see included a number of supported WiFi clients, etc. - might be a good place to start.  Would also be nice to have a cross-reference as to which support various features such as which wireless bands and WiFi versions are supported in AP mode, which support external antennas, etc.

I previously had high hopes of using a "utility" Pi that I carry in my bag as a "can meet any need" device.  Given these issues, I had to also include a separate "travel router" for serving as an AP.  It easily serves 12+ WiFi clients, but the CPU was lacking for NAT + routing.  I could use the 2 together, and use the travel router as just a "dumb AP", and then let the Pi do the rest of the networking - but it would be nice to consolidate this all together.

I'm also interested in any specific product recommendations here.

guillain, 2020-12-23:
Yep,

Since my message I've buyed the USB key [BossTrend AC1200](https://www.amazon.fr/gp/product/B01IEU7UZ0/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1) based on the chipset RTL8812AU and now I've ~15 devices connected without issue.

It works like a charm with the driver https://github.com/cilynx/rtl88x2bu.git.
Just one thing to monitor, after reboot an high process consumption:
- RTW_CMD_THREAD
- writing commands to Realtek-based network devices
- ~25% of CPU
- 2/3 minutes

If I've new I'll let you now there.

Enjoy and take care.

ericbeaudry, 2020-12-23:
> > buy a USB dongle (which might give you higher performance as well).
> 
> @pelwell using a usb dongle is considering an alternative solution for me.
> 
> Could you please to share which usb dongle can handle more wifi clients, with brand and model specs?
> 
> Thank you very much!

Got hit by that new limitation too. The maximum Wifi connection I've reached on a Pi4 is 7 as opposed to about 30 on a Pi3 18 months ago. I have no idea where and why that limitation got added but it's rendering the Pi pretty useless in my scenario. 
On the bright side, I ordered a couple of external antennas with the RT5370 chipset (no driver to install with this chipset) and the first I plugged in is already sustaining 21 steady connections. I intend to push and see the limit in the next couple of days. Here's the antenna that looks promising [RT5370 Antenna](https://www.amazon.ca/gp/product/B01KWQAQ00)

ziesemer, 2020-12-23:
@guillain  and @ericbeaudry  - Could each of you please share which Pi and OS versions you are seeing your latest results with?

guillain, 2020-12-23:
Hi Ziemer,
You've it there: https://github.com/raspberrypi/linux/issues/3010#issuecomment-748450967

martignoni, 2020-12-23:
@ericbeaudry says:

> Got hit by that new limitation too. The maximum Wifi connection I've reached on a Pi4 is 7 as opposed to about 30 on a Pi3 18 months ago.

This leads to think there's indeed something wrong either in Pi4 hardware or in Raspberry Pi OS or drivers (which is the purpose of this issue). Humbly suggesting we stick to this, and talk about dongles elsewhere (e.g. on [RPi forums](https://www.raspberrypi.org/forums/viewforum.php?f=36)).

@ziesemer: I'm getting poor results with RPi4 since many versions of Raspberry Pi OS, and disappointing results with RPi 3B too (see https://github.com/raspberrypi/linux/issues/3010#issue-454860067). But I could not test on a large scale if the new wifi drivers shipped with last image (version 2020-12-02) have changed anything. I'd be glad to have some feedback about this.

ericbeaudry, 2020-12-23:
> @ericbeaudry says:
> 
> > Got hit by that new limitation too. The maximum Wifi connection I've reached on a Pi4 is 7 as opposed to about 30 on a Pi3 18 months ago.
> 
> This leads to think there's indeed something wrong either in Pi4 hardware or in Raspberry Pi OS or drivers (which is the purpose of this issue). Humbly suggesting we stick to this, and talk about dongles elsewhere (e.g. on [RPi forums](https://www.raspberrypi.org/forums/viewforum.php?f=36)).
> 
> @ziesemer: I'm getting poor results with RPi4 since many versions of Raspberry Pi OS, and disappointing results with RPi 3B too (see [#3010 (comment)](https://github.com/raspberrypi/linux/issues/3010#issue-454860067)). But I could not test on a large scale if the new wifi drivers shipped with last image (version 2020-12-02) have changed anything. I'd be glad to have some feedback about this.

Agreed! Still the fact that an external RT5370 is capable of handling over 21 (3 times more) simultaneous Wifi clients is "interesting" to say the least.

I'm also having the issue on a clean install of ubuntu 20.04. The 7th wifi client was crashing the Wifi interface for good and required a reboot. Following the instructions provided #3849. I replaced the firmware with the one provided by Cypress and now the 7th client is not crashing the wifi interface anymore but I'm still unable to add more than 7 clients. Disconnecting one, opens up a slot that gets used by the next device requesting a connection. 

Hopefully someone with a deeper understanding of the wifi stack common to both raspbian and ubuntu will be able to dig a bit more and find the culprit.

waiter21, 2021-01-01:
I found this thread as I'm experiencing exactly the same thing. I'm running a Rpi3 B+as an access point. I can connect 7 devices.  When I attempt to connect an 8th device,  the wifi crashes, all 7 devices that were connected are kicked off, and I need to reboot to recover.

The devices connecting are low bandwidth, ESP32 sensors, transferring MQTT data to a Mosquito server on the same Pi as the AP.

I'm following for any resolution.  I can also help with this issue, but I'm nowhere near tech savey on the Rpi as many of you.

ericbeaudry, 2021-01-02:
> I found this thread as I'm experiencing exactly the same thing. I'm running a Rpi3 B+as an access point. I can connect 7 devices. When I attempt to connect an 8th device, the wifi crashes, all 7 [...]

The procedure describe in #3849 helps in mitigating the crashes. It won't give you more connections but at least your Pi won't need a reboot every time you get more than 7 clients. Since this issue doesn't seem to attract much attention, I personally decided to dump the onboard WiFi and go with an external antenna. It has proven rock solid for 10 days now in a setup pretty similar to yours. A bunch of sensors sending telemetry over MQTT hosted on the Pi itself.

pelwell, 2021-01-02:
The firmware files don't contain the whole image, but rather patches to the built-in ROM image. This means that every change takes up more of the limited patch RAM. The only way to get more free RAM, and hence more clients, is to revert to older firmware with more bugs and vulnerabilities.

ericbeaudry, 2021-01-02:
So you're saying that this is by design and nothing (other than reverting to older firmware) can be done. This is a rather chocking revelation. If this is the case, why is the issue still open 6 months later? Can it be documented properly and closed so people stop wasting time trying to make this work?

waiter21, 2021-01-02:
I downloaded fw_test_fix.bin (617k).

I assume this will also work on a Rpi 3 B+

Next question,  I'm not exactly sure how to implement/make the changes.  I'm a little apprehensive about attempting to do this without knowing what I'm doing. Is there a How-To on how to make these firmware updates?

ericbeaudry, 2021-01-02:
> I downloaded fw_test_fix.bin (617k).
> 
> I assume this will also work on a Rpi 3 B+
> 
> Next question, I'm not exactly sure how to implement/make the changes. I'm a little apprehensive about attempting to do this without knowing what I'm doing. Is there a How-To on how to make these firmware updates?

I'm no expert either. So I simply followed the steps provided by pelwell on dec 1st https://github.com/raspberrypi/linux/issues/3849#issuecomment-736834802

Maybe give him a moment to get back to you before attempting this procedure. He definitely knows a lot more than us!

waiter21, 2021-01-02:
OK,  Thanks.  7 clients is workable for now.  I discovered this problem when I also connected my smartphone so I could monitor MQTT messages.

I also ordered a USB RT5370.  When it gets here I'll test to see if I can get past 7 clients with it.  This may be my fix if the on board wifi cannot handle more than 7.

ezimmert, 2021-01-04:
I just conducted tests today with a rpi 3b+ and a rpi 4 with the latest image of Rapsberry Pi OS Lite and a RT5370 USB dongle and in both cases I managed to connect 30 devices (mix of iPads and Android tablets) without any problem.

rafaelpascale, 2021-01-28:
I tested with 7 clients (ESP32) on raspberry 4 and not worked, always lost one device
With raspberry 3b+ worked with 20 clients 

Will be the beaglebone black work better than raspbery 4 for MQTT connections ?

freakbsd, 2021-03-01:
Same problem here, 8 clients connected, no more.
RPI4 (4g ram), Buster updated.

In the end, @pelwell is right here https://github.com/raspberrypi/linux/issues/3010#issuecomment-748484728, so if you're just looking to increase the number of connected clients, then take a look at this table https://github.com/iiab/iiab/issues/823.

Take 7.45.18.0 (follow the link) and replace /var/lib/firmware/brcmfmac43455-sdio.bin (of course, make a backup first). Reboot.

I've done it 2 days ago and until now I see no issues, stability seems okay, 20 clients connected.

freakbsd, 2021-03-08:
Still stable, 23 clients connected.

Choupinounours, 2021-03-16:
> Same problem here, 8 clients connected, no more.
> RPI4 (4g ram), Buster updated.
> 
> In the end, @pelwell is right here [#3010 (comment)](https://github.com/raspberrypi/linux/issues/3010#issuecomment-748484728), so if you're just looking to increase the number of connected clients, then take a look at this table [iiab/iiab#823](https://github.com/iiab/iiab/issues/823).
> 
> Take 7.45.18.0 (follow the link) and replace /var/lib/firmware/brcmfmac43455-sdio.bin (of course, make a backup first). Reboot.
> 
> I've done it 2 days ago and until now I see no issues, stability seems okay, 20 clients connected.

I also tried today and it worked with 20 clients ! Thanks a lot !

ziesemer, 2021-12-10:
Please see also:

- https://github.com/iiab/iiab/issues/2853
  - https://github.com/iiab/iiab/issues/823
- https://github.com/RPi-Distro/firmware-nonfree/tree/bullseye/debian/config/brcm80211/cypress
  - > The expected number of supported clients using this firmware is now 19.

And to further answer one of my prior questions here - it looks like the limit being worked against here is 800 KB SRAM built-in to the wireless chip.  (So the 1/2/4/8 GB of RAM available on the Raspberry Pi itself has nothing to do with the limits here.)

JamesH65, 2021-12-10:
Correct. AIUI, that SRAM on the wireless chip is used both for data storage whilst in use, but also additional features, and bug fixes  i.e. additional code over and above the ROM based code on the chip itself. So each time a feature is added or a bug fix made, the amount of SRAM available for runtime variables decreases.

johnthedev97, 2021-12-17:
I believe I am also running into this issue with Raspberry pi 4 8 GB RAM.
I started troubleshooting when my newly purchased chromebook failed to connect giving an error message "too-many-stas".

But rest of the clients continued working. But today morning it seems to have failed completely with none of the clients working. I see the below error in the syslog
```
Dec 16 22:52:45 raspberrypi kernel: [110590.099295] mmc1: Timeout waiting for hardware interrupt.
Dec 16 22:52:45 raspberrypi kernel: [110590.099360] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
Dec 16 22:52:45 raspberrypi kernel: [110590.099375] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Dec 16 22:52:56 raspberrypi kernel: [110600.339419] mmc1: Timeout waiting for hardware interrupt.
Dec 16 22:52:56 raspberrypi kernel: [110600.339464] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
Dec 16 22:52:56 raspberrypi kernel: [110600.339479] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Dec 16 22:52:56 raspberrypi kernel: [110600.339611] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Dec 16 22:53:06 raspberrypi kernel: [110610.579552] mmc1: Timeout waiting for hardware interrupt.
Dec 16 22:53:06 raspberrypi kernel: [110610.579795] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
Dec 16 22:53:06 raspberrypi kernel: [110610.579806] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Dec 16 22:53:06 raspberrypi kernel: [110610.579929] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Dec 16 22:53:06 raspberrypi kernel: [110610.579939] ieee80211 phy0: brcmf_cfg80211_get_station: GET STA INFO failed, -110
Dec 16 22:53:16 raspberrypi kernel: [110620.819669] mmc1: Timeout waiting for hardware interrupt.
Dec 16 22:53:16 raspberrypi kernel: [110620.819769] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
Dec 16 22:53:16 raspberrypi kernel: [110620.819786] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Dec 16 22:53:16 raspberrypi kernel: [110620.821176] brcmfmac: brcmf_sdio_hdparse: HW header checksum error
Dec 16 22:53:16 raspberrypi kernel: [110620.821190] brcmfmac: brcmf_sdio_rxfail: terminate frame

...

Dec 17 10:35:39 raspberrypi kernel: [152764.265029] ieee80211 phy0: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
Dec 17 10:35:39 raspberrypi kernel: [152764.265049] ieee80211 phy0: brcmf_cfg80211_get_station: GET STA INFO failed, -110
Dec 17 10:36:28 raspberrypi hostapd: wlan0: STA xx:xx:xx:xx:xx:xx WPA: group key handshake failed (RSN) after 4 tries
Dec 17 10:36:31 raspberrypi kernel: [152816.025715] ieee80211 phy0: brcmf_cfg80211_del_station: SCB_DEAUTHENTICATE_FOR_REASON failed -110
Dec 17 10:36:32 raspberrypi kernel: [152816.985825] ieee80211 phy0: brcmf_netdev_wait_pend8021x: Timed out waiting for no pending 802.1x packets
Dec 17 10:36:34 raspberrypi kernel: [152819.545738] ieee80211 phy0: send_key_to_dongle: wsec_key error (-110)
Dec 17 10:36:37 raspberrypi hostapd: wlan0: STA xx:xx:xx:xx:xx:xx WPA: group key handshake failed (RSN) after 4 tries
Dec 17 10:36:37 raspberrypi kernel: [152822.105778] ieee80211 phy0: brcmf_cfg80211_change_station: Setting SCB (de-)authorize failed, -110
Dec 17 10:36:39 raspberrypi kernel: [152824.675813] ieee80211 phy0: brcmf_cfg80211_del_station: SCB_DEAUTHENTICATE_FOR_REASON failed -110
```

I do have multiple services running in the pi, but I monitor the system and it looks well from resource perspective. And the error message in the logs above kind of line up with with where all the network traffic seems to stop in my monitoring dashboard
[image]

Any pointers or help is appreciated. I had about 7 or 8 clients connected to AP. Now nothing is connecting, ethernet is working. And I am using the wireless device only in AP mode (not using client mode simultaneously), it is setup to route/NAT the traffic to eth0

JamesH65, 2021-12-17:
Does the wireless system work if you use it in client mode (perhaps try a new SD with the latest Bullseye on it)? Just to check the HW is working in a normal mode.

johnthedev97, 2021-12-17:
@JamesH65 Yes it is able to work in client mode after I reconfigured it.

@pelwell sorry I don't know how to find the firmware version. Can you please share any links/documentation on that?

pelwell, 2021-12-17:
Try `dmesg | grep brcmfmac` (shortly after boot, in case you have a lot of other kernel messages).

Your kernel log fragment above does suggest a low-level problem with the SDIO bus, rather than limited resources, which leads me to ask the usual questions about kernel versions (`uname -a`) and power supplies - consider them asked.

johnthedev97, 2021-12-17:
```
[    6.364457] brcmfmac: F1 signature read @0x18000000=0x15264345
[    6.389705] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    6.391379] usbcore: registered new interface driver brcmfmac
[    6.730614] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    6.730742] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    6.750406] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229 (617f1f5 CY) FWID 01-2dbd9d2e
[    8.907097] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
```
```
Linux raspberrypi 5.10.82-v7l+ #1497 SMP Fri Dec 3 16:28:40 GMT 2021 armv7l GNU/Linux
```
I purchased a kit from Labists in Amazon and the power supply came with it. The exact product no longer seems to be available now, but it was "LABISTS Raspberry Pi 4 Model B 4 GB Ultimatives Kit mit 32GB Class10 Micro SD-Karte, 5,1V 3,0A USB-C EIN/Aus-Schaltnetzteil, 3 Premium Kupfer Kühlkö" (amazon.de)

By the way, after I reconfigured it to AP mode after testing as requested by @JamesH65 now it is working again. But I have only 4 clients now

chradams, 2022-01-14:
network_event_handler(): WiFi disconnect, Reason: 2 - AUTH_EXPIRE

New device are rejected even if it is a phone after 8/9 device are connected/

denissabramovs, 2022-01-14:
Same here, onl 7-8 connected clients and then it starts failing new connections with a timeout.
Using raspberry pi 4 Model B with:
```
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

```
[  401.960202] wlp4s0: deauthenticating from 6c:5a:b0:cd:1e:d1 by local choice (Reason: 3=DEAUTH_LEAVING)
[  401.990688] wlp4s0: authenticate with dc:a6:32:f6:ed:ec
[  401.993986] wlp4s0: send auth to dc:a6:32:f6:ed:ec (try 1/3)
[  402.122791] wlp4s0: send auth to dc:a6:32:f6:ed:ec (try 2/3)
[  402.226518] wlp4s0: send auth to dc:a6:32:f6:ed:ec (try 3/3)
[  402.334791] wlp4s0: authentication with dc:a6:32:f6:ed:ec timed out
[  403.859591] wlp4s0: authenticate with dc:a6:32:f6:ed:ec
[  403.863292] wlp4s0: send auth to dc:a6:32:f6:ed:ec (try 1/3)
[  403.990752] wlp4s0: send auth to dc:a6:32:f6:ed:ec (try 2/3)
[  404.094497] wlp4s0: send auth to dc:a6:32:f6:ed:ec (try 3/3)
[  404.198499] wlp4s0: authentication with dc:a6:32:f6:ed:ec timed out
[  406.052859] wlp4s0: authenticate with dc:a6:32:f6:ed:ec
[  406.055582] wlp4s0: send auth to dc:a6:32:f6:ed:ec (try 1/3)
[  406.182423] wlp4s0: send auth to dc:a6:32:f6:ed:ec (try 2/3)
[  406.286469] wlp4s0: send auth to dc:a6:32:f6:ed:ec (try 3/3)
[  406.394446] wlp4s0: authentication with dc:a6:32:f6:ed:ec timed out
[  408.519057] wlp4s0: authenticate with dc:a6:32:f6:ed:ec
[  408.522179] wlp4s0: send auth to dc:a6:32:f6:ed:ec (try 1/3)
[  408.650531] wlp4s0: send auth to dc:a6:32:f6:ed:ec (try 2/3)
[  408.754430] wlp4s0: send auth to dc:a6:32:f6:ed:ec (try 3/3)
[  408.858366] wlp4s0: authentication with dc:a6:32:f6:ed:ec timed out
[  419.721574] wlp4s0: authenticate with dc:a6:32:f6:ed:ec
[  419.724305] wlp4s0: send auth to dc:a6:32:f6:ed:ec (try 1/3)
[  419.854234] wlp4s0: send auth to dc:a6:32:f6:ed:ec (try 2/3)
[  419.958295] wlp4s0: send auth to dc:a6:32:f6:ed:ec (try 3/3)
[  420.062260] wlp4s0: authentication with dc:a6:32:f6:ed:ec timed out
[  427.861527] wlp4s0: authenticate with 6c:5a:b0:cd:1e:d1
[  427.864501] wlp4s0: send auth to 6c:5a:b0:cd:1e:d1 (try 1/3)
[  427.891161] wlp4s0: authenticated
[  427.894166] wlp4s0: associate with 6c:5a:b0:cd:1e:d1 (try 1/3)
[  427.897942] wlp4s0: RX AssocResp from 6c:5a:b0:cd:1e:d1 (capab=0x1811 status=0 aid=6)
[  427.907055] wlp4s0: associated
[  428.001772] IPv6: ADDRCONF(NETDEV_CHANGE): wlp4s0: link becomes ready
```

pelwell, 2022-01-14:
Have you tried the "minimal" firmware? It supports up to 19 clients.

https://github.com/RPi-Distro/firmware-nonfree/tree/bullseye/debian/config/brcm80211/cypress

denissabramovs, 2022-01-14:
> Have you tried the "minimal" firmware? It supports up to 19 clients.
> 
> https://github.com/RPi-Distro/firmware-nonfree/tree/bullseye/debian/config/brcm80211/cypress

Not yet, i just started investigating this issue. I have a few wifi dongles, will try them instead of built-in adapter. If that will help, i'll report back.

denissabramovs, 2022-01-14:
> Have you tried the "minimal" firmware? It supports up to 19 clients.
> 
> https://github.com/RPi-Distro/firmware-nonfree/tree/bullseye/debian/config/brcm80211/cypress

Daaamn, it is really working. **Thank you very much, for sharing this**, but in any case, i need to consider some better wifi-adapter, this will be fine as a temporary fix. Solution is not for production for sure.

XECDesign, 2022-01-14:
@pelwell Any idea how frequently that firmware is used? Since apt updates to the package will overwrite the user's changes, I was wondering whether it's worth adding and documenting a mechanism to switch between them properly (https://wiki.debian.org/DebianAlternatives).

pelwell, 2022-01-14:
If there is a recognised way to support such user choices then that would be awesome - the priority was to get the firmware out there.

chradams, 2022-01-14:
With a Raspberry Pi Zero W `Linux 5.10.60+ armv6l` getting about 10 clients. 9 ESP32 devices and a Windows 10 desktop.
The 10th ESP32 is reporting a `WiFi disconnect, Reason: 2 - AUTH_EXPIRE`
During this interval an Android phone also is unable to connect as the 10th or 11th device.
Goal would be for 30 clients on a Raspberry Pi AP.
Might try the suggested FW to at least observe results.

pelwell, 2022-01-14:
The Zero W is a different WLAN chip - less capable than the 43455, and not compatible with its firmware.

pelwell, 2022-01-14:
> Goal would be for 30 clients on a Raspberry Pi AP.

You can set your goal as high as you like, but it's not achievable on what is essentially a 10 year old mobile phone class WLAN chip.

denissabramovs, 2022-01-14:
> > > Have you tried the "minimal" firmware? It supports up to 19 clients.
> > > https://github.com/RPi-Distro/firmware-nonfree/tree/bullseye/debian/config/brcm80211/cypress
> > 
> > 
> > Daaamn, it is really working. **Thank you very much, for sharing this**, but in any case, i need to consider some better wifi-adapter, this will be fine as a temporary fix. Solution is not for production for sure.
> 
> Can I ask why you think this is not suitable for production?

Because it is not yet tested by me for some period of time. Time will show. :)

briantbutton, 2023-07-06:
Just to say:  @pelwell, thank for 'cyfmac43455-sdio-minimal.bin'.  Exactly what the doctor ordered.  Grateful for that.  We popped it in and will start testing today.
For the record, our environment is Ubuntu 22.04, the installation is done with scripts and we do not allow unattended upgrades.  So this will be great if it works as intended.

dave-andersen, 2023-10-27:
btw, just to make this concrete for others searching: We ran into this bug recently as well with a slightly different manifestation, using a cm4. We're having the CM4 be both a wifi client and running hostapd.

(1) If the chip is in station (STA) mode and connected to an AP, then only one client can connect to the CM4. Other clients are instantly rejected.

(2) If the chip is in STA but searching for an AP - but not connected - then several clients can connect but they get transiently disconnected every minute or so as the chip tries again to find its AP to associate with.

(3) If the chip is NOT in STA mode / not trying to associate, then several clients can connect.

We have yet to try the minimized firmware mentioned above to see if it fixes either the disconnect behavior or allows more clients to connect while in station mode. But this may explain why a few people in this thread observed as little as 1 client able to connect to hostap. I hadn't seen (2) reported previously but I assume it's related to a lack of memory resources on the chip as well.

martignoni, 2023-10-27:
@dave-andersen IIRC, STA and AP mode **must** use the same channel for AP+STA to work correctly. But I suppose you've already checked this.

[5 more replies]