raspberrypi/linux #3013: Kernel 4.19.42+ dtoverlay=i2c-rtc,ds1307 not working

gsnw, 2019-06-16:
**Describe the bug**
config.txt Parameter dtoverlay=i2c-rtc,ds1307=on stopped working after Kernel 4.19.42+ Update

**To reproduce**
Add config.txt Parameter dtoverlay=i2c-rtc,ds1307 or dtoverlay=i2c-rtc,ds1307=on
Reboot and update Kernel from 4.14.98+ to 4.19.42+

**Expected behaviour**
_Kernel 4.14.98+ Output:_
kernel: [   18.758598] rtc-ds1307 1-0068: registered as rtc0

**Actual behaviour**
The system does not generate an RTC device with DS1307 connected.

**System**

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
  - Raspberry Pi 1 B+
  - Raspberry Pi 3 B+

* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2018-04-18
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, d6c238c1b2b1b070a574d3e25048ca442e3e221f, stage2

* Which firmware version (`vcgencmd version`)?
Mar 27 2019 15:46:32 
Copyright (c) 2012 Broadcom
version 2e98b31d18547962e564bdf88e57b3df7085c29b (clean) (release) (start_cd)

* Which kernel version (`uname -a`)?
Linux test 4.19.42+ #1219 Tue May 14 21:16:38 BST 2019 armv6l GNU/Linux

**Logs**
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.42+ (dom@dom-XPS-13-9370) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1219 Tue May 14 21:16:38 BST 2019
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Model B Rev 2
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x1e400000
[    0.000000] On node 0 totalpages: 126976
[    0.000000]   Normal zone: 1116 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 126976 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x49c with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 125860
[    0.000000] Kernel command line: coherent_pool=1M bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=9f2d13a6-3a9c-49a1-af84-f7bf3f200d7e rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait rootdelay=5 net.ifnames=0
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 483208K/507904K available (6932K kernel code, 633K rwdata, 2068K rodata, 456K init, 797K bss, 16504K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xdf800000 - 0xff800000   ( 512 MB)
                   lowmem  : 0xc0000000 - 0xdf000000   ( 496 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (6934 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   ( 456 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 634 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 798 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 25044 entries in 74 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000028] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000084] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000192] bcm2835: system timer (irq = 27)
[    0.000792] Console: colour dummy device 80x30
[    0.001429] console [tty1] enabled
[... 198 lines cut ...]
[   14.075144] [vc_sm_connected_init]: failed to initialize shared memory service
[   14.075153] [vc_sm_connected_init]: failed, ret -1
[   14.093122] media: Linux media interface: v0.10
[   14.293749] videodev: Linux video capture interface: v2.00
[   14.381450] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[   14.434458] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[   14.464346] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[   14.466171] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[   14.508403] bcm2835_mmal_vchiq: Failed to open VCHI service connection (status=-1)
[   15.099433] snd_bcm2835: module is from the staging directory, the quality is unknown, you have been warned.
[   15.231496] bcm2835_audio soc:audio: card created with 8 channels
[   16.174082] usbcore: registered new interface driver uas
[   16.402426] sd 0:0:0:0: Attached scsi generic sg0 type 0
[   18.032326] Adding 2097148k swap on /dev/sda2.  Priority:-2 extents:1 across:2097148k FS
[   20.049698] EXT4-fs (sda4): mounted filesystem with ordered data mode. Opts: (null)
[   22.828628] EXT4-fs (sda5): mounted filesystem with ordered data mode. Opts: (null)
[   23.933770] EXT4-fs (sda3): mounted filesystem with ordered data mode. Opts: (null)
[   24.642814] systemd-journald[64]: Received request to flush runtime journal from PID 1
[   27.186132] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   27.465631] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   28.124068] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[   29.728437] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC1E1
[   35.888742] uart-pl011 20201000.serial: no DMA platform data

**Additional context**

pelwell, 2019-06-17:
The generation of module aliases for I2C devices changed in 4.19. Previously, I2C aliases (e.g. "i2c:ds1307") were generated for all I2C devices, even those declared in Device Tree (i.e. from an overlay), but as of 4.19 the compatible strings in Device Tree (e.g. "maxim,ds1307") are converted to OF module aliases (e.g. "of:N*T*Cmaxim,ds1307"). In most cases this change is harmless (or even beneficial), but it does mean that the compatible strings have to be correct. In the case of the ds1307 mode of the i2c-rtc overlay, the string is wrong - it should have been "dallas,ds1307". Only the newer DS3231 (the device I have...) has a "maxim" compatible string.

The correct fix is to update the overlay to include the correct compatible string, but that will take a while to propagate through the system to a Raspbian package update. In the meantime, adding `rtc-ds1307` to the end of `/etc/modules` (or perhaps using "dtoverlay=i2c-rtc,ds3231") should work around the problem.

brabl2, 2019-06-18:
Adding `rtc-ds1307` to the `/etc/modules` works for me as a temporary fix.

gsnw, 2019-06-19:
I believe the workaround with dtoverlay=i2c-rtc,ds3231 works, too.

```
[   21.116034] rtc-ds1307 1-0068: registered as rtc0
[   32.794637] rtc rtc0: __rtc_set_alarm: err=-22
```

A request with `hwclock -r` gives a clock response

Trotter73, 2019-07-08:
Hi,

dtoverlay=i2c-sensor,lm75 also appears to be broken, I assume by the same mechanism, shall I raise a new issue for this ?

pelwell, 2019-07-08:
Yes please - it makes tracking easier.

gsnw, 2019-07-13:
@pelwell The issues can be closed from my side. Thanks for the correction and quick help

I have test the config.txt parameter with dtoverlay=i2c-rtc,ds1307 and it works with new raspbian buster and the default kernel 4.19.57+

`[   25.055118] rtc-ds1307 1-0068: registered as rtc0`

Which OS and version (cat /etc/rpi-issue)?
Raspberry Pi reference 2019-07-10
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 175dfb027ffabd4b8d5080097af0e51ed9a4a56c, stage2

Which firmware version (vcgencmd version)?
Jul  9 2019 14:41:21 
Copyright (c) 2012 Broadcom
version 6c3fe3f096a93de3b34252ad98cdccadeb534be2 (clean) (release) (start_cd)

Which kernel version (uname -a)?
Linux test 4.19.57+ #1244 Thu Jul 4 18:42:50 BST 2019 armv6l GNU/Linux

closed 2019-07-13 by pelwell