raspberrypi/linux #3014: MT7612U driver is not working.

artificerpi, 2019-06-16:
**Describe the bug**
I wanted to use a wifi dongle on raspberry pi 3 with mt7612u driver, but failed to make it work.

**To reproduce**
*  Fresh install latest stretch-lite image and update kernel : `4.19.42-v7+`
* Download mt7612u wifi driver bin file into `/lib/firmware`
* Insert wifi dongle, and run `sudo usb_modeswtich ...` command to switch into wifi mode.
* Run `ifconfig` and `dmesg` to see results.

**Expected behaviour**
New wlan interface should show up, and no errors in dmesg messge.

**Actual behaviour**
No new wlan interface can be found, and dmesg got:
```
[   33.873306] usb 1-1.5: New USB device found, idVendor=0e8d, idProduct=7612, bcdDevice= 1.00
[   33.873320] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   33.873329] usb 1-1.5: Product: 802.11ac WLAN
[   33.873338] usb 1-1.5: Manufacturer: MediaTek Inc.
[   33.873347] usb 1-1.5: SerialNumber: 000000000
[   34.121756] usb 1-1.5: reset high-speed USB device number 5 using dwc_otg
[   34.252792] mt76x2u 1-1.5:1.0: ASIC revision: 76120044
[   34.271279] mt76x2u: probe of 1-1.5:1.0 failed with error -22
```

**System**
```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi3B+
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

andrum99, 2019-06-16:
Where exactly is the driver from?

artificerpi, 2019-06-16:
The driver is included in the firmware, installed by `sudo apt dist-upgrade`. see https://github.com/raspberrypi/linux/pull/2890. And bin file is from  https://github.com/ulli-kroll/mt7612u/tree/master/mcu/bin.  I've checked this driver is working on ubuntu 18.04

andrum99, 2019-06-16:
Does it work with the stock vendor supplied firmware, or is that the same as the one on that github repo?

artificerpi, 2019-06-16:
I've tried both, neither is working on raspberry pi. But bin files can work on my laptop.

lategoodbye, 2019-06-17:
This driver has been introduced with 4.19 and still contains a lot of bugs for non-x86 platforms.
Are you able to test rpi-5.1?
Which wifi dongle are you using?

cybex-dev, 2019-06-17:
I have the same problem as @artificerpi.

```
$ uname -a
Linux gateway 4.19.50-v7+ #1234 SMP Thu Jun 13 11:06:37 BST 2019 armv7l GNU/Linux
```

I have posted the module (driver) code on [my github - cybex-dev](https://github.com/cybex-dev/MediaTek_MT7612). This is the module code given by the manufacturer, as is.

_Please note however that the manufacturer code does not seem to compile correctly._

cybex-dev, 2019-06-22:
any updates for us?

artificerpi, 2019-06-26:
@lategoodbye Thanks. It works on rpi-5.1.y

```
[  116.410847] usb 1-1.4: New USB device found, idVendor=0e8d, idProduct=7612, bcdDevice= 1.00
[  116.410864] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  116.410877] usb 1-1.4: Product: 802.11ac WLAN
[  116.410890] usb 1-1.4: Manufacturer: MediaTek Inc.
[  116.410903] usb 1-1.4: SerialNumber: 000000000
[  116.659813] usb 1-1.4: reset high-speed USB device number 5 using dwc_otg
[  116.791276] mt76x2u 1-1.4:1.0: ASIC revision: 76120044
[  116.823546] mt76x2u 1-1.4:1.0: ROM patch build: 20140408060640a
[  116.975644] mt76x2u 1-1.4:1.0: Firmware Version: 0.0.00
[  116.975654] mt76x2u 1-1.4:1.0: Build: 1
[  116.975661] mt76x2u 1-1.4:1.0: Build Time: 201406241830____
[  118.211340] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[  118.213166] usbcore: registered new interface driver mt76x2u
[  519.023105] wlan1: authenticate with **:**:**:**:**:**
[  520.821733] wlan1: send auth to **:**:**:**:**:** (try 1/3)
[  520.827379] wlan1: authenticated
[  520.830158] wlan1: associate with **:**:**:**:**:** (try 1/3)
[  520.837322] wlan1: RX AssocResp from **:**:**:**:**:** (capab=0xc11 status=0 aid=1)
[  520.842802] wlan1: associated
[  520.912608] cryptd: max_cpu_qlen set to 1000
[  520.993817] wlan1: Limiting TX power to 30 (30 - 0) dBm as advertised by **:**:**:**:**:**
[  521.000561] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
```

closed 2019-07-01 by artificerpi

JFrandon, 2019-07-24:
I heve the same issue
@artificerpi How do I install "rpi-5.1.y" ?

I have a Raspberry Pi 2B kernel 4.19.59-v7+

pelwell, 2019-07-24:
You would have to build it yourself (see https://www.raspberrypi.org/documentation/linux/kernel/building.md), but if someone can identify the necessary commits (and if they cherry-pick successfully) we would be open to back-porting them to rpi-4.19.y.

lategoodbye, 2019-07-24:
The driver was / is under heavy reworking, so i expect a lot of changes between 4.19 and 5.1.

Y2Kot, 2019-09-18:
I've update my kernel to 5.1 version, however I got 
in dmesg | grep mt76
```
[    6.067522] mt76x0u 1-1.4:1.0: ASIC revision: 76100002 MAC revision: 76502000
[    6.078647] mt76x0u 1-1.4:1.0: mediatek/mt7610e.bin not found, switching to mediatek/mt7610u.bin
[    6.078693] mt76x0u 1-1.4:1.0: Direct firmware load for mediatek/mt7610u.bin failed with error -2
[    6.103586] mt76x0u: probe of 1-1.4:1.0 failed with error -2
[    6.103735] usbcore: registered new interface driver mt76x0u
```
in lsusb:
`Bus 001 Device 004: ID 148f:761a Ralink Technology, Corp. MT7610U ("Archer T2U" 2.4G+5G WLAN Adapter`
how can I install driver?

pelwell, 2019-09-18:
You have an MT7610, not an MT7612U, so this is the wrong place for your question. However, the Linux firmware repo appears to have firmwares that should work: https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek

Y2Kot, 2019-09-18:
sorry. My wrong. Two last questions: how can I install bins from you url? And should I update to 5.3 to make it work?

pelwell, 2019-09-18:
In your browser, click on the file you want, then click on '(plain)' to download it. And stay away from 5.3 for now - it's not really ready yet; 5.2 should be fine.

kmahyyg, 2019-10-16:
I can confirm that there must be some necessary modification in Kernel, at least version 5.3 is what we want.

- EDUP AC-1616 (MT7612U confirmed by manufacturer) is working only on my laptop with Arch Linux with kernel 5.3.6, none of my friends' 5.2.0 (don't work at all) or 4.19 in raspbian(no AP mode support) is working.

So may I ask if there's any support chart or progressing chart for 5.3.x kernel? Can we have AP mode support in kernel 4.19.x in raspbian?

artificerpi, 2019-10-18:
@kmahyyg I think you should watch codes changes in [upstream linux kernel ](https://github.com/torvalds/linux/tree/master/drivers/net/wireless/mediatek/mt76/mt76x2). AP mode is not supported in 4.19.x, and it's not stable. I can confirm that starting from 5.1.12-v7+, mt7612u is working fine in Raspbian.

kmahyyg, 2019-10-18:
> @kmahyyg I think you should watch codes changes in [upstream linux kernel ](https://github.com/torvalds/linux/tree/master/drivers/net/wireless/mediatek/mt76/mt76x2). AP mode is not supported in [...]

Thanks for your reply. That's why I comment here. I think there might be some schedule about the kernel upgrade in raspbian since rpi doesn't have mainline kernel support. Currently, except my RPI, all my devices were upgraded to 5.3.7 kernel. Sorry for disturbing. Thank you for your guys hard work.

JamesH65, 2019-10-19:
We do run behind mainline and although there is a mainline kernel that is specifically for Pi, it doesn't have all the features we need for our devices, so we have to patch in all the required changes. That takes a while, to do it and to test it and to fix any problems, which is why we are on 4.19. Newer kernels are being tested. Not sure which is our next release, or when that will be. 5.3 is currently very flakey.

pelwell, 2019-10-19:
If we had to ship a newer kernel today it would be 5.2. As James said, there are a few issues with our 5.3 fork.

evilhard2, 2024-01-29:
Hi
I installed kali linux on ORANGE PI 5. I updated kali. But I still have the same problem I can't get my alfa awuso36acm usb key to work.

When I am in a terminal and I type lsusb I see the mediatek mt7612u. but nothing wants to work.

Thanks you for helping me.