raspberrypi/linux #3019: SACK and MSS network vulnerabilities (CVE-2019-11477, CVE-2019-11479)?

ziesemer, 2019-06-23:
I'm not seeing any fixes committed or released for Raspbian so far.  Are these in-progress somewhere, or is Raspbian believed to not be vulnerable for some reason?

From https://www.kb.cert.org/vuls/id/905115/ , I'd understand that the following should be applicable here:

- CVE-2019-11477: SACK Panic (Linux >= 2.6.29). A sequence of specifically crafted selective acknowledgements (SACK) may trigger an integer overflow, leading to a denial of service or possible kernel failure (panic).
- CVE-2019-11479: Excess Resource Consumption Due to Low MSS Values (all Linux versions). The default maximum segment size (MSS) is hard-coded to 48 bytes which may cause an increase of fragmented packets. This vulnerability may create a resource consumption problem in both the CPU and network interface, resulting in slowness or denial of service.

JamesH65, 2019-06-23:
I think both those are fixed in our latest kernel, which is currently a higher version than specified in these reports. So ensure you are up to date. At least, that's how I read the reports, I could be wrong!

https://security-tracker.debian.org/tracker/CVE-2019-11477
https://security-tracker.debian.org/tracker/CVE-2019-11479

ziesemer, 2019-06-23:
> I think both those are fixed in our latest kernel, which is currently a higher version than specified in these reports. So ensure you are up to date. At least, that's how I read the reports, I could [...]

I don't see how.  These were not disclosed by Netflix until almost a week ago on 2019-06-17.  I haven't seen a kernel update in over a month, and https://github.com/raspberrypi/firmware/tree/stable hasn't been updated since 2019-05-17.

With the latest updates applied, I'm running `4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux`.  I am no expert in how everything here tracks from upstream - but in each of the above Debian trackers, 4.19.52 is referenced as the closed fixed stable kernel release, which would be < 4.19.42 by my count?  (Even https://github.com/raspberrypi/firmware/tree/master was updated to just 4.19.50 on 2019-06-13, and I would expect is also missing the fixes.)

I'm also unable to locate the referenced 3b4929f65b0d8249f19a50245cd88ed1a2f78cff in the commit log here?  (Actually, this links back to something here that was committed somewhere on 2019-05-17 - but again, I'm no expert in how the actual project is tracking from upstream.)  Regardless, I do not see anything relevant committed here as visible from https://github.com/raspberrypi/linux/commits/rpi-4.19.y .

JamesH65, 2019-06-24:
Report dates seem much earlier than 17/5. https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2019-11477 is dated 23/4.

pelwell, 2019-06-24:
The fixes are in Linux 4.19.52. We've been a bit busy for the last few weeks, but we'll merge these fixes ASAP.

popcornmix, 2019-06-24:
rpi-4.19.y source tree has been updated. I'll push an rpi-update kernel later today.

ziesemer, 2019-06-24:
> We've been a bit busy for the last few weeks

Yes - just BREAKING THE INTERNET with the surprise news of an awesome [Raspberry Pi 4](https://www.raspberrypi.org/products/raspberry-pi-4-model-b)!  (Looks like the official site is currently down - and not surprising given the news - so some actual breakage here.  :-)

pelwell, 2019-06-24:
Did you see that our blog is currently hosted on Pi 4s? https://blog.mythic-beasts.com/2019/06/22/raspberry-pi-on-raspberry-pi/

> We’ve set up 14 Pi 4s for PHP processing for the main website (56 cores, 56GB RAM), two for static file serving (8 cores, 8GB RAM) and two for memcached (8 cores / 8GB RAM). Late on Friday night we started moving production traffic from the existing virtual machines to the Pi 4 cluster, completing the move shortly after midnight. Every page from the blog after Sat 22nd June has been generated on a Raspberry Pi 4.

P33M, 2019-06-24:
> Yes - just BREAKING THE INTERNET with the surprise news of an awesome [Raspberry Pi 4](https://www.raspberrypi.org/products/raspberry-pi-4-model-b)! (Looks like the official site is currently down - [...]

Amusingly, this isn't our problem. We use Cloudflare to cache the main page and they're the ones (today, of all days) having issues.

popcornmix, 2019-06-24:
rpi-update kernel should contain these fixes.
They should appear in apt kernel soon assuming no issues arise from kernel bump.

ziesemer, 2019-07-10:
I'm still not seeing any kernel updates come through on Stretch here from apt.  (I haven't installed Buster anywhere to check that one yet.)  Are there any updates available to share here, or another place that I should better be tracking the progress from this?  (Sure, I could do an rpi-update, but the larger concern is making these available to everyone via apt.)

JamesH65, 2019-07-10:
@XECDesign Any schedule of this?

pelwell, 2019-07-10:
The fixes were part of Linux 4.19.52, which was picked up in the firmware of 2019-06-24 ("kernel: Bump to 4.19.55"). We're in the process of finalising the next mainline kernel release, which will include these patches.

mati75, 2019-07-10:
firmware 2019-06-20 have kernel 4.19.50: https://github.com/raspberrypi/firmware/blob/88ca9081f5e51cdedd16d5dbc85ed12a25123201/extra/uname_string 

2019-06-24 wasn't released.

pelwell, 2019-07-10:
It was released via rpi-update, but not as a mainline Raspbian package, hence my comment about finalising the next one.

ziesemer, 2019-07-14:
Closing as I've now received firmware `libraspberrypi-bin (1.20190709~stretch-1)` with kernel `4.19.57-v7+ #1244 SMP Thu Jul 4 18:45:25 BST 2019 armv7l GNU/Linux` via apt.

closed 2019-07-14 by ziesemer