raspberrypi/linux #3020: Pi4/fkms kernel warnings: vc4_crtc_atomic_flush and drm_vblank_put

6by9, 2019-06-24:
This is more a placeholder to say that we are aware of the two kernel warnings that trigger whenever the screen mode changes or blanks (ie screensaver) with the Pi4. eg:
```
[12296.546207] WARNING: CPU: 0 PID: 0 at /home/dave/Pi/linux2711/drivers/gpu/drm/drm_vblank.c:1026 drm_vblank_put+0xcc/0xf0 [drm]
[12296.546215] Modules linked in: fuse rfcomm bnep hci_uart btbcm serdev bluetooth ecdh_generic 8021q garp stp llc binfmt_misc evdev vc4 drm_kms_helper brcmfmac v3d brcmutil snd_soc_core gpu_sched snd_compress snd_pcm_dmaengine syscopyarea sha256_generic drm sysfillrect cfg80211 sysimgblt drm_panel_orientation_quirks fb_sys_fops snd_bcm2835(C) rfkill snd_pcm snd_timer raspberrypi_hwmon bcm2835_codec(C) hwmon snd v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media argon_mem fixed uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[12296.546607] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        WC        4.19.50-v7l+ #9
[12296.546615] Hardware name: BCM2835
[12296.546648] [<c0211d44>] (unwind_backtrace) from [<c020ca94>] (show_stack+0x20/0x24)
[12296.546668] [<c020ca94>] (show_stack) from [<c0928cf0>] (dump_stack+0xc8/0x114)
[12296.546688] [<c0928cf0>] (dump_stack) from [<c0220c68>] (__warn+0xf4/0x120)
[12296.546705] [<c0220c68>] (__warn) from [<c0220d80>] (warn_slowpath_null+0x4c/0x58)
[12296.546889] [<c0220d80>] (warn_slowpath_null) from [<bf441a48>] (drm_vblank_put+0xcc/0xf0 [drm])
[12296.547228] [<bf441a48>] (drm_vblank_put [drm]) from [<bf441a90>] (drm_crtc_vblank_put+0x24/0x28 [drm])
[12296.547452] [<bf441a90>] (drm_crtc_vblank_put [drm]) from [<bf766110>] (vc4_crtc_handle_page_flip+0x50/0x60 [vc4])
[12296.547559] [<bf766110>] (vc4_crtc_handle_page_flip [vc4]) from [<bf767924>] (vc4_crtc_irq_handler+0xd0/0x14c [vc4])
[12296.547618] [<bf767924>] (vc4_crtc_irq_handler [vc4]) from [<c027a9d8>] (__handle_irq_event_percpu+0x4c/0x218)
[12296.547634] [<c027a9d8>] (__handle_irq_event_percpu) from [<c027abd0>] (handle_irq_event_percpu+0x2c/0x68)
[12296.547647] [<c027abd0>] (handle_irq_event_percpu) from [<c027ac5c>] (handle_irq_event+0x50/0x74)
[12296.547662] [<c027ac5c>] (handle_irq_event) from [<c027e758>] (handle_fasteoi_irq+0xb0/0x194)
[12296.547680] [<c027e758>] (handle_fasteoi_irq) from [<c0279ae0>] (generic_handle_irq+0x30/0x44)
[12296.547698] [<c0279ae0>] (generic_handle_irq) from [<c027a21c>] (__handle_domain_irq+0x60/0xc4)
[12296.547716] [<c027a21c>] (__handle_domain_irq) from [<c0202240>] (gic_handle_irq+0x48/0x88)
[12296.547730] [<c0202240>] (gic_handle_irq) from [<c02019bc>] (__irq_svc+0x5c/0x7c)
[12296.547739] Exception stack(0xc1001ef0 to 0xc1001f38)
[12296.547750] 1ee0:                                     00000000 2f08c000 40000093 40000093
[12296.547762] 1f00: c1000000 c1004dbc c1004e04 00000001 c109588a c1004d80 00000001 c1001f4c
[12296.547772] 1f20: c10051c4 c1001f40 c02f7d84 c02093b0 40000013 ffffffff
[12296.547790] [<c02019bc>] (__irq_svc) from [<c02093b0>] (arch_cpu_idle+0x30/0x4c)
[12296.547809] [<c02093b0>] (arch_cpu_idle) from [<c094630c>] (default_idle_call+0x34/0x48)
[12296.547829] [<c094630c>] (default_idle_call) from [<c0250310>] (do_idle+0xe0/0x17c)
[12296.547849] [<c0250310>] (do_idle) from [<c0250628>] (cpu_startup_entry+0x28/0x2c)
[12296.547864] [<c0250628>] (cpu_startup_entry) from [<c0940034>] (rest_init+0xa0/0xc0)
[12296.547882] [<c0940034>] (rest_init) from [<c0e00f28>] (start_kernel+0x42c/0x4bc)
[12296.547892] ---[ end trace 25abbe8a70e0d92b ]---
[12896.813813] ------------[ cut here ]------------
[12896.813902] WARNING: CPU: 0 PID: 750 at /home/dave/Pi/linux2711/drivers/gpu/drm/vc4/vc4_firmware_kms.c:869 vc4_crtc_atomic_flush+0x90/0x9c [vc4]
[12896.813912] Modules linked in: fuse rfcomm bnep hci_uart btbcm serdev bluetooth ecdh_generic 8021q garp stp llc binfmt_misc evdev vc4 drm_kms_helper brcmfmac v3d brcmutil snd_soc_core gpu_sched snd_compress snd_pcm_dmaengine syscopyarea sha256_generic drm sysfillrect cfg80211 sysimgblt drm_panel_orientation_quirks fb_sys_fops snd_bcm2835(C) rfkill snd_pcm snd_timer raspberrypi_hwmon bcm2835_codec(C) hwmon snd v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media argon_mem fixed uio_pdrv_genirq uio i2c_dev ip_tables x_tables ipv6
[12896.814411] CPU: 0 PID: 750 Comm: Xorg Tainted: G        WC        4.19.50-v7l+ #9
[12896.814419] Hardware name: BCM2835
[12896.814454] [<c0211d44>] (unwind_backtrace) from [<c020ca94>] (show_stack+0x20/0x24)
[12896.814476] [<c020ca94>] (show_stack) from [<c0928cf0>] (dump_stack+0xc8/0x114)
[12896.814496] [<c0928cf0>] (dump_stack) from [<c0220c68>] (__warn+0xf4/0x120)
[12896.814512] [<c0220c68>] (__warn) from [<c0220d80>] (warn_slowpath_null+0x4c/0x58)
[12896.814575] [<c0220d80>] (warn_slowpath_null) from [<bf766490>] (vc4_crtc_atomic_flush+0x90/0x9c [vc4])
[12896.814711] [<bf766490>] (vc4_crtc_atomic_flush [vc4]) from [<bf725214>] (drm_atomic_helper_commit_planes+0x190/0x258 [drm_kms_helper])
[12896.814845] [<bf725214>] (drm_atomic_helper_commit_planes [drm_kms_helper]) from [<bf767b1c>] (vc4_atomic_complete_commit+0x60/0x35c [vc4])
[12896.814948] [<bf767b1c>] (vc4_atomic_complete_commit [vc4]) from [<bf767f68>] (vc4_atomic_commit+0x150/0x180 [vc4])
[12896.815185] [<bf767f68>] (vc4_atomic_commit [vc4]) from [<bf4343bc>] (drm_atomic_commit+0x58/0x60 [drm])
[12896.815544] [<bf4343bc>] (drm_atomic_commit [drm]) from [<bf436158>] (drm_mode_atomic_ioctl+0x468/0xaf4 [drm])
[12896.815886] [<bf436158>] (drm_mode_atomic_ioctl [drm]) from [<bf422998>] (drm_ioctl_kernel+0x94/0xd0 [drm])
[12896.816220] [<bf422998>] (drm_ioctl_kernel [drm]) from [<bf422d70>] (drm_ioctl+0x1d8/0x3a8 [drm])
[12896.816397] [<bf422d70>] (drm_ioctl [drm]) from [<c03ae4c8>] (do_vfs_ioctl+0x9c/0x7d8)
[12896.816417] [<c03ae4c8>] (do_vfs_ioctl) from [<c03aec48>] (ksys_ioctl+0x44/0x6c)
[12896.816435] [<c03aec48>] (ksys_ioctl) from [<c03aec88>] (sys_ioctl+0x18/0x1c)
[12896.816452] [<c03aec88>] (sys_ioctl) from [<c0201000>] (ret_fast_syscall+0x0/0x28)
[12896.816461] Exception stack(0xda57bfa8 to 0xda57bff0)
[12896.816473] bfa0:                   026c0930 bee07328 0000000b c03864bc bee07328 00000000
[12896.816485] bfc0: 026c0930 bee07328 c03864bc 00000036 024f8558 027a04b8 02577910 0000002d
[12896.816495] bfe0: b6ccb094 bee072fc b6cb3720 b693f51c
[12896.816577] ---[ end trace 25abbe8a70e0d92c ]---
```
These are harmless and due to the timing on when we are producing vblank events being subtly wrong. We are investigating these already.

stschake, 2019-06-24:
Can I ask what the thought process was behind going with the firmware kms for Raspbian? Were there some hardware tweaks to the HVS in the Rpi4?

I figure there must be for 4k support, if we can get some information on those, I'd love to get those into the mainline KMS.

popcornmix, 2019-06-24:
This is all 3d related rather than HVS.

The 3d hardware in Pi0-3 has a GLES2 driver written for the GPU (the original chip is was designed for didn't even have an arm core).

Since then an arm side driver has been written (open source) and to use it rather than the GPU driver you use fkms.

Pi4 has a newer version of the 3d hardware (to support 4k, and newer standards like GLES3) which has never had a GPU driver written for it. So fkms is the only option that supports 3d on Pi4.

stschake, 2019-06-24:
Sorry, I was unclear. In Pi 0-3, you had the old V3D 2.x, but you also had the HVS with full atomic KMS support - no need for fkms. Pi 4 has a V3D 4.2, but does it also have the old HVS from VC4, something else entirely, or no compositor hardware at all anymore?

popcornmix, 2019-06-24:
The HVS is present and very similar to previous Pi models. (Support for 4K and HDR is added).

I think you may be talking about the kms driver (arm side HVS/HDMI driver).
That is being worked on but it probably a couple of months away.

popcornmix, 2019-06-24:
(note: the fkms driver is now much more fully featured. It supports multiple displays and mode setting. There's not an really any missing functionality compared to kms).

6by9, 2019-06-25:
It's a modified HVS, pixel valve, and a new HDMI block. The firmware drivers for these were written anyway for device bringup.

FKMS also has the benefit of retaining DispmanX's ability to add overlays from the camera, omxplayer, etc, therefore avoided a hard requirement of rewriting all those apps (twice, as displaying stuff through X is different from console).
As popcornmix says, FKMS has been significantly extended to support overlay layers, extended the supported formats list, added mode setting. The main thing missing would be fully accurate timestamps.

PeterOGB, 2019-06-27:
I'm seeing these kernel warnings on a twin headed machine.  HDMI-1 fails to come out of screen-save, and sometimes HDMI-0 then goes blank if the screen saver kicks in again.  Will disabling the screen saver avoid the problem in the short term ?

6by9, 2019-06-27:
As per the paragraph at the end of the first comment 
> These are harmless and due to the timing on when we are producing vblank events being subtly wrong. We are investigating these already.

These warnings are nothing to do with the actual mode switching, just the reporting of vblank events up the stack.
We are tracking a firmware side issue over HDMI1 - see https://github.com/raspberrypi/firmware/issues/1159, and in particular https://github.com/raspberrypi/firmware/issues/1159#issuecomment-506370578 for where our suspicions are currently lying.

Absolutely EVERYONE will see these messages in the kernel log, which is why I created this issue before anyone else reported it. I could have just nuked them, but they do correctly indicate an issue that we haven't resolved, so that would really be the wrong thing to do.

PeterOGB, 2019-06-27:
I've not seen any mention of the screen saver behaviour so I was trying to give you more information, but it seems that was unwelcome.....

6by9, 2019-06-27:
Very first paragraph of the first post

> This is more a placeholder to say that we are aware of the two kernel warnings that trigger whenever the screen mode changes or blanks _**(ie screensaver)**_ with the Pi4. eg:

Your post seemed to be a "me too" which is not helpful. Apologies if I was too terse - things are rather busy at the moment.

PeterOGB, 2019-06-27:
I meant I'd not seen any  mention when I emailed James, he pointed me here, so I just repeated what I had told him as I assumed that's why he'd pointed me here... Sorry if I misunderstood...

popcornmix, 2023-01-24:
@6by9 okay to close?

6by9, 2023-01-24:
Should be resolved, yes.

closed 2023-01-24 by 6by9