raspberrypi/linux #3021: Kernel missing support for aufs-dkms

bmweller, 2019-06-24:
Unable to install package aufs-dkms

apt install aufs-dkms

```
Setting up aufs-dkms (4.19+20190211-1) ...
Removing old aufs-4.19+20190211 DKMS files...

------------------------------
Deleting module version: 4.19+20190211
completely from the DKMS tree.
------------------------------
Done.
Loading new aufs-4.19+20190211 DKMS files...
It is likely that 4.19.50+ belongs to a chroot's host
Building for 4.19.50+, 4.19.50-v7+ and 4.19.50-v7l+
Building initial module for 4.19.50+
Error! Bad return status for module build on kernel: 4.19.50+ (armv6l)
Consult /var/lib/dkms/aufs/4.19+20190211/build/make.log for more information.
```

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? PiZeroW
* Which OS and version 
```
Raspberry Pi reference 2019-06-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 150e25c4f8123a4c9c63e8dca1b4737fa6c1135c, stage2
```
* Which firmware version 
```
vcgencmd version
Jun 20 2019 16:13:08 
Copyright (c) 2012 Broadcom
```
* Which kernel version
```
Linux raspberrypi 4.19.50+ #896 Thu Jun 20 16:09:52 BST 2019 armv6l GNU/Linux
```

make.log
``` 
DKMS make.log for aufs-4.19+20190211 for kernel 4.19.50+ (armv6l)
Mon 24 Jun 16:48:08 BST 2019
make: Entering directory '/var/lib/dkms/aufs/4.19+20190211/build'
-I/var/lib/dkms/aufs/4.19+20190211/build/include -DCONFIG_AUFS_FS_MODULE -UCONFIG_AUFS -DCONFIG_AUFS_BRANCH_MAX_127 -DCONFIG_AUFS_SBILIST
make -C /lib/modules/4.19.50+/build M=/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs EXTRA_CFLAGS="-I/var/lib/dkms/aufs/4.19+20190211/build/include -DCONFIG_AUFS_FS_MODULE -UCONFIG
_AUFS -DCONFIG_AUFS_BRANCH_MAX_127 -DCONFIG_AUFS_SBILIST" modules
make[1]: Entering directory '/usr/src/linux-headers-4.19.50+'
  GEN     /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/conf.str
  CC [M]  /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/module.o
In file included from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/branch.h:33,
                 from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/aufs.h:38,
                 from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/module.c:25:
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/super.h:134:2: error: unknown type name ‘vfs_readf_t’
  vfs_readf_t  si_xread;
  ^~~~~~~~~~~
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/super.h:135:2: error: unknown type name ‘vfs_writef_t’
  vfs_writef_t  si_xwrite;
  ^~~~~~~~~~~~
In file included from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/aufs.h:38,
                 from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/module.c:25:
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/branch.h:243:20: error: unknown type name ‘vfs_readf_t’; did you mean ‘key_ref_t’?
 ssize_t xino_fread(vfs_readf_t func, struct file *file, void *buf, size_t size,
                    ^~~~~~~~~~~
                    key_ref_t
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/branch.h:245:21: error: unknown type name ‘vfs_writef_t’; did you mean ‘key_ref_t’?
 ssize_t xino_fwrite(vfs_writef_t func, struct file *file, void *buf,
                     ^~~~~~~~~~~~
                     key_ref_t
In file included from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/aufs.h:46,
                 from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/module.c:25:
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/file.h: In function ‘au_vm_prfile_set’:
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/file.h:334:7: error: ‘struct vm_area_struct’ has no member named ‘vm_prfile’; did you mean ‘vm_file’?
  vma->vm_prfile = file;
       ^~~~~~~~~
       vm_file
In file included from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/aufs.h:57,
                 from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/module.c:25:
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/vfsub.h: In function ‘vfsub_update_time’:
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/vfsub.h:233:9: error: implicit declaration of function ‘update_time’; did you mean ‘pte_same’? [-Werror=implicit-function-declaration
]
  return update_time(h_inode, ts, flags);
         ^~~~~~~~~~~
         pte_same
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:304: /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/module.o] Error 1
make[1]: *** [Makefile:1517: _module_/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.19.50+'
make: *** [Makefile:47: fs/aufs/aufs.ko] Error 2
make: Leaving directory '/var/lib/dkms/aufs/4.19+20190211/build'
```
* Additional Context
It appears kernel patches required for aufs module installation are not included in the RPI kernel distribution 
https://github.com/sfjro/aufs4-standalone/tree/aufs4.19.17%2B
```
 apply ./aufs4-base.patch to your kernel source files.
- apply ./aufs4-mmap.patch too.
- apply ./aufs4-standalone.patch too.
```

pelwell, 2019-06-25:
We're unlikely to support a niche feature that has been rejected by the Linux devs and that requires custom patches to be applied. You can strengthen your case by explaining what AUFS is, why our users want it, and why it is better than any alternative solutions. As a general rule, features that increase the size of the base kernel or that slow operation even when the feature isn't used are very rarely accepted, so you might want to address the issue of resource utilisation in your request.

hostingnuggets, 2019-06-30:
@pelwell If it is not possible to install the `aufs-dkms` package on Raspbian buster then it's simply not possible to install the official `docker-ce` package from docker.com [1] because `aufs-dkms` is a dependency of the `docker-ce` package. I think a lot of people will want to run a modern version of Docker on their new Raspi4 so this should be valid enough reason IMHO to get things working with AUFS on Raspbian. It used to work nicely on Raspbian stretch. So why not buster?

[1] https://docs.docker.com/install/linux/docker-ce/debian/

jeanibarz, 2019-07-01:
Can anyone explain me how to install aufs-dkms on a Raspberry Pi 3B+ ? I have the same error :

`It is likely that 4.19.50-v7+ belongs to a chroot's hostError! Bad return status for module build on kernel: 4.19.50+ (armv7l)
Consult /var/lib/dkms/aufs/4.19+20190211/build/make.log for more information.
dpkg: erreur de traitement du paquet aufs-dkms (--configure) :
 installed aufs-dkms package post-installation script subprocess returned error exit status 10
Des erreurs ont été rencontrées pendant l'exécution :
 aufs-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
`

And yes it seems that docker-ce doesn't work because of that 👎

PhilippHeuer, 2019-07-03:
Here all commands i used to install docker ce 18.09.0 on Raspian Buster:

```bash
sudo apt-get update
apt-get install apt-transport-https ca-certificates software-properties-common
curl -fsSL https://yum.dockerproject.org/gpg | sudo apt-key add -
echo "deb https://download.docker.com/linux/raspbian/ stretch stable" > /etc/apt/sources.list.d/docker.list
apt install docker-ce
systemctl enable docker.service
systemctl start docker.service
```

You can't use the official debian repository, there is a special one for raspian: https://download.docker.com/linux/raspbian/

oliverkurth, 2019-07-04:
@PhilippHeuer That does not work either. There is no buster in https://download.docker.com/linux/raspbian/ , so I tried stretch. Looks like it still wants to build aufs, and I am getting the same error:

```
Setting up aufs-dkms (4.19+20190211-1) ...
Loading new aufs-4.19+20190211 DKMS files...
It is likely that 4.19.50-v7+ belongs to a chroot's host
Building for 4.19.50+, 4.19.50-v7+ and 4.19.50-v7l+
Building initial module for 4.19.50+
Error! Bad return status for module build on kernel: 4.19.50+ (armv7l)
Consult /var/lib/dkms/aufs/4.19+20190211/build/make.log for more information.
dpkg: error processing package aufs-dkms (--configure):
 installed aufs-dkms package post-installation script subprocess returned error exit status 10
```

oliverkurth, 2019-07-04:
Well, it does not seem to matter, it works anyway. I can run containers now.

Looks like the solution is to install with `--no-install-recommends`. This is from here: https://github.com/docker/for-linux/issues/709

PhilippHeuer, 2019-07-04:
Thanks for that, i manually uninstalled it because i thought that it was leftover by my earlier tries.
Added `--no-install-recommends` to my docker ce installation step.

Good to have a working script to reproduce my installation now.

pelwell, 2019-07-04:
So where does that leave this issue?

hostingnuggets, 2019-07-05:
@NobodyXu thanks, that totally makes sense! I didn't notice that the aufs-dkms pacakge is only recommended and not mandatory. So it's all good as one can skip installing the recommended packages.

bmweller, 2019-07-05:
Thank you, closing this request

closed 2019-07-05 by bmweller

netwninja, 2019-08-09:
> Here all commands i used to install docker ce 18.09.0 on Raspian Buster:
> 
> ```shell
> sudo apt-get update
> apt-get install apt-transport-https ca-certificates software-properties-common
> curl -fsSL https://yum.dockerproject.org/gpg | sudo apt-key add -
> echo "deb https://download.docker.com/linux/raspbian/ stretch stable" > /etc/apt/sources.list.d/docker.list
> apt install --no-install-recommends docker-ce
> systemctl enable docker.service
> systemctl start docker.service
> ```
> 
> You can't use the official debian repository, there is a special one for raspian: https://download.docker.com/linux/raspbian/
> You will not need `aufs-dkms` in that case.

I just tested it on RaspPi 4 with Raspbain Lite on it. It worked

Adel-Magebinary, 2019-09-05:
I was running zfs on my ubuntu 18.04.

adel@adelsstudio:/var/lib/docker$ sudo nano /lib/systemd/system/docker.service

I have changed the storage driver to zfs and the docker booted correctly.

ExecStart=/usr/bin/dockerd **--storage-driver=zfs** -H fd:// --containerd=/run/containerd/containerd.sock 

adel@adelsstudio:/var/lib/docker$ sudo systemctl daemon-reload
adel@adelsstudio:/var/lib/docker$ sudo systemctl restart docker

Ivan61, 2019-12-22:
This is my conclusion.
- aufs is not supported by Raspbian since Linux devs had been rejected it.
- But **aufs-tools** is still listed in recommends package of **docker-ce**
Check this: `apt show docker-ce docker-ce-cli containerd.io`

This is Docker's problem. [Docker has an independent release repository for Raspbian](https://download.docker.com/linux/raspbian/dists/buster/stable/). They should remove `aufs-toos` from the recommends list in their release for Raspbian.

Before they fix it, add a [preference control file for APT](https://manpages.debian.org/buster/apt/apt_preferences.5.en.html) can prevent `aufs-tools` from being installed.
```
pi@raspberrypi:~/linux $ cat /etc/apt/preferences.d/aufs-tools 
Package: aufs-tools
Pin: version *
Pin-Priority: -1
```

Let's see the difference.

Run `apt install docker-ce docker-ce-cli containerd.io` without setting this control file:
```
pi@raspberrypi:~ $ sudo apt install docker-ce docker-ce-cli containerd.io
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  aufs-dkms aufs-tools cgroupfs-mount dkms libltdl7 pigz raspberrypi-kernel-headers
Suggested packages:
  aufs-dev python3-apport menu
The following NEW packages will be installed:
  aufs-dkms aufs-tools cgroupfs-mount containerd.io dkms docker-ce docker-ce-cli libltdl7 pigz raspberrypi-kernel-headers
0 upgraded, 10 newly installed, 0 to remove and 58 not upgraded.
Need to get 0 B/84.3 MB of archives.
After this operation, 425 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```

Run `apt install docker-ce docker-ce-cli containerd.io`  with the apt control file:
```
pi@raspberrypi:~ $ sudo apt install docker-ce docker-ce-cli containerd.io
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  cgroupfs-mount libltdl7 pigz
Recommended packages:
  aufs-tools
The following NEW packages will be installed:
  cgroupfs-mount containerd.io docker-ce docker-ce-cli libltdl7 pigz
0 upgraded, 6 newly installed, 0 to remove and 58 not upgraded.
Need to get 0 B/59.1 MB of archives.
After this operation, 261 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```

I'll recommend to do this rather than set `--no-install-recommends`.

StuartIanNaylor, 2020-01-10:
Its not just Docker though the problem is the complete lack of tools for overlayFS.
Its just such a bad situation as Aufs does have a number of tools the merge down and snapshot tools that are missing with overlayFS but Aufs has this problem with not being compiled in due to what is mainly politics.

You can do some really cool things with union filesystems such as using a zram ram upper over a large sd card lower and periodic merge down to create extremely low write resiliant file systems.
You can even create large file caches with ssd over hardrives.

I am not sure why overlayFS dev stopped with just docker but seems like it did.

MikePadge, 2020-02-19:
@pelwell I just ended up here because the auto installer for Nextcloud debian pi is attempting to install aufs-dkms 

https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh 

I have no idea about the history here, and can do the patches. Just thought I'd let yall know that they appear to be currently leaning on that.

pietroberto, 2020-05-05:
DKMS make.log for aufs-4.19+20190211 for kernel 4.19.97+ (armv7l)
mar  5 mag 2020, 18.37.12, CEST
make: ingresso nella directory "/var/lib/dkms/aufs/4.19+20190211/build"
-I/var/lib/dkms/aufs/4.19+20190211/build/include -DCONFIG_AUFS_FS_MODULE -UCONFIG_AUFS -DCONFIG_AUFS_BRANCH_MAX_127 -DCONFIG_AUFS_SBILIST
make -C /lib/modules/4.19.97+/build M=/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs EXTRA_CFLAGS="-I/var/lib/dkms/aufs/4.19+20190211/build/include -DCONFIG_AUFS_FS_MODULE -UCONFIG_AUFS -DCONFIG_AUFS_BRANCH_MAX_127 -DCONFIG_AUFS_SBILIST" modules
make[1]: ingresso nella directory "/usr/src/linux-headers-4.19.97+"
  CC [M]  /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/module.o
  CC [M]  /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/sbinfo.o
  CC [M]  /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/super.o
  CC [M]  /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/branch.o
In file included from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/branch.h:33,
                 from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/aufs.h:38,
                 from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/sbinfo.c:23:
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/super.h:134:2: error: unknown type name ‘vfs_readf_t’
  vfs_readf_t  si_xread;
  ^~~~~~~~~~~
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/super.h:135:2: error: unknown type name ‘vfs_writef_t’
  vfs_writef_t  si_xwrite;
  ^~~~~~~~~~~~
In file included from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/branch.h:33,
                 from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/aufs.h:38,
                 from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/super.c:27:
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/super.h:134:2: error: unknown type name ‘vfs_readf_t’
  vfs_readf_t  si_xread;
  ^~~~~~~~~~~
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/super.h:135:2: error: unknown type name ‘vfs_writef_t’
  vfs_writef_t  si_xwrite;
  ^~~~~~~~~~~~
In file included from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/branch.h:33,
                 from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/aufs.h:38,
                 from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/module.c:25:
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/super.h:134:2: error: unknown type name ‘vfs_readf_t’
  vfs_readf_t  si_xread;
  ^~~~~~~~~~~
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/super.h:135:2: error: unknown type name ‘vfs_writef_t’
  vfs_writef_t  si_xwrite;
  ^~~~~~~~~~~~
In file included from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/aufs.h:38,
                 from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/sbinfo.c:23:
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/branch.h:243:20: error: unknown type name ‘vfs_readf_t’
 ssize_t xino_fread(vfs_readf_t func, struct file *file, void *buf, size_t size,
                    ^~~~~~~~~~~
In file included from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/aufs.h:38,
                 from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/super.c:27:
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/branch.h:243:20: error: unknown type name ‘vfs_readf_t’; did you mean ‘key_ref_t’?
 ssize_t xino_fread(vfs_readf_t func, struct file *file, void *buf, size_t size,
                    ^~~~~~~~~~~
                    key_ref_t
In file included from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/aufs.h:38,
                 from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/module.c:25:
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/branch.h:243:20: error: unknown type name ‘vfs_readf_t’; did you mean ‘key_ref_t’?
 ssize_t xino_fread(vfs_readf_t func, struct file *file, void *buf, size_t size,
                    ^~~~~~~~~~~
                    key_ref_t
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/branch.h:245:21: error: unknown type name ‘vfs_writef_t’; did you mean ‘key_ref_t’?
 ssize_t xino_fwrite(vfs_writef_t func, struct file *file, void *buf,
                     ^~~~~~~~~~~~
                     key_ref_t
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/branch.h:245:21: error: unknown type name ‘vfs_writef_t’; did you mean ‘pte_write’?
 ssize_t xino_fwrite(vfs_writef_t func, struct file *file, void *buf,
                     ^~~~~~~~~~~~
                     pte_write
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/branch.h:245:21: error: unknown type name ‘vfs_writef_t’; did you mean ‘key_ref_t’?
 ssize_t xino_fwrite(vfs_writef_t func, struct file *file, void *buf,
                     ^~~~~~~~~~~~
                     key_ref_t
In file included from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/branch.h:33,
                 from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/aufs.h:38,
                 from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/branch.c:25:
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/super.h:134:2: error: unknown type name ‘vfs_readf_t’
[... 58 lines cut ...]
         file_update_time
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/vfsub.h:233:9: error: implicit declaration of function ‘update_time’; did you mean ‘pte_same’? [-Werror=implicit-function-declaration]
  return update_time(h_inode, ts, flags);
         ^~~~~~~~~~~
         pte_same
In file included from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/aufs.h:57,
                 from /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/sbinfo.c:23:
/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/vfsub.h: In function ‘vfsub_update_time’:
[... cut ...]
make[2]: *** [scripts/Makefile.build:303: /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/super.o] Error 1
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:303: /var/lib/dkms/aufs/4.19+20190211/build/fs/aufs/branch.o] Error 1
make[1]: *** [Makefile:1522: _module_/var/lib/dkms/aufs/4.19+20190211/build/fs/aufs] Error 2
make[1]: uscita dalla directory "/usr/src/linux-headers-4.19.97+"
make: *** [Makefile:47: fs/aufs/aufs.ko] Error 2
make: uscita dalla directory "/var/lib/dkms/aufs/4.19+20190211/build"

Skillvendor, 2020-06-09:
> Here all commands i used to install docker ce 18.09.0 on Raspian Buster:
>
> ```shell
> sudo apt-get update
> apt-get install apt-transport-https ca-certificates software-properties-common
> curl -fsSL [...]

This is indeed what solved my problem with aufs-dkms. Adding --no-install-recommends to your docker install might fix your issue, it did fix mine