raspberrypi/linux #3022: cat /proc/cpuinfo | grep Hardware on a Pi 4 returns BCM2835, should be BCM2711?

Gadgetoid, 2019-06-24:
Grepping for the hardware version in /proc/cpuinfo reveals BCM2835 on a Pi 4. Should this not be BCM2711?

To reproduce simply: `cat /proc/cpuinfo | grep Hardware`

This should? show: ```Hardware        : BCM2711```

But instead shows: ```Hardware        : BCM2835```

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.50-v7l+ #895 SMP Thu Jun 20 16:03:42 BST 2019 armv7l GNU/Linux
```
```
pi@raspberrypi:~ $ cat /proc/cpuinfo | grep Revision
Revision        : c03111
```
```
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
```
pi@raspberrypi:~ $ vcgencmd version
Jun 20 2019 16:04:31
Copyright (c) 2012 Broadcom
version 407b1da8fa3d1a7108cb1d250f5064a3420d2b7d (clean) (release) (start)
```

popcornmix, 2019-06-24:
It shows BCM2835 on Pi0, Pi1, Pi2 and Pi3.
You should use the [revision code](https://www.raspberrypi.org/documentation/hardware/raspberrypi/revision-codes/README.md) to identify the processor.

Gadgetoid, 2019-06-24:
Oh lord forgive me. Everything old is new again. I wrote a library for this 4 years ago - https://github.com/pimoroni/RPi.version

closed 2019-06-25 by Gadgetoid

kobzardmytro, 2020-01-31:
ubuntu@server:~$ cat /proc/device-tree/model
Raspberry Pi 4 Model B Rev 1.1ubuntu@server:~$ 
ubuntu@server:~$ 
ubuntu@server:~$ cat /proc/cpuinfo | grep Hardware
Hardware	: BCM2835
ubuntu@server:~$ cat /proc/cpuinfo | grep Revision
Revision	: b03111
ubuntu@server:~$ uname -a
Linux server.local 4.19.97-v8-27 #27 SMP PREEMPT Mon Jan 20 10:37:34 PST 2020 aarch64 aarch64 aarch64 GNU/Linux
ubuntu@server:~$ cat /etc/os-release
NAME="Ubuntu"
VERSION="18.04.3 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.3 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
ubuntu@server:~$ vcgencmd version
Jan 16 2020 15:57:11 
Copyright (c) 2012 Broadcom
version d844d7ce19c1696e0f6110aab106e250c3099147 (clean) (release) (start)
ubuntu@server:~$

pelwell, 2020-01-31:
The 64-bit kernel doesn't have a machine description concept, so the BCM2835 string is hard-coded. That's unlikely to change.

kobzardmytro, 2020-01-31:
Thanks I got it.

JamesH65, 2020-01-31:
> The 64-bit kernel doesn't have a machine description concept, so the BCM2835 string is hard-coded. That's unlikely to change.

It would be feasible in that bit of code to check the revision code, and change the machine name to match - I have a PR for it somewhere that I never submitted. ot sure why it was not done like that originally. Is there a real reason?

pelwell, 2020-01-31:
Choose one or more of the following:
1. Hardware is shown as BCM2835 on all 32-bit Pis. It's more of an "Architecture" field - there's a "Model" string for the detail.
2. /proc/cpuinfo doesn't normally have any of this information on a 64-bit build, so this is all stuff we have to maintain downstream.
3. Following on from 2, we'd like to be able to stop supporting this at some point and expect people to use other means to identify the hardware, so putting effort into making this more appealing is counter-productive.
4. Because it was easy and I'm lazy.

kobzardmytro, 2020-01-31:
I understand.

dougbertwashere, 2020-09-09:
I have 2 Pi 4 Bs each with 4GB, where cpuinfo output has Hardware showing 'BCM2711'  
While my 8GB Pi displays BCM2835.  Running HypriotOS

pelwell, 2020-09-09:
I would guess that your 8GB Pi is in 64-bit mode.

dougbertwashere, 2020-09-09:
thanks for the reply, but it is running 32bit hypriotOS, 

$ uname -m
armv7l

JamesH65, 2020-09-09:
My Pi 4 4GB running Raspbian shows 2835....

pelwell, 2020-09-10:
Yes, we know that. As I said above:
> The 64-bit kernel doesn't have a machine description concept, so the BCM2835 string is hard-coded. That's unlikely to change.

The reporting of BCM2711 as a separate machine type was added after 4.19 - what kernel version are you running on the 8GB board? It's also a back-port that doesn't appear in the upstream 5.4, so what the kernel reports is version dependent. 

The take-aways are:
* A 5.4 (or later) kernel built from our tree will report `BCM2711` in an AArch32 build.
* Everything else reports `BCM2835`.

dougbertwashere, 2020-09-11:
thank you for that information

I have a 6 pi cluster. 
I started with 1 to 5 pi's running on 5.4.51 kernels and my Pi 4 B 4gb displays 'BCM2711' properly

then I added a the sixth node, a 8gb node.
checking the kernel I found that the sixth is running 4.19.118  so there is the cause of the issue

and I need to upgrade that OS/kernel to match the others

thanks again