raspberrypi/linux #3024: RPi 4 kernel 4.19.56 Compile error

Dark-Sky, 2019-06-26:
I tried to compile the kernel 4.19.56 kernel for the RPi 4 and and the compile errors out with this. I searched the include with grep and did not get any return on the undeclared. Possible missing header file?

`drivers/staging/vc04_services/vc-sm-cma/vc_sm.c: In function 'cache_op_to_func':
drivers/staging/vc04_services/vc-sm-cma/vc_sm.c:1271:10: error: 'dmac_inv_range' undeclared (first use in this function)
 1271 |   return dmac_inv_range;
      |          ^~~~~~~~~~~~~~
drivers/staging/vc04_services/vc-sm-cma/vc_sm.c:1271:10: note: each undeclared identifier is reported only once for each function it appears in
drivers/staging/vc04_services/vc-sm-cma/vc_sm.c:1274:10: error: 'dmac_clean_range' undeclared (first use in this function)
 1274 |   return dmac_clean_range;
      |          ^~~~~~~~~~~~~~~~
drivers/staging/vc04_services/vc-sm-cma/vc_sm.c:1277:10: error: 'dmac_flush_range' undeclared (first use in this function)
 1277 |   return dmac_flush_range;
      |          ^~~~~~~~~~~~~~~~
drivers/staging/vc04_services/vc-sm-cma/vc_sm.c: At top level:
drivers/staging/vc04_services/vc-sm-cma/vc_sm.c:1473:9: error: redefinition of 'struct vc_sm_cma_ioctl_clean_invalid_block'
 1473 |  struct vc_sm_cma_ioctl_clean_invalid_block {
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from drivers/staging/vc04_services/vc-sm-cma/vc_sm.c:57:
./include/linux/broadcom/vc_sm_cma_ioctl.h:92:9: note: originally defined here
   92 |  struct vc_sm_cma_ioctl_clean_invalid_block {
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
drivers/staging/vc04_services/vc-sm-cma/vc_sm.c: In function 'vc_sm_cma_compat_ioctl':
drivers/staging/vc04_services/vc-sm-cma/vc_sm.c:1497:1: warning: control reaches end of non-void function [-Wreturn-type]
 1497 | }`

pelwell, 2019-06-26:
It builds for me. Which ARCH and defconfig are you using?

Dark-Sky, 2019-06-26:
bcm2711_defconfig

When I did a clone I only pulled the rpi-4.19.y branch. Should I have pulled the whole tree?

pelwell, 2019-06-26:
That shouldn't matter, but beware of .zip downloads - they have been known to be unreliable.

Dark-Sky, 2019-06-26:
I don't do zips; got burned a long time ago. lol.

I did find it in one of the undeclared in a header in arch/include/asm but have not checked the others. If they are all in arch/include/asm then maybe a include is needed in the affected .c file?

```
[ray@ray-pc arch]$ grep -r dmac_inv_range
arm/mm/proc-syms.c:EXPORT_SYMBOL(dmac_inv_range);
arm/include/asm/glue-cache.h:#define dmac_inv_range			__glue(_CACHE,_dma_inv_range)
arm/include/asm/cacheflush.h:#define dmac_inv_range			cpu_cache.dma_inv_range
arm/include/asm/cacheflush.h:extern void dmac_inv_range(const void *, const void *);
```

Dark-Sky, 2019-06-26:
They are all in arch/include/asm. The compiler just does not know to look there.

[ray@ray-pc arch]$ grep -r dmac_inv_range
arm/mm/proc-syms.c:EXPORT_SYMBOL(dmac_inv_range);
arm/include/asm/glue-cache.h:#define dmac_inv_range			__glue(_CACHE,_dma_inv_range)
arm/include/asm/cacheflush.h:#define dmac_inv_range			cpu_cache.dma_inv_range
arm/include/asm/cacheflush.h:extern void dmac_inv_range(const void *, const void *);
[ray@ray-pc arch]$ grep -r dmac_clean_range
arm/mm/proc-syms.c:EXPORT_SYMBOL(dmac_clean_range);
arm/include/asm/glue-cache.h:#define dmac_clean_range		__glue(_CACHE,_dma_clean_range)
arm/include/asm/cacheflush.h:#define dmac_clean_range		cpu_cache.dma_clean_range
arm/include/asm/cacheflush.h:extern void dmac_clean_range(const void *, const void *);
[ray@ray-pc arch]$ grep -r dmac_flush_range
arm/mm/proc-syms.c:EXPORT_SYMBOL(dmac_flush_range);
arm/mm/dma-mapping.c:				dmac_flush_range(ptr, ptr + PAGE_SIZE);
arm/mm/dma-mapping.c:			dmac_flush_range(ptr, ptr + size);
arm/include/asm/glue-cache.h:#define dmac_flush_range		__glue(_CACHE,_dma_flush_range)
arm/include/asm/cacheflush.h:#define dmac_flush_range		cpu_cache.dma_flush_range
arm/include/asm/cacheflush.h:extern void dmac_flush_range(const void *, const void *);

Dark-Sky, 2019-06-26:
Here is where the include was removed:

https://github.com/raspberrypi/linux/commit/30e54a37c6f90626afc5598110c19badeb2e44fd#diff-b0dd46012f58b6eac3e41f9ed1a0b859

pelwell, 2019-06-26:
You are using ARCH=arm, aren't you?

Dark-Sky, 2019-06-26:
ARCH=arm64

steev, 2019-06-26:
Indeed, arm64 doesn't have the defines, they were added back as part of https://github.com/raspberrypi/linux/commit/fe4d96ba28fb4c3eae1c488f2f1c14e100c43871 but doesn't seem they were migrated to arm64 as well

pelwell, 2019-06-26:
Hence my question two hours ago...

You should know by now that 64-bit is currently a niche use case, so it helps when you are upfront about using it.

It should probably be updated to use the dma_ops, but @6by9 may have other ideas.

6by9, 2019-06-26:
The cache flushing stuff is all down to the Pi3 HEVC hw assisted decode which uses the QPUs, VPUs, and I think some H264 blocks. Flushing the entire buffer (which is what the dma_ops do) is too expensive, and the code has been optimised to within an inch to get the required performance. 
The use of these cache ops is an abuse, hence having to do the revert, but is needed in the above case.

We could just make the ioctls and supporting code conditional on being ARM (not ARM64).

Strit, 2019-06-26:
Just wondering.
If arm64 is such a niche case, why start doing 64-bit commits like https://github.com/raspberrypi/linux/commit/cdb78ce891f6c6367a69c0a46b5779a58164bd4b?
Was getting my hopes up when I saw that commit. :smiley:

pelwell, 2019-06-26:
Because:

a) it's a matter of when, not if, we produce an official 64-bit build, and we have to start somewhere, and
b) we aren't the only people building kernels from this repo, and it's important to share any work we've done to avoid duplicated (and hence wasted) effort.

digitallyelite, 2019-06-26:
@6by9 I tried to carve out the bits in drivers/staging/vc04_services/vc-sm-cma/vc_sm.c but unfortunately, it seems a tiny bit more involved as drivers/staging/vc04_services/vchiq-mmal/mmal-vchiq.c seems to have some dependencies there that are a bit above my skill level.

digitallyelite, 2019-06-26:
> Because:
> 
> a) it's a matter of when, not if, we produce an official 64-bit build, and we have to start somewhere, and

Good to hear when, not if.  Since it has been 3 years since the first 64bit raspberrypi hardware was released

pelwell, 2019-06-27:
Nobody has ever come up with a compelling argument that justifies the extra effort of supporting another build. That the arm64 kernel doesn't support FIQs, ruling out our downstream dwc_otg USB performance enhancements, is the nail in the coffin for mainstream Raspbian 64-bit Pi3. Pi4, with it's standard XHCI USB3 and larger address range, tips the balance in my opinion, but there is still significant work to be done before we could consider going live with a 64-bit kernel (with 32-bit userspace).

Whether to then add a full 64-bit userspace Raspbian variant is a separate decision.

agherzan, 2019-06-28:
This is also a regression for RPi3 64 bit support.

agherzan, 2019-06-28:
This broke when 30e54a37c6f90626afc5598110c19badeb2e44fd brought in cache control ioctls using cacheflush. This implementation depends on the DMA Cache Coherency methods which don't exist and seems  to have never existed on arm64.

agherzan, 2019-07-01:
We can close this as https://github.com/raspberrypi/linux/commit/9d1deec93fa8b1b4953ff5e9210349f3c85b9a8d dropped the bits that were failing at compile using the default defconfig.

d11wtq, 2019-07-03:
This may not be the place to ask, but the official kernel compilation docs are out of date and don't include RPi4 info from what I can see. Is this the correct way to compile for the RPi4 (on the RPi4)?

32-bit:

```
make bcm2711_defconfig

make -j4 zImage modules dtbs
sudo make modules_install

sudo cp arch/arm/boot/dts/*.dtb /boot/
sudo cp arch/arm/boot/dts/overlays/*.dtb* /boot/overlays/
sudo cp arch/arm/boot/dts/overlays/README /boot/overlays/

sudo cp arch/arm/boot/zImage /boot/my-kernel.img
```

64-bit:

```
make ARCH=arm64 bcm2711_defconfig

make ARCH=arm64 -j4 zImage modules dtbs
sudo make modules_install

sudo cp arch/arm/boot/dts/*.dtb /boot/
sudo cp arch/arm/boot/dts/overlays/*.dtb* /boot/overlays/
sudo cp arch/arm/boot/dts/overlays/README /boot/overlays/

sudo cp arch/arm/boot/zImage /boot/my-64bit-kernel.img
```

Then add `kernel=my-kernel.img` on the same line before everything else on /boot/cmdline.txt.

Asking because I haven't tried 64-bit but on my RPi4, trying this several times a few days ago the kernel would compile but then it Pi simply would fail to boot (a few initial blinks and then a static green LED). Changing the cmdline.txt back meant the Pi would still fail to boot (presumably due to destructive overwrites on /boot and in the modules). The steps I followed were based on the RPi3 steps from https://www.raspberrypi.org/documentation/linux/kernel/building.md but using bcm2711 instead of 2709. Are there undocumented manual steps to configure needed modules for the Pi4?

d11wtq, 2019-07-03:
Just realized I'm an idiot. I was editing cmdline.txt and not config.txt :facepalm:

Dark-Sky, 2019-07-04:
@d11wtq 

A lot of good info in this thread and it's links in it.

https://github.com/raspberrypi/linux/issues/3032

Dark-Sky, 2019-07-04:
@d11wtq 

Lots of good info in this thread and it's links in it.

https://github.com/raspberrypi/linux/issues/3032

d11wtq, 2019-07-04:
@Dark-Sky awesome thanks!

Dark-Sky, 2019-07-04:
For got to mention that gentoo basic config in of a link in that thread was very basic with a lot of stuff disabled. For us doing a make bcm2711_defconfig in the root of the RPi 4.19 tree was a better starting point. Make sure you do a git pull fird because some commits was made today you will need.

agherzan, 2019-07-04:
@d11wtq Checkout this https://andrei.gherzan.ro/2019/07/04/raspbian-rpi4-64.html . This should give you all you need.

JamesH65, 2019-07-30:
Closing, as suggested here https://github.com/raspberrypi/linux/issues/3024#issuecomment-507302458

closed 2019-07-30 by JamesH65