raspberrypi/linux #3025: RPi 3 B+ doesn't process unicast promiscuous packets

aviramha, 2019-06-27:
I'm trying to use the RPi to process and analyze port mirroring traffic. During my tests, it seems that certain packets are seen as RX errors on the ifconfig, and I can't see those packets using tcpdump. In my setup I use the RPi Ethernet port for the port mirroring and another interface is USB Ethernet one. When I switched cables and made the USB Ethernet the port mirroring interface the bug didn't occur.
I initially saw #2566 but tried to do rpi-update with no success.

**To reproduce**

1. Connect RPi 3 B+ Ethernet port to Linux computer B.
2. On the RPi (replace eth0 with ethernet interface if differ), execute tcpdump -nvvei eth0
3. Use scapy or any other tool to send raw packet on the interface from computer B:
`from scapy.all import *
import base64
p = base64.b64decode(b'ACWQl0oGAAwpJ39+CABFAAPnCl5AAIAGAADAqGdwaChCRfVUAFAnrEUMl/K7CVAYgWDWXwAAR0VUIC9lbi11cy9ob21lcGFnZS9fc2MvanMvYTczNjM0NDItY2VjNjc0OTcvZGlyZWN0aW9uPWx0ci5sb2NhbGVzPWVuLXVzLnRoZW1lcz1zdGFydC5kcGk9cmVzb2x1dGlvbjF4L2EwLTQ4NjZjMy1mODVkOTQ3YS8yMy0zNTMwYzMtNGNjMWQ2ZmIvMTMtZjU3MDEyLTY4ZGRiMmFiL2IxLWUwNzU1Ni1jZjUzZGIzMi9maW5hbmNlLWpzLTE1LWJiNGYwMGIyNDhkZTYzODYwNjI4OTQ0ZTYwZDk2ZC1lNTc4MDk3MD92ZXI9Mi4wLjUzOTUuMjQxMDIgSFRUUC8xLjENCkFjY2VwdDogYXBwbGljYXRpb24vamF2YXNjcmlwdCwgKi8qO3E9MC44DQpSZWZlcmVyOiBodHRwOi8vcmVuZGVyaW5nLXdlc3R1cy10ZXN0LWljZWxhYi1wcmltZS5jbG91ZGFwcC5uZXQvDQpBY2NlcHQtTGFuZ3VhZ2U6IGVuLVVTDQpYLURvd25sb2FkLUluaXRpYXRvcjogc2NyaXB0PSJkb2MgREZEODsgc3BlY3VsYXRpdmU7IGh0bWwgdG9rZW5pemVyIEM5MTgiDQpVc2VyLUFnZW50OiBNb3ppbGxhLzUuMCAoV2luZG93cyBOVCA2LjE7IFRyaWRlbnQvNy4wOyBydjoxMS4wKSBsaWtlIEdlY2tvIFBUU1QvMTg0DQpBY2NlcHQtRW5jb2Rpbmc6IGd6aXAsIGRlZmxhdGUNCkhvc3Q6IHJlbmRlcmluZy13ZXN0dXMtdGVzdC1pY2VsYWItcHJpbWUuY2xvdWRhcHAubmV0DQpDb25uZWN0aW9uOiBLZWVwLUFsaXZlDQpDb29raWU6IHByaW1lX3VzZXJuYW1lPW1zbnByaW1lOyBwcmltZV9wYXNzd29yZD1BbXAzZCE7IFByZWZlcmVuY2VzTXNuPWV5SkliMjFsVUdGblpTSTZleUpUZEhKcGNHVnpJanBiWFN3aVRXVlRkSEpwY0dWTmIyUjFiR1Z6SWpwYlhTd2lUV0Z5YTJWMFEyOXVabWxuZFhKaGRHbHZiaUk2ZXlKTllYSnJaWFFpT2lKbGJpMVZVeUlzSWxOMWNIQnlaWE56VUhKdmJYQjBJanBtWVd4elpTd2lVSEpsWm1WeWNtVmtUR0Z1WjNWaFoyVkRiMlJsSWpvaVpXNHRWVk1pTENKRGIzVnVkSEo1UTI5a1pTSTZJbFZUSW4xOWZRMg0KDQo=')
sendp(p, iface='INTERFACE)'
`
4. You wont see the packet on tcpdump, and if you run ifconfig you'll see that RX error increased.

**Expected behaviour**
TCPDump or any raw capturing tool should receive the packet.

**Actual behaviour**
Packet seems to be considered as error and tcpdump doesn't see the packet.

**System**

[debuginfo.log](https://github.com/raspberrypi/linux/files/3333641/debuginfo.log)

**Additional context**
I ran raspinfo after switching cables (so now I use the RPi ethernet for regular networking and the USB dongle for the port tapping)

SudhamaniB, 2019-08-07:
Hi,

I tried to replicate the issue with
Kernel Version: Linux raspberrypi 4.19.60-v7+ #2 SMP Thu Aug 1 21:03:18 IST 2019 armv7l GNU/Linux.

I could see no RX packet error when I sent Raw packets from Host PC to RPi3B+
Which OS and version , model you have seen this error?

Regards,
Sudhamani.

aviramha, 2019-08-07:
Linux version 4.19.55-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #906 SMP Mon Jun 24 21:00:57 BST 2019

> Hi,
> 
> I tried to replicate the issue with
> Kernel Version: Linux raspberrypi 4.19.60-v7+ #2 SMP Thu Aug 1 21:03:18 IST 2019 armv7l GNU/Linux.
> 
> I could see no RX packet error when I sent Raw packets from Host PC to RPi3B+
> Which OS and version , model you have seen this error?
> 
> Regards,
> Sudhamani.

Linux version 4.19.55-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #906 SMP Mon Jun 24 21:00:57 BST 2019

SudhamaniB, 2019-10-21:
> Linux version 4.19.55-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #906 SMP Mon Jun 24 21:00:57 BST 2019
>
>
> Linux version 4.19.55-v7+ (dom@buildbot) (gcc [...]

Are you still facing this issue with latest kernel?

JamesH65, 2020-02-13:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

artztech, 2020-05-08:
hello,

yeah, @aviramha , still not working on rpi3 b+.?  i'm facing same issue on #2566. any update from rpi team or microchip?

root@raspberrypi:~# uname -a
Linux raspberrypi 5.4.35-v7+ #1314 SMP Fri May 1 17:41:46 BST 2020 armv7l GNU/Li                                                                                        nux
root@raspberrypi:~# ifconfig eth0 promisc up
root@raspberrypi:~# ifconfig eth0 down
root@raspberrypi:~# ifconfig eth0 up
root@raspberrypi:~# ifconfig
eth0: flags=4355<UP,BROADCAST,PROMISC,MULTICAST>  mtu 1500
        ether b8:27:eb:85:5e:1d  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 28  bytes 6858 (6.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 2322  bytes 1303071 (1.2 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2322  bytes 1303071 (1.2 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.10.10.71  netmask 255.255.255.0  broadcast 10.10.10.255
        ether b8:27:eb:d0:0b:48  txqueuelen 1000  (Ethernet)
        RX packets 1199  bytes 82016 (80.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 1103  bytes 146345 (142.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

aviramha, 2020-05-10:
> hello,
> 
> yeah, @aviramha , still not working on rpi3 b+.? i'm facing same issue on #2566. any update from rpi team or microchip?
> 
> Linux raspberrypi 5.4.35-v7+ #1314 SMP Fri May 1 17:41:46 BST 2020 armv7l GNU/Li nux
> root@raspberrypi:~# ifconfig eth0 promisc root@raspberrypi:~# ifconfig eth0 down
> root@raspberrypi:~# ifconfig eth0 up root@raspberrypi:~# ifconfig
> eth0: flags=4355<UP,BROADCAST,PROMISC,MULTICAST> mtu 1500
> ether b8:27:eb:85:5e:1d txqueuelen 1000 (Ethernet)
> RX packets 0 bytes 0 (0.0 B)
> RX errors 0 dropped 0 overruns 0 frame 0
> TX packets 28 bytes 6858 (6.6 KiB)
> TX errors 0 dropped 0 overruns 0 carrier 0 collisions 0
> 
> lo: flags=73<UP,LOOPBACK,RUNNING> mtu 65536
> inet 127.0.0.1 netmask 255.0.0.0
> loop txqueuelen 1000 (Local Loopback)
> RX packets 2322 bytes 1303071 (1.2 MiB)
> RX errors 0 dropped 0 overruns 0 frame 0
> TX packets 2322 bytes 1303071 (1.2 MiB)
> TX errors 0 dropped 0 overruns 0 carrier 0 collisions 0
> 
> can we use another driver suitable with this chipset for performing testing?

Hi,
I ended up using an external USB for port tapping, didn't investigate any further..