raspberrypi/linux #3026: USB3-SATA controller doesn't work when connected to a USB3 port.

janost, 2019-06-27:
There are a few similar issues here, however all of them are for older Pis with USB2 only, and my issue is specifically related to USB3 on the Raspi 4.

**Describe the bug**
Orico USB3-SATA enclosure doesn't work when connected to one of the Raspberry Pi 4's USB3 ports. According to some google searches, some USB3-SATA controllers have buggy UAS implementation in their firmware, so people usually have success disabling it on their machines, however it doesn't work in my case. Also, the enclosure works fine even with UAS enabled on other systems (Windows, Arch Linux on a laptop, etc).

Affected device:
```
Bus 001 Device 005: ID 152d:0578 JMicron Technology Corp. / JMicron USA Technology Corp. JMS567 SATA 6Gb/s bridge
```

**To reproduce**
Connect the device to a Raspi4's USB3 port with a SATA SSD inside and try to write some data to it (for example using `dd`).

**Expected behaviour**
The data gets written properly.

**Actual behaviour**
The writing process hangs for a long time or indefinitely, data won't get written properly and various error messages get spammed to `dmesg`. The device is basically unusable.

**System**
Raspi 4 4GB with a Samsung EVO+ 64GiB micro SD.
Running the latest Raspbian system and firmware.
```
root@raspberrypi:~# uname -a
Linux raspberrypi 4.19.56-v7l+ #1242 SMP Wed Jun 26 17:41:03 BST 2019 armv7l GNU/Linux
root@raspberrypi:~# /opt/vc/bin/vcgencmd version
Jun 26 2019 17:42:42
Copyright (c) 2012 Broadcom
version 1186f932808ed601ddd583a30a3ce055477b1a26 (clean) (release) (start)
```

**Logs**
[USB2: connecting the device](https://pastebin.com/Es9yxEJc)
[USB2: successful write test with `dd`](https://pastebin.com/u03YJVrg)

[USB3 with UAS: connecting the device](https://pastebin.com/3DL4xwzn)
[USB3 with UAS: failed write test with `dd`](https://pastebin.com/bviytcYW)
[USB3 with UAS: dmesg after write test](https://pastebin.com/jKAdmXad)

[USB3 without UAS: connecting the device](https://pastebin.com/VvFC7bS3)
USB3 without UAS: write test with `dd` hangs indefinitely
[USB3 without UAS: dmsg after write test](https://pastebin.com/5cU5BLzx)

**Additional context**
The device works fine when connected to other systems, even with UAS enabled.

pelwell, 2019-06-27:
Does this enclosure have a separate power supply or are you trying to power it from the Pi's USB3 port alone?

janost, 2019-06-27:
It doesn't have a separate power supply, it gets power from the USB port. However, I'd like to emphasize that the device works properly when connected to one of the Raspi's USB2 ports.

pelwell, 2019-06-27:
Do you have a powered hub you can connect through?

janost, 2019-06-27:
I'm going to test it with the built-in hub in my monitor and will report back soon.

janost, 2019-06-27:
It does look like a power issue, with a powered hub, it works.
```
root@raspberrypi:/mnt/ssd1# dd if=/dev/zero of=/mnt/ssd1/testfile bs=4M count=1024
1024+0 records in
1024+0 records out
4294967296 bytes (4.3 GB, 4.0 GiB) copied, 30.0171 s, 143 MB/s
```

However I discovered a probably unrelated issue. While writing to the disk, some kernel threads are heavily loading 2-3 CPU cores, even causing high network latency on the ethernet interface while the disk operation is running.
Should I open a separate issue about that?

pelwell, 2019-06-27:
> Should I open a separate issue about that?

Yes, please - it helps others to find your issuek and means we can close the two issues independently.

closed 2019-06-27 by janost