raspberrypi/linux #3027: AEOTEC USB Zwave module not creating /dev/ttyACM0

markcarline, 2019-06-27:
When i connect one of these:

https://www.amazon.co.uk/Aeotec-Z-Stick-Z-Wave-create-gateway/dp/B00YETCNOE/ref=asc_df_B00YETCNOE/?tag=googshopuk-21&linkCode=df0&hvadid=311081482938&hvpos=1o1&hvnetw=g&hvrand=15354715739214754224&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=1006626&hvtargid=pla-563872593762&psc=1&th=1&psc=1

I normally get this device:

/dev/ttyACM0

which is used for serial comms to zwave home automation via home assistant.

Connecting it to the Pi4 + Buster i see no such device, infact when i connect via lsusb i see no change before / after connecting.

Any ideas?

pelwell, 2019-06-27:
Do you get the same lack of results in the USB2 and USB3 ports?

markcarline, 2019-06-27:
Yes. I tried both usb2 and usb3 and I'm using a genuine pi 4 for power supply.

pelwell, 2019-06-27:
Are there any relevant messages in the kernel log? Unplug the device, `sudo dmesg -C` (to clear out the old messages), plug it in, wait a moment, then `sudo dmesg > usb.txt`.

markcarline, 2019-06-27:
Hmm - i tried a fresh install of buster and got this:

[  398.804524] usb 1-1.1: new full-speed USB device number 4 using xhci_hcd
[  398.941260] usb 1-1.1: New USB device found, idVendor=0658, idProduct=0200, bcdDevice= 0.00
[  398.941278] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  399.010369] cdc_acm 1-1.1:1.0: ttyACM0: USB ACM device
[  399.012948] usbcore: registered new interface driver cdc_acm
[  399.012958] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters

and i see /dev/ttyACM0

something is odd about my original install.  Let me reinstall HA over the weekend and i'll report back.

Thanks!

pelwell, 2019-06-27:
"HA"?

markcarline, 2019-06-27:
HA - Home Assistant.  It uses zwave via the serial port /dev/ttyACM0

markcarline, 2019-06-27:
ok so.... i tried my "fresh" install and i can't recreate that message above.  Seems very temperamental.  I wonder if its power related.

pelwell, 2019-06-27:
Hmm - it _could_ be power related, but I would expect some kind of response to a device insertion, not nothing.

markcarline, 2019-06-27:
Hmm - I tried the MicroSD card which wouldnt work on the Pi4 on my Pi3B+ and works flawlessly, plugged in + out + in + out....

   81.081896] usb 1-1.3: new full-speed USB device number 6 using dwc_otg
[   81.215187] usb 1-1.3: New USB device found, idVendor=0658, idProduct=0200, bcdDevice= 0.00
[   81.215205] usb 1-1.3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[   81.289184] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
[   81.290193] usbcore: registered new interface driver cdc_acm
[   81.290201] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[   95.383586] usb 1-1.3: USB disconnect, device number 6
[  102.641555] usb 1-1.3: new full-speed USB device number 7 using dwc_otg
[  102.774812] usb 1-1.3: New USB device found, idVendor=0658, idProduct=0200, bcdDevice= 0.00
[  102.774830] usb 1-1.3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  102.775964] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device
[  110.231628] usb 1-1.3: USB disconnect, device number 7
[  115.331396] usb 1-1.3: new full-speed USB device number 8 using dwc_otg
[  115.464944] usb 1-1.3: New USB device found, idVendor=0658, idProduct=0200, bcdDevice= 0.00
[  115.464962] usb 1-1.3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  115.466096] cdc_acm 1-1.3:1.0: ttyACM0: USB ACM device

markcarline, 2019-06-28:
I thought I would check with aeotec as to the certification of the pi4 am I mentioned to check if a module is loaded, they said:

-----
​We have tested using RPi V3 in this case, but we have not yet tested RPi 4. One item to absolutely make sure is that the cdc acm module is installed from the Linux kernel which is required to actuate as a ttyACM0 device.  I'll look into acquiring a RPi V4 to see if i can find any issues here.

Cheers,
Chris Cheng
Field Application Engineer
Aeotec Inc.
www.Aeotec.com
-----

I'm going to test more over the weekend, any ideas about the module they mentioned?

pelwell, 2019-06-28:
The module is in Raspbian images - see your kernel logs above.

Can you try connecting via a powered hub?

markcarline, 2019-06-28:
I currently only have the official raspberry pi 4 power supply, can you recommend a powered hub? I'll buy one. Thanks!

pelwell, 2019-06-28:
I wasn't expecting you to buy one, especially since I don't know if power is the cause of the problem, but the general rule of thumb round here is that Amazon Basics cables and adapters are safe bets.

bieniu, 2019-06-28:
I had exactly same issue on RPi 3B after upgrade kernel to version raspberrypi-kernel_1.20181112-1 (November 4). Downgrade kernel fixed the issue. I haven't updated the kernel since then.

markcarline, 2019-06-28:
> I had exactly same issue on RPi 3B after upgrade kernel to version raspberrypi-kernel_1.20181112-1 (November 4). Downgrade kernel fixed the issue. I haven't updated the kernel since then.

interesting but but i am guessing i can't do that with the PI4?

bieniu, 2019-06-28:
> interesting but but i am guessing i can't do that with the PI4?
I think not.

markcarline, 2019-06-28:
so... I found something interesting.  If i connect it to one of the USB 2 ports (i tried all four actually) i get nothing in the dmesg logs.

but then when i connect a four port hub (that could be powered - but isnt) i see these:

[ 1269.486927] usb 1-1.4: new high-speed USB device number 15 using xhci_hcd
[ 1269.617433] usb 1-1.4: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.11
[ 1269.617448] usb 1-1.4: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[ 1269.617461] usb 1-1.4: Product: USB 2.0 Hub
[ 1269.618998] hub 1-1.4:1.0: USB hub found
[ 1269.619082] hub 1-1.4:1.0: 4 ports detected
[ 1270.046942] usb 1-1.4.1: new high-speed USB device number 16 using xhci_hcd
[ 1270.277312] usb 1-1.4.1: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.11
[ 1270.277327] usb 1-1.4.1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[ 1270.277340] usb 1-1.4.1: Product: USB 2.0 Hub
[ 1270.278633] hub 1-1.4.1:1.0: USB hub found
[ 1270.278720] hub 1-1.4.1:1.0: 4 ports detected

then when i connect the Aeotec USB stick to one of the back two USB ports I then see:

[ 1329.418219] usb 1-1.4.1.1: new full-speed USB device number 17 using xhci_hcd
[ 1329.654873] usb 1-1.4.1.1: New USB device found, idVendor=0658, idProduct=0200, bcdDevice= 0.00
[ 1329.654890] usb 1-1.4.1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 1329.657705] cdc_acm 1-1.4.1.1:1.0: ttyACM0: USB ACM device

It now works 100% of the time.

odd

So your suggestion of using a hub @pelwell you might be onto something.

pelwell, 2019-06-28:
I think the Pi budget will stretch to one of these - I'll get one on order.

markcarline, 2019-06-28:
ok cool - I'm sure its something simple but from what i've read about this stick a few people have reported that updated BIOS on NUC PCs has fixed similar issues so maybe this is firmware related.

i remember seeing something similat with the pi2s back in the day and @ghollingworth i think helped fix it - some kind of timing issue with usb (I sent my hub in and they put it into some kind of usb tool.

anyway - i'm up and running with ZWave, Home Assistant and a super cool Pi4 now 💯 :)

markcarline, 2019-07-03:
By the way I haven't to notice this article which talked about you firmware changes to the USB chip, could this issue be fixed as part of those changes?....

https://www.tomshardware.com/news/raspberry-pi-4-firmware-update-tested,39791.html

pelwell, 2019-07-03:
I'm expecting the ZWave module to arrive any day now so I'll be able to try the new firmware, but I wouldn't expect it to fix this issue. Plus, the power saving features are causing problems with some devices. However, that mainly seems to affect high throughput to SSDs, etc., and it is possible to revert to the original version, so feel free to try for yourself: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=243500&p=1490467#p1490467

daaaaan, 2019-07-03:
I can confirm the new firmware **doesn't** help the issue.

P33M, 2019-07-03:
Looking at some USB analyzer traces, the zwave module doesn't even signal a USB attach event. On a VLI expansion card in a PC, I see an attach event but spurious SE1s (illegal bus states) which makes me think that there's some oddness in the way the device is pulling D+/D-. Will require probing with a scope to find out what is happening.

daaaaan, 2019-07-03:
Not sure if its relevant but on the rpi3 my stick exhibits no issue.

markcarline, 2019-07-03:
This is the USB hub that when I connect it to the pi 4 and the zwave USB device everything works fine:

https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.co.uk%2Fulk%2Fitm%2F283514195231

systemofapwne, 2019-07-04:
I might have a similar problem with a "WiMOD iM871A-usb". Whenever I plug in the stick, a new device is created in /dev. That stick has been configured to receive W-MBUS data. On my PC, it shows me data popping in when I connect to it via putty. But on the pi4 with Raspbian Buster, if I do "cat" on that device, the output is quiet (while on stretch, it was outputting received data).
I could reproduce a similar behaviour with a self-made "nano-cul" stick.

pelwell, 2019-07-04:
There is something peculiar electrically about the Zwave dongle. When you plug it into the Pi 4 (through a USB analyser) the Pi tries to talk USB3 to it, and gets no response because the device is only USB2-capable. This shouldn't happen. Repeating the same test with a PC USB3 port as host you see the same USB3 chatter to start with, then after a few hundred milliseconds it falls back to USB3.

This suggests that the dongle has the electrical signature of a USB3 device and the VIA USB3 host controller doesn't get past that to try USB2. (I'm not sure that behaviour is required by the USB3 spec but it seems sensible)

systemofapwne, 2019-07-04:
This is what I get with my stick:
[ 6951.473076] usb 1-1.4: new high-speed USB device number 8 using xhci_hcd
[ 6951.625698] usb 1-1.4: New USB device found, idVendor=05e3, idProduct=0610, bcdDevice= 6.54
[ 6951.625712] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 6951.625723] usb 1-1.4: Product: USB2.1 Hub
[ 6951.625732] usb 1-1.4: Manufacturer: GenesysLogic
[ 6951.634230] hub 1-1.4:1.0: USB hub found
[ 6951.634533] hub 1-1.4:1.0: 4 ports detected
[ 6951.973078] usb 1-1.4.4: new full-speed USB device number 9 using xhci_hcd
[ 6952.130569] usb 1-1.4.4: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[ 6952.130579] usb 1-1.4.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 6952.130587] usb 1-1.4.4: Product: WiMOD iM871A-usb
[ 6952.130594] usb 1-1.4.4: Manufacturer: Silicon Labs
[ 6952.130601] usb 1-1.4.4: SerialNumber: 00B93F17
[ 6952.146717] cp210x 1-1.4.4:1.0: cp210x converter detected
[ 6952.152214] usb 1-1.4.4: cp210x converter now attached to ttyUSB1

For this test, I connected it to a hub this time. Yet, it fails to work, even though it gets added to /dev: There is no serial throughput whatsoever.

systemofapwne, 2019-07-04:
Just to make sure, that I did not mess anything up when I upgraded from stretch to buster: I just flashed buster (lite) from iso, made an apt-get update/upgrade/dist-upgrade and tried it again:
"sudo cat /dev/ttyUSB1" does not print anything, while this stick definitely receives data and is supposed to print it here. There is definitly something messed up.

fribse, 2019-07-09:
> There is something peculiar electrically about the Zwave dongle. When you plug it into the Pi 4 (through a USB analyser) the Pi tries to talk USB3 to it, and gets no response because the device is [...]

Wouldn't that fit with those that have used a hub in between, if it's a USB2 hub, that it will not get confused, and just go to USB2?

pelwell, 2019-07-09:
Yes, it does fit the hypothesis.

TerrorSource, 2019-07-11:
Question remains, who is causing this and how to solve this? The Pi4 or the Aeotec stick?
- Aeotec stick is using the wrong chips as it seems
- Pi 4 cannot past the USB3 chip on the stick to fallback on USB2.0
I cannot find any other issue with this stick on USB3.0. I cannot imagen that no-one ever used this stick on a USB3.0 port before the Pi4.

pelwell, 2019-07-11:
There is clearly room for improvement on both sides, but the ZWave dongle appears to be non-compliant. I haven't seen a requirement for USB3 controllers and hubs to revert to USB2 after a while, but other controllers do it otherwise nobody would be able to use the Aeotec device.

We have a number of things to work through with manufacturer of the Pi4 USB3 device, and the question of USB3 to USB2 fallback is one of them but far from the top priority.

TerrorSource, 2019-07-13:
Just a quick update. Using a USB 2.0 hub is a good workaround

Bought the cheapest USB 2.0 hub i could find ( https://www.action.com/nl-nl/p/usb-poort-/ )
Connect the USB 2.0 hub, saw the extra device via dmesg
Connected Aeotec Z-wave stick in the USB 2.0 hub. Is now visible also and gets ttyACM0 port assigned.

Mastiffen, 2019-07-23:
Are there any official word on this? I have tried connectign a powered USB 3.0 Hub without using the power to the USB 2.0 ports, and that works halfway. It creates the device, and it connects, but only partly. The messages that come in are not correct.

Edit: It started working as it should after a few boots and now have been working correctly for some days.

P33M, 2019-07-23:
The device is non-compliant and causes issues with certain USB3.0 ports. See:
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=245031#p1502030

The recommended workaround is to connect a USB2.0 hub between the Pi and the device.

closed 2019-07-23 by P33M

markcarline, 2019-07-26:
UPDATE:  The ticket i opened with AEOTEC about this issue mentioned:

_The Z-Stick Gen5 does support USB3.0, while we believe that USB 3.0 port of the RPi4 has poor USB current drive capability. Try using a high power adapter to power the RPi4 (5VDC, 3.5A, 17.5W).
​The only other alternative is to try an external USB hub between RPi 4 and Z-Stick Gen5._

Mastiffen, 2019-07-26:
That does not explain why it works when I use the USB 3.0 hub without the power, but doesn't work when I plug in the power on the same hub.

JamesH65, 2019-07-26:
They are talking rubbish. I've seen the traces on a scope - their device is almost certainly non-compliant, and to implicate the Pi4 current capacity is passing the buck.

markcarline, 2019-07-26:
I have updated my case with AEOTEC and asked them to comment on this guthub issue with their comments.

P33M, 2019-07-26:
I dismantled the device to find out why it's broken. There is a discrete pull-up resistor connected between D+ and Vbus directly.

Mastiffen, 2019-07-26:
@P33M Is it some kind of a "soft" short circuit? If it's a short circuit then there's no enough power without power to the hub to actually make the hub understand that it's being short circuited. Is that about right?

alucryd, 2019-08-01:
FWIW my test results are a bit contradictory to what I read here. A 2.0 hub I had lying around didn't help at all, instead I ended up plugging the zwave stick into my desktop monitor hub. Both 2.0 and 3.0 ports on the hub are working fine, and I can connect the hub to either 2.0 or 3.0 ports on the rpi without issue.

pelwell, 2019-08-01:
Out of interest, does the Aeotec dongle work in the USB 2 hub when it's connected to some other Pi or PC?

alucryd, 2019-08-01:
Yes, the dongle was correctly discovered when I plugged the hub into my laptop.

roberttuttle, 2019-08-02:
My current results on this issue:
- Wall-powered USB 3.0 hub into RPi 4 USB 3.0 port + Aeotec stick on hub = working (dmesg reports both hub and stick working ... /dev/ttyACM0 activated)
- Bus-powered USB 3.0 hub into RPi 4 USB 3.0 port + Aeotec stick on hub = not working (hub reports fine via dmesg ... Aeotec stick ... nothing)
- Tried both options above on USB 2.0 port on RPi 4 ... same results for both (working via wall power / not working via bus power)
- Tried the reported "trick" of plugging in hub and then plugging in stick directly to USB 2.0 on RPi 4 ... nothing

I have a pure old-school USB 2.0 hub (wall and bus power options) arriving from Amazon tomorrow to try which based on reports above should do the trick for just $5 (USB 3.0 hub with power is overkill)
https://www.amazon.com/gp/product/B07F8SVY35/ref=ppx_yo_dt_b_asin_title_o00_s00

roberttuttle, 2019-08-05:
UPDATE: USB 2.0 hub (see above) did not work (neither wall nor bus powered). Ordered a 4-port USB 3.0 hub with adapter power options to try out.
https://www.amazon.com/gp/product/B00TPMEOYM/ref=ppx_yo_dt_b_asin_title_o00_s00

This one worked ... plugged into the USB 3.0 port on the RPi. It is interesting that it works with both wall power ... 0.5A adapter ... and bus power from the RPi). 

So, the "just plug in a USB 2.0 adapter in between the Aeotec stick and the RPi" is not necessarily a 100% workaround.

fribse, 2019-08-05:
Hey Robert.
How about on reboot, does it get auro mounted? 
I would consider this Hub just for that.

roberttuttle, 2019-08-05:
> Hey Robert.
> How about on reboot, does it get auro mounted?
> I would consider this Hub just for that.

Yes ... rebooted dozens of times now and /dev/ttyACM0 gets discovered/activated 100% each time. For me ... YMMV. (Raspbian Buster, RPi 4).

Mastiffen, 2019-08-06:
The Plexgear Desktop 500 works 100 % on the Pi 4 (of course with Buster, since that's the only version of Rasbian that works on the 4) when it's not powered. Not at all when it's powered.

markcarline, 2019-08-06:
This is the one I use, it's cheap and just works:

https://rover.ebay.com/rover/0/0/0?mpre=https%3A%2F%2Fwww.ebay.co.uk%2Fulk%2Fitm%2F362166919258

fribse, 2019-08-07:
I've switched to HassIO now. From the 1 reboot so far it seems that it now keeps the USB stick mounted. I have ordered the USB3 hub anyways. My problem for the other suggestions is that they are not easily available in my area. And even the one that @roberttuttle uses I had to get from Germany.

Mastiffen, 2019-08-07:
@fribse HassIO on Buster? I thought that was still not working right.

pelwell, 2019-08-07:
@fribse are you saying that running different software on your Pi 4 allows the ZWave dongle to work?

fribse, 2019-08-07:
@Mastiffen and @pelwell.
I was running Raspbian Buster with docker and Hassio on it.
I've now changed it to the 64-bit hassio beta image.
It now automounts, it still has to be attached to a USB2 hub as far as the 2(!) reboots say. Next test will be power down etc. to see what happens.

bolausson, 2019-08-07:
> UPDATE: USB 2.0 hub (see above) did not work (neither wall nor bus powered). Ordered a 4-port USB 3.0 hub with adapter power options to try out. [...]

I have a rather old "Speedlink Nobilé active 4-Port USB 3.0-Hub with AC-Adapter"
https://www.amazon.de/Speedlink-aktiver-Netzteil-3-0-Anschl%C3%BCsse-kompatibel/dp/B00Q6PLUHA

And the workaround does not work with this one.

So it really seems to be a gamble which hub might work...

Cheers,
Bjoern

bolausson, 2019-08-07:
> UPDATE: USB 2.0 hub (see above) did not work (neither wall nor bus powered). Ordered a 4-port USB 3.0 hub with adapter power options to try out. [...]

What is the exact product code of the manufacturer of the USB 3.0 hub you are using?
In the description from the Amazon Link you provided it says:

> Sabrent 4-Port USB 3.0 Hub with Individual LED Lit Power Switches, Included 5V/2.5A Power Adapter (HB-UMP3)

If I search the same on Amazon.de I only find:

> Sabrent 4-Port USB 3.0 Hub mit einzelnen Power Schalter und LEDs (HB-UM43)

https://www.amazon.de/dp/B00JX1ZS5O/ref=cm_sw_em_r_mt_dp_U_SIUsDbGBZTS3Y
 
Now the big question is - will it work or not?!?

fribse, 2019-08-09:
@bolausson It's the same, they just bundle it with a PSU in the US apparently.

bolausson, 2019-08-09:
@fribse I ordered the Sabrent USB-Hub as well and it should arrive today... will report if it works or not.

The "Speedlink Nobilé active 4-Port USB 3.0-Hub with AC-Adapter" neither works via USB-2 or USB-3 cable and it doesn't matter if it is connected to the USB-2 or USB-3 port of the Pi. It doesn't make a difference if the USB-Hub is powered via a PSU or not...

Mastiffen, 2019-08-09:
The Plexgear Desktop 500 still works 100 %, but I don't know if that's available everywhere. Here's the link so you can see what it looks like:

https://www.kjell.com/no/produkter/data-og-nettverk/kabler-og-adaptere/usb/usb-huber/plexgear-desktop-500-usb-3-0-hub-med-stromforsyning-4-veis-p69306

[67 more replies]