raspberrypi/linux #3028: Heavy writes to USB3 SSD causing network latency spikes

janost, 2019-06-27:
While debugging my issue with a USB3-SATA controller (#3026) I found that heavy writes to an USB3 SSD are causing some kernel threads to spike the CPU and increase network latency, which is observable because of the noticeable lag in the SSH session and can be confirmed with an ICMP ping.

I'm not exactly sure if this is a bug or normal behaviour for linux, and unfortunately I don't have similar low-power devices to test their behaviour in USB3 write-heavy situations.

The issue can be toned down a bit by adjusting `vm.dirty_ratio` and `vm.dirty_background_ratio`, however it's still significant.

What's weird is it doesn't seem to affect the network when the SSD is formatted to `f2fs`, and is less serious with `xfs` (smaller latency spikes and less frequently).

The attached screenshot shows writing to an `ext4` formatted USB3 SSD. The latency spikes are not caused by an unrelated network issue, there's a gigabit ethernet link between the Pi and the computer.

[image]

pelwell, 2019-06-27:
Can you repeat your testing with `total_mem=3072` in config.txt? This temporarily disables the top 1GB of RAM, which requires special handling from the PCIe driver (and hence affects USB).

janost, 2019-06-27:
@pelwell Disabling the top 1GiB RAM seems to fix the issue, the network performance is unaffected during heavy USB3 writes.

Is this something that can fixed in software without disabling the top 1GiB RAM?

pelwell, 2019-06-27:
It would require the identification of a better place to insert the usage of "bounce buffers", avoiding the need to dally in an atomic context, but a better distribution of interrupt affinities may help to reduce the impact on latencies.

P33M, 2019-07-02:
Can you try migrating the ethernet IRQs to different cores?
```
pi@raspberrypi:~$ cat /proc/interrupts
           CPU0       CPU1       CPU2       CPU3
 16:          0          0          0          0     GICv2  29 Level     arch_timer
 17:      41124      14496      10648      11508     GICv2  30 Level     arch_timer
 22:        999          0          0          0     GICv2 114 Level     DMA IRQ
 30:       1784          0          0          0     GICv2  65 Level     fe00b880.mailbox
 33:       7253          0          0          0     GICv2 153 Level     uart-pl011
 36:          0          0          0          0     GICv2  72 Level     dwc_otg, dwc_otg_pcd, dwc_otg_hcd:usb3
 37:          0          0          0          0     GICv2 169 Level     brcmstb_thermal
 38:     328053          0          0          0     GICv2 158 Level     mmc1, mmc0
 44:          0          0          0          0     GICv2 106 Level     v3d
 46:      17234          0          0          0     GICv2 189 Level     eth0
 47:          0          0          0          0     GICv2 190 Level     eth0
 52:         33          0          0          0     GICv2  66 Level     VCHIQ doorbell
 55:       8711          0          0          0  Brcm_MSI 524288 Edge      xhci_hcd
FIQ:              usb_fiq
IPI0:          0          0          0          0  CPU wakeup interrupts
IPI1:          0          0          0          0  Timer broadcast interrupts
IPI2:       6856      38844      11303      30037  Rescheduling interrupts
IPI3:        470       1345       2658       1177  Function call interrupts
IPI4:          0          0          0          0  CPU stop interrupts
IPI5:      16040        623        684       1514  IRQ work interrupts
IPI6:          0          0          0          0  completion interrupts
Err:          0
```
`# echo 4 > /proc/irq/47/smp_affinity && echo 8 > /proc/irq/46/smp_affinity`

janost, 2019-07-02:
@P33M unfortunately I no longer have the board, so I can't test it any further.

janost, 2020-03-05:
@P33M I got another 4GB board and changing interrupts for `eth0` seems to almost completely mitigate the issue. There's a slight bump in latency, and throughput is somewhat slower during USB writes, however it's much better. Thank you.