raspberrypi/linux #3032: aarch64 bcm2711_defconfig with VideoCore disabled fails to get to early console

hvenev, 2019-06-28:
aarch64 `bcm2711_defconfig` with `CONFIG_BCM_VIDEOCORE` disabled (see https://github.com/raspberrypi/linux/issues/3031) fails to get to early console with the kernel command line `earlycon=pl011,0xfe201000 console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootfstype=f2fs rootwait`. The `kernel7l.img` provided in https://github.com/raspberrypi/firmware with the same kernel command line and firmware boots just fine (and indeed starts printing to the console earlier than when `earlycon` is not set).

hvenev, 2019-06-28:
This happens both on current rpi-4.19.y (e8a66b4f610b3a20bae8f706256d230135916c26) with firmware from https://github.com/raspberrypi/firmware/commit/bcf40b5c2b94178c7564fb451098d44968e44af5 and raspberrypi-kernel_1.20190620-1 (883a1a5bf14abc9b3a014a3907772cd0b3129c8a) with firmware from 1.20190620+1 (https://github.com/raspberrypi/firmware/commit/88ca9081f5e51cdedd16d5dbc85ed12a25123201).

pelwell, 2019-06-28:
2711 v8 support is a work in progress. In particular, the v8 ARM stubs need GIC support.

hvenev, 2019-06-28:
Isn't early console initialized before the GIC?

pelwell, 2019-06-28:
Perhaps, but the L2 cache latencies need to be set before either of them.

hvenev, 2019-06-29:
OK... How do I reproduce the results from cdb78ce891f6c6367a69c0a46b5779a58164bd4b? Is it enough to set `total_mem=1024`?

agherzan, 2019-07-01:
@pelwell I see that https://github.com/raspberrypi/tools/commit/7f4a937e1bacbc111a22552169bc890b4bb26a94#diff-d9d684ceaf48229906f8a534b4cac276 includes gic support for v8 arm stubs. Isn't this included?

pelwell, 2019-07-01:
It's not built into the firmware yet, but if you build it yourself and add `armstub=...` to config.txt it should work.

agherzan, 2019-07-01:
That was what I suspected. Thanks. I"ll try that. Are there any other known issues even with this stub added?

pelwell, 2019-07-01:
Not that I'm aware of, but as you can tell this isn't one of the use cases we tested pre-launch. The two related config.txt settings are `arm_64bit` and `enable_gic`: the former should be set automatically if `kernel8.img` is loaded, and the latter is set if the dtb includes an enabled GIC node.

diegosueiro, 2019-07-02:
@agherzan and @pelwell ,

I can confirm that the 64-bit kernel boots having the following in the config.txt:
```
kernel=kernel8.img                                                              
arm_64bit=1
device_tree=bcm2711-rpi-4-b.dtb
enable_gic=1                                       
armstub=armstub8-gic.bin
```
I used Yocto and meta-raspberrypi (master branches) with some additions to have a raspberrypi4-64 machine and building the kernel and using the firmwares in the head of its branches (rpi-4.19.y and master).

The tools I had to compile separately to produce the `armstub8-gic.bin`.

agherzan, 2019-07-02:
Hi @diegosueiro. You'll have this in meta-raspberrypi master today. Also, I don't think you need the device tree set. Also I'd expect kernel8 to be picked up automatically too.

diegosueiro, 2019-07-02:
@agherzan,

I was just extra cautious with the settings. 😉

agherzan, 2019-07-02:
@diegosueiro have you limit the total memory as well?

diegosueiro, 2019-07-02:
@agherzan, as far as I remember I haven't touch any memory configuration.

agherzan, 2019-07-02:
@diegoherranz Do you happen to have a 1G board? Also, when you tested, did it mount the rootfs and boot the distro?

diegosueiro, 2019-07-02:
@agherzan, yes I'm using the 1GB and I have the full system booted. But I'm using systemd instead of sysvinit.

I can send you my changes later.

agherzan, 2019-07-02:
@pelwell Do you have any idea why kernel will hang at waiting for the rootfs?

```
[    2.246724] Waiting for root device /dev/mmcblk0p2...
[    2.260587] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[    2.267794] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.275006] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.283493] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[    2.290703] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.339258] random: fast init done
[    2.361344] mmc1: new high speed SDIO card at address 0001
```

pelwell, 2019-07-02:
Is this with >1GB? The SDHCI controller only sees the bottom 1GB of RAM. There may be a way of telling Linux about this restriction, but at the time I played with the 64-bit kernel I couldn't make it work without artificially limiting the RAM.

agherzan, 2019-07-02:
@pelwell It is indeed on a 4G. When you say that you only made it work with limiting the ram, did you use `total_mem=1024`? Or have you limited in linux as well?

pelwell, 2019-07-02:
I used `total_mem=1024`, which is propagated to Linux via the Device Tree memory setting.

agherzan, 2019-07-02:
That's what I assumed but sadly I get this with that setting on 1024.

rfinnie, 2019-07-02:
> That's what I assumed but sadly I get this with that setting on 1024.

Confirmed same on my Pi 4 2GB ("mmc1: queuing unknown CIS tuple 0x80"), with bcm2711_defconfig.  total_mem=1024 doesn't seem to have an effect on MMC.  I also tried putting the rootfs on a USB flash drive, which it does find:

```
[    2.092135] Waiting for root device UUID=ab214f01-fe36-4818-8450-ad7c795f0f42...
...
[    3.948868] scsi 0:0:0:0: Direct-Access     Samsung  Flash Drive      1100 PQ: 0 ANSI: 6
[    3.957703] sd 0:0:0:0: [sda] 125313283 512-byte logical blocks: (64.2 GB/59.8 GiB)
[    3.965639] sd 0:0:0:0: [sda] Write Protect is off
[    3.978096] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[    5.398480]  sda: sda1
[    5.402732] sd 0:0:0:0: [sda] Attached SCSI removable disk
```

but doesn't seem to mount as root despite e.g. CONFIG_USB_STORAGE being included.  I haven't debugged beyond that.

sakaki-, 2019-07-03:
Thanks all for the hints in this thread. Have used it to bring the [gentoo-on-rpi3-64bit](https://github.com/sakaki-/gentoo-on-rpi3-64bit) userland up on an RPi4 (brief notes [here](https://www.raspberrypi.org/forums/viewtopic.php?p=1491136#p1491136)).

Screenshot:
[image]

pelwell, 2019-07-03:
Nice work, and thanks for sharing!

If anyone figures out how to persuade the SD interface to work in a 2GB or 4GB system before we get round to it then please share for the good of all Pi 4 users. The solution will be a software configuration change that enables software bounce buffers rather than finding a hardware switch - there is no magic bit that will create the missing address lines.

agherzan, 2019-07-03:
@sakaki- With the above PR merged you should be able to switch from the USB rootfs workaround.

pelwell, 2019-07-03:
...provided you don't try to use more than 1GB.

agherzan, 2019-07-03:
@pelwell he limits it as well.

And one more thing @sakaki- . I also hit the issue with the wifi. It might be the same issue with the NRAM configuration for an external clock input (that was silently slid into raspbian). I'm yet to figure out what was that exactly.

https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-kernel/linux-firmware-rpidistro/linux-firmware-rpidistro/0001-brcmfmac43455-sdio.txt-Follow-raspbian-change-for-bo.patch
https://github.com/RPi-Distro/firmware-nonfree/issues/3

pelwell, 2019-07-03:
> @pelwell he limits it as well.

Yes, I think she does - I was aiming my comment at the wider audience. 8-)

sakaki-, 2019-07-04:
@agherzan,

just tested your changes, and can confirm on both counts: [setting](https://github.com/raspberrypi/linux/commit/395a6a8b68fed103cb1c75d8e224cd4ff2b89bf0) `CONFIG_MMC_SDHCI_IPROC=y` fixes the SD-card issue (`/dev/mmcblk0` now visible and can use e.g. `/dev/mmcblk0p2` as rootfs, directly or via UID), and [reverting](https://github.com/RPi-Distro/firmware-nonfree/pull/4) to `boardflags3=0x44200100` (in `/lib/firmware/brcm/brcmfmac43455-sdio.txt`) makes WiFi available.

Many thanks!

aiminickwong, 2019-07-04:
@sakaia  I have do all of above with my 4G pi used aarch_bcm2711_deconfig, but I also  get error.
 
--------------------------------------------------------------

```
kernel will hang at waiting for the rootfs?
[    2.246724] Waiting for root device /dev/mmcblk0p2...
[    2.260587] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[    2.267794] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.275006] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.283493] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[    2.290703] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.339258] random: fast init done
[    2.361344] mmc1: new high speed SDIO card at address 0001
```
Could you share you latest kernel build full config file?

pelwell, 2019-07-04:
Did you add `total_mem=1024` to config.txt? This is necessary until we figure out the necessary runes to support the limited address range of the SDHCI controller.

aiminickwong, 2019-07-04:
@pelwell  I had done this point, but I found my pi4 USB port was not supplied by power when system boot start，so I think maybe a problem with the compiled kernel option, the same way and the system img on my Pi3B was ok except Pi4, could you share your kernel full config file for me ? Thanks.

aiminickwong, 2019-07-04:
@pelwell  In my kernel config, I had defined the "SDHCI"

```
CONFIG_MMC_SDHCI=y
CONFIG_MMC_SDHCI_IO_ACCESSORS=y
CONFIG_MMC_SDHCI_PCI=y
CONFIG_MMC_SDHCI_ACPI=y
CONFIG_MMC_SDHCI_PLTFM=y
CONFIG_MMC_SDHCI_OF_ARASAN=y
CONFIG_MMC_SDHCI_OF_AT91=y
CONFIG_MMC_SDHCI_OF_DWCMSHC=y
CONFIG_MMC_SDHCI_CADENCE=y
CONFIG_MMC_SDHCI_IPROC=y
```

pelwell, 2019-07-04:
With the current rpi-4.19.y source tree (`git pull --rebase` to update), just `make ARCH=arm bcm2711_defconfig` - I'm not using any custom configuration settings.

sakaki-, 2019-07-04:
I've just uploaded a **mini-tarball of the necessary files to get a 64-bit boot on a Pi4** (modulo all the current caveats and health warnings that attach to that). You can use it to easily adapt an existing 32-bit (such as Raspbian Buster) or 64-bit (such as gentoo-on-rpi3-64bit) image for testing use on a Pi4, under a 64-bit kernel - no compilation required. Lots of stuff will probably not work. There is a 1GiB memory clamp applied. Not production quality - use at your own risk, and just as a convenience (to avoid having to compile the 64-bit kernel, armstub8-gic.bin etc.)

Full instructions [here](https://www.raspberrypi.org/forums/viewtopic.php?p=1492322#p1492322).

aiminickwong, 2019-07-04:
@sakaki @pelwell  Very thanks of you, that's true of my kernel config problem, It's work now. :)

agherzan, 2019-07-04:
I wrong a small guide on all the needed bits to migrate a Raspbian (and not only) to 64b kernel. See this: https://andrei.gherzan.ro/2019/07/04/raspbian-rpi4-64.html . It helps for any other distro integration.

pelwell, 2019-07-04:
Thanks, Andrei - that will be very helpful to the community.

cpdef, 2019-07-04:
yes Sakakis guide is written pretty well, i could migrate my old raspberry pi 3 image to the raspberry 4 using these instructions

lategoodbye, 2019-07-05:
Did anyone tried to set

sdhci_pltfm_data.quirks2 = SDHCI_QUIRK2_BROKEN_64_BIT_DMA

in sdhci-iproc to possibly workaround the 1 GB limitation (didn't test it)?

TheEnbyperor, 2019-07-07:
Thanks for the suggestion but it unfortunately doesn't work

TheEnbyperor, 2019-07-07:
There also seems to be more things that break with >1GiB of ram. These lines show up in the kernel boot log with 4GiB but not with 1GiB.
```
[    0.197837] bcm2835-dma: probe of fe007000.dma failed with error -5
```
```
[    0.200874] raspberrypi-firmware soc:firmware: Request 0x00000001 returned status 0x00000000
[    0.204730] raspberrypi-firmware soc:firmware: Request 0x00000003 returned status 0x00000000
[    0.208814] raspberrypi-firmware soc:firmware: Request 0x00030046 returned status 0x00000000
```
```
[    0.864869] raspberrypi-firmware soc:firmware: Request 0x00030043 returned status 0x00000000
[    0.864907] raspberrypi-exp-gpio soc:firmware:gpio: Failed to get GPIO 0 config (-22 80)
[    0.864975] raspberrypi-firmware soc:firmware: Request 0x00030043 returned status 0x00000000
[    0.865004] raspberrypi-exp-gpio soc:firmware:gpio: Failed to get GPIO 1 config (-22 81)
[    0.865066] raspberrypi-firmware soc:firmware: Request 0x00030043 returned status 0x00000000
[    0.865094] raspberrypi-exp-gpio soc:firmware:gpio: Failed to get GPIO 2 config (-22 82)
[    0.865154] raspberrypi-firmware soc:firmware: Request 0x00030043 returned status 0x00000000
[    0.865182] raspberrypi-exp-gpio soc:firmware:gpio: Failed to get GPIO 3 config (-22 83)
[    0.865240] raspberrypi-firmware soc:firmware: Request 0x00030043 returned status 0x00000000
[    0.865268] raspberrypi-exp-gpio soc:firmware:gpio: Failed to get GPIO 4 config (-22 84)
[    0.865327] raspberrypi-firmware soc:firmware: Request 0x00030043 returned status 0x00000000
[    0.865354] raspberrypi-exp-gpio soc:firmware:gpio: Failed to get GPIO 5 config (-22 85)
[    0.865413] raspberrypi-firmware soc:firmware: Request 0x00030043 returned status 0x00000000
[    0.865441] raspberrypi-exp-gpio soc:firmware:gpio: Failed to get GPIO 6 config (-22 86)
[    0.865500] raspberrypi-firmware soc:firmware: Request 0x00030043 returned status 0x00000000
[    0.865527] raspberrypi-exp-gpio soc:firmware:gpio: Failed to get GPIO 7 config (-22 87)
[    0.866380] [brcmvirt_gpio_probe]: failed to dma_alloc_coherent(4096)
[    0.866418] brcmvirt-gpio: probe of soc:virtgpio failed with error -12
[    0.866808] brcm-pcie fd500000.pcie: could not get clock
```
```
[    0.925306] raspberrypi-firmware soc:firmware: Request 0x00040013 returned status 0x00000000
[    0.925337] bcm2708_fb soc:fb: Unable to determine number of FB's. Assuming 1
[    0.925359] bcm2708_fb soc:fb: FB found 1 display(s)
[    0.925527] bcm2708_fb soc:fb: cannot allocate DMA CBs
[    0.925551] bcm2708_fb soc:fb: probe failed, err -12
[    0.925582] bcm2708_fb: probe of soc:fb failed with error -12
[    0.926796] raspberrypi-firmware soc:firmware: Request 0x00030030 returned status 0x00000000
```
```
[    0.994800] raspberrypi-firmware soc:firmware: Request 0x00028001 returned status 0x00000000
[    0.995361] usbcore: registered new interface driver usb-storage
[    0.995547] mousedev: PS/2 mouse device common for all mice
[    0.996332] bcm2835-wdt bcm2835-wdt: Broadcom BCM2835 watchdog timer
[    0.996507] raspberrypi-firmware soc:firmware: Request 0x00030007 returned status 0x00000000
[    0.996537] bcm2835-cpufreq:bcm2835_cpufreq_get_clock:93: Failed to get clock (-22)
[    0.996602] raspberrypi-firmware soc:firmware: Request 0x00030004 returned status 0x00000000
[    0.996629] bcm2835-cpufreq:bcm2835_cpufreq_get_clock:93: Failed to get clock (-22)
[    0.996652] bcm2835-cpufreq: min=0 max=0
[    0.996728] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[    0.996756] bcm2835-cpufreq:bcm2835_cpufreq_get_clock:93: Failed to get clock (-22)
[    0.996778] cpufreq: cpufreq_online: ->get() failed
[    0.996911] raspberrypi-firmware soc:firmware: Request 0x00030007 returned status 0x00000000
[    0.996938] bcm2835-cpufreq:bcm2835_cpufreq_get_clock:93: Failed to get clock (-22)
[    0.997000] raspberrypi-firmware soc:firmware: Request 0x00030004 returned status 0x00000000
[    0.997026] bcm2835-cpufreq:bcm2835_cpufreq_get_clock:93: Failed to get clock (-22)
[    0.997049] bcm2835-cpufreq: min=0 max=0
[    0.997118] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[    0.997145] bcm2835-cpufreq:bcm2835_cpufreq_get_clock:93: Failed to get clock (-22)
[    0.997167] cpufreq: cpufreq_online: ->get() failed
[    0.997291] raspberrypi-firmware soc:firmware: Request 0x00030007 returned status 0x00000000
[    0.997318] bcm2835-cpufreq:bcm2835_cpufreq_get_clock:93: Failed to get clock (-22)
[    0.997380] raspberrypi-firmware soc:firmware: Request 0x00030004 returned status 0x00000000
[    0.997406] bcm2835-cpufreq:bcm2835_cpufreq_get_clock:93: Failed to get clock (-22)
[    0.997429] bcm2835-cpufreq: min=0 max=0
[    0.997497] raspberrypi-firmware soc:firmware: Request 0x00030002 returned status 0x00000000
[    0.997524] bcm2835-cpufreq:bcm2835_cpufreq_get_clock:93: Failed to get clock (-22)
[    0.997546] cpufreq: cpufreq_online: ->get() failed
[    0.997667] raspberrypi-firmware soc:firmware: Request 0x00030007 returned status 0x00000000
[    0.997694] bcm2835-cpufreq:bcm2835_cpufreq_get_clock:93: Failed to get clock (-22)
[... 17 lines cut ...]
```
[    1.005826] raspberrypi-firmware soc:firmware: Request 0x00030004 returned status 0x00000000
```
```
[    2.583347] raspberrypi-firmware soc:firmware: Request 0x00030043 returned status 0x00000000
[    2.591831] raspberrypi-exp-gpio soc:firmware:gpio: Failed to get GPIO 4 config (-22 84)
[... cut ...]
[    2.683375] raspberrypi-firmware soc:firmware: Request 0x00028001 returned status 0x00000000
[    2.692712] raspberrypi-firmware soc:firmware: Request 0x00028001 returned status 0x00000000
```
```
[    2.712170] mmc1: queuing unknown CIS tuple 0x80 (2 bytes)
[    2.720981] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.728108] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.736523] mmc1: queuing unknown CIS tuple 0x80 (7 bytes)
[    2.743670] mmc1: queuing unknown CIS tuple 0x80 (3 bytes)
[    2.758219] usb usb1-port1: couldn't allocate usb_device
```

j-selby, 2019-07-07:
The failure to send requests via the MMC mailbox is due to those failed coherent memory allocations - this isn't just an issue for MMC operations but for other modules as above (e.g. GPIO). Noting that there is a fairly large coherent pool (1M by default) and that increasing this doesn't fix the issue, there is something else going on.

lategoodbye, 2019-07-07:
I'm getting a similiar behavior reported by @TheEnbyperor while using multi_v7_defconfig on my RPI 4.
I opened a separate issue #3057 .

TheEnbyperor, 2019-07-07:
That seems to be the same root cause, so maybe it's worth concentrating on this issue until/when it appears that the causes differ

agherzan, 2019-07-07:
@lategoodbye Why are you using that defconfig and not the bcm2711 one?

lategoodbye, 2019-07-07:
The kernel must work with different configurations. The minimum for a ARM v7 board like the RPI 4 is to start with multi_v7_defconfig. This configuration is also the base for multiple distributions like Fedora. So the problem isn't the configuration, it's the driver or the devicetree implementation.

agherzan, 2019-07-07:
Sure. That is definitely the case. I was just curious why started from there.

j-selby, 2019-07-08:
Adding `coherent_pool=20M cma=40M@20M` to your kernel command line does seem to get things further along (i.e I got a partition table from the SD card), but still throws errors in a few places. Does seem that the kernel is putting DMA memory too high?

lategoodbye, 2019-07-08:
Maybe there is a problem in the regions `ranges` and `dma-ranges` defined under the soc node in the devicetree.

lategoodbye, 2019-07-08:
@pelwell I hope, i got this right: sdhost controller is limited to 40 bit and emmc2 to 32 bit?

pelwell, 2019-07-08:
SDHOST, like the original Arasan EMMC, is a slave device - it has no bus master capability - so the question is meaningless. The SDHCI-compliant EMMC2 in 2711 has an effective 30-bit range.

Googulator, 2019-07-12:
I was able to get my 4GB Pi to boot using this commit: https://github.com/Googulator/linux/commit/64d06963c3ba7b60c068a70839c2400cff3a39d8

plus the command line arguments "coherent_pool=16M cma=256M@16M swiotlb=force"

Still some DMA errors visible though.

j-selby, 2019-07-12:
@Googulator see the above comment from me - thats enough to get it to boot with DMA errors, so not sure what your commit actually does there.

Strit, 2019-07-12:
It seems to be related to this bit in https://github.com/raspberrypi/linux/drivers/mmc/host/sdhci.c
From line 960 onwards.
```
if (host->flags & SDHCI_REQ_USE_DMA) {
		int sg_cnt = sdhci_pre_dma_transfer(host, data, COOKIE_MAPPED);

		if (sg_cnt <= 0) {
			/*
			 * This only happens when someone fed
			 * us an invalid request.
			 */
			WARN_ON(1);
			host->flags &= ~SDHCI_REQ_USE_DMA;
		} else if (host->flags & SDHCI_USE_ADMA) {
			sdhci_adma_table_pre(host, data, sg_cnt);

			sdhci_writel(host, host->adma_addr, SDHCI_ADMA_ADDRESS);
			if (host->flags & SDHCI_USE_64_BIT_DMA)
				sdhci_writel(host,
					     (u64)host->adma_addr >> 32,
					     SDHCI_ADMA_ADDRESS_HI);
		} else {
			WARN_ON(sg_cnt != 1);
			sdhci_writel(host, sdhci_sdma_address(host),
				     SDHCI_DMA_ADDRESS);
		}
	}
```
I get this error after a while:
```
sdhci-iproc fe340000.emmc2: Cannot do DMA to address 0x00000000edf11000
```
Hope this is helpful to some narrowing it down.
Maybe this >1GB RAM stuff needs it's own issue?

cpdef, 2019-07-12:
btw. the link is dead i guess it should be [https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/mmc/host/sdhci.c](https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/mmc/host/sdhci.c)

pcwalton, 2019-07-13:
Perhaps it's possible to disable SD card DMA by adding a quirk here: https://elixir.bootlin.com/linux/latest/source/drivers/mmc/host/sdhci-pci-core.c#L1607

There are `SDHCI_QUIRK_BROKEN_DMA` and `SDHCI_QUIRK_BROKEN_ADMA` flags: https://elixir.bootlin.com/linux/latest/source/drivers/mmc/host/sdhci.h#L388

This will obviously negatively impact performance, but it's better than losing 75% of the RAM.

pcwalton, 2019-07-13:
Setting `debug_quirks` to `SDHCI_QUIRK_BROKEN_DMA | SDHCI_QUIRK_BROKEN_ADMA` makes the DMA error messages go away, but then there's the message "Timeout waiting for hardware interrupt". Also USB seems to be broken.

I think digging into the SDHCI driver specifically is not the answer. Someone needs to look into the memory map differences between the working 32 bit configuration and the broken 64 bit one from the perspective of the kernel.

j-selby, 2019-07-13:
@pcwalton I think you will find that is all mostly placebo - all you need to get something going is setting the CMA range to a low address, as everything else just hides errors or pretends that everything is working. In my testing USB also breaks (though it works with an aarch64 userspace with the memory limiter enabled).

github-usr-name, 2019-07-14:
@j-selby Based on lines 3558-3572 of sdhci.c, I believe that @pcwalton is correct; the quirk flags disable DMA and revert to PIO:

```
   3558     if ((host->quirks & SDHCI_QUIRK_BROKEN_DMA) &&
   3559         (host->flags & SDHCI_USE_SDMA)) {
   3560         DBG("Disabling DMA as it is marked broken\n");
   3561         host->flags &= ~SDHCI_USE_SDMA;
   3562     }
   3563 
   3564     if ((host->version >= SDHCI_SPEC_200) &&
   3565         (host->caps & SDHCI_CAN_DO_ADMA2))
   3566         host->flags |= SDHCI_USE_ADMA;
   3567 
   3568     if ((host->quirks & SDHCI_QUIRK_BROKEN_ADMA) &&
   3569         (host->flags & SDHCI_USE_ADMA)) {
   3570         DBG("Disabling ADMA as it is marked broken\n");
   3571         host->flags &= ~SDHCI_USE_ADMA;
   3572     }
```

[5 more replies]

closed 2020-04-22 by hvenev