raspberrypi/linux #3033: /dev/pts/x  x=0,1,2,3,4...  can't send / received data in Pi 4B

tonylab2001, 2019-06-29:
/dev/pts/x   x=0,1,2,3,4... can't send / received data in Pi 4B 

use 4.19.50-v71 + #895    Jun 20 2019 

same image for Pi3B/3B+ no problem,  but Raspberry pi 4B can't send & received data

 try  sudo chmod 777 /dev/pts/x     still the same . 

try cat  /dev/pts/x    can't received data,  try cutecom no data as well.

tonylab2001, 2019-07-02:
attach photo  for example

tonylab2001, 2019-07-02:
example

tonylab2001, 2019-07-02:
[image]

pelwell, 2019-07-04:
/dev/pts isn't something I've come across before. Please give a clear example of how you are configuring and using it, showing the differences between success and failure.

tonylab2001, 2019-07-04:
> 
> 
> /dev/pts isn't something I've come across before. Please give a clear example of how you are configuring and using it, showing the differences between success and failure.

[image]
[image]

1,  load picom1  2. load picom2  3 . check all port status  and check /dec/pts/x   in picom 1,2 
4. load cutecom open in /dev/pts/x    5.  load cutecom open in /dev/pts/x  
same image run PI3B+ no problem, in Pi4B no data send and received .

pelwell, 2019-07-04:
The screenshot include mention of an I2C UART - is there some additional hardware required?

tonylab2001, 2019-07-04:
> 
> 
> The screenshot includes mention of an I2C UART - is there some additional hardware required?

We use this i2c UART board
http://www.pridopia.co.uk/rs-pi-i2c-rs232-rs485.html

pelwell, 2019-07-04:
What I2C->UART devices are fitted to that board? The images are too small to tell.

tonylab2001, 2019-07-04:
sc16is752  -->  2 Port for RS232, 2 Port for RS485

tonylab2001, 2019-07-04:
> 
> 
> What I2C->UART devices are fitted to that board? The images are too small to tell.

sc16is752 --> 2 Port for RS232, 2 Port for RS485