raspberrypi/linux #3035: RPi4: Patch to get KVM working

jan-kiszka, 2019-06-30:
So far only informative:
As the RPi4 is moving closer to common ARM SoC designs, proper virtualization is finally in reach. I got KVM working, though we currently need a patch for the 32-bit variant, see https://marc.info/?l=kvm&m=156190795627075&w=2. Once that has been reviewed and approved upstream, this kernel can pick it up as well. The patch was tested successfully with the 4.19-rpi branch.

popcornmix, 2019-07-01:
Let us know when it merges.
If upstream patches have been tested and seen to be useful we're usually happy to cherry-pick them.

macromorgan, 2019-10-21:
Is there a way to enable KVM for the default config for 64bit while we wait for the 32bit to get merged, if it does?

jan-kiszka, 2019-10-21:
32-bit KVM is considered dead on ARM by the maintainer, and there are also complications with the patch I sent. So 64-bit is the better way forward.

macromorgan, 2020-01-10:
Just want to raise the issue again, especially since the 64 bit kernel seems to be pretty stable for me on Raspbian.  Is there a way we can get KVM enabled in the default configs?  Sadly it'll have to be a point of differentiation from the 32 bit builds from what I see above, but it would help me and probably others who like to dabble with KVM on their Raspberry Pis by not having to build their own kernels.  Thank you.

macromorgan, 2020-01-21:
Sorry about the dupe, wasn't sure if I was doing the feature request correctly (since this one is tagged for a patch that looks like is no longer maintained and won't/shouldn't be added to the 32 bit kernel).

pelwell, 2020-01-21:
I take your point - it's not clear cut either way.

You could further the cause by telling us (on the other thread) a few reasons why more than a handful of Pi users might want KVM.

pelwell, 2020-01-21:
Sorry, post here - this _is_ the other thread.

macromorgan, 2020-01-21:
KVM would allow users to virtualize operating systems more quickly on the Raspberry Pi (4).  While this technically would work on the Pi 2 2nd revision, Pi 3, and Pi 4 all with the 64 bit kernel, it is on the Pi 4 where memory constraints are such that it is feasible.

When running Debian/Raspbian Linux in QEMU on a Pi 4, even with the same architecture (armhf or aarch64), it's far too slow to be usable unless KVM is enabled.  With KVM enabled I can easily run Raspbian Desktop within a VM on my Raspberry Pi 4.

The assumption (on my part) is that by enabling this you will enable others to experiment with KVM more easily, and that by doing this only for the 64 bit kernel (which is the only kernel that supports it anyway as far as I know) it will not have widespread impact on the greater community in the event of any issues.

jan-kiszka, 2020-01-22:
I think we are only a few kernel releases away from proper upstream support for the RPi 4. Then major distro kernels will start to work as well, and distros do not ask why you want KVM. They carry this as default feature for full system virtualization, container isolation, etc.

pelwell, 2020-02-06:
https://github.com/raspberrypi/linux/pull/3456 is a first attempt at a KVM configuration for a 64-bit Pi 4 kernel.

pelwell, 2020-02-27:
Did anybody get round to trying my patch (#3456) to enable KVM in the Pi 4 64-bit kernel?

pelwell, 2020-03-03:
The patch is now merged to the current and upcoming kernel branches, but remember that this only affects 64-bit Pi 4 kernel builds.