raspberrypi/linux #3036: kernel trace net/wireless/sme.c:945 cfg80211_roamed+0x240/0x260

AloisKlingler, 2019-07-01:
I guess this happens caused by wifi roaming. I do not use ad-hoc mode, but have different APs with the same SSID running, at least two of them are reachable by this raspberry pi w zero.

wpa_supplicant.conf:
`ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=AT

network={
        ssid="mySSID"
        scan_ssid=1
        psk="myPSK"
        key_mgmt=WPA-PSK
        pairwise=CCMP
        group=CCMP
}
`

the traces look always the same:
`[  318.397409] ------------[ cut here ]------------
[  318.398656] WARNING: CPU: 0 PID: 45 at net/wireless/sme.c:945 cfg80211_roamed+0x240/0x260 [cfg80211]
[  318.398673] Modules linked in: bnep hci_uart btbcm serdev bluetooth ecdh_generic vc4 drm_kms_helper drm drm_panel_orientation_quirks brcmfmac snd_soc_core brcmutil snd_compress snd_pcm_dmaengine snd_pcm sha256_generic snd_timer snd raspberrypi_hwmon syscopyarea hwmon sysfillrect sysimgblt fb_sys_fops cfg80211 rfkill i2c_bcm2835 bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem v4l2_common bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6
[  318.398923] CPU: 0 PID: 45 Comm: kworker/0:2 Tainted: G         C        4.19.42+ #1219
[  318.398930] Hardware name: BCM2835
[  318.399358] Workqueue: events brcmf_fweh_event_worker [brcmfmac]
[  318.399424] [<c0017e54>] (unwind_backtrace) from [<c0014d9c>] (show_stack+0x20/0x24)
[  318.399468] [<c0014d9c>] (show_stack) from [<c06afcf0>] (dump_stack+0x20/0x28)
[  318.399507] [<c06afcf0>] (dump_stack) from [<c0024388>] (__warn+0xf4/0x11c)
[  318.399534] [<c0024388>] (__warn) from [<c00244e0>] (warn_slowpath_null+0x4c/0x58)
[  318.400751] [<c00244e0>] (warn_slowpath_null) from [<bf228584>] (cfg80211_roamed+0x240/0x260 [cfg80211])
[  318.402399] [<bf228584>] (cfg80211_roamed [cfg80211]) from [<bf496f10>] (brcmf_notify_roaming_status+0x19c/0x258 [brcmfmac])
[  318.402959] [<bf496f10>] (brcmf_notify_roaming_status [brcmfmac]) from [<bf4a0b30>] (brcmf_fweh_call_event_handler+0x40/0x7c [brcmfmac])
[  318.403738] [<bf4a0b30>] (brcmf_fweh_call_event_handler [brcmfmac]) from [<bf4a0df4>] (brcmf_fweh_event_worker+0x288/0x52c [brcmfmac])
[  318.404051] [<bf4a0df4>] (brcmf_fweh_event_worker [brcmfmac]) from [<c003d108>] (process_one_work+0x130/0x3a0)
[  318.404087] [<c003d108>] (process_one_work) from [<c003d3ac>] (worker_thread+0x34/0x530)
[  318.404113] [<c003d3ac>] (worker_thread) from [<c0042eb0>] (kthread+0x11c/0x158)
[  318.404137] [<c0042eb0>] (kthread) from [<c00090ac>] (ret_from_fork+0x14/0x28)
[  318.404149] Exception stack(0xcacc7fb0 to 0xcacc7ff8)
[  318.404163] 7fa0:                                     00000000 00000000 00000000 00000000
[  318.404178] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  318.404191] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  318.404204] ---[ end trace a7f312ce1f360eb6 ]---
`

Best regards
Alois

lategoodbye, 2019-07-01:
Does this warning occur imediately after connecting to wifi?
Except of this warning, are there any consequences to the wifi connection?
Could you describe the access points (hardware)?

AloisKlingler, 2019-07-01:
hello,

no, it does not immediately after connecting to wifi. I could not see any time dependency yet.
The wifi connection to the device is sometimes very laggy. I could not narrow down yet if the trace is anyhow correlated to the connection issue. it recovers the connection to a stable state without rebooting. Neither the RPI, nor any of the access points change position.

I have two Zyxel NWA1123-ACv2 (2.4Ghz and 5GHz):
`          Cell 03 - Address: 60:31:97:F9:00:F4
                    Channel:9
                    Frequency:2.452 GHz (Channel 9)
                    Quality=50/70  Signal level=-60 dBm  
                    Encryption key:on
                    ESSID:""
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 6 Mb/s
                              9 Mb/s; 12 Mb/s; 18 Mb/s
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 50ms ago
                    IE: Unknown: 0000
                    IE: Unknown: 010882848B960C121824
                    IE: Unknown: 030109
                    IE: Unknown: 050400010000
                    IE: Unknown: 0706434820010D14
                    IE: Unknown: 2A0102
                    IE: Unknown: 32043048606C
                    IE: Unknown: 2D1AAD0103FFFF0000000000000000000001000000000406E6A70C00
                    IE: Unknown: 3D1609000500000000000000000000000000000000000000
                    IE: Unknown: 4A0E14000A002C01C800140005001900
                    IE: Unknown: 7F080100000000000040
                    IE: Unknown: DD180050F2020101000003A4000027A4000042435E0062322F00
                    IE: Unknown: DD0900037F01010000FF7F
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: WPA Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
          Cell 04 - Address: 62:F4:97:F9:00:F5
                    Channel:9
                    Frequency:2.452 GHz (Channel 9)
                    Quality=50/70  Signal level=-60 dBm  
                    Encryption key:on
                    ESSID:""
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 6 Mb/s
                              9 Mb/s; 12 Mb/s; 18 Mb/s
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 50ms ago
                    IE: Unknown: 0000
                    IE: Unknown: 010882848B960C121824
                    IE: Unknown: 030109
                    IE: Unknown: 050400010000
                    IE: Unknown: 0706434820010D14
                    IE: Unknown: 2A0102
                    IE: Unknown: 32043048606C
                    IE: Unknown: 2D1AAD0103FFFF0000000000000000000001000000000406E6A70C00
                    IE: Unknown: 3D1609000500000000000000000000000000000000000000
                    IE: Unknown: 4A0E14000A002C01C800140005001900
                    IE: Unknown: 7F080100000000000040
                    IE: Unknown: DD180050F2020101000003A4000027A4000042435E0062322F00
                    IE: Unknown: DD0900037F01010000FF7F
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: WPA Version 1
                        Group Cipher : CCMP
[... 78 lines cut ...]
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 6 Mb/s
                              9 Mb/s; 12 Mb/s; 18 Mb/s
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 40ms ago
                    IE: Unknown: 000C000000000000000000000000
                    IE: Unknown: 010882848B960C121824
                    IE: Unknown: 030102
                    IE: Unknown: 050401020000
                    IE: Unknown: 2A0104
                    IE: Unknown: 32043048606C
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : TKIP
                        Pairwise Ciphers (2) : CCMP TKIP
                        Authentication Suites (1) : PSK
                    IE: WPA Version 1
                        Group Cipher : TKIP
                        Pairwise Ciphers (2) : CCMP TKIP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 2D1A4E101BFFFF000000000000000000000100000000000000000000
                    IE: Unknown: 3D1602001700000000000000000000000000000000000000
                    IE: Unknown: 7F080000000200000040
                    IE: Unknown: DD180050F2020101000003A4000027A4000042435E0062322F00
`

lategoodbye, 2019-07-01:
Could you please rearrange all your outputs, because they are hard to read.
What connection issue? You provided a kernel trace, but never mentioned a explicit connection issue. Does the connection issue occur before the trace or at the same time?

AloisKlingler, 2019-07-01:
I am sorry I saw the misarranged code blocks but tought it's a browser issue on my side. :-)
I edited them so they look readable now.

As I am not sure that my connectivity issue is related to the traces I did not mention it on starting the issue.

The wifi connection to the device is sometimes very laggy. I can see with tcpdump a lot of retransmissons, duplicate packets, but I could not find a causing part yet.
The result is, e.g., I cannot connect to ssh during this period. If I get a shell, the MOTD does not come through, the connection does not response any more. "dmesg" output looks like I am connected over a serial cable and stalls erratic.
pinging the box works (with normal wifi-latency), with small and with large packets, the whole time flawless.
The connection state recovers stable connection without rebooting, during the unstable connection there are not occuring traces during the issue. That's why I am not sure if the laggy connection is related to the traces.

The device is headless and the connectivity issue happens irregularly, sometimes it needs days to occur, sometimes it is occuring several times a day. I have not found a workaround yet. Rebooting does not help if the connection is laggy (so after the reboot it is still laggy - also without traces).

Thanks. :-)

Alois

AloisKlingler, 2019-07-22:
nobody? :-(

pallas, 2019-09-13:
I've seen the same backtrace, & long after the device has come up. It did not break network connectivity.
```
[162216.813332] ------------[ cut here ]------------
[162216.813505] WARNING: CPU: 3 PID: 7156 at net/wireless/sme.c:945 cfg80211_roamed+0x224/0x238 [cfg80211]
[162216.813511] Modules linked in: bnep hci_uart btbcm serdev bluetooth ecdh_generic joydev brcmfmac brcmutil hid_microsoft sha256_generic cfg80211 bcm2835_v4l2(C) bcm2835_codec(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig v4l2_common rfkill videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common raspberrypi_hwmon snd_bcm2835(C) hwmon snd_pcm videodev snd_timer media snd vc_sm_cma(C) evdev fixed uio_pdrv_genirq uio ip_tables x_tables ipv6
[162216.813636] CPU: 3 PID: 7156 Comm: kworker/3:2 Tainted: G         C        4.19.66-v7+ #1253
[162216.813638] Hardware name: BCM2835
[162216.813742] Workqueue: events brcmf_fweh_event_worker [brcmfmac]
[162216.813767] [<80111f38>] (unwind_backtrace) from [<8010d4b0>] (show_stack+0x20/0x24)
[162216.813778] [<8010d4b0>] (show_stack) from [<808191e0>] (dump_stack+0xd4/0x118)
[162216.813791] [<808191e0>] (dump_stack) from [<801209c8>] (__warn+0x104/0x11c)
[162216.813799] [<801209c8>] (__warn) from [<80120b18>] (warn_slowpath_null+0x50/0x58)
[162216.813927] [<80120b18>] (warn_slowpath_null) from [<7f38edec>] (cfg80211_roamed+0x224/0x238 [cfg80211])
[162216.814127] [<7f38edec>] (cfg80211_roamed [cfg80211]) from [<7f558cb8>] (brcmf_notify_roaming_status+0x1b0/0x240 [brcmfmac])
[162216.814289] [<7f558cb8>] (brcmf_notify_roaming_status [brcmfmac]) from [<7f5629a4>] (brcmf_fweh_call_event_handler+0x48/0x88 [brcmfmac])
[162216.814447] [<7f5629a4>] (brcmf_fweh_call_event_handler [brcmfmac]) from [<7f562c08>] (brcmf_fweh_event_worker+0x224/0x49c [brcmfmac])
[162216.814534] [<7f562c08>] (brcmf_fweh_event_worker [brcmfmac]) from [<8013bf0c>] (process_one_work+0x170/0x458)
[162216.814544] [<8013bf0c>] (process_one_work) from [<8013c250>] (worker_thread+0x5c/0x5a4)
[162216.814553] [<8013c250>] (worker_thread) from [<8014253c>] (kthread+0x138/0x168)
[162216.814562] [<8014253c>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
[162216.814566] Exception stack(0xb63f1fb0 to 0xb63f1ff8)
[162216.814571] 1fa0:                                     00000000 00000000 00000000 00000000
[162216.814577] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[162216.814582] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[162216.814669] ---[ end trace afc3e43e83209e2a ]---
```

pallas, 2019-09-13:
It's specifically in this code:
```
	if (!info->bss) {
		info->bss = cfg80211_get_bss(wdev->wiphy, info->channel,
					     info->bssid, wdev->ssid,
					     wdev->ssid_len,
					     wdev->conn_bss_type,
					     IEEE80211_PRIVACY_ANY);
	}

	if (WARN_ON(!info->bss))
		return;
```

jmoverley, 2019-10-02:
Seeing same issues here with RPi zero-w

[115122.460692] brcmfmac: brcmf_update_bss_info: wl dtim_assoc failed (-52)
[115122.461148] ------------[ cut here ]------------
[115122.462374] WARNING: CPU: 0 PID: 22850 at net/wireless/sme.c:752 __cfg80211_connect_result+0x388/0x400 [cfg80211]
[115122.462389] Modules linked in: bnep hci_uart btbcm serdev bluetooth ecdh_generic 8021q garp stp llc brcmfmac brcmutil sha256_generic cfg80211 rfkill raspberrypi_hwmon hwmon snd_bcm2835(C) snd_pcm snd_timer bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem snd v4l2_common bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops i2c_bcm2835 videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6
[115122.462604] CPU: 0 PID: 22850 Comm: kworker/u2:2 Tainted: G         C        4.19.66+ #1253
[115122.462612] Hardware name: BCM2835
[115122.463674] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[115122.463928] [<c0017edc>] (unwind_backtrace) from [<c0014e24>] (show_stack+0x20/0x24)
[115122.463960] [<c0014e24>] (show_stack) from [<c06b9750>] (dump_stack+0x20/0x28)
[115122.463996] [<c06b9750>] (dump_stack) from [<c0024450>] (__warn+0xf4/0x11c)
[115122.464020] [<c0024450>] (__warn) from [<c00245a8>] (warn_slowpath_null+0x4c/0x58)
[115122.465157] [<c00245a8>] (warn_slowpath_null) from [<bf2860cc>] (__cfg80211_connect_result+0x388/0x400 [cfg80211])
[115122.467225] [<bf2860cc>] (__cfg80211_connect_result [cfg80211]) from [<bf259f88>] (cfg80211_process_wdev_events+0x13c/0x1c8 [cfg80211])
[115122.469226] [<bf259f88>] (cfg80211_process_wdev_events [cfg80211]) from [<bf25a054>] (cfg80211_process_rdev_events+0x40/0x9c [cfg80211])
[115122.471215] [<bf25a054>] (cfg80211_process_rdev_events [cfg80211]) from [<bf254258>] (cfg80211_event_work+0x24/0x2c [cfg80211])
[115122.472265] [<bf254258>] (cfg80211_event_work [cfg80211]) from [<c003d248>] (process_one_work+0x130/0x3a0)
[115122.472301] [<c003d248>] (process_one_work) from [<c003d4ec>] (worker_thread+0x34/0x530)
[115122.472326] [<c003d4ec>] (worker_thread) from [<c0042ff0>] (kthread+0x11c/0x158)
[115122.472350] [<c0042ff0>] (kthread) from [<c00090ac>] (ret_from_fork+0x14/0x28)
[115122.472363] Exception stack(0xd1f3bfb0 to 0xd1f3bff8)
[115122.472376] bfa0:                                     00000000 00000000 00000000 00000000
[115122.472391] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[115122.472404] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[115122.472417] ---[ end trace 154d98e6989517a8 ]---

is there a why to limit BSSIDs allowed to connect to?

andrey-utkin, 2019-11-01:
Also having this, with rpi4 running Sakaki's genpi64 https://github.com/sakaki-/gentoo-on-rpi-64bit
```
[Fri Nov  1 17:08:24 2019] ------------[ cut here ]------------
[Fri Nov  1 17:08:24 2019] WARNING: CPU: 0 PID: 30449 at net/wireless/sme.c:945 cfg80211_roamed+0x1f8/0x6b0 [cfg80211]
[Fri Nov  1 17:08:24 2019] Modules linked in: rfcomm lz4 lz4_compress aes_neon_bs aes_neon_blk crypto_simd cryptd aes_arm64 algif_skcipher af_alg hci_uart btbcm serdev 8021q
garp stp llc ipv6 bnep bluetooth ecdh_generic brcmfmac brcmutil vc4 sha256_generic drm_kms_helper cfg80211 cec snd_soc_core bcm2835_codec(C) bcm2835_v4l2(C) rfkill v4l2_mem2m
em snd_compress v3d snd_pcm_dmaengine bcm2835_mmal_vchiq(C) v4l2_common syscopyarea videobuf2_dma_contig videobuf2_vmalloc sysfillrect snd_bcm2835(C) gpu_sched sysimgblt vide
obuf2_memops fb_sys_fops drm videobuf2_v4l2 snd_pcm spidev videobuf2_common drm_panel_orientation_quirks snd_timer videodev snd raspberrypi_hwmon media hwmon rpivid_mem vc_sm
_cma(C) spi_bcm2835 uio_pdrv_genirq uio fixed
[Fri Nov  1 17:08:24 2019] CPU: 0 PID: 30449 Comm: kworker/0:2 Tainted: G        WC        4.19.80-v8-3492a1b00349-p4-bis+ #2
[Fri Nov  1 17:08:24 2019] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[Fri Nov  1 17:08:24 2019] Workqueue: events brcmf_fil_bsscfg_int_get [brcmfmac]
[Fri Nov  1 17:08:24 2019] pstate: 40000005 (nZcv daif -PAN -UAO)
[Fri Nov  1 17:08:24 2019] pc : cfg80211_roamed+0x1f8/0x6b0 [cfg80211]
[Fri Nov  1 17:08:24 2019] lr : cfg80211_roamed+0x1f8/0x6b0 [cfg80211]
[Fri Nov  1 17:08:24 2019] sp : ffffff8012abbb90
[Fri Nov  1 17:08:24 2019] x29: ffffff8012abbb90 x28: ffffffc0f25942e0
[Fri Nov  1 17:08:24 2019] x27: ffffff80010eb620 x26: ffffffc0f1211380
[Fri Nov  1 17:08:24 2019] x25: 0000000000000000 x24: ffffffc0f1473000
[Fri Nov  1 17:08:24 2019] x23: ffffffc0f25942e0 x22: 00000000006000c0
[Fri Nov  1 17:08:24 2019] x21: ffffffc0f1474008 x20: ffffffc0f1473000
[Fri Nov  1 17:08:24 2019] x19: ffffff8012abbc40 x18: 0000000000000000
[Fri Nov  1 17:08:24 2019] x17: 0000000000000000 x16: 0000000000000000
[Fri Nov  1 17:08:24 2019] x15: 000000006e68ef58 x14: 0000000000000000
[Fri Nov  1 17:08:24 2019] x13: 0000000000000000 x12: ffffff8008f25000
[Fri Nov  1 17:08:24 2019] x11: ffffff8008e1f000 x10: ffffff8008f250e0
[Fri Nov  1 17:08:24 2019] x9 : 0000000000000000 x8 : 0000000000000001
[Fri Nov  1 17:08:24 2019] x7 : ffffff8008f25000 x6 : ffffffc0fbf640a0
[Fri Nov  1 17:08:24 2019] x5 : ffffffc0fbf640a0 x4 : 00000040f3180000
[Fri Nov  1 17:08:24 2019] x3 : 0000000000000002 x2 : 1ff7cf7fcd6c2700
[Fri Nov  1 17:08:24 2019] x1 : 0000000000000000 x0 : 0000000000000024
[Fri Nov  1 17:08:24 2019] Call trace:
[Fri Nov  1 17:08:24 2019]  cfg80211_roamed+0x1f8/0x6b0 [cfg80211]
[Fri Nov  1 17:08:24 2019]  0xffffff80010c6498
[Fri Nov  1 17:08:24 2019]  0xffffff80010c671c
[Fri Nov  1 17:08:24 2019]  brcmf_fil_bsscfg_int_get+0xdc/0x5d0 [brcmfmac]
[Fri Nov  1 17:08:24 2019]  brcmf_fil_bsscfg_int_get+0x314/0x5d0 [brcmfmac]
[Fri Nov  1 17:08:24 2019]  process_one_work+0x1ec/0x460
[Fri Nov  1 17:08:24 2019]  worker_thread+0x48/0x430
[Fri Nov  1 17:08:24 2019]  kthread+0x12c/0x130
[Fri Nov  1 17:08:24 2019]  ret_from_fork+0x10/0x1c
[Fri Nov  1 17:08:24 2019] ---[ end trace f04a3f647b7b1ce5 ]---
```

bevinhex, 2020-04-04:
Same here, regular PC

Broezel, 2024-02-13:
Hi there,

the "roaming" oops seems to be still present. I get the oops regularly on Zero W and also Zero 2W
with different kernels up to 6.6.16. The trace looks like the same started from Alois.
In most cases the device keeps reachable, but sometimes ...
I also noticed, that after an oops the value-line "wlan0:" in /proc/net/wireless disappears.
We are using a D-Link wifi network (DWC 2000 + 20x DWL2600AP).

```
Jan 05 14:27:18 rpz2 kernel: ------------[ cut here ]------------
Jan 05 14:27:18 rpz2 kernel: WARNING: CPU: 0 PID: 11794 at net/wireless/sme.c:1181 cfg80211_roamed+0x3a8/0x408 [cfg80211]
Jan 05 14:27:18 rpz2 kernel: Modules linked in: nf_tables nfnetlink dm_mod tcp_diag inet_diag spidev brcmfmac brcmutil sha256_generic libsha256 cfg80211 8>
Jan 05 14:27:18 rpz2 kernel: CPU: 0 PID: 11794 Comm: kworker/0:4 Tainted: G        W          6.1.66+ #1707
Jan 05 14:27:18 rpz2 kernel: Hardware name: BCM2835
Jan 05 14:27:18 rpz2 kernel: Workqueue: events brcmf_fweh_event_worker [brcmfmac]
Jan 05 14:27:18 rpz2 kernel:  unwind_backtrace from show_stack+0x18/0x1c
Jan 05 14:27:18 rpz2 kernel:  show_stack from dump_stack_lvl+0x34/0x58
Jan 05 14:27:18 rpz2 kernel:  dump_stack_lvl from __warn+0x7c/0xf8
Jan 05 14:27:18 rpz2 kernel:  __warn from warn_slowpath_fmt+0x60/0xcc
Jan 05 14:27:18 rpz2 kernel:  warn_slowpath_fmt from cfg80211_roamed+0x3a8/0x408 [cfg80211]
Jan 05 14:27:18 rpz2 kernel:  cfg80211_roamed [cfg80211] from brcmf_bss_roaming_done+0x160/0x234 [brcmfmac]
Jan 05 14:27:18 rpz2 kernel:  brcmf_bss_roaming_done [brcmfmac] from brcmf_notify_roaming_status+0x40/0x64 [brcmfmac]
Jan 05 14:27:18 rpz2 kernel:  brcmf_notify_roaming_status [brcmfmac] from brcmf_fweh_event_worker+0x280/0x570 [brcmfmac]
Jan 05 14:27:18 rpz2 kernel:  brcmf_fweh_event_worker [brcmfmac] from process_one_work+0x204/0x488
Jan 05 14:27:18 rpz2 kernel:  process_one_work from worker_thread+0x34/0x53c
Jan 05 14:27:18 rpz2 kernel:  worker_thread from kthread+0xc0/0xe4
Jan 05 14:27:18 rpz2 kernel:  kthread from ret_from_fork+0x14/0x30
Jan 05 14:27:18 rpz2 kernel: Exception stack(0xe0c21fb0 to 0xe0c21ff8)
Jan 05 14:27:18 rpz2 kernel: 1fa0:                                     00000000 00000000 00000000 00000000
Jan 05 14:27:18 rpz2 kernel: 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Jan 05 14:27:18 rpz2 kernel: 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Jan 05 14:27:18 rpz2 kernel: ---[ end trace 0000000000000000 ]---

```

lategoodbye, 2024-02-13:
@Broezel Can you please provide more information about the roaming setup:

- What type of Wifi roaming does the D-Link wifi network use (ESSID like Alois)?
- How long does it take after boot to trigger this warning?
- Is the Raspberry Pi in the range of two roaming APs?
- Do you use wpa_supplicant and is there a specific setting?

Broezel, 2024-02-14:
> @Broezel Can you please provide more information about the roaming setup:
> 
>     * What type of Wifi roaming does the D-Link wifi network use (ESSID like Alois)?
> 
```
          Cell 01 - Address: 6C:19:8F:7D:9F:D5
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=36/70  Signal level=-74 dBm  
                    Encryption key:on
                    ESSID:"*********"
                    Bit Rates:5.5 Mb/s; 11 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Bit Rates:6 Mb/s; 9 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 85246890ms ago
                    IE: Unknown: 0009766F736C616D697363
                    IE: Unknown: 01088B9618243048606C
                    IE: Unknown: 030101
                    IE: Unknown: 0706444520010D14
                    IE: Unknown: 2A0100
                    IE: Unknown: 2F0100
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : TKIP
                        Pairwise Ciphers (2) : CCMP TKIP
                        Authentication Suites (1) : PSK
                    IE: Unknown: 32020C12
                    IE: Unknown: 2D1A7C181FFFFF000000000000000000000000000000000000000000
                    IE: Unknown: 3D1601080000000000000000000000000000000000000000
                    IE: Unknown: 4A0E14000A002C01C800140005001900
                    IE: Unknown: 7F0101
                    IE: Unknown: DD090010180200000C0000
                    IE: WPA Version 1
                        Group Cipher : TKIP
                        Pairwise Ciphers (2) : CCMP TKIP
                        Authentication Suites (1) : PSK
                    IE: Unknown: DD180050F2020101800003A4000027A4000042435E0062322F00
                    IE: Unknown: DD0A0002BC016C198F7D9FC0

```
>
>     * How long does it take after boot to trigger this warning?
> 
This varies between a few hours and several days. Yesterday evening i've just added "brcmfmac.debug=0x100000"
to the command line and made a reboot. Currently no messages so far.
>
>     * Is the Raspberry Pi in the range of two roaming APs?
> 
Yes. 
>     * Do you use wpa_supplicant and is there a specific setting?
>
Here are our settings:

```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=DE
network={
    key_mgmt=WPA-PSK
    ssid="********"
    psk=************************************
    proto=RSN
    pairwise=TKIP
    group=TKIP
    scan_ssid=1
}

```

Broezel, 2024-02-14:
... with "brcmfmac.debug=0x100000":
```
[85929.249372] ------------[ cut here ]------------
[85929.249418] WARNING: CPU: 0 PID: 13250 at net/wireless/sme.c:1181 cfg80211_roamed+0x3a8/0x408 [cfg80211]
[85929.252915] Modules linked in: nf_tables nfnetlink dm_mod tcp_diag inet_diag spidev brcmfmac brcmutil sha256_generic libsha256 zram zsmalloc 8021q garp stp llc cfg80211 rfkill raspberrypi_hwmon i2c_bcm2835 spi_bcm2835 fixed uio_pdrv_genirq uio fuse ip_tables x_tables
[85929.253233] CPU: 0 PID: 13250 Comm: kworker/0:1 Not tainted 6.1.21+ #1642
[85929.253269] Hardware name: BCM2835
[85929.253287] Workqueue: events brcmf_fweh_event_worker [brcmfmac]
[85929.254018]  unwind_backtrace from show_stack+0x18/0x1c
[85929.254076]  show_stack from dump_stack_lvl+0x34/0x58
[85929.254120]  dump_stack_lvl from __warn+0x7c/0xf8
[85929.254168]  __warn from warn_slowpath_fmt+0x60/0xcc
[85929.254219]  warn_slowpath_fmt from cfg80211_roamed+0x3a8/0x408 [cfg80211]
[85929.257671]  cfg80211_roamed [cfg80211] from brcmf_bss_roaming_done+0x15c/0x230 [brcmfmac]
[85929.261971]  brcmf_bss_roaming_done [brcmfmac] from brcmf_notify_roaming_status+0x40/0x64 [brcmfmac]
[85929.263024]  brcmf_notify_roaming_status [brcmfmac] from brcmf_fweh_event_worker+0x280/0x570 [brcmfmac]
[85929.264067]  brcmf_fweh_event_worker [brcmfmac] from process_one_work+0x204/0x48c
[85929.264656]  process_one_work from worker_thread+0x34/0x53c
[85929.264702]  worker_thread from kthread+0xc0/0xe4
[85929.264748]  kthread from ret_from_fork+0x14/0x34
[85929.264779] Exception stack(0xe18f5fb0 to 0xe18f5ff8)
[85929.264804] 5fa0:                                     00000000 00000000 00000000 00000000
[85929.264828] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[85929.264850] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[85929.265033] ---[ end trace 0000000000000000 ]---

```

pelwell, 2024-02-14:
`brcmfmac.debug=0x100000` enables the printing of messages from the firmware to the kernel log; you should look for lines beginning `brcmfmac: CONSOLE:`. It won't change the contents of the WARNING dumps.

ganymed, 2024-05-01:
I had the same on a Raspberry Pi Zero 2 W.  The Raspberry Pi had lost its IP address and with that also internet connectivity, and showed the dmesg warning as above.  "sme.c:1181 cfg80211_roamed..."  

sudo ifconfig wlan0 down
sudo ifconfig wlan0 up

brought the IP address and internet connectivity back without a reboot or any other command.  This is for a device that is running headless and should stay up without intervention for years, so while this maybe helps for debugging, it is not a solution for me yet.

Spider84, 2024-06-01:
Same Warning on my Raspbery Pi
```
[19171.499565] WARNING: CPU: 3 PID: 1123 at net/wireless/sme.c:1181 cfg80211_roamed+0x380/0x3d4 [cfg80211]
[19171.500198] Modules linked in: brcmfmac brcmutil sha256_generic cfg80211 bcm2835_codec(C) raspberrypi_hwmon v4l2_mem2mem bcm2835_v4l2(C) bcm2835_isp(C) videobuf2_vmalloc bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common videodev rfkill i2c_bcm2835 bonding snd_bcm2835(C) mc vc_sm_cma(C) fuse ip_tables x_tables ipv6 overlay vc4 snd_soc_hdmi_codec snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd drm_display_helper drm_dma_helper drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm drm_panel_orientation_quirks backlight cec uio_pdrv_genirq uio fixed
[19171.500564] CPU: 3 PID: 1123 Comm: kworker/3:2 Tainted: G        WC         6.1.21-v7+ #1642
[19171.500584] Hardware name: BCM2835
[19171.500597] Workqueue: events brcmf_fweh_event_worker [brcmfmac]
[19171.500961]  unwind_backtrace from show_stack+0x18/0x1c
[19171.501001]  show_stack from dump_stack_lvl+0x68/0x8c
[19171.501028]  dump_stack_lvl from __warn+0x84/0x124
[19171.501054]  __warn from warn_slowpath_fmt+0xac/0x170
[19171.501080]  warn_slowpath_fmt from cfg80211_roamed+0x380/0x3d4 [cfg80211]
[19171.501597]  cfg80211_roamed [cfg80211] from brcmf_bss_roaming_done+0x168/0x214 [brcmfmac]
[19171.502396]  brcmf_bss_roaming_done [brcmfmac] from brcmf_notify_roaming_status+0x40/0x64 [brcmfmac]
[19171.503020]  brcmf_notify_roaming_status [brcmfmac] from brcmf_fweh_event_worker+0x22c/0x4f0 [brcmfmac]
[19171.503646]  brcmf_fweh_event_worker [brcmfmac] from process_one_work+0x244/0x508
[19171.503975]  process_one_work from worker_thread+0x48/0x558
[19171.504007]  worker_thread from kthread+0xe4/0x100
[19171.504036]  kthread from ret_from_fork+0x14/0x34
[19171.504058] Exception stack(0xb88f9fb0 to 0xb88f9ff8)
[19171.504074] 9fa0:                                     00000000 00000000 00000000 00000000
[19171.504090] 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[19171.504106] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[19171.504227] ---[ end trace 0000000000000000 ]---
```