raspberrypi/linux #3037: Cannot compile bcm2835-mmal-vchiq.ko since updating to raspberrypi-kernel_1.20190620-1 tag

dflogeras, 2019-07-01:
I am trying to upgrade my from-source built kernel from the previous raspberrypi-kernel_1.20190517-1 tag to the latest (4.19.x) raspberrypi-kernel_1.20190620-1 tag.

After switching tags in my git clone, and then doing a 'make oldconfig', I rebuild.  The process stops with:

```
ERROR: "vc_sm_cma_import_dmabuf" [drivers/staging/vc04_services/vchiq-mmal/bcm2835-mmal-vchiq.ko] undefined!
ERROR: "vc_sm_cma_free" [drivers/staging/vc04_services/vchiq-mmal/bcm2835-mmal-vchiq.ko] undefined!
ERROR: "vc_sm_cma_int_handle" [drivers/staging/vc04_services/vchiq-mmal/bcm2835-mmal-vchiq.ko] undefined!
make[3]: *** [/mnt/data/dave/pi_kern/scripts/Makefile.modpost:92: __modpost] Error 1
make[2]: *** [/mnt/data/dave/pi_kern/Makefile:1229: modules] Error 2
```
I notice that between prior and current versions, CONFIG_BCM_VC_SM_CMA seems to no longer be able to be built as a module.  Since I am building CONFIG_BCM2835_VCHIQ, CONFIG_SND_BCM2835, CONFIG_VIDEO_BCM2835, CONFIG_BCM2835_VCHIQ_MMAL all as modules, this may be related to its inability to find the symbols defined in the (now) built-in CONFIG_BCM_VC_SM_CMA.

I'd prefer to keep these as modules, since not all my systems have/need camera/video hardware.  If I build all as built-ins the problem is worked-around.

6by9, 2019-07-01:
Correct - it is not currently possible to build vc-sm-cma as a module.

There is ongoing work on the V4L2 and vc-sm-cma drivers to support allocation as well as import. I had believed that using the CMA functions directly was the correct approach as that is how the Android ION allocator works. The CMA functions are not exported, therefore you can not build as a module if you wish to use them, hence vc-sm-cma became a bool.

I've now found that using cma directly doesn't work, so it's being reworked to use dma_alloc_coherent and associated calls. I won't be looking into build issues over the current driver.

dflogeras, 2019-07-01:
OK thanks for the info, I'll build statically for now and wait patiently for a future update.  Cheers

closed 2019-07-01 by dflogeras